networkupstools/nut #2670: USB HID UPS / APC HID UPS shutdown command attempted order

Dummy0815, 2024-11-06:
Hi we are using an APC USB UPS with an Try Contact I/O card (AP9613) with an external key switch, to switch the UPS and therefor the connected systems with this key ON/OFF.
The APC is "simulating" an LOW BAT with key position OFF and triggers a UPS ON with the other position.
Everything works fine with the 'old' generation of APC (SUA750RMI1U, SUA750I, ..) but with the newer generation (SMT1500RMI2U) we have the issue, that if key OFF (low bat) triggers the system shut down, but the last command "upsdrvctl shutdown -> driver -k" sends an "shutdown.reboot" to the UPS. This causes the UPS to turn the outlets delayed off (which is what we want), but issuing a restart, which causes the UPS to turn on the outlets again (as main power is on the the whole time).
Reason seems to be that the 'old' APC's support the shutdown commands:
"shutdown.reboot", "shutdown.return", "shutdown.stayoff" (and load.off.delay)
While the newer ones only support:
"shutdown.reboot" (and load.off.delay)

So according the NUT documentation the driver -k tries to issue in a fixed order "shutdown.return", "shutdown.reboot", "load.off.delay".
So old APC is shutdown with "shutdown.return" which is Ok as the UPS stays down as the main power doesn't change.
While the newer, as they down support the first command, are doing an "shutdown.reboot" causing the issue we have.

So my question is there any parameter which influences the attempted command sequence? So that we could configure try: 'load.off.delay' first than, "shutdown.return", than "shutdown.reboot"?
As an workaround is there any chance to send a 'load.off.delay' in the final system shutdown stage instead of "upsdrvctl shutdown"?

For reference the output of an old vs an newer APC:
########################
upsc shups
..
device.mfr: American Power Conversion
device.model: Smart-UPS 750
driver.name: usbhid-ups
driver.version.data: APC HID 0.95
driver.version.internal: 0.38

upscmd -l shups
..
load.off.delay - Turn off the load with a delay (seconds)
shutdown.reboot - Shut down the load briefly while rebooting the UPS
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off

########################
upsc  shups
..
device.mfr: American Power Conversion
device.model: Smart-UPS 1500
driver.name: usbhid-ups
driver.version: 2.7.2
driver.version.data: APC HID 0.95

upscmd -l shups
..
load.off.delay - Turn off the load with a delay (seconds)
shutdown.reboot - Shut down the load briefly while rebooting the UPS

jimklimov, 2024-11-06:
Hello, thanks for the interesting question.

First of all, please note that you seem to be running NUT v2.7.2 which is over 10 years old and would not be directly updated, at least by core project. You might build your own if required, though (e.g. to apply a custom patch). Current NUT codebase aims to be buildable wherever it worked before, so https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests might help you follow that path.

Looking at even current code, the command sequence in https://github.com/networkupstools/nut/blob/d244d73b1c90d5a461b39023bd4900f9eacd935d/drivers/usbhid-ups.c#L969-L993 is hard-coded. I suppose a new driver configuration variable to list the suitable shutdown commands in order can make sense here.

Note that normally NUT daemons (drivers, upsmon, upsd...) are killed as any other services in the system going down, before it gets to the part where it calls `upsdrvctl shutdown -> driver -k`, so you would not be able to simply `upscmd` the needed operation. Generally it depends on your OS and its NUT integration, in some cases you might be able to modify the shutdown sequence so that NUT daemons survive until power-cycling. In others, the OS kills off all remaining processes before the endgame.

jimklimov, 2024-11-10:
Tinkering about this...

jimklimov, 2024-11-19:
@Dummy0815 : cheers, posted a PR with prospective change for this feature. Testing would be very welcome; see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about making a custom build of NUT for experiments (can be run straight from build workspace).

Dummy0815, 2024-11-25:
@jimklimov: Hi sorry, I've just returned from sick leave and still have to catch up. 
Thank you very much for your efforts and I will try this as soon as possible.

Dummy0815, 2024-12-02:
@jimklimov: Hi sorry took a while, but now I've done some test.
And it works like a charm.
I've tested it with 3 different APC UPS:
SUA750RMI1U (older model)
SMT1500RMI1U and SMT1500RMI2U  (newer models).
and were able to control the shutdown behavior the way I needed it.
Great work and many thank

closed 2024-12-02 by Dummy0815

jimklimov, 2024-12-02:
Just to clarify: you configured `sdcommands` in `ups.conf` section of each device for the operation order applicable to that device, and it worked (tried the "correct" shutdown command first for the respective device and did not try others after success of this one, and covered different such "first" commands for different devices)?

For historic reference, can you please post relevant excerpts from your `ups.conf` here? :)

jimklimov, 2024-12-02:
Also, which scenario(s) did you check?

Did this work with e.g.:
* `upscmd -u ... -p ... -w upsname shutdown.default`, or
* `usbhid-ups -a upsname -k`, or
* `upsdrvctl shutdown`,
* or also a fully-fledged low-battery FSD with `nutshutdown` hook script (or equivalent) called in the real OS endgame?

Dummy0815, 2024-12-02:
Yes as far as I can tell SUA750RMI1U handles shutdown.reboot correct. Independent of main power it does a shutdown and if there is no main power it keeps off until main power is restored and if main power stays on it immediately delivers power again.
(Same behavior as SMT1500RMI1U)

Yes the command shutdown.return is never issued to SMT1500RMI1U as it doesn't support it.

But the big improvement through your changes is, as you said, now we have the control. :-)
And if a UPS implements a shutdown.xy in a different way, there is now at least a good chance, to pick one that does what you want :-)

jimklimov, 2024-12-02:
I've caught an older error, now fixed in that branch, about driver-to-driver communications: the new instance called with `drivername -k` or `upsdrvctl shutdown` did not actually wait for reply from the older daemonized one, and tried to connect to the device anyway and pass its own copy of the logic implementation of the shutdown command (potentially even with a different `sdcommands` customization than what the running daemon knows).

Hopefully that is fixed now, and only one attempt at commanding the UPS would happen.

PR #2686 (2024-12-07): Introduce `sdcommands` option and `shutdown.default` INSTCMD to all drivers
Closes: #2670

Started from discussion in the issue linked above, about drivers hard-coding some operations as their late-shutdown implementation -- which may not be the best fit for a particular supported device model.

This PR aims to allow one to configure (a generally comma-separated list of) `sdcommands` option with arbitrary INSTCMD values supported by the device-specific part of driver code (if any). In testing e.g. beeper commands can be used instead of powering off the load.

It also aliases the call to `upsdr [... cut]