networkupstools/nut #2672: Is it advisable to place the PIPEFN and LOCKFN for upssched in /var/lib/nut ?

ashrude, 2024-11-06:
Hi, I've been trying to wrap my head around where the PIPEFN and LOCKFN files should go. My understanding from reading the FHS is that /run is the best place for runtime data, but I don't want to have a init script to create /run/nut with the correct permissions for the nut user.

I've seen people using /etc/nut but again, according to the FHS it's for static data only.

/var/lib/nut is the home directory for the nut user (at least in alpine), it seems to me that this would be the best place to put PIPEFN and LOCKFN. Is there a better directory or a reason not to use /var/lib/nut?

Thanks, Ashley.

jimklimov, 2024-11-06:
Can't really see why not. If systemd spread its tentacles there, you can take advantage of `systemd-tmpfiles` for creating such locations (in `/run` too) before starting NUT services. As you can see in `scripts/`, making those from init scripts is also a common approach...

One benefit of a temporary location is not-abusing persistent storage for files you don't need persisted, and it helps to have them disappear during reboot and so rule out confusion with obsolete PID files etc.

ashrude, 2024-11-07:
I didn't think about the obsolete files problem. I'll use systemd-tmpfiles to create /run/nut and it looks like the openrc equivalent is checkpath in start_pre().

Thanks!

closed 2024-11-07 by ashrude

jimklimov, 2024-11-07:
Well, obsolete files *should not* be a problem, vanilla NUT init-scripts go to some lengths to remove e.g. `/etc/killpower` while starting the OS, etc. - but packages often deliver something of their own and overlook nuances.

In current code (though maybe not part of a release yet... or was it in 2.8.2 already?...) daemons that can use PID files to signal their older running instances or kill them off, now try to check that the discovered PID number from a file (or command-line option) resolves to their program name, for example.