networkupstools/nut #2681: Reconnect broke NUT plugin for Unraid by Fix #2671

masterwishx, 2024-11-16:
After compiled latest master with pr #2671 ,, it seems nut not working as usual and reconnect every time .
this fix should solve #2643 #2644 issues but seems not working as expected.

@desertwitch also broked NUT Plugin for Unraid :  

[image]

```
Nov 16 11:24:28 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:31 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:35 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:38 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:41 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:44 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:48 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:51 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:54 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:24:57 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:01 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:04 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:07 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:10 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:14 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:17 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:20 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:24 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:27 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:30 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:34 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:37 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:40 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:44 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:47 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:50 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
Nov 16 11:25:54 server usbhid-ups[8198]: [D4:8198]  device has been disconnected, try to reconnect
```

masterwishx, 2024-11-16:
[bad nut 16.11.24.txt](https://github.com/user-attachments/files/17784152/bad.nut.16.11.24.txt)

desertwitch, 2024-11-16:
Hmm, first thing that comes to mind is that the first data collection round(s) may still return 0 HID objects (maybe while the UPS is still initializing the connection, collecting data, ...). This would make it reconnect, only to again see 0 HID objects in the first data round(s), reconnect again - resulting in an infinite loop as we see above?

masterwishx, 2024-11-16:
> Hmm, first thing that comes to mind is that the early data collection round(s) may still return 0 HID objects (maybe while the UPS is still initializing the connection, collecting data, ...). This would make it reconnect, only to again see 0 HID objects in the first data round(s), reconnect again - resulting in an infinite loop as we see above? Perhaps we should exclude this reconnection condition in the early stages of the connection, and e.g. only consider it when HID elements have been seen at least `x` times or in the last `x` previous round(s)?

for now Maybe to make this fix for APC model only for what its fixed?

im not sure why i have 0 HID objects, but by function : 

```
Error handler for usb_get/set_* functions.
 * 0 unknown or temporary failure (ignored)

```

https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/libusb1.c#L871

https://github.com/networkupstools/nut/blob/0f6ae03925b139f64e8f45576dd9d5623cd1fd34/drivers/libusb1.c#L838C1-L841C4

Befor fix i had: 

```
Oct 23 16:05:09 server usbhid-ups[32153]: [D1:32153] upsdrv_updateinfo...
Oct 23 16:05:10 server usbhid-ups[32153]: [D2:32153] nut_libusb_get_interrupt: Connection timed out
Oct 23 16:05:10 server usbhid-ups[32153]: [D1:32153] Got 0 HID objects...
Oct 23 16:05:10 server usbhid-ups[32153]: [D1:32153] Quick update...
Oct 23 16:05:10 server usbhid-ups[32153]: [D4:32153] Entering libusb_get_report
Oct 23 16:05:10 server usbhid-ups[32153]: [D3:32153] Report[get]: (3 bytes) => 26 05 02
Oct 23 16:05:10 server usbhid-ups[32153]: [D5:32153] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
```

After: 

```
Nov 16 11:05:49 server usbhid-ups[10186]: [D1:10186] upsdrv_updateinfo...
Nov 16 11:05:49 server usbhid-ups[10186]: [D1:10186] Got to reconnect!
Nov 16 11:05:49 server usbhid-ups[10186]: [D5:10186] send_to_all: SETINFO driver.state "reconnect.trying"
Nov 16 11:05:49 server usbhid-ups[10186]: [D6:10186] send_to_all: write 40 bytes to socket 16 succeeded (ret=40): SETINFO driver.state "reconnect.trying"
Nov 16 11:05:49 server usbhid-ups[10186]: [D4:10186] Closing comm_driver previous handle
Nov 16 11:05:49 server usbhid-ups[10186]: [D4:10186] ===================================================================
Nov 16 11:05:49 server usbhid-ups[10186]: [D4:10186]  device has been disconnected, try to reconnect
Nov 16 11:05:49 server usbhid-ups[10186]: [D4:10186] ===================================================================
Nov 16 11:05:49 server usbhid-ups[10186]: [D4:10186] Opening comm_driver ...
Nov 16 11:05:49 server usbhid-ups[10186]: [D2:10186] Checking device 1 of 11 (1D6B/0003)
Nov 16 11:05:49 server usbhid-ups[10186]: [D1:10186] nut_libusb_open: invalid libusb bus number 0

```

masterwishx, 2024-11-16:
> Perhaps we should exclude this reconnection condition in the early stages of the connection, and e.g. only consider it when HID elements have been seen at least `x` times or in the last `x` previous round(s)?

Seems I have 0 hid object and timeout on like every update but then quick update, but after fix I have reconnect on every update...

jimklimov, 2024-11-16:
I see :\ <facepalm>

Would it help to make this a toggle (with `0=>reconnect` behavior off by default), and/or a counter of how many 0-entries in a row we've got (similar to throttling elsewhere)?

masterwishx, 2024-11-16:
> I see :\
> 
> Would it help to make this a toggle (with `0=>reconnect` behavior off by default), and/or a counter of how many 0-entries in a row we've got (similar to throttling elsewhere)?

Not shure becose I have on every update 0 hid
:(

masterwishx, 2024-11-16:
[image]

desertwitch, 2024-11-16:
> [image]
> 
> 

May not fix your case, but at least your UPS is functional (iirc?) and doesn't lock up requiring manual reconnect as the initial PR author's. So for such cases as his a configurational toggle of this behavior could be helpful, I think.

masterwishx, 2024-11-16:
> > I see :\ 
> > Would it help to make this a toggle (with `0=>reconnect` behavior off by default), and/or a counter of how many 0-entries in a row we've got (similar to throttling elsewhere)?
> 
> I'm definitely in a favor of a no-surprise, default-off toggle solution (counter seems good, can be -1 as default for disabled / 0 instant reconnect / 1 one time in a row / 2 two times in a row, ...)

What is mean for row? i have 0 HID for circle when update

masterwishx, 2024-11-16:
> May not fix your case, probably,

Not a problem its working for now ,just i think maybe realted some commands maybe not working all time but maybe will find fix later

desertwitch, 2024-11-16:
> What is mean for row? i have 0 HID for circle when update

Means for some UPS could be:

zero_hid_reconnect 3

3 HID elements
3 HID elements
0 HID elements (no reconnect)
3 HID elements
0 HID elements
0 HID elements
0 HID elements
(3 in a row - reconnect)
3 HID elements

masterwishx, 2024-11-16:
> > > > I see :\
> > 
> > 
> > > > Would it help to make this a toggle (with `0=>reconnect` behavior off by default), and/or a counter of how many 0-entries in a row we've got (similar to throttling elsewhere)?
> > 
> > 
> > > 
> > 
> > 
> > > I'm definitely in a favor of a no-surprise, default-off toggle solution (counter seems good, can be -1 as default for disabled / 0 instant reconnect / 1 one time in a row / 2 two times in a row, ...)
> > 
> > 
> > What is mean for row? i have 0 HID for circle when update
> 
> Means for some UPS could be:
> 
> zero_hid_reconnect 3
> 
> 4 HID elements 4 HID elements 0 HID elements (no reconnect) 4 HID elements 0 HID elements 0 HID elements 0 HID elements (3 in a row - reconnect) 5 HID elements
> 
> Like three consecutive zero HID element cycles.

Looks Fine to me!

masterwishx, 2024-11-16:
i have like every cicle 0 HID : 

```
Nov 16 12:00:44 DaRKNaS usbhid-ups[32357]: [D1:32357] upsdrv_updateinfo...
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] nut_libusb_get_interrupt: Connection timed out
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D1:32357] Got 0 HID objects...
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D1:32357] Quick update...
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D4:32357] Entering libusb_get_report
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[get]: (3 bytes) => 26 05 02
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.BatterySystem.Charger.ChargerType, Type: Feature, ReportID: 0x26, Offset: 0, Size: 8, Value: 5
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] hu_find_infoval: found Constant Charge (CC) (value: 5)
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[buf]: (3 bytes) => 26 05 02
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.BatterySystem.Charger.Status, Type: Feature, ReportID: 0x26, Offset: 8, Size: 8, Value: 2
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] ABM Status is 2
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Lookup [2] failed for [battery.charger.type.status]
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[buf]: (3 bytes) => 26 05 02
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.BatterySystem.Charger.Status, Type: Feature, ReportID: 0x26, Offset: 8, Size: 8, Value: 2
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] ABM numeric status: 2
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] ABM string status: floating
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] ABM is -1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D4:32357] Entering libusb_get_report
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[get]: (5 bytes) => 0a ff ff ff ff
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 2147483647, LogMin = -1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00001001, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: -1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D4:32357] Entering libusb_get_report
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[get]: (5 bytes) => 09 ff ff ff ff
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 2147483647, LogMin = -1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00001001, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 32, Value: -1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D4:32357] Entering libusb_get_report
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[get]: (6 bytes) => 01 01 00 01 00 01
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] hu_find_infoval: found online (value: 1)
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] process_boolean_info: online
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[buf]: (6 bytes) => 01 01 00 01 00 01
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D2:32357] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] process_boolean_info: !dischrg
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D3:32357] Report[buf]: (6 bytes) => 01 01 00 01 00 01
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Unit = 00000000, UnitExp = 0
Nov 16 12:00:45 DaRKNaS usbhid-ups[32357]: [D5:32357] Exponent = 0
[... cut ...]
Nov 16 12:00:47 DaRKNaS usbhid-ups[32357]: [D2:32357] nut_libusb_get_interrupt: Connection timed out
Nov 16 12:00:47 DaRKNaS usbhid-ups[32357]: [D1:32357] Got 0 HID objects...
Nov 16 12:00:47 DaRKNaS usbhid-ups[32357]: [D1:32357] Quick update...
Nov 16 12:00:47 DaRKNaS usbhid-ups[32357]: [D4:32357] Entering libusb_get_report

```

you mean someone have `0 HID`  3 time printed one after another without any printed lines between: 

```
upsdrv_updateinfo...
 nut_libusb_get_interrupt: Connection timed out
 Got 0 HID objects...
 Got 0 HID objects... 
 Got 0 HID objects...
```

jimklimov, 2024-11-16:
Ack, sorry about that, I'll poke a counter solution then.

PR #2682 (2024-11-17): Fix fallout of #2671 and complete the feature with optional `interrupt_pipe_no_events_tolerance` setting
Now users of devices that can be caught hanging out and slacking off by a continuous stream of "Got 0 HID Objects" from interrupts can configure a toggle to say how much is enough, and reconnect the driver. Legacy default is to not bother, and assume the device just has nothing to urgently report.

Closes: #2681
Follow-up to: #2671

CC for testing: @desertwitch @masterwishx @michal4132
closed 2024-11-17 by jimklimov