networkupstools/nut #2683: Extracting shutdown command from "sniffed" USB traffic

ErikDB87, 2024-11-17:
In light of issue #2666, I "sniffed" USB traffic from APC's `PowerChute Serial Shutdown` software (with the help of this guide (https://www.youtube.com/watch?v=0MC-D_oNzbk)). I hope to be able to find out the shutdown command for my APC "Back-UPS BX750MI FW:295202G -302202G". APC's `PowerChute Serial Shutdown` software is able to shut the UPS down, so there must be a command, is my reasoning.

**Any help on figuring this command out from the pcap files, is welcome!** I haven't got the slightest idea on how to go about it...

I was able to generate two logs. The scenario was both initially the same: I unplugged the UPS from the electricity grid, and the `PowerChute Serial Shutdown` software started closing down my (Windows) laptop. Then I tried two scenarios:

- `2024-11-14-APC-BX750MI-1.pcap`: I let everything run its course, and the UPS shut down, a little over a minute after my laptop had shut down completely (so maybe two minutes after initiating the process, as referenced on that forum I linked somewhere above?). I then plugged the UPS back in, and it restarted (of course).
- `2024-11-14-APC-BX750MI-2.pcap`: As soon as Windows notified me of the shutdown that was going to occur, I plugged the UPS back in. I was pleasantly surprised that the same process as in scenario 1 happened, and the UPS power cycled.

The only caveat is of course that I don't know whether the `PowerChute Serial Shutdown` software shut down `USBPcap` prior to giving its final command - the holy grail - to the UPS... I don't think this is the case: I saw `2024-11-14-APC-BX750MI-2.pcap` "growing" even after I plugged the UPS back in.

(I can't upload the log files, because they're not an allowed file type, so here's a link to a directory on my Google Drive, where they can be found: https://drive.google.com/drive/folders/12xLZH4YWJGXEjNETs2YBJix4boIUVDAE?usp=sharing.)

Thanks in advance to any and all! ;)

ErikDB87, 2024-11-19:
I made another "sniff" log (`2024-11-19-APC-BX750MI.pcap`, via above mentioned link), this time with the setting that my Windows system would be shut down after 1 second of battery usage, and I unplugged the USB cable as soon as Windows notified me of the shutdown. That way I hoped to keep the log file shorter, and the shutdown command as close to the end of it as possible. The output was still 712 entries long, though...

ErikDB87, 2025-06-11:
Looks like this is not a very popular approach, and I don't like lingering open issues. :)

jimklimov, 2025-06-11:
Actually, I think I'll keep this issue open, so it is not completely lost from sight. There are quite a few things to address in NUT backlog, and maybe some future visitor one day will find and fix this.

But thanks for reporting, and for trying to clean up as many other projects like to do :)

faluc-dev, 2026-08-16:
We've now bench-tested the second APC in our fleet — a **Back-UPS XS 700U** (fw `924.Z5 .I`, USB fw `Z5`, same `051d:0002` USB id, a ~2022-generation unit) — with the same protocol as the BX750MI work above. Headline: **the value-sensitivity and the fix in PR #3566 apply to this older generation too**, but the surrounding semantics differ enough that they seem worth recording.

Same core situation as the BX:

- The descriptor has no `DelayBeforeStartup` anywhere, and `APCDelayBeforeReboot` (report 0x40, 8-bit) sits under `UPS.APCGeneralCollection` — so the stock chain lands on `shutdown.reboot` = "10", which is ACKed and never executed. **`0x40=1` executes** (bench-proven repeatedly, load observed). The PR's added `shutdown.return` table line would make default `upsdrvctl shutdown` work on this family as well.
- Two extra usages the BX lacks, both unmapped by `apc-hid` and untested by us (no hidraw on the macOS bench): `UPS.APCGeneralCollection.APCDelayBeforeShutdown` (report 0x41, 16-bit) and `UPS.APCGeneralCollection.DelayBeforeShutdown` (report 0x42, 16-bit).

Where the XS diverges from the BX-2025 firmware:

| Behavior | BX750MI (2025 fw) | XS 700U (924.Z5) |
|---|---|---|
| `0x40=1` written on line | silently discarded | **arms** and runs to expiry |
| `0x40=1` grace | ~2 min | **~60 s** |
| 0x40 expiry with AC present | n/a (never armed) | **~0.5 s output cycle** (camera-verified) |
| 0x40 expiry on battery | cut; re-energize on AC return | same (return ~2 s after AC) |
| `0x15` (`load.off.delay`) expiry on line | silent no-op | **cuts** |
| Off-state after a 0x15 cut | auto-return on AC | **sticky** — AC edges do NOT re-energize; only the front-panel button does |

Two practical consequences:

1. **`sdcommands = load.off.delay` advice is firmware-dependent on this family.** On the BX it's the right answer; on the XS it's a stranding hazard — the cut is honored on line *and* on battery, and the resulting off-state survives AC return, so a `load.off.delay` killpower leaves the load dark until someone presses the button. (Battery-*exhaustion* off is different: that state does auto-restart on AC return — observed on this unit in a real outage.) `shutdown.reboot` with value 1 is the only self-recovering kill on the XS, which is another argument for fixing the default chain (the PR) rather than steering users to per-device `sdcommands`.
2. The 0x40 race behavior differs in mechanism but not in outcome: on the XS the armed timer is not cancelled by AC return — expiry with AC present is a ~0.5 s output cycle, on battery it's a cut-until-AC. Either way every shut-down host eventually gets an AC edge, same net effect as the BX's race-immune cycle.

Bench-method notes for anyone testing these units:

- `ups.status` is unreliable around timer operations (`OL DISCHRG` on line during countdowns, `OB`/`OL` blips both ways, `OL` reported while on battery in the off-state), and `input.voltage` holds a stale reading after AC loss (decays via ~140 V). Our ground truth was a filament desk lamp filmed with a phone camera: the eye sees millisecond transfer-relay dips the 30/60 fps video integrates over, so "dark frames on video" cleanly separates a real output cycle (the 0.5 s event above) from relay transfer clicks.
- The XS's USB iSerial carries two trailing spaces (`"3B2022X70450  "`) — a `serial` matcher in `ups.conf` needs to account for them.

Full HID walk and per-second poller logs from the bench are archived; happy to run any other write you'd like ruled in or out on either unit.