networkupstools/nut #2685: Eaton 9SX Model usbhid-ups: Eaton battery.voltage value is missing

ALEXNETHUNTER, 2024-11-19:
I have Eaton 9SX 1000i and im not getting battery.voltage value.
Could it have a fix like for Eaton 5P / 5PX / 5SC series?

<details>
<summary>Client Logs</summary>

battery.capacity: 9.00
battery.charge: 100
battery.charge.low: 0
battery.charge.restart: 0
battery.charger.status: floating
battery.energysave.delay: 300
battery.protection: yes
battery.runtime: 1530
battery.runtime.low: 180
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 9SX 1000i
device.serial: *********
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.product: Eaton 9SX
driver.parameter.productid: FFFF
driver.parameter.serial: *********
driver.parameter.synchronous: auto
driver.parameter.vendor: EATON
driver.parameter.vendorid: 0463
driver.state: quiet
driver.version: 2.8.2
driver.version.data: MGE HID 1.46
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.bypass.current: 0.00
input.bypass.frequency: 50.0
input.bypass.voltage: 226.7
input.frequency: 50.0
input.frequency.nominal: 50
input.transfer.high: 276
input.transfer.low: 140
input.voltage: 226.7
input.voltage.nominal: 230
outlet.1.autoswitch.charge.low: 1
outlet.1.delay.shutdown: 999999
outlet.1.delay.start: 3
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.autoswitch.charge.low: 1
outlet.2.delay.shutdown: 999999
outlet.2.delay.start: 6
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 1.50
output.frequency: 50.0
output.frequency.nominal: 50
output.powerfactor: 0.83
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
[... 438 lines cut ...]
Nov 19 21:14:20 nut usbhid-ups[43346]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO output.powerfactor "0.85"
Nov 19 21:14:20 nut usbhid-ups[43346]: [D4] Entering libusb_get_report
Nov 19 21:14:20 nut usbhid-ups[43346]: [D3] Report[get]: (4 bytes) => 82 01 01 01
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Unit = 00000000, UnitExp = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Exponent = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D2] Path: UPS.OutletSystem.Outlet.[2].PresentStatus.SwitchOn/Off, Type: Feature, ReportID: 0x82, Offset: 8, Size: 8, Value: 1
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] hu_find_infoval: found on (value: 1)
Nov 19 21:14:20 nut usbhid-ups[43346]: [D4] Entering libusb_get_report
Nov 19 21:14:20 nut usbhid-ups[43346]: [D3] Report[get]: (9 bytes) => 81 01 04 01 01 01 04 02 01
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Unit = 00000000, UnitExp = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Exponent = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.Switchable, Type: Feature, ReportID: 0x81, Offset: 56, Size: 8, Value: 1
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] hu_find_infoval: found yes (value: 1)
Nov 19 21:14:20 nut usbhid-ups[43346]: [D3] Report[buf]: (4 bytes) => 82 01 01 01
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Unit = 00000000, UnitExp = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] Exponent = 0
Nov 19 21:14:20 nut usbhid-ups[43346]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.SwitchOn/Off, Type: Feature, ReportID: 0x82, Offset: 16, Size: 8, Value: 1
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] hu_find_infoval: found on (value: 1)
Nov 19 21:14:20 nut usbhid-ups[43346]: [D5] send_to_all: SETINFO driver.state "quiet"
Nov 19 21:14:20 nut usbhid-ups[43346]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"

</details>

jimklimov, 2024-11-19:
Sounds like https://github.com/networkupstools/nut/pull/2442 (merged after 2.8.2 release). Can you try custom-building NUT (see wiki e.g. about in-place builds) from master branch with a similar fix added?

ALEXNETHUNTER, 2024-11-20:
I've built from master, but it makes no sense, since 9sx model is not in the list in _mge-hid.c_:

```
	EATON_5P = 0x500,			/* Eaton 5P / 5PX / 5SC series */
	EATON_9E = 0x900			/* Eaton 9E entry-level series */
} models_type_t;
```

Additional parametes appeared:

```
device.usb.version: 01.29
driver.parameter.interrupt_pipe_no_events_tolerance: -1
input.bypass.switch.off: off
input.bypass.switch.on: disabled
input.eco.switchable: normal
input.transfer.bypass.high: 265
input.transfer.bypass.low: 184
input.transfer.forced: enabled
input.transfer.frequency.bypass.range: 5
input.transfer.hysteresis: 10
outlet.1.designator: Output
outlet.1.protect.status: protected
outlet.2.protect.status: protected
```

<details>
<summary>New Client Logs</summary>

battery.capacity: 9.00
battery.charge: 100
battery.charge.low: 0
battery.charge.restart: 0
battery.charger.status: floating
battery.energysave.delay: 300
battery.protection: yes
battery.runtime: 1545
battery.runtime.low: 180
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 9SX 1000i
device.serial: GC01M36086
device.type: ups
device.usb.version: 01.29
driver.debug: 6
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.product: Eaton 9SX
driver.parameter.productid: FFFF
driver.parameter.serial: xxxxxxxxx
driver.parameter.synchronous: auto
driver.parameter.vendor: EATON
driver.parameter.vendorid: 0463
driver.state: quiet
driver.version: 2.8.2.1303-1303-gb6241458b
driver.version.data: MGE HID 1.51
driver.version.internal: 0.58
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.bypass.current: 0.00
input.bypass.frequency: 50.0
input.bypass.switch.off: off
input.bypass.switch.on: disabled
input.bypass.voltage: 230.2
input.eco.switchable: normal
input.frequency: 50.0
input.frequency.nominal: 50
[... 579 lines cut ...]
Nov 20 14:46:09 nut usbhid-ups[87487]: [D2] Path: UPS.OutletSystem.Outlet.[1].Status, Type: Feature, ReportID: 0x83, Offset: 0, Size: 8, Value: 2
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] hu_find_infoval: found protected (value: 2)
Nov 20 14:46:09 nut usbhid-ups[87487]: [D4] Entering libusb_get_report
Nov 20 14:46:09 nut usbhid-ups[87487]: [D3] Report[get]: (9 bytes) => 81 01 04 01 01 01 04 02 01
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Unit = 00000000, UnitExp = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Exponent = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.Switchable, Type: Feature, ReportID: 0x81, Offset: 56, Size: 8, Value: 1
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] hu_find_infoval: found yes (value: 1)
Nov 20 14:46:09 nut usbhid-ups[87487]: [D3] Report[buf]: (4 bytes) => 82 01 01 01
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Unit = 00000000, UnitExp = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Exponent = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.SwitchOn/Off, Type: Feature, ReportID: 0x82, Offset: 16, Size: 8, Value: 1
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] hu_find_infoval: found on (value: 1)
Nov 20 14:46:09 nut usbhid-ups[87487]: [D3] Report[buf]: (4 bytes) => 83 02 02 02
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Unit = 00000000, UnitExp = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] Exponent = 0
Nov 20 14:46:09 nut usbhid-ups[87487]: [D2] Path: UPS.OutletSystem.Outlet.[3].Status, Type: Feature, ReportID: 0x83, Offset: 16, Size: 8, Value: 2
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] hu_find_infoval: found protected (value: 2)
Nov 20 14:46:09 nut usbhid-ups[87487]: [D5] send_to_all: SETINFO driver.state "quiet"
Nov 20 14:46:09 nut usbhid-ups[87487]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"

</details>

arnaudquette-eaton, 2024-11-21:
@jimklimov please refrain yourself from assigning to me ;)
I'm willing to help, not to un-retire :D

masterwishx, 2024-11-28:
> Sounds like #2442 (merged after 2.8.2 release). Can you try custom-building NUT (see wiki e.g. about in-place builds) from master branch with a similar fix added?

> I've built from master, but it makes no sense, since 9sx model is not in the list in _mge-hid.c_:
> 
> ```
> 	EATON_5P = 0x500,			/* Eaton 5P / 5PX / 5SC series */
> 	EATON_9E = 0x900			/* Eaton 9E entry-level series */
> } models_type_t;
> ```

Yes ,seems should be added :  

EATON_9SX = 0x901 not shure what hex number ...
also for `static models_name_t mge_model_names ()` , etc  for this will work same as was in #2380

masterwishx, 2024-11-28:
> I've built from master, but it makes no sense, since 9sx model is not in the list in _mge-hid.c_:
> 
> ```
> 	EATON_5P = 0x500,			/* Eaton 5P / 5PX / 5SC series */
> 	EATON_9E = 0x900			/* Eaton 9E entry-level series */
> } models_type_t;
> ```
> 
> Additional parametes appeared:
> 
> ```
> device.usb.version: 01.29
> driver.parameter.interrupt_pipe_no_events_tolerance: -1
> input.bypass.switch.off: off
> input.bypass.switch.on: disabled
> input.eco.switchable: normal
> input.transfer.bypass.high: 265
> input.transfer.bypass.low: 184
> input.transfer.forced: enabled
> input.transfer.frequency.bypass.range: 5
> input.transfer.hysteresis: 10
> outlet.1.designator: Output
> outlet.1.protect.status: protected
> outlet.2.protect.status: protected
> ```
> 
> New Client Logs
> New Debug Logs

Please check #2698

masterwishx, 2024-11-29:
@ALEXNETHUNTER as your issue should be solved by pr #2698, another question : 

Very strange but you should have next vars for ECO mode:

```
input.transfer.eco.high
input.transfer.eco.low
input.transfer.frequency.eco.range
```
Are they missing? 
Can you enter to ECO mode by buttons on ups? 
Did you posted full debug log? 
If not please post it.

ALEXNETHUNTER, 2024-11-30:
Full upc output:

<details>
<summary>Debug Logs</summary>

```

battery.capacity: 9.00
battery.charge: 100
battery.charge.low: 0
battery.charge.restart: 0
battery.charger.status: floating
battery.energysave.delay: 300
battery.protection: yes
battery.runtime: 1560
battery.runtime.low: 180
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 9SX 1000i
device.serial: xxxxxxxxx
device.type: ups
device.usb.version: 01.29
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.product: Eaton 9SX
driver.parameter.productid: FFFF
driver.parameter.serial: xxxxxxxxx
driver.parameter.synchronous: auto
driver.parameter.vendor: EATON
driver.parameter.vendorid: 0463
driver.state: quiet
driver.version: 2.8.2.1303-1303-gb6241458b
driver.version.data: MGE HID 1.51
driver.version.internal: 0.58
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.bypass.current: 1.40
input.bypass.frequency: 50.0
input.bypass.switch.off: off
input.bypass.switch.on: disabled
input.bypass.voltage: 227.6
input.eco.switchable: ECO
input.frequency: 50.0
input.frequency.nominal: 50
input.transfer.bypass.high: 265
input.transfer.bypass.low: 184
input.transfer.forced: enabled
input.transfer.frequency.bypass.range: 5
input.transfer.high: 276
input.transfer.hysteresis: 10
input.transfer.low: 140
input.voltage: 227.6
input.voltage.nominal: 230
outlet.1.autoswitch.charge.low: 1
outlet.1.delay.shutdown: 999999
outlet.1.delay.start: 3
outlet.1.desc: PowerShare Outlet 1
outlet.1.designator: Output
outlet.1.id: 1
outlet.1.protect.status: protected
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.autoswitch.charge.low: 1
outlet.2.delay.shutdown: 999999
outlet.2.delay.start: 6
outlet.2.desc: PowerShare Outlet 2
[... 507 lines cut ...]
Nov 30 11:31:44 nut usbhid-ups[954]: [D4] Entering libusb_get_report
Nov 30 11:31:44 nut usbhid-ups[954]: [D3] Report[get]: (9 bytes) => 81 01 04 01 01 01 04 02 01
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Exponent = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.Switchable, Type: Feature, ReportID: 0x81, Offset: 56, Size: 8, Value: 1
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] hu_find_infoval: found yes (value: 1)
Nov 30 11:31:44 nut usbhid-ups[954]: [D3] Report[buf]: (4 bytes) => 82 01 01 01
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Exponent = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D2] Path: UPS.OutletSystem.Outlet.[3].PresentStatus.SwitchOn/Off, Type: Feature, ReportID: 0x82, Offset: 16, Size: 8, Value: 1
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] hu_find_infoval: found on (value: 1)
Nov 30 11:31:44 nut usbhid-ups[954]: [D3] Report[buf]: (4 bytes) => 83 02 02 02
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] Exponent = 0
Nov 30 11:31:44 nut usbhid-ups[954]: [D2] Path: UPS.OutletSystem.Outlet.[3].Status, Type: Feature, ReportID: 0x83, Offset: 16, Size: 8, Value: 2
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] hu_find_infoval: found protected (value: 2)
Nov 30 11:31:44 nut usbhid-ups[954]: [D5] send_to_all: SETINFO driver.state "quiet"
Nov 30 11:31:44 nut usbhid-ups[954]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"

```

</details>

masterwishx, 2024-11-30:
> Full upc output:
> 
> Debug Logs
> Yes, those values are missing, here is output part for input.transfer and eco Set USP in ECO mode
> 
> ```
> input.transfer.bypass.high: 265
> input.transfer.bypass.low: 184
> input.transfer.forced: enabled
> input.transfer.frequency.bypass.range: 5
> input.transfer.high: 276
> input.transfer.hysteresis: 10
> input.transfer.low: 140
> input.eco.switchable: ECO
> ```
> 
> What do you mean 'full debug mode' (how long do i need to collect output logs) ?
> 
> I enabled debug
> 
> ```
> upsrw -u admin -p pass -s driver.debug=6 eaton
> ```
> 
> and got output with command
> 
> `journalctl -flu nut-driver`
> 
> Debug Logs

Thanks a lot, you entered to ECO mode by nut commands /rw values or by ups switches?

masterwishx, 2024-11-30:
> Yes, those values are missing, here is output part for input.transfer and eco
> Set USP in ECO mode

Thanks
i asked becose when added bypass/ECO mode also added to check volt/freq limits , but i made new pr #2684 where in this case for ECO mode we use default limits, so if you dont have it its ok

ALEXNETHUNTER, 2024-11-30:
> Thanks a lot, you entered to ECO mode by nut commands /rw values or by ups switches?

I switched up thrue ups

masterwishx, 2024-11-30:
> ECO

yep its fine you can have issue to enter by commands/r/w value for now but can try it ?

masterwishx, 2024-11-30:
if you can debug log all proccess of enter /exit from eco ?

masterwishx, 2024-11-30:
> I switched up thrue ups

if enter to eco by command not working just use ups buttons but you can exit from eco by commands ?

```
bypass.start
ecomode.disable
bypass.stop
```

if not just use button of ups again , but need to be logged all proccess enter /exit ECO mode . Thanks for help

ALEXNETHUNTER, 2024-11-30:
Finally I found all needed commands, especially for ECO mode,  to do all stuff via cli:

1. Debug ON
`upsrw -u admin -p pass -s driver.debug=6 eaton`

2. Enable bypass mode:
`upsrw -u admin -p pass -s input.bypass.switch.on=on eaton`

3. Enable ECO mode:
`upsrw -u admin -p pass -s input.eco.switchable=enable eaton`

4. Disable bypass mode: 
`upsrw -u admin -p pass -s input.bypass.switch.off=off eaton`

5. Disable ECO mode:
`upsrw -u admin -p pass -s input.eco.switchable=normal eaton`

6. Debug OFF
`upsrw -u admin -p pass -s driver.debug=0 eaton`

<details>
<summary>Debug Logs</summary>

Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 15:59:33 nut usbhid-ups[2861]: [D2] Path: UPS.OutletSystem.Outlet.[3].Status, Type: Feature, ReportID: 0x83, Offset: 16, Size: 8, Value: 2
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] hu_find_infoval: found protected (value: 2)
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.state "quiet"
Nov 30 15:59:33 nut usbhid-ups[2861]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
Nov 30 15:59:38 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Nov 30 15:59:38 nut usbhid-ups[2861]: [D2] entering main_setvar(driver.debug, 0) for [eaton] on socket 10
Nov 30 15:59:38 nut usbhid-ups[2861]: [D1] NOTE: Will disable verbose debug now, due to socket protocol request
Nov 30 15:59:38 nut usbhid-ups[2861]: Applying debug level 0 received during run-time via socket protocol, ignoring other settings
Nov 30 16:09:24 nut usbhid-ups[2861]: Applying debug level 6 received during run-time via socket protocol, ignoring other settings
Nov 30 16:09:24 nut usbhid-ups[2861]: [D1] debug level is '6'
Nov 30 16:09:24 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.debug "6"
Nov 30 16:09:24 nut usbhid-ups[2861]: [D6] send_to_all: write 25 bytes to socket 10 succeeded (ret=25): SETINFO driver.debug "6"
Nov 30 16:09:30 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling PING with 1 args
Nov 30 16:09:30 nut usbhid-ups[2861]: [D2] send_to_one: sending PONG
Nov 30 16:09:30 nut usbhid-ups[2861]: [D5] send_to_one: PONG
Nov 30 16:09:30 nut usbhid-ups[2861]: [D6] send_to_one: write 5 bytes to socket 10 succeeded (ret=5): PONG
Nov 30 16:09:31 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Nov 30 16:09:31 nut usbhid-ups[2861]: [D2] entering main_setvar(input.bypass.switch.on, on) for [eaton] on socket 10
Nov 30 16:09:31 nut usbhid-ups[2861]: [D2] shared main_setvar() does not handle variable input.bypass.switch.on, proceeding to driver-specific handler
Nov 30 16:09:31 nut usbhid-ups[2861]: [D1] setvar(input.bypass.switch.on, on)
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] hu_find_valinfo: found on (value: 1)
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D3] Report[set]: (2 bytes) => 59 01
Nov 30 16:09:31 nut usbhid-ups[2861]: [D4] Set report succeeded
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] setvar: SUCCEED
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.state "updateinfo"
Nov 30 16:09:32 nut usbhid-ups[2861]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] upsdrv_updateinfo...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] nut_libusb_get_interrupt: Connection timed out
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] Got 0 HID objects (520 times in a row, tolerance is -1)...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] Full update...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:09:32 nut usbhid-ups[2861]: [D3] Report[get]: (6 bytes) => 06 64 9f 06 00 00
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
Nov 30 16:09:32 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:09:32 nut usbhid-ups[2861]: [D3] Report[get]: (2 bytes) => 12 00
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Exponent = 0
[... 2108 lines cut ...]
Nov 30 16:10:33 nut usbhid-ups[2861]: [D2] Path: UPS.BatterySystem.Charger.ABMEnable, Type: Feature, ReportID: 0x22, Offset: 0, Size: 8, Value: 1
Nov 30 16:10:33 nut usbhid-ups[2861]: [D2] ABM is enabled
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] Lookup [1] failed for [battery.charger.status]
Nov 30 16:10:33 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:10:33 nut usbhid-ups[2861]: [D3] Report[get]: (4 bytes) => 26 04 03 12
[... cut ...]
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:10:33 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.PercentLoad, Type: Feature, ReportID: 0x07, Offset: 56, Size: 8, Value: 33
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO ups.load "33"
Nov 30 16:10:33 nut usbhid-ups[2861]: [D6] send_to_all: write 22 bytes

ALEXNETHUNTER, 2024-11-30:
> Also please check pr #2698 that fix your current issue with battery.voltage

Didn't get what I need to do with this, there is CI problem.

closed 2024-11-30 by jimklimov

masterwishx, 2024-11-30:
> > Also please check pr #2698 that fix your current issue with battery.voltage
> 
> Didn't get what I need to do with this, there is CI problem.

just use master as it was merged already

masterwishx, 2024-11-30:
> there is CI problem

its just internal CI issue

ALEXNETHUNTER, 2024-11-30:
> @ALEXNETHUNTER as your issue should be solved by pr #2698, another question :
>
> Very strange but you should have next vars for ECO mode:
>
> ```
> input.transfer.eco.high
> input.transfer.eco.low [...]

```
input.transfer.bypass.high: 265
input.transfer.bypass.low: 184
input.transfer.forced: enabled
input.transfer.frequency.bypass.range: 5
input.transfer.high: 276
input.transfer.hysteresis: 10
input.transfer.low: 140
```

Maybe for this need on more fix for 9SX models. I can create a ticket. Is any logs requires?

masterwishx, 2024-11-30:
> ```
> input.transfer.bypass.high: 265
> input.transfer.bypass.low: 184
> input.transfer.forced: enabled
> input.transfer.frequency.bypass.range: 5
> input.transfer.high: 276
> input.transfer.hysteresis: 10 [...]

If you don't have HID paths for ECO limit, there is no fix for it, but it's not a big problem as we use default limits.

masterwishx, 2024-11-30:
@ALEXNETHUNTER I will check your log to get more info about.

masterwishx, 2024-11-30:
> * Enable bypass mode:
>   `upsrw -u admin -p pass -s input.bypass.switch.on=on eaton`
> * Enable ECO mode:
>   `upsrw -u admin -p pass -s input.eco.switchable=enable eaton`
> * Disable bypass mode:
>   `upsrw -u admin -p pass -s input.bypass.switch.off=off eaton`

Please confirm that ups switched to ECO mode only after last `input.bypass.switch.off=off` command  or after `input.eco.switchable=enable`?

> * Enable ECO mode:
>   `upsrw -u admin -p pass -s input.eco.switchable=enable eaton`

exit from ECO: 
```
 Disable ECO mode:
upsrw -u admin -p pass -s input.eco.switchable=normal eaton
```

or : 

```
Disable bypass mode:
upsrw -u admin -p pass -s input.bypass.switch.off=off eaton

Disable ECO mode:
upsrw -u admin -p pass -s input.eco.switchable=normal eaton
```

ALEXNETHUNTER, 2024-11-30:
Disable bypass mode 
then 
Disable ECO mode

in those sequence of commands i wrote

masterwishx, 2024-11-30:
> Disabled bypass mode then Disabled ECO mode
> 
> in those sequence of commands i wrote

Thanks, so ups switched to bypass after 2 commands:
Enter to bypass
Enable eco mode right?

ALEXNETHUNTER, 2024-11-30:
> > Disabled bypass mode then Disabled ECO mode
> > in those sequence of commands i wrote
> 
> Thanks, so ups switched to ECO mode after 2 commands: Enter to bypass Enable eco mode right?

Yes

ALEXNETHUNTER, 2024-11-30:
Do you need  to be upd switched to ECO mode after bypass off?

masterwishx, 2024-11-30:
> Do you need ups to be switched to ECO mode after bypass off?

No, just in my case I have 9E, when I made pr for ECO/Bypass mode, and enabled ECO mode 

I think it was switched to ECO after bypass off only. (3 commands: bypass on, enable eco, bypass off ). 
And now can't exit from Eco mode. Also don't have buttons on ups for it. 

But your steps looks more right. 
So that why I wanted you to test it. 
So I need to try it again, maybe after shutdown ups it will goes to online again.

masterwishx, 2024-12-01:
@ALEXNETHUNTER can you please tell in what os you are using nut ?
how did you saw the ups was switched to eco mode , by ups itself or by nut ?
im asking becouse i cant see your ups was changed ups status to ecomode after 2 commands and at all in log!

`Path: UPS.PowerConverter.Input.[5].PresentStatus.Used, Type: Feature, ReportID: 0x3b, Offset: 32, Size: 8, Value: 0`

should be `1` when status changed

ALEXNETHUNTER, 2024-12-01:
Ubuntu 24.04.1 LTS
Using Grafana oversaw Input and Output Voltage difference, with double checking in UPS Settings - In/Out Settings-HighEfficiencyMode checkmark appeating/disappearing whie swithcing on/off.

My mis

I found 

`Nov 30 16:09:33 nut usbhid-ups[2861]: [D2] Path: UPS.PowerConverter.Input.[2].PresentStatus.Used, Type: Feature, ReportID: 0x38, Offset: 24, Size: 8, Value: 1`

<details>
<summary>Here Full Debug Logs</summary>

Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 15:59:33 nut usbhid-ups[2861]: [D2] Path: UPS.OutletSystem.Outlet.[3].Status, Type: Feature, ReportID: 0x83, Offset: 16, Size: 8, Value: 2
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] hu_find_infoval: found protected (value: 2)
Nov 30 15:59:33 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.state "quiet"
Nov 30 15:59:33 nut usbhid-ups[2861]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
Nov 30 15:59:38 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Nov 30 15:59:38 nut usbhid-ups[2861]: [D2] entering main_setvar(driver.debug, 0) for [eaton] on socket 10
Nov 30 15:59:38 nut usbhid-ups[2861]: [D1] NOTE: Will disable verbose debug now, due to socket protocol request
Nov 30 15:59:38 nut usbhid-ups[2861]: Applying debug level 0 received during run-time via socket protocol, ignoring other settings
Nov 30 16:09:24 nut usbhid-ups[2861]: Applying debug level 6 received during run-time via socket protocol, ignoring other settings
Nov 30 16:09:24 nut usbhid-ups[2861]: [D1] debug level is '6'
Nov 30 16:09:24 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.debug "6"
Nov 30 16:09:24 nut usbhid-ups[2861]: [D6] send_to_all: write 25 bytes to socket 10 succeeded (ret=25): SETINFO driver.debug "6"
Nov 30 16:09:30 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling PING with 1 args
Nov 30 16:09:30 nut usbhid-ups[2861]: [D2] send_to_one: sending PONG
Nov 30 16:09:30 nut usbhid-ups[2861]: [D5] send_to_one: PONG
Nov 30 16:09:30 nut usbhid-ups[2861]: [D6] send_to_one: write 5 bytes to socket 10 succeeded (ret=5): PONG
Nov 30 16:09:31 nut usbhid-ups[2861]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Nov 30 16:09:31 nut usbhid-ups[2861]: [D2] entering main_setvar(input.bypass.switch.on, on) for [eaton] on socket 10
Nov 30 16:09:31 nut usbhid-ups[2861]: [D2] shared main_setvar() does not handle variable input.bypass.switch.on, proceeding to driver-specific handler
Nov 30 16:09:31 nut usbhid-ups[2861]: [D1] setvar(input.bypass.switch.on, on)
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] hu_find_valinfo: found on (value: 1)
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:31 nut usbhid-ups[2861]: [D3] Report[set]: (2 bytes) => 59 01
Nov 30 16:09:31 nut usbhid-ups[2861]: [D4] Set report succeeded
Nov 30 16:09:31 nut usbhid-ups[2861]: [D5] setvar: SUCCEED
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO driver.state "updateinfo"
Nov 30 16:09:32 nut usbhid-ups[2861]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] upsdrv_updateinfo...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] nut_libusb_get_interrupt: Connection timed out
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] Got 0 HID objects (520 times in a row, tolerance is -1)...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D1] Full update...
Nov 30 16:09:32 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:09:32 nut usbhid-ups[2861]: [D3] Report[get]: (6 bytes) => 06 64 9f 06 00 00
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
Nov 30 16:09:32 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:09:32 nut usbhid-ups[2861]: [D3] Report[get]: (2 bytes) => 12 00
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.RemainingCapacityLimitSetting, Type: Feature, ReportID: 0x12, Offset: 0, Size: 8, Value: 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:09:32 nut usbhid-ups[2861]: [D3] Report[get]: (2 bytes) => 0e 00
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Unit = 00000000, UnitExp = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:09:32 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.RestartLevel, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 0
[... cut ...]
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] Exponent = 0
Nov 30 16:10:33 nut usbhid-ups[2861]: [D2] Path: UPS.PowerSummary.PercentLoad, Type: Feature, ReportID: 0x07, Offset: 56, Size: 8, Value: 33
Nov 30 16:10:33 nut usbhid-ups[2861]: [D5] send_to_all: SETINFO ups.load "33"
Nov 30 16:10:33 nut usbhid-ups[2861]: [D6] send_to_all: write 22 bytes to socket 10 succeeded (ret=22): SETINFO ups.load "33"
Nov 30 16:10:33 nut usbhid-ups[2861]: [D4] Entering libusb_get_report
Nov 30 16:10:33 nut usbhid-ups[2861]:

masterwishx, 2024-12-01:
> Ubuntu 24.04.1 LTS Using Grafana oversaw Input and Output Voltage difference, same time hearing click, also double checking in UPS Settings - In/Out Settings-HighEfficiencyMode checkmark appearing/disappearing while switching on/off
> 
> I found
> 
> `Nov 30 16:09:33 nut usbhid-ups[2861]: [D2] Path: UPS.PowerConverter.Input.[2].PresentStatus.Used, Type: Feature, ReportID: 0x38, Offset: 24, Size: 8, Value: 1`

Thanks, OK then but still strange you should have : 
`Path: UPS.PowerConverter.Input.[5].PresentStatus.Used, Type: Feature, ReportID: 0x3b, Offset: 32, Size: 8, Value: 1`

masterwishx, 2024-12-01:
@ALEXNETHUNTER if you are using grafana , check dashboard i made for NUT :  
https://grafana.com/grafana/dashboards/20846-nut-ups-telegraf/

ALEXNETHUNTER, 2024-12-01:
> @ALEXNETHUNTER if you are using grafana , check dashboard i made for NUT : https://grafana.com/grafana/dashboards/20846-nut-ups-telegraf/

I'm using exactly your dashboard :)

ALEXNETHUNTER, 2024-12-01:
> @ALEXNETHUNTER if you are using grafana , check dashboard i made for NUT : https://grafana.com/grafana/dashboards/20846-nut-ups-telegraf/

Also would be great to calculate separately Power Consumption and using Output watts in calculation field. That is I'm trying to do, no luck in influx queries so far (.
I have difference between input/output watts. Output is always more for 18-60 watts then Input. It's a real consumption of consumers. 
I think that is diffference because of a UPS double convertions. 

My scheme is: 

Input from Wall -> PowerStation -> UPS.  And PS shows on display IN/OUT for example 368 Watt.  Grafana shows Input 324 Watt / Output 368 Watt

masterwishx, 2024-12-01:
> Output is always more for 18-60 watts then Input. It's a real consumption of consumers.
> I think that is diffference because of a UPS double convertions.

Sure the inverter  of ups take some power .

> I have difference between input/output watts 

Not sure from where you got input power , powerstation ?

masterwishx, 2024-12-01:
in UPS we have Power (VA) and Real Power (W)

ALEXNETHUNTER, 2024-12-01:
> Not sure from where you got input power , powerstation ?
From UPS Load Graph - Watts stat (Real Power Rating * Ups Load)

ALEXNETHUNTER, 2024-12-01:
Aha, different is because Output Watts is calculating using output voltage witch is 230 Volts, and Real Power Rating is using RAW input Volts using nominal power that is calculating with real input volts from wall

masterwishx, 2024-12-01:
> UPS Load (Watts) - 297

It's not input, it's ups load % * real power rating if you will check it.
Mostly important I think real power, also power factor.
Pf show the difference of VA and watts.
So in eco mode pf is greater becose ups not loosing power on inverter.

ALEXNETHUNTER, 2024-12-01:
So when using UPS convertor, which is the most using cause of unstable wall input, in result we have not right power consumption and money statistic

masterwishx, 2024-12-01:
> we have not right power consumption and

It's right power consumption, just ups take some of it on by self, so when inverter is on more power needs for ups.
You can find more info on it in Google.

But you right it cost also money, so why I wanted to switch on ECO mode Becouse less power for ups itself.

Also it's depend in ups, my unit 9E2000i has 1600w real power and 2000va(w) power
So it has 0.8power factor when online, when 
I'm in ECO mode it has close to 0.9 power factor Becouse inverter is off but still it consume some power.

Your model 9sx should have better power factor so it loose less power.

ALEXNETHUNTER, 2024-12-13:
> Maybe Becouse you switched back to online very quickly?

I don't think so.

As I sad earlier, UPS has no any 'eco' parameter exept :  input.eco.switchable: ECO / (normal).
So Grafana nas no data nor upsc.

masterwishx, 2024-12-13:
> UPS.PowerConverter.Input.[5].PresentStatus.Used

but you have in log : `UPS.PowerConverter.Input.[5].PresentStatus.Used` = 0 

it should switch to =1 then will change status

ALEXNETHUNTER, 2024-12-13:
maybe wrong collected logs...but when I executed on/off commands I also checked the output:  input.eco.switchable: it was different ECO or normal

masterwishx, 2024-12-13:
> maybe wrong collected logs...but when I executed on/off commands I also checked the output: input.eco.switchable: it was different ECO or normal

Yes it's variable to switch input mode.
But then also should change status also.
I'm sorry to bother you just wanted to check if all working fine with ECO mode.
Becouse in my case not all fine, can't exit from eco mode to online :(

masterwishx, 2025-03-28:
@ZivertX as we added new commands for automatic ecomode on/off and other changes how eco mode work will be cool if you can check main as was merged.

As mentioned befor my 9E stuck in ecomode
#2719, so I can't test it.

If you have time please test if ecomode auto functions working fine.

Thanks

masterwishx, 2025-04-04:
> > [@ZivertX](https://github.com/ZivertX) as we added new commands for automatic ecomode on/off and other changes how eco mode work will be cool if you can check main as was merged.
> > As mentioned befor my 9E stuck in ecomode [#2719](https://github.com/networkupstools/nut/issues/2719), so I can't test it.
> > If you have time please test if ecomode auto functions working fine.
> > Thanks
> 
> Hi. I've updated from master. But my mind is blowing at the weekend and everyday news in my country. What exactly do you want me to do?

Just to check new functions:
`experimental.ecomode.start.auto`
`experimental.ecomode.stop.auto`
Like you was checked befor the ecomode:
https://github.com/networkupstools/nut/issues/2685#issuecomment-2509021983 
But now only by one command to enter and exit to/from ecomode. 

I'm feeling we are in same country :)

ZivertX, 2025-04-04:
`experimental.ecomode.start.auto`
 `experimental.ecomode.stop.auto` 

upsc command does not appear in those lines.

masterwishx, 2025-04-04:
@jimklimov ?

masterwishx, 2025-04-04:
> `experimental.ecomode.start.auto` `experimental.ecomode.stop.auto`
> 
> upsc command does not appear in those lines.

strange, are you sure you have latest master ? 
https://github.com/networkupstools/nut/blob/4f84c7dc650782da5d7047a93ddf036229df58f6/data/cmdvartab#L263-L264

masterwishx, 2025-04-04:
> `experimental.ecomode.start.auto` `experimental.ecomode.stop.auto`
> 
> upsc command does not appear in those lines.

if so ,can you post what commands list you have ?

jimklimov, 2025-04-04:
For commands, `upscmd` not `upsc`, e.g.
```
# upscmd -l eco650
Instant commands supported on UPS [eco650]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
...
```

Alas, the UPS I have here (Eaton Ellipse 650 ECO) is not of the type supported by this feature, I think.

masterwishx, 2025-04-04:
Yep its added to commands upscmd

ZivertX, 2025-04-04:
yes.

upscmd -l eaton | grep experimental.ecomode

```
experimental.ecomode.disable - Take the UPS out of High Efficiency (aka ECO) mode
experimental.ecomode.enable - Put UPS in High Efficiency (aka ECO) mode
experimental.ecomode.start.auto - Put UPS in Bypass mode then High Efficiency (aka ECO) mode
experimental.ecomode.stop.auto - Take the UPS out of High Efficiency (aka ECO) mode after exiting Bypass mode 
```

ZivertX, 2025-04-04:
Switch to bypass to see if eco mode activated automatically?

masterwishx, 2025-04-05:
> Did not get feature action
> 
> Switch to bypass to see if eco mode activated automatically or Switch to eco to see if bypass activated automatically?

No, that what is description say.
experimental.ecomode.start.auto will switch to bypass then to ecomode.

So we need to test if command when you run:
experimental.ecomode.start.auto
1. Will switch to bypass
2. Will switch to ecomode

So you don't need to run 2 commands like was befor(switch to bypass then to ecomode)

And also back to online :
experimental.ecomode.stop.auto
1.will switch from bypass
2.will switch from ecomode

So there is only one command to switch to ecomode and back to online

ZivertX, 2025-04-05:
```
nut@nut ~ % upsrw -u admin -p pass -s driver.debug=6 eaton                     
OK

nut@nut ~ % upscmd -u admin -p pass eaton experimental.ecomode.start.auto
OK

nut@nut ~ % upscmd -u admin -p pass eaton experimental.ecomode.stop.auto 
OK

nut@nut ~ % upsrw -u admin -p pass -s driver.debug=0 eaton               
OK

```

Full upc output:

<details>
<summary>Debug Logs</summary>

```
Apr 05 12:01:58 nut usbhid-ups[954]: [D5] send_to_one: PONG
Apr 05 12:01:58 nut usbhid-ups[954]: [D6] send_to_one: write 5 bytes to socket 10 succeeded (ret=5): PONG
Apr 05 12:02:05 nut usbhid-ups[954]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling PING with 1 args
Apr 05 12:02:05 nut usbhid-ups[954]: [D2] send_to_one: sending PONG
Apr 05 12:02:05 nut usbhid-ups[954]: [D5] send_to_one: PONG
Apr 05 12:02:05 nut usbhid-ups[954]: [D6] send_to_one: write 5 bytes to socket 10 succeeded (ret=5): PONG
Apr 05 12:02:05 nut usbhid-ups[954]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Apr 05 12:02:05 nut usbhid-ups[954]: [D2] entering main_setvar(driver.debug, 0) for [eaton] on socket 10
Apr 05 12:02:05 nut usbhid-ups[954]: [D1] NOTE: Will disable verbose debug now, due to socket protocol request
Apr 05 12:02:05 nut usbhid-ups[954]: Applying debug level 0 received during run-time via socket protocol, ignoring other settings
Apr 05 12:10:04 nut usbhid-ups[954]: Applying debug level 6 received during run-time via socket protocol, ignoring other settings
Apr 05 12:10:04 nut usbhid-ups[954]: [D1] debug level is '6'
Apr 05 12:10:04 nut usbhid-ups[954]: [D5] send_to_all: SETINFO driver.debug "6"
Apr 05 12:10:04 nut usbhid-ups[954]: [D6] send_to_all: write 25 bytes to socket 10 succeeded (ret=25): SETINFO driver.debug "6"
Apr 05 12:10:06 nut usbhid-ups[954]: [D5] send_to_all: SETINFO driver.state "updateinfo"
Apr 05 12:10:06 nut usbhid-ups[954]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"
Apr 05 12:10:06 nut usbhid-ups[954]: [D1] upsdrv_updateinfo...
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] nut_libusb_get_interrupt: Connection timed out
Apr 05 12:10:07 nut usbhid-ups[954]: [D1] Got 0 HID objects (3987 times in a row, tolerance is -1)...
Apr 05 12:10:07 nut usbhid-ups[954]: [D1] Quick update...
Apr 05 12:10:07 nut usbhid-ups[954]: [D4] Entering libusb_get_report
Apr 05 12:10:07 nut usbhid-ups[954]: [D3] Report[get]: (2 bytes) => 22 01
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Exponent = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] Path: UPS.BatterySystem.Charger.ABMEnable, Type: Feature, ReportID: 0x22, Offset: 0, Size: 8, Value: 1
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] ABM status is: enabled (1)
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Lookup [1] failed for [battery.charger.abm.status]
Apr 05 12:10:07 nut usbhid-ups[954]: [D4] Entering libusb_get_report
Apr 05 12:10:07 nut usbhid-ups[954]: [D3] Report[get]: (4 bytes) => 26 04 04 13
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Exponent = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] Path: UPS.BatterySystem.Charger.Mode, Type: Feature, ReportID: 0x26, Offset: 8, Size: 8, Value: 4
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Lookup [4] failed for [battery.charger.mode.status]
Apr 05 12:10:07 nut usbhid-ups[954]: [D3] Report[buf]: (4 bytes) => 26 04 04 13
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Exponent = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] Path: UPS.BatterySystem.Charger.Status, Type: Feature, ReportID: 0x26, Offset: 16, Size: 8, Value: 19
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Lookup [19] failed for [battery.charger.type.status]
Apr 05 12:10:07 nut usbhid-ups[954]: [D3] Report[buf]: (4 bytes) => 26 04 04 13
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Unit = 00000000, UnitExp = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] Exponent = 0
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] Path: UPS.BatterySystem.Charger.Mode, Type: Feature, ReportID: 0x26, Offset: 8, Size: 8, Value: 4
Apr 05 12:10:07 nut usbhid-ups[954]: [D2] ABM charger status is: resting (4)
Apr 05 12:10:07 nut usbhid-ups[954]: [D3] Report[buf]: (4 bytes) => 26 04 04 13
Apr 05 12:10:07 nut usbhid-ups[954]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
[... 934 lines cut ...]
Apr 05 12:10:52 nut usbhid-ups[954]: [D5] send_to_all: SETINFO driver.state "quiet"
Apr 05 12:10:52 nut usbhid-ups[954]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
Apr 05 12:10:56 nut usbhid-ups[954]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling INSTCMD with 2 args
**Apr 05 12:10:56 nut usbhid-ups[954]: [D2] entering main_instcmd(experimental.ecomode.stop.auto, (null)) for [eaton] on socket 10
Apr 05 12:10:56 nut usbhid-ups[954]: [D2] shared main_instcmd() does not handle command experimental.ecomode.stop.auto, proceeding to driver-specific handler
[... cut ...]
Apr 05 12:10:58 nut usbhid-ups[954]: [D6] send_to_one: write 5 bytes to socket 10 succeeded (ret=5): PONG
Apr 05 12:11:00 nut usbhid-ups[954]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling SET with 3 args
Apr 05 12:11:00 nut usbhid-ups[954]: [D2] entering main_setvar(driver.debug, 0) for [eaton] on socket 10
Apr 05 12:11:00 nut usbhid-ups[954]: [D1] NOTE: Will disable verbose debug now, due to socket protocol request
Apr 05 12:11:00 nut usbhid-ups[954]: Applying debug level 0 received during run-time via socket protocol, ignoring other settings

```
</details>

ZivertX, 2025-04-05:
But after all USP still in eco mode (see in graphs), input and output aligns

```
Broadcast message from nut@nut (somewhere) (Sat Apr  5 12:15:46 2025):         
                                                                               
UPS eaton@localhost: in ECO mode (as defined by vendor)
```

ZivertX, 2025-04-05:
disabled ecomode using: 
`upscmd -u admin -p adminpass eaton experimental.ecomode.disable`

masterwishx, 2025-04-05:
> But after all USP still in eco mode (see in graphs), input and output aligns
>
> ```
> Broadcast message from nut@nut (somewhere) (Sat Apr  5 12:15:46 [...]

Thanks for helping, its better to have the full log.

@jimklimov  Maybe you can help here, im not sure why we have: 
```
Apr 05 12:10:10 nut usbhid-ups[954]: [D1] instcmd(experimental.ecomode.start.auto, [NULL])**
Apr 05 12:10:10 nut usbhid-ups[954]: [D3] instcmd: using Path 'UPS.PowerConverter.Input.[5].Switchable'
Apr 05 12:10:10 nut usbhid-ups[954]: [D3] hu_find_valinfo: no matching HID value for this INFO_* value (1)
...

Apr 05 12:10:52 nut usbhid-ups[954]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
Apr 05 12:10:56 nut usbhid-ups[954]: [D6] sock_arg: Driver on /var/state/ups/usbhid-ups-eaton is now handling INSTCMD with 2 args
**Apr 05 12:10:56 nut usbhid-ups[954]: [D2] entering main_instcmd(experimental.ecomode.stop.auto, (null)) for [eaton] on socket 10
Apr 05 12:10:56 nut usbhid-ups[954]: [D2] shared main_instcmd() does not handle command experimental.ecomode.stop.auto, proceeding to driver-specific handler
Apr 05 12:10:56 nut usbhid-ups[954]: [D1] instcmd(experimental.ecomode.stop.auto, [NULL])**
Apr 05 12:10:56 nut usbhid-ups[954]: [D3] instcmd: using Path 'UPS.PowerConverter.Input.[5].Switchable'
Apr 05 12:10:56 nut usbhid-ups[954]: [D3] hu_find_valinfo: no matching HID value for this INFO_* value (0)

```
Also i cant see when switching to Bypass before ecomode

masterwishx, 2025-04-05:
> graphs while testing:

Did you saw maybe when ups entered to bypass befor enter to ecomode?
When used `experimental.ecomode.start.auto`

@jimklimov do you see something unusual in debug?

ZivertX, 2025-04-05:
> Did you saw maybe when ups entered to bypass befor enter to ecomode?

It's really strange behaviour.

12:10:04  Applied debug level 6

then at

12:10:10  hit in cli ecomode.start.auto
12:10:56  hit in cli ecomode.stop.auto

12:11:00 applied debug level 0 (disabled)

and at 12:15:46 i saw console message but didn't thought that ups could be in eco/bypass mode
after about 20 minutes I checked the graphs and realized that ups somehow is working in that mode, then disabled it.
All clocks are synced.

[33 more replies]

PR #2698 (2024-11-30): Eaton 9SX/9PX Fix for battery.voltage
Eaton 9SX/9PX Fix for battery.voltage and battery.voltage.nominal by adding 9SX/9PX model to mge_model_names []
Fix #2685