networkupstools/nut #2690: Incorrect output Voltage Phasak Protekt 7621 2160va

PedroCavaleiro, 2024-11-24:
I'm running the latest version of NUT and on my Phasak UPS (Phasak Protekt 7621 2160va) the output voltage seems to be wrong.
This is the output by `upsc phasak2160@localhost` command.

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1
battery.runtime: 1080
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 26.0
battery.voltage.nominal: 24
device.mfr: 1
device.model: 2160
device.serial: 000000000000
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.serial: 000000000000
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 50.0
input.transfer.high: 290
input.transfer.low: 165
input.voltage: 246.0
input.voltage.nominal: 230
output.frequency: 50.0
output.voltage: 15.6
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 19
ups.mfr: 1
ups.model: 2160
ups.productid: 0601
ups.realpower.nominal: 1400
ups.serial: 000000000000
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

Although I only intend to really monitor the following values
```
battery.charge: 100
battery.voltage: 26.0
input.voltage: 246.0
output.voltage: 15.6 <-- incorrect value
ups.load: 19
ups.status: OL
```
The output as I marked is incorrect, I assume this might be other sensor, the LCD on my UPS measures the output voltage correctly so I'm assuming is something to do with the driver

What steps can I take to further debug this and eventually fix it 

This is my ups.conf
```
[phasak2160]
        driver = usbhid-ups
        port = auto
        desc = "Phasak Protekt 7621 2160VA 1400W"
        vendorid = 0764
        productid = 0601
        serial = 000000000000
```

jimklimov, 2024-11-24:
2.8.1 is actually not the latest NUT version - maybe one last packaged by your distro? Current release is 2.8.2, and half a year of development on master branch since. Can you try building the current code to check how it behaves (and also become prepared to tweak that source while troubleshooting the issue)?

Check wiki article on "in-place" builds. You can run the resulting driver from the build workspace then, without overwriting anything in the system.

jimklimov, 2024-11-24:
Just in case, did any other versions of NUT report the value better (so we can compare code differences)?

CPS mapping may also compensate for broken HID encoding (LogMin/LogMax issue #1024). On one hand, maybe on your device you should turn off the fix if it misfires. On another - maybe make sure it does try to fix the broken descriptors.

Bump up driver debugging verbosity to check for more details (see also wiki).

PedroCavaleiro, 2024-11-24:
I will try upgrade the NUT version then I’m gonna look into the CPS mapping and debug the driver
If anything I’ll try the in-place build at this stage I might some help 

I’ll report back with my findings as soon as I replace one of the batteries that I found out was leaking

PedroCavaleiro, 2025-05-01:
Sorry for the delay but it took way to long for me to get the batteries required the supplier took a lot of time, however just today I tested the voltage it appears to be correct on version 2.8.3

I'll reopen if needed

closed 2025-05-01 by PedroCavaleiro