networkupstools/nut #2691: RIello IDG 1600 - Driver shows partial data

ddppddpp, 2024-11-24:
My Riello IDG 1600 doesn't show battery capacity and therefore is can't be used for a reasonable power down automation. I've tested this on various bsd/ubuntu/debian boxes and with an in-place build from current source code.
Current setup - Proxmox VE box with testing package 2.8.1-3.3
Output
```
upsc ups@localhost
Init SSL without certificate database
battery.capacity: 9
battery.voltage: 27.0
battery.voltage.nominal: 24
device.mfr: RPS S.p.a.
device.model: UNV1           
device.serial:                 
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.1
driver.version.internal: 0.11
driver.version.usb: libusb-1.0.26 (API: 0x100010a)
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.10
input.voltage: 230
output.frequency: 50.10
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 229
output.voltage.nominal: 230
ups.delay.reboot: 5
ups.delay.shutdown: 5
ups.firmware: SWM038-02-02
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: UNV1           
ups.power.nominal: 1600
ups.productid: 5500
ups.realpower.nominal: 960
ups.serial:                 
ups.status: OL
ups.vendorid: 04b4
```
configuration
```
[ups]
  driver = riello_usb
  port = auto
  desc = "Riello IDG 1600"
  #vendorid = "04B4"
  #productid = "5500"
```

jimklimov, 2024-11-24:
Some other readings are way off too (e.g. 4096-1=4095).

Can you run the driver with bumped debug verbosity to check what it gets
for those fields? Check in source if it has battery.charge at all? You may
have to estimate from battery.voltage otherwise (e.g. port "guesstimation"
from other drivers; I think some tickets on that were backlogged).

jimklimov, 2024-11-28:
Also, quite many devices do not report charge (level nor voltage) regardless of computer-side software, but can raise an alarm status to require an imminent shutdown based on some criteria built into the controller (which may be or not be in our control as e.g. settable thresholds).

Did you check if NUT with your UPS actually initiates a shutdown when battery runs low?

Note that (as detailed in FAQ I think) you can simulate a shutdown by setting a non-disruptive `SHUTDOWNCMD` in `upsmon.conf` (e.g. to `echo` a message to `wall` or a log file), and watch the UPS yourself to re-plug it to avoid the loss of power to the server (or not run any useful load from the UPS while testing).

For complex shutdowns, e.g. some systems going down earlier or later based on on-battery time, you would need to study `upssched` as the handler called from `upsmon` on each client.