networkupstools/nut #2695: Error: Driver not connected

fedewalsyn1993, 2024-11-26:
Hi,
I am running the program on Windows Server 2022 and I only get this message

"PS C:\Program Files (x86)\NUT\bin> .\upsc.exe ups
Error: Driver not connected"

[image]

MODE = standalone

My configuration
[ups]
        driver = "usbhid-ups.exe"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        product = "Smart-UPS 1500 RM FW:667.19.I USB FW:11.03"
        serial = "AS1237232001"
        vendor = "American Power Conversion"
        # bus = "002"
        # device = "001"

I used zadig to install 4 drivers : libusbk, libusb-win32 (v1.4.0.0), WinUSB (v6.1.7600.16385), USB Serial (CDC)

[image]

[image]

I am waiting for your response
thanks

jimklimov, 2024-11-26:
Can you please try starting the driver (stop the NUT service and run `usbhid-ups -DDDDDD -d1 -a ups`) to see what it complains about?

It may also help to simplify the matching (leave only IDs without strings, if you only have one UPS there).

jimklimov, 2024-11-26:
Well, that much seemed to work - lots of meaningful SETINFO events. Not sure why data dump did not get into log, maybe something missed about Windows buffer flushing?..

So the next question is if the driver starts (and keeps working) as part of the service. And if `upsd` sees it.

Try adding `debug_min=6` to `ups.conf` (and maybe `DEBUG_MIN 3` to `upsd.conf`), I think service messages can end up in Event Log.

fedewalsyn1993, 2024-11-27:
I used this
[image]
it don´t start a service 

I saw this on event visor of windows

Erroring application name: usbhid-ups.exe, version: 0.0.0.0, timestamp: 0x6745b2fe
Failed module name: libusb-1.0.dll, version: 1.0.27.11882, timestamp: 0x65c4d7a3
Exception code: 0xc0000005
Error offset: 0x0000000000002358
Failed process identifier: 0xf0
Failed application start time: 0x01db40c6cc7172fa
Erroring application path: C:\Program Files (x86)\NUT\bin\usbhid-ups.exe
Failed module path: C:\Program Files (x86)\NUT\bin\libusb-1.0.dll
Report ID: 3bcc3d44-f540-44b6-9b19-aba29777bd84
Full name of the errored package: 
Relative application identifier of the failed package:

I configured this "Try adding debug_min=6 to ups.conf (and maybe DEBUG_MIN 3 to upsd.conf)"
and this is de result

PS C:\Program Files (x86)\NUT\bin> .\usbhid-ups.exe -a ups
Network UPS Tools 2.8.2.1303.38-1341-g6dea4d4dd (development iteration after 2.8.2) - Generic HID driver 0.58
USB communication driver (libusb 0.1) 0.48
   0.000000     [D1] debug level is '6'
   0.001143     [D5] send_to_all: SETINFO driver.debug "6"
   0.002855     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.005332     [D1] Can not become_user(<null>): not implemented on this platform
   0.009867     [D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.014725     WaitNamedPipe : 2
: No error [El sistema no puede encontrar el archivo especificado. ]
   0.025118     [D1] Request for other driver to exit returned code -1
   0.028200     [D1] Socket dialog with the other driver instance (may be absent) failed: No error [El sistema no puede encontrar el archivo especificado. ]
   0.033432     [D5] send_to_all: SETINFO device.type "ups"
   0.037564     [D5] send_to_all: SETINFO driver.state "init.device"
   0.041432     [D1] upsdrv_initups (non-SHUT)...
   0.042899     [D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.48')
   0.198858     [D3] usb_busses=000001F672EF7D30
   0.201243     [D2] Checking device (8086/1C26) (001/000)
   0.202888     [D1] Failed to open device (8086/1C26), skipping: Function not implemented
   0.205895     [D2] Checking device (8087/0024) (001/001)
   0.207598     [D1] Failed to open device (8087/0024), skipping: Function not implemented
   0.210447     [D2] Checking device (8087/0024) (003/001)
   0.213576     [D1] Failed to open device (8087/0024), skipping: Function not implemented
   0.216750     [D2] Checking device (8086/1C2D) (003/000)
   0.218940     [D1] Failed to open device (8086/1C2D), skipping: Function not implemented
   0.222485     [D2] Checking device (1033/0194) (002/000)
   0.224706     [D1] Failed to open device (1033/0194), skipping: Function not implemented
   0.231003     [D2] Checking device (051D/0002) (002/001)
   0.270192     [D2] - VendorID: 051d
   0.271892     [D2] - ProductID: 0002
   0.273213     [D2] - Manufacturer: American Power Conversion
   0.276404     [D2] - Product: Smart-UPS 1500 RM FW:667.19.I USB FW:11.03
   0.278715     [D2] - Serial Number: AS1237232001
   0.280244     [D2] - Bus: 002
   0.281239     [D2] - Device: 001
   0.282242     [D2] - Device release number: 0006
   0.284453     [D2] Trying to match device
   0.285680     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.288104     [D3] match_function_regex: matching a device...
   0.290546     [D2] Device matches
   0.293958     [D3] nut_libusb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.300190     [D2] Retrieved HID descriptor (expected 9, got 9)
   0.302365     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 28 04
   0.304765     [D3] HID descriptor length (method 1) 1064
   0.306345     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.309035     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 28 04
   0.312151     [D3] HID descriptor length (method 2) 1064
   0.313910     [D2] HID descriptor length 1064
   0.450125     [D2] Report Descriptor size = 1064
[... 2024 lines cut ...]
   8.223266     [D5] hu_find_infoval: found !dischrg (value: 0)
   8.224993     [D5] process_boolean_info: !dischrg
   8.226424     [D3] Report[buf]: (3 bytes) => 07 0c 00
   8.227827     [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   8.229770     [D5] Unit = 00000000, UnitExp = 0
   8.231042     [D5] Exponent = 0
   8.231941     [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x07, Offset: 0, Size: 1, Value: 0
   8.240591     [D5] hu_find_infoval: found !chrg (value: 0)
   8.242328     [D5] process_boolean_info: !chrg
   8.243587     [D3] Report[buf]: (3 bytes) => 07 0c 00
   8.245083     [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   8.246945     [D5] Unit = 00000000, UnitExp = 0
   8.248272     [D5] Exponent = 0
   8.250126     [D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 4, Size: 1, Value: 0
   8.255333     [D5] hu_find_infoval: found !lowbatt (value: 0)
   8.257080     [D5] process_boolean_info: !lowbatt
   8.258533     [D5] send_to_all: SETINFO ups.status "OL"
   8.260032     [D5] send_to_all: DATAOK
   8.261144     [D5] send_to_all: SETINFO driver.state "init.quiet"
   8.263079     [D2] dstate_init: sock \\.\pipe\usbhid-ups-ups open on handle 00000000000000A8
   8.269745     [D5] send_to_all: SETINFO driver.parameter.pollinterval "2"
   8.272070     [D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   8.274278     [D5] send_to_all: SETINFO device.mfr "American Power Conversion"
   8.276434     [D5] send_to_all: SETINFO device.model "Smart-UPS 1500 RM"
   8.278342     [D5] send_to_all: SETINFO device.serial "AS1237232001"

jimklimov, 2024-11-28:
> I used <driver start command>, it don´t start a service

Yes, it did only start the driver, run one data collection cycle, and exit its program.

The Windows service can be registered or unregistered with `nut.exe` wrapper program, which is then also used as the Windows service handler to typically start driver(s), data server `upsd` and client `upsmon` (or just `upsmon` on non-NUT-server systems). The service lifecycle is managed by Windows commands then, like `net start/stop`, or by Service manager UI clicking.

> what do you think? is this correct?

Well, if the driver still fails to start as part of service - this is not fully correct.
If it does start standalone (as your user account), this is not a device compatibility or NUT config problem however.

Can you check if after stopping the manually launched driver and restarting the NUT Windows service, the driver still fails to start?
* if "yes" - maybe it is something about the context the NUT service runs in, might not have HW access or something; if debugging is still enabled, Event viewer may have driver debug log messages leading up to the crash - notably, your earlier message indicated "Failed module name: libusb-1.0.dll, version: 1.0.27.11882" so it may be some work in the library we use going wrong depending on circumstances.
* if "no" (e.g. service works now) - can you post how the configuration files changed overall? Maybe there is some issue solved by enabling debugging (is that the only change?), there was sadly a precedent about this. I believe your config change did not include commenting away device-matching criteria (e.g. detailed vendor/product strings - does not seem to be the case per `upsc` output), right?

closed 2025-04-03 by jimklimov