networkupstools/nut #2710: Phoenix Contact firmware version is not consistent on same model

RikyPlaza, 2024-12-04:
Hello,
I noticed that the firmware version modbus register value is not consistent on different devices of the same model:

-	QUINT4 device 1 -> 305
-	QUINT4 device 2-> 513
-	QUINT device 1-> 544
-	QUINT device 2 -> 592

I'm already in contact with Phoenix Contact to establish if there is a specific range of firmwares or another way to make the driver understand which model is connected.

RikyPlaza, 2024-12-05:
I've also noticed that even if UPSModel variable is not set the driver will use the QUINT addresses. This because in the models enum QUINT is 0 and the UPSModel var is initialized to 0. I'll add a NONE value in the enum and manage it in all switch statements.

jimklimov, 2024-12-10:
Thanks for the follow-up!

RikyPlaza, 2024-12-11:
Update: A new detection way has been found. Development is in progress.
I've also noticed that i made a mistake reading the alarms registers (two readings for same register).

RikyPlaza, 2024-12-11:
Opened pull request #2716.

RikyPlaza, 2024-12-17:
@jimklimov Hello, from my side everything is ok.

Thank you!

closed 2024-12-17 by RikyPlaza