networkupstools/nut #2714: APC plugged into ASUSTOR using NUT server without credentials.

GuruSR, 2024-12-08:
Home Assistant can see the APC through the ASUSTOR no problem, the NUT-Monitor from lestat.st could also see the APC from the ASUSTOR on the same Raspberry Pi 4B where this client cannot because it seems the client requires an actual login, but the ASUSTOR only lists LAN1 and the IP address of itself, nowhere is there any information on the login, though defaults of asustor 11111 don't work.  The NUT-Monitor proved there is a lack of login required to talk to it, unsure if it is using a guest sign in or not.

If it can bypass a login, is there a means to do this?  I've looked through the documentation, everywhere lists requiring a username and password, whereas this doesn't seem to be the case with the ASUSTOR.

Would really like to get this setup (if possible a UI capable of showing status or a conky means), since it isn't plugged into the system attempting to talk to it.

jimklimov, 2024-12-09:
Can you please rephrase the question into more shorter sentences?

Some clarification questions OTOH:

* Into which machine is the UPS data cable plugged? Is it USB, Serial, Networked (SNMP) etc.?
* Do you have control over that machine to see the NUT configuration files?
* Which version of NUT is involved where?
* Is the mentioned Raspberry a client or the NUT server?
* What do you want to do with the UPS that has you concerned about logins? Specifically, data-reading is anonymous. Logins are needed for managerial operations (set device/driver variables, run instant commands) or for a client like `upsmon` to subscribe for a long session.

Also note that the NUT-Monitor was contributed to NUT codebase a decade ago, so is maintained and available in https://github.com/networkupstools/nut/tree/master/scripts/python (other forks of the lib and/or app that went other ways also exist).

GuruSR, 2024-12-10:
Clarifications:

- The UPS unit is plugged into the ASUSTOR Raid via USB as it says in the description.
- The NUT configuration on the ASUSTOR is "on" or "off", it lists "LAN1" as the "?name?", unsure, no documentation I could find on it.
- Version 2.7.4-13.
- The Raspberry Pi is a client as the ASUSTOR is running the NUT server for it to see, the ASUSTOR's UPS is powering the Raspberry Pi, why I was using the NUT-Monitor to keep an eye on it, because an earlier version of the nut-client refused to work without credentials (read the man pages on this, very minimal information).
- The NUT-Monitor (was in use until I upgraded linux versions), was able to see the ASUSTOR's provided UPS without credentials.  I want to do the same with nut-client, with the hopes of also adding stats to conky on-screen (if possible).
- I am not looking for a repeat of NUT-Monitor (even if it is being maintained), as I want to reduce what is already running, as I have quite a lot of services going now, I am looking to use the nut-client without credentials, but don't see any instructions on how to proceed.

jimklimov, 2024-12-12:
One more clarification: by `nut-client` you mean the service into which the `upsmon` daemon is wrapped? Different packaging varies creatively about that, so better to ask. 

There are tons of documentation on this one. IIRC it requires credentials by current docs, although the code may still support a legacy `MONITOR` line format. Maybe you can get away passing empty strings `""` as user name and password.

GuruSR, 2024-12-12:
Using nut-client, not server (nut-server isn't running).

I did try the monitor with "" for both credentials and it fails, as much as I've tried, I couldn't get the nut-client to start without credentials.  What legacy monitor line format have you seen?

jimklimov, 2024-12-12:
Looking at current code, at https://github.com/networkupstools/nut/blob/master/clients/upsmon.c#L1711-L1717 - now all 5 arguments of the `MONITOR` line are required in practice. Also per https://github.com/networkupstools/nut/blob/master/clients/upsmon.c#L537 the `LOGIN` is always tried.

jimklimov, 2024-12-12:
Note: what I assume you mean by `nut-client(.service)` is a wrap for `upsmon` client program that is primarily responsible for shutdown of the host it runs on. For other monitoring you'd want other clients, e.g. marry `upsc` and MRTG, or keep a trail of `upslog`, or put a browser showing `upscgi` renditions on the conky screen...

jimklimov, 2024-12-12:
> Version 2.7.4-13

That's on ASUSTOR? Or on Raspberry? Either way, note 2.7.4 is from March 2016...

GuruSR, 2024-12-16:
I was finally able to get the nut-client service to see it.  I think, upsc shows it, so it should be fine now, just have to deal with it not setting a pid.
Hmm, I commented out the PID file so that it isn't being created, now no errors!