networkupstools/nut #2717: [HCL] Cyber Power System UT1050EG supported by usbhid-ups

strainu, 2024-12-11:
```
$ upsc upsb
Init SSL without certificate database
Error: Unknown UPS
admin@jarod[~]$ upsc ups
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 5044
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: UT1050EG
device.serial:
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 499.0
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 233.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 8
ups.mfr: CPS
ups.model: UT1050EG
ups.productid: 0501
ups.realpower.nominal: 630
ups.serial:
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

```
$ upsrw ups
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 295

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 167

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```

```
$ upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

Have yet to figure out how to run tests, since I have it on a Truenas server, which do not configure the user correctly.

strainu, 2024-12-12:
Note that, while not tested, the following models should also work:
* UT1050EIG
* UT1200EG
* UT1200EIG

jimklimov, 2024-12-16:
NOTE: Input frequency seems off by a factor of 10 (499.0 I guess should be in ~50Hz range), but voltages seem surprisingly correct :) in regard to issues addressed in #2718, #439, #169... so maybe there are different firmwares with same vendor/product IDs in play, that we may want to differentiate.

Can you please check if fixes in current NUT codebase (2.8.2 or master, particularly a custom build of PR #2718) do not actually break input/output voltage reporting on your device?

Stasiz, 2025-01-05:
I confirm that it works fine with UT650EG.
**Config:**
```
$ cat /etc/ups/ups.conf
maxretry = 3
[cyberpower]
  driver = usbhid-ups
  port = auto
  vendorid = 0764
  productid = 0501
  desc = "UPS CyberPower UT650EG"
```
**OS and package:**
```
$ cat /etc/os-release
NAME="Oracle Linux Server"
VERSION="8.10"
ID="ol"
ID_LIKE="fedora"
VARIANT="Server"
VARIANT_ID="server"
VERSION_ID="8.10"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Oracle Linux Server 8.10"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:oracle:linux:8:10:server"
HOME_URL="https://linux.oracle.com/"
BUG_REPORT_URL="https://github.com/oracle/oracle-linux"
ORACLE_BUGZILLA_PRODUCT="Oracle Linux 8"
ORACLE_BUGZILLA_PRODUCT_VERSION=8.10
ORACLE_SUPPORT_PRODUCT="Oracle Linux"
ORACLE_SUPPORT_PRODUCT_VERSION=8.10
```
```
$ rpm -qa nut
nut-2.8.2-1.el8.x86_64
```
**Native app:**
```
$ pwrstat -version
version:
pwrstat version 1.4.1
```
```
$ pwrstat -status
The UPS information shows as following:

        Properties:
                Model Name................... UT650EG
                Firmware Number.............. BF01701GC11.y
                Rating Voltage............... 230 V
                Rating Power................. 360 Watt

        Current UPS status:
                State........................ Normal
                Power Supply by.............. Utility Power
                Utility Voltage.............. 236 V
                Output Voltage............... 238 V
                Utility Frequency............ 499.0 Hz
                Battery Capacity............. 100 %
                Remaining Runtime............ 38 min.
                Load......................... 57 Watt(16 %)
                Line Interaction............. None
                Test Result.................. Unknown
                Last Power Event............. None
```
**NUT:**
```
$ upsc cyberpower
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 2172
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.7
battery.voltage.nominal: 12
device.mfr: CPS
device.model: UT650EG
device.serial:
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: updateinfo
driver.version: 2.8.2
driver.version.data: CyberPower HID 0.80
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
input.frequency: 499.0
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 236.0
input.voltage.nominal: 230
output.voltage: 238.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 15
ups.mfr: CPS
ups.model: UT650EG
ups.productid: 0501
ups.realpower.nominal: 360
ups.serial:
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
```
$ upsrw -l cyberpower
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[driver.debug]
Current debug verbosity level of the driver program
Type: NUMBER
Value: 0

[driver.flag.allow_killpower]
Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
Type: NUMBER
Value: 0

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 295

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 167

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```
```
$ upscmd -l cyberpower
Instant commands supported on UPS [cyberpower]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

jimklimov, 2025-01-05:
@Stasiz : Thanks for the report, especially for the bit about `pwrstat` - interesting to see they also see frequency as 499 HZ (presuming native value is off by 10x, so 49.9).

Stasiz, 2025-01-05:
**nut-issue-2717**
```
$ upsc cyberpower
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 10734
battery.runtime.low: 300
battery.status: 100%
battery.type: PbAcid
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: CPS
device.model: UT650EG
device.serial:
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.7.4.11104.1-11105-g75efd0c5e
driver.version.data: CyberPower HID 0.83
driver.version.internal: 0.60
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
input.frequency: 50.0
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 238.0
input.voltage.nominal: 230
output.voltage: 240.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 120
ups.firmware: BF01701GC
ups.load: 0
ups.mfr: CPS
ups.model: UT650EG
ups.power.nominal: 0
ups.productid: 0501
ups.realpower.nominal: 360
ups.serial:
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

jimklimov, 2025-01-05:
Thanks! This looks better :)

jimklimov, 2025-01-05:
Strange that some numbers line `ups.load` and `battery.runtime` look so different (although the change is somewhat consistent). Did you disconnect the load while testing?

Stasiz, 2025-01-05:
Yes, I tested it without load.

PR #2752 (2025-01-05): drivers/cps-hid.c, NEWS.adoc: try to fix frequency scaling dynamically
Closes: #2717

Some CPS devices report frequency off by a factor of 10 (also in the vendor's own `pwrstat` tool) as e.g. `499.0`. This PR should try to detect the scaling range and adjust the reading.

EDUCATED GUESS: Probably the underlying problem is a yet another mis-use of USB HID report format, with an interim change to reporting in 0.1HZ units in some firmwares, but not updating the exponent in the report.
closed 2025-01-05 by jimklimov