networkupstools/nut #2719: Eaton 9E2000i (USB) can't return to Online from ECO mode

masterwishx, 2024-12-13:
Using Unraid 6.12.14 plugin + lasted master branch compiled. Libusb 1.0.26

Maybe related to libusb version or to bug in firmware of ups.

When I made pr #2637 for add bypass /ECO I was enabled ECO mode by 3 commands:

bypass.on
ecomode.enable
bypass.off

Then after Blackouts throughout the house for several hours ups returned to online.

Then enabled again same way now can't exit to online.

So now I have both
 input.bypass.switch.on = disabled
 input.bypass.switch.off = disabled

However I got Success on commands can't enter to bypass or exit eco. Maybe related to #2644
Also can't switch to bypass by buttons on ups. 

Tryed to power off ups for some hours, but then start to eco mode again. 

@arnaudquette-eaton I'm very sorry to bother you, but if you would you like to help, will cool if you can check if it bug in firmware again in 9E or some other issue. 

Becouse no buttons on ups to enable eco mode, when I buyed ups and asked Eaton support how to enable eco mode they told me to buy snmp card....

Tested by user on Eaton 9sx in https://github.com/networkupstools/nut/issues/2685#issuecomment-2509021983

bypass.on + ecomode.enable = enable eco
bypass.off + ecomode.disable = back to online

jimklimov, 2024-12-13:
@masterwishx : Just wanted to confirm the instant commands you use, Can't see `bypass.on` in code, and there are `bypass.start` and `bypass.stop` in `docs/nut-names.txt`.

masterwishx, 2024-12-13:
> @masterwishx : Just wanted to confirm the instant commands you use, Can't see `bypass.on` in code, and there are `bypass.start` and `bypass.stop` in `docs/nut-names.txt`.

Yes sorry (that was from memory) I will change it. I used /tryed  both commands and r/w variables

masterwishx, 2025-02-03:
@arnaudquette-eaton Very sorry to ask you again but Please if you will have time to check if there is issue in Eaton 9E firmware for USB , i can't enter back to Online from ECO Mode after all work was Done, if you check posts above :  

Have now in ECO Mode Enabled : 
```
"UPS.PowerConverter.Input.[2].SwitchOnControl" = 0
"UPS.PowerConverter.Input.[2].SwitchOffControl" = 0
```

So sequence of `input.bypass.switch.off=off + input.eco.switchable=normal  -> back to online` that was tested on 9SX user is not working for 9E , and im stuck on ECO Mode...

```
"UPS.PowerConverter.Input.[2].SwitchOffControl" = 1 + "UPS.PowerConverter.Input.[5].Switchable"=0
```

masterwishx, 2025-03-17:
@jimklimov  , sorry for offtop .
i made a page `https://github.com/networkupstools/nut/wiki/NUT-in-Grafana-Dashboard`
if you can check when you have time if looks fine. 
Thanks

jimklimov, 2025-03-17:
> [@jimklimov](https://github.com/jimklimov) , sorry for offtop . i made a page `https://github.com/networkupstools/nut/wiki/NUT-in-Grafana-Dashboard` if you can check when you have time if looks fine. Thanks

Looks good, thanks! I've updated the name to group with "Monitoring with MRTG" so solutions are nearby in the list, so the URL became https://github.com/networkupstools/nut/wiki/Monitoring-NUT-with-Grafana-Dashboard

And updated the Markdown a bit, so the lists look like lists :)

masterwishx, 2025-03-17:
> Looks good, thanks! I've updated the name to group with "Monitoring with MRTG"

Yep i saw only after added it , so was no time to update it. Thanks

masterwishx, 2025-10-27:
> Using Unraid 6.12.14 plugin + lasted master branch compiled. Libusb 1.0.26
> 
> Maybe related to libusb version or to bug in firmware of ups.
> 
> When I made pr [#2637](https://github.com/networkupstools/nut/pull/2637) for add bypass /ECO I was enabled ECO mode by 3 commands:
> 
> ```
> bypass.start                    "UPS.PowerConverter.Input.[2].SwitchOnControl" = 1
> ecomode.enable                  "UPS.PowerConverter.Input.[5].Switchable" = 1
> bypass.stop                     "UPS.PowerConverter.Input.[2].SwitchOffControl" = 1
> ```
> 
> or (with same HID Paths and hid_values)
> 
> ```
> input.bypass.switch.on=on + input.eco.switchable=ECO + input.bypass.switch.off=off
> ```
> 
> Then after Blackouts throughout the house for several hours ups returned to online.
> 
> Then enabled again same way now can't exit to online.
> 
> Seems we can switch by `input.eco.switchable` to ECO/normal only when `input.bypass.switch.on=on` or `input.bypass.switch.off=off`.
> 
> but now I have both:
> 
> ```
>  input.bypass.switch.on = disabled
>  input.bypass.switch.off = disabled
> ```
> 
> However I got Success on commands can't enter to bypass or exit eco. Maybe related to [#2644](https://github.com/networkupstools/nut/issues/2644) Also can't switch to bypass by buttons on ups.
> 
> Tryed to power off ups for some hours, but then start to eco mode again.
> 
> [@arnaudquette-eaton](https://github.com/arnaudquette-eaton) I'm very sorry to bother you, but if you would you like to help, will cool if you can check if it bug in firmware again in 9E or some other issue.
> 
> Becouse no buttons on ups to enable eco mode, when I buyed ups and asked Eaton support how to enable eco mode they told me to buy snmp card....
> 
> Tested by user on Eaton 9sx in [#2685 (comment)](https://github.com/networkupstools/nut/issues/2685#issuecomment-2509021983)
> 
> that looks more right:
> 
> ```
> input.bypass.switch.on=on + input.eco.switchable=ECO  -> enable ecomode
> input.bypass.switch.off=off + input.eco.switchable=normal  -> back to online
> ```

@jimklimov any chance that @ericclappier-eaton can help here?

masterwishx, 2025-11-10:
@ericclappier-eaton do you have any news maybe?

masterwishx, 2026-09-21:
> hey there, sorry for the silence. I'm now too far away from Opensource, including at work. [@ericclappier-eaton](https://github.com/ericclappier-eaton) can you forward this to the right (9E) team, or at worst through the support?

Hi , it seems no answer from him about year, in case you will have time to send it to (9E) team or support will be cool .
I have also no answer from Eaton support at [EatonCareIL@Eaton.com](mailto:EatonCareIL@Eaton.com) about year ago :)