networkupstools/nut #2720: DDLs for Ippon Back Basic series (650/850 + 1500/2200)

dvdesolve, 2024-12-13:
Personally I have experience with 650, 850, 1500 and 2200 models. According to the [website](https://ippon.ru/catalog/item/ippon-650-Euro-383323/) there are a bunch of models which can be grouped like this:
1. 650 VA (Schuko CEE7 and IEC320 С13 outlets, 1 battery 7.2 Ah)
2. 850 VA (Schuko CEE7 and IEC320 С13 outlets, 1 battery 9 Ah)
3. 1050 VA (Schuko CEE7 and IEC320 С13 outlets, 1 battery 9 Ah)
4. 1200 VA + LCD (Schuko CEE7 outlets, 2 batteries 7.2 Ah)
5. 1500 VA (Schuko CEE7 and IEC320 С13 outlets, 2 batteries 9 Ah)
6. 2200 VA (Schuko CEE7 and IEC320 С13 outlets, 2 batteries 9 Ah)

They differ in nominal power ratings, number and form-factor of outlets, charging times, number of batteries and some minor details. Seems like 650/850/1050 models are just siblings with different power ratings, the same true for 1500/2200. 1200 LCD model stands separate from them at the first glance.

IIRC 650/850 and 1500/2200 works the same in my environments. The only differences between these groups (from the NUT's point of view) are charging times and number of batteries. DDLs for 650/850 and 1500/2200 pairs are below. NB: dump for 1500/2200 is done on older NUT (production server) while 650/850 has been dumped on my working PC with latest NUT from git repo.

### Ippon__Back_Basic_650_850__nutdrv_qx__2.8.2.1__01.dev
```
# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Fri, 13 Dec 2024 11:52:19 +0300
# DEVICE:EOC

# :; upsc ippon
battery.charge: 100
battery.charge.low: 25
battery.voltage: 13.8
battery.voltage.high: 13.6
battery.voltage.low: 10.4
battery.voltage.nominal: 12
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.chargetime: 36000
driver.parameter.default.battery.voltage.high: 13.6
driver.parameter.default.battery.voltage.low: 10.4
driver.parameter.override.battery.charge.low: 25
driver.parameter.override.battery.voltage.nominal: 12
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.protocol: q1
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.state: quiet
driver.version: 2.8.2.1
driver.version.data: Q1 0.08
driver.version.internal: 0.39
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 50.1
input.voltage: 231.4
input.voltage.fault: 231.4
output.voltage: 231.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 7
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: ippon
# [driver.debug]
# Current debug verbosity level of the driver program
# Type: NUMBER
# Value: 0
#
# [driver.flag.allow_killpower]
# Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
# Type: NUMBER
# Value: 0
#
# [ups.delay.shutdown]
# Interval to wait after shutdown with delay command (seconds)
# Type: RANGE NUMBER
# Option: "12-600" SELECTED
#
# [ups.delay.start]
# Interval to wait before (re)starting the load (seconds)
# Type: RANGE NUMBER
# Option: "0-599940" SELECTED
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: ippon
# Instant commands supported on UPS [ippon]:
#
# beeper.toggle - Toggle the UPS beeper
# driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
# driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
# driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
# driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
# load.off - Turn off the load immediately
# load.on - Turn on the load immediately
# shutdown.default - Run the driver-defined UPS shutdown sequence (as opposed to user-configured 'sdcommands')
# shutdown.return - Turn off the load and return when power is back
# shutdown.stayoff - Turn off the load and remain off
# shutdown.stop - Stop a shutdown in progress
# test.battery.start - Start a battery test
# test.battery.start.deep - Start a deep battery test
# test.battery.start.quick - Start a quick battery test
# test.battery.stop - Stop the battery test
# DEVICE:EOC
```

### Ippon__Back_Basic_1500_2200__nutdrv_qx__2.8.2.1__01.dev
```
# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Fri, 13 Dec 2024 11:55:10 +0300
# DEVICE:EOC

# :; upsc ippon
battery.charge: 100
battery.charge.low: 25
battery.voltage: 27.7
battery.voltage.high: 27.2
battery.voltage.low: 20.8
battery.voltage.nominal: 24
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.chargetime: 21600
driver.parameter.default.battery.voltage.high: 27.2
driver.parameter.default.battery.voltage.low: 20.8
driver.parameter.override.battery.charge.low: 25
driver.parameter.override.battery.voltage.nominal: 24
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.protocol: q1
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.state: quiet
driver.version: 2.8.2
driver.version.data: Q1 0.08
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 49.6
input.voltage: 230.3
input.voltage.fault: 230.3
output.voltage: 230.3
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 10
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: ippon
# [driver.debug]
# Current debug verbosity level of the driver program
# Type: NUMBER
# Value: 0
#
# [driver.flag.allow_killpower]
# Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
# Type: NUMBER
# Value: 0
#
# [ups.delay.shutdown]
# Interval to wait after shutdown with delay command (seconds)
# Type: RANGE NUMBER
# Option: "12-600" SELECTED
#
# [ups.delay.start]
# Interval to wait before (re)starting the load (seconds)
# Type: RANGE NUMBER
# Option: "0-599940" SELECTED
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: ippon
# Instant commands supported on UPS [ippon]:
#
# beeper.toggle - Toggle the UPS beeper
# driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
# driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
# driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
# driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
# load.off - Turn off the load immediately
# load.on - Turn on the load immediately
# shutdown.return - Turn off the load and return when power is back
# shutdown.stayoff - Turn off the load and remain off
# shutdown.stop - Stop a shutdown in progress
# test.battery.start - Start a battery test
# test.battery.start.deep - Start a deep battery test
# test.battery.start.quick - Start a quick battery test
# test.battery.stop - Stop the battery test
# DEVICE:EOC
```

jimklimov, 2025-09-28:
Technically yes, sure. Practically maintenance of HCL and DDL is a lot of boring routine that needs attention to detail, so when I at least have some time for NUT and there are real bugs pending, they usually win :\

dvdesolve, 2025-09-30:
Oh, sorry me! I understand, thank you a lot!