networkupstools/nut #2724: Windows build - information request

RikyPlaza, 2024-12-17:
Hello,
I saw that the windows built binaries are available in the AppVeyor page. I would like to know if there is any way to trigger the build-process manually without creating any pull request

jimklimov, 2024-12-17:
Currently, no - at least not for the general public. Account owners can trigger (re?-)builds on their repositories, though.

You can go about this by creating an AppVeyor account and letting it build your fork of NUT, or by building it yourself - that's fairly documented in the codebase and scripting and makefiles. You can do so natively on Windows with MSYS2, or on Linux (tried Debian/Ubuntu) with cross-mingw toolkit.

RikyPlaza, 2024-12-17:
@jimklimov Thank you, could you give me some advice about the toolchain to use and the --host / --target / --build flag combination to use to cross compile it on Ubuntu systems?

Thank you again

jimklimov, 2024-12-17:
For gory details, including prerequisites (packaged or custom-built - e.g. libsnmp was missing in MSYS2 IIRC) see https://github.com/networkupstools/nut/blob/master/scripts/Windows/README.adoc and https://github.com/networkupstools/nut/blob/master/scripts/Windows/build-mingw-prereqs.sh as well as common https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt

The build routine and settings (more or less one well-tried path is hardcoded so far) is in https://github.com/networkupstools/nut/blob/master/scripts/Windows/build-mingw-nut.sh

The way I go about it locally is to run `./ci_build.sh` "as is" under MSYS2, and it builds natively on the platform using its tools (GNU/clang toolkits built for WIN32), or I run `BUILD_TYPE=cross-windows-mingw ./ci_build.sh` on a cross-build prepared host, and it takes care of the out-of-tree build under `./scripts/Windows/nut_build` and proto area under `./scripts/Windows/nut_install` by calling the other scripts.

jimklimov, 2024-12-17:
Note you can also use `BUILD_TYPE=cross-windows-mingw{32,64}` for specific targeting, otherwise it tries to default based on what seems available/reasonable.

RikyPlaza, 2024-12-18:
@jimklimov Hi thank you for the references.
I'm trying to cross compile it under ubuntu system. Following the instruction at https://github.com/networkupstools/nut/blob/master/scripts/Windows/README.adoc i'm getting an error building the pthread library. i've set up all the mentioned variables but when i esecute the make command i get the following output:

`root@osboxes:~/nut-win-deps/x86_64-w64-mingw32/pthreads-w32-2-8-0-release# make -f GNUmakefile "CROSS=$ARCH-" $BUILD_FLAG GC-inlined
make: unrecognized option '--build=x86_64-linux-gnu'
`
I cannot find suggestions on this error...

jimklimov, 2024-12-18:
Not immediately sure... I see it was unpacked but not built in my Ubuntu worker; maybe the instructions were botched in copy-paste, checking.

At face value, the option is for `configure` scripts, not `make` itself. Maybe just the `$BUILD_FLAG` should not be in this line.

Also, generally you should not build programs as `root` "just in case" - the final `make install` or direct `mkdir/cp` instructions are prefixed with `sudo` where needed.

jimklimov, 2024-12-18:
Huh, in the document, the "pthread library" chapter starts with a note:
> The MinGW distribution in Ubuntu 21.10 already includes pthread files, so the build instructions below were not relevant for this component. 

So I think this one you can skip.

FWIW, without the `$BUILD_FLAG`, `make` fails at code issues here:
````
pthread.h:307:8: error: redefinition of ‘struct timespec’
...
/usr/share/mingw-w64/include/sys/timeb.h:91:8: note: originally defined here
````

As well as some (non-fatal) warnings about `cast to pointer from integer of different size` for `*-inlined` code.

RikyPlaza, 2024-12-18:
@Yes i already tried to remove the $BUILD_FLAG. Righit now i'll try to build the project on MSYS2 as you mentioned...
I'll take care to check the cross compilation on ubuntu systems and keep you posted, but i don't know yet when i can do it...

Thank you!

jimklimov, 2024-12-18:
Actually they have a `HAVE_STRUCT_TIMESPEC` fencing which you can add to `config.h` and the build does pass.

jimklimov, 2024-12-18:
I'll bump the text of instructions with this bit.

jimklimov, 2024-12-18:
@RikyPlaza : I've updated the document a bit; if you pass the rest of instructions successfully and get a NUT for Windows build completed and working, please let us know here and feel free to close the ticket.

...or if more docs clarifications are needed, with your or my PRs, so be it :^)

RikyPlaza, 2024-12-18:
@jimklimov Hi.
Thank you for the new instuctions. I had some other fails building the dependencies... right now I'm trying to salve them but I'm not full dedicated to this...
If I manage to have a Windows build I'll update the docs with the steps I had to follow. 

PS: I was unable to build it on MSYS2 and AppVeyor aswell. I did not investigate MSYS2 issues yet...
AppVeyor was failing due to timeout (60 minutes) using the default repository yaml file so I don't really know how to do it...
Regards.

jimklimov, 2024-12-18:
Did it fail at artifact compression, after 7zip?

RikyPlaza, 2024-12-19:
@jimklimov Yes, i'll try to remove the tests and let the project only build nut.

I've done some improvement to the documentation of the cross build, but i'm stuck building the freetype2 library. Here's the failure log:
`configure: error: native C compiler is not working
make: *** [builds/unix/detect.mk:91: setup] Error 1
make: .includefreetypefreetype.h: No such file or directory
config.mk:26: builds/unix/unix-cc.mk: No such file or directory
make: *** No rule to make target 'builds/unix/unix-cc.mk'.  Stop.
[sudo] password for osboxes: 
make: .includefreetypefreetype.h: No such file or directory
config.mk:26: builds/unix/unix-cc.mk: No such file or directory
make: *** No rule to make target 'builds/unix/unix-cc.mk'.  Stop.
`
Do you have any suggestions?

jimklimov, 2024-12-19:
Don't remember that one. If there's a `config.log` it would have more details about what it ran and how that failed. For example, unexpected `compiler not working` may be in fact issues about free space or some broken ccache pernissions.

RikyPlaza, 2024-12-20:
@jimklimov Hi, finally i got it working. I manage to have a windows build on Ubuntu virtual machine but i'll not be able to test it untill 3rd week of january...
I've updated the documentation on a dedicated branch of my fork, would you like me to open the pull request to merge the changes or should i wait to test it and than open the pull request?

RikyPlaza, 2024-12-20:
By the way, i also got the binaries from AppVeyor removig the test scripts and merging the after_test scripts with the after_build ones.
Thank you :)

From my side this issue can be closed.

jimklimov, 2024-12-20:
Well, I guess a PR would be welcome. No harm in opening one early, at least to pass all the docs CI checks. Although as any author, you'd probably have some second thoughts and refinement in the coming days - maybe do open after that phase... :D

RikyPlaza, 2024-12-21:
@jimklimov Hi, got it!
In the meanwhile, if you want, you can take a look at this: 
https://github.com/RikyPlaza/nut/blob/Windows_cross_build_documentation/scripts/Windows/README.adoc

So if more clarifications or specifications are needed we will do everything with one PR.

Regrads and thank you for the support

closed 2024-12-21 by RikyPlaza