networkupstools/nut #2725: Support for CyberPower LX1500GU3

faspina, 2024-12-17:
Running  CyberPower LX1500GU3 with Nut Server and USB cable.

usbhid-ups  2.8.2 stable. 

Load is reporting  as 0

jimklimov, 2024-12-17:
Is it expected to be significantly higher? There can be some data conversion issues in a driver, or the device not noticing a load under 10% or so.

jimklimov, 2024-12-17:
Can you post a debug log from the driver initialization, verbosity 6?

Any chance of running older (2.7.4) and/or newer (master-branch) driver builds, to rule out regressions and already-solved issues? See wiki hot links about "in-place" builds of current code, and data-dumping.

faspina, 2024-12-17:
> Is it expected to be significantly higher? There can be some data conversion issues in a driver, or the device not noticing a load under 10% or so.

The Battery Run time is reporting as 7 hour 30 minute so I don't think the driver is reporting correctly at all

faspina, 2024-12-17:
> Can you post a debug log from the driver initialization, verbosity 6?
> 
> Any chance of running older (2.7.4) and/or newer (master-branch) driver builds, to rule out regressions and already-solved issues? See wiki hot links about "in-place" builds of current code, and data-dumping.

I am able to run Preview, 2.8.2, 2.8.1, 2.8.0 and 2.7.4 in Unraid plug in all with similar results 

[nut-debug-20241217112728.zip](https://github.com/user-attachments/files/18169232/nut-debug-20241217112728.zip)

here is the debug out put file from the unraid plug in

jimklimov, 2024-12-17:
Thanks for checking, I'll try to take a look when I get to a PC.

What sort of load is there? By numbers,  it is a 1500VA UPS? Some Raspberry or small NAS under say 100W might be under the detection threshold, depends on UPS. Well under - might allow some hours too...

faspina, 2024-12-17:
You can close I am an idiot, server plugged into the non Battery Backup slot,  must have happened when I reconfigured.

closed 2024-12-17 by faspina