networkupstools/nut #2727: Cannot setup battery.charge and temperature sensor for Green Cell PowerProof 2000VA UPS05

hajar97, 2024-12-18:
In the table of supported UPSs GreenCell only has 1 listed with poor support. I am happy to provide any data and do thorough testing for mine one to make it work with NUT. 

With the configuration below I got most of the features going, but with some still significant limitations, which I would like to overcome:
1. I couldn't find a way to read batter.charge that comes from the UPS. Only estimate it using min and max voltage. I know UPS does its own battery charge estimation that can be seen through its native software. 
2. I couldn't find a way to read temperature. Again, a measurement that it provides through its native app.
3. I was only able to install 2.8.0 version of NUT that is available via apt-get install on my Proxmox Ubuntu. While I did manage to compile 2.8.2 version of NUT and install it, installation installs it into totally different folders that apt-get install (/usr/local/ups instead of /usr/sbin) and then I can only run it from those folders using -u root flag. Without it, but using root account, I always run into all kinds of Permission denied issues. And even when running it -u root flag I cannot then check UPS diagnostics using upsc command as I get "Error: Connection failure: Connection refused"

So my main questions are:
1. Is there a simple way to upgrade to 2.8.2 so it works as simply as the version installed via apt-get install nut without having to fight through error over error?
2. Or if 2.8.2 isn't actually going to increase my chances with limitations 1 and 2, then happy to stick with 2.8.0 for now, but then how do I best overcome those limitations?

Any help would be hugely appreciated. 

My current ups.conf file:

```
[gc2000va]
  driver = nutdrv_qx # blazer_usb #if doesn't work try usbhid-ups
  protocol = megatec
  # norating
  novendor
  # noscanlangid
  # vendorid = "0001"
  # bus = "001"
  # productid = "0000
  # product = "MEC0003"
  port = auto
  desc = "Green Cell PowerProof 2000VA UPS05"
  chargetime = 28800
  default.battery.type = PbAcid
  default.battery.capacity.nominal = 18.0
  default.battery.charge.low = 10
  default.battery.charge.warning = 50
  default.battery.voltage.nominal = 24
  default.battery.voltage.low = 21.00
  default.battery.voltage.high = 27.60
  # default.battery.packs = 2
  default.input.frequency.nominal= 50.0
  default.input.voltage.nominal= 230.0
  default.output.current.nominal = 8.0
  default.output.frequency.nominal = 50.0
  default.output.voltage.nominal = 230.0
  default.ups.mfr = "Green Cell"
  default.ups.model = "PowerProof 2000VA UPS05"
  default.ups.mfr.date = "2024/12/17"
  default.ups.vendor = "Fry's Electronics"
  default.ups.product = "MEC0003"
  default.ups.power.nominal = "2000"
  default.ups.realpower.nominal = "1200"
```

hajar97, 2024-12-18:
oh and another inconsistency is that I know this UPS has 2 batteries of 12v each. But if I set battery.pack to 2 and battery.voltage.nominal to 12 (so the sum would come to 24), then battery.charge estimation always shows 100% no matter how low actual battery.voltage drops with all the other settings as above. If I remove battery.packs and change battery.voltage.nominal to 24 then battery.charge estimation starts working correctly. Not sure what I got wrong there.

Just in case, attaching a screenshot of some tech specs of the UPS. 
[image]
[image]

jimklimov, 2024-12-18:
Cheers, and welcome to the NUT community!

Regarding the custom build, check the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests instructions (you probably have build prerequisites already, but better revise the linked documentation too). This should take care of using same configuration paths and account names as packaged NUT, although not all binaries may land into same locations (maybe good or bad, depending on how eager you want to "forget" the packaged variant). Note you can probe the device by calling the driver program from the build workspace (possibly via `sudo`) without installing anything freshly built into the system.

If nothing else, newer NUT codebase should be more inclined to troubleshooting and debugging, see also https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity -- this might help you trace the actual queries to and replies from the device.

There are many dialects of Megatec Q<x> (where X is a number) protocol family, you might want to pick and try other values of `protocol=...` to see if they pick up more info about your device (see `nutdrv_qx --help` for up-to-date list). Temperature is commonly reported in the replies, and given that this is a text protocol, you would see it in debug logs even if the selected driver code path does not.

hajar97, 2024-12-18:
Thank you, Jim. Great to be part of the community. Hopefully my quest to solve above challenges will also be useful to others. Regarding the link you shared, I have actually tried to follow it previously, but got lost through the maze of many steps and things to consider. When building I used configure --with-all, so I guess all prerequisites are already in place. I use root user and there is no sudo command on my Linux. Make install is successful too. But then how to move all that newly built 2.8.2 package over 2.8.0 is a mystery. They definitely do not install into the same folders as the packaged NUT.  

I also tried to run the driver program from my build workspace (I don't have sudo, running using root account directly), but strangely that fails using all kinds of permissions errors, which seems odd to me given that I am doing everything running a root account. The only way to overcome this is by using -u root, then things launch. But then I cannot get upsc to give me any readings always failing with "Error: Connection failure: Connection refused". 

I really thought compiling it will be the biggest challenge, and then after make install everything will land nicely. But seems I was wrong. It is from make install where all the insurmountable issues really start to pop up...

Is there perhaps a particular section in the file I should follow more closely in your view? I am suspecting it must be the "Replacing a NUT deployment" but then it is referring to some many things that I don't quite understand or issues like use of sudo while I don't have sudo, that I am worried that I may damage my proxmox setup in the end if I keep poking at it without really knowing what I am doing...

jimklimov, 2024-12-18:
You can look at packaging recipes in your distro, they will have a wall-long list of `configure` script options. The "in-place" build support tries to guess the critical ones from available system files and path-name existence.

hajar97, 2024-12-18:
Ok, so where do I even start? Do I have to start all the way with git command, getting source and starting compiling all over again, or is there a way to do something with the tar.gz package which I already downloaded and built?

hajar97, 2024-12-18:
> There are many dialects of Megatec Q (where X is a number) protocol family, you might want to pick and try other values of `protocol=...` to see if they pick up more info about your device (see `nutdrv_qx --help` for up-to-date list). Temperature is commonly reported in the replies, and given that this is a text protocol, you would see it in debug logs even if the selected driver code path does not.

Ok, I have tried all protocols and subdrivers and here are my findings:
* All of these protocols seem to work: q1, megatec, zinto, ablerex. There is no visible difference in info they pickup
* All of these subdrivers work: krauler, fabula, hunnox with any of the above protocols. There is no visible difference in info they pickup
* All of the other protocols and subdrivers give the following error: Device not supported!
* The only exception is hunnox protocol which gives this error: Can't initialise data from the UPS

Any clues or further suggestions what to try?

jimklimov, 2024-12-18:
The `tar.gz` package is a release snapshot. The 2.8.2 one should contain the needed `configure` script abilities for in-place builds, then you won't require `autogen.sh` from those instructions. Not sure you'd have `ci_build.sh` either (IIRC wiki took a shortcut with that for some examples, not sure).

jimklimov, 2024-12-18:
Further suggestions were to enable debug so you'd see what the device says to queries used.

hajar97, 2024-12-18:
how do I enable debug and what should I look for and where when I enable it?

Upgrading to 2.8.2 seems too complicated right now. If not strictly required, I’d leave it until packaged releases have 2.8.2. Not sure why years went by and they are still on 2.8.0...

jimklimov, 2024-12-19:
https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity

hajar97, 2024-12-19:
Thank you, will take a look and report back. In the meantime, do you have a suggestion how I should change my settings above in order to be able to specify battery.pack=2 and have battery.charge correctly estimated? If I enable battery.pack=2 now then my battery.charge estimation goes to permanent 100%.

https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity

jimklimov, 2024-12-19:
Double-checking with https://networkupstools.org/docs/release-notes.chunked/NUT_Release_Notes.html#_release_notes_for_nut_2_8_2_what_8217_s_new_since_2_8_1 - there were fixes to battery charge accounting in different releases after 2.8.0, e.g.:
* https://github.com/networkupstools/nut/issues/1279
* https://github.com/networkupstools/nut/issues/2325
* https://github.com/networkupstools/nut/pull/2324

jimklimov, 2024-12-19:
> ...available via apt-get install on my Proxmox Ubuntu...
> Not sure why years went by and they are still on 2.8.0...

Well, some distros are still on 2.7.4 that is nearing 8 years old (facepalm). To their moderate defence, it was the latest official NUT release till April 2022, but... still...

Many long-term lines of Debian/Ubuntu based distros pick a version of a delivered project to package, and patch it ad nauseam with minor fixes or bugfix backports. On one hand, it does improve stability for end users (few interim surprises as major changes only when you go up to newer distro major version), on another they are left behind as the project upstream gets improved.

You might have better chances of newer packaged code with "experimental" or "sid" rolling package repositories, at the cost of potentially being the first to discover new bugs too.

Also, I think Proxmox ISOs are based on Debian proper (7.x was on bullseye/11; not sure OTOH what 8.x is based on). So if you really have Ubuntu, that was via added packages that you installed into a somewhat independent of distro of your choice and obligation to manage ;)

hajar97, 2024-12-19:
> Also, I think Proxmox ISOs are based on Debian proper (7.x was on bullseye/11; not sure OTOH what 8.x is based on). So if you really have Ubuntu, that was via added packages that you installed into a somewhat independent of distro of your choice and obligation to manage ;)

Eh sorry, my mistake. You're absolutely right. I am on latest Proxmox 8.3.2 with Debian GNU/Linux 12 (bookworm). It is absolutely not Ubuntu. I had too many Ubuntu Linuxes in the past and totally mixed it up.

Do you know if there is an experimental packages for Debian 12, such that I could find and install latest NUT 2.8.2? Or if not, do you if there is a simple way to fix the nightmare issues on Debian that simple make install is giving me vs the prepackaged version?

jimklimov, 2024-12-19:
Well, part of the fix can be to `make uninstall` from that workspace, so it would remove the files it added. A problem with that could be that if it infringed on path names delivered by packaging too, it would remove them as well (but I gather `apt` has a `fix` operation somewhere to restore missing files). Could be prudent to back up your configs though.

Otherwise, I mostly have practice with git-originated builds and automation that ended up as "in-place" build logic, and that path was walked by many others since. So I suggest to not go about inventing new wheels where we have one, and complete the instructions from that wiki page :)

hajar97, 2024-12-19:
Ok, I’ll see when I can find another full day to pick that fight up. Already spend a day on it, just not prepared to invest as much again. My main issue is not walking the path. My main issue that each step of the way is faced with problems and as I try to resolve them I am worried that I may make a permanent damage to the proxmox setup. To avoid it I then first need to create a VM to safely walk that path and in case I break something I don’t damage all other months long configurations I did. So one thing requires another and another and it all adds up to requiring too much time. apt-git install nut is so much more straightforward compared to this. Anyway…

Would you have any suggestions how to address the battery.packs question from above?

jimklimov, 2024-12-19:
Just recently the matter of packs was raised and solved in issue #2730 - take a look there. But it may rely on the newer code...

hajar97, 2024-12-20:
Thanks, that helped.

I got around running debug at level 6. Below is output. It seems what NUT can read gets "--.-" for temperature. So I guess there is no temperature from this device then?

Anything else relevant that catches your eye and you recommend to do or set some specific NUT settings?

```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000696     Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-gc2000va.pid exists)! Terminating other driver!
   5.000811     [D1] Saving PID 8760 into /run/nut/nutdrv_qx-gc2000va.pid
   5.000845     [D5] send_to_all: SETINFO device.type "ups"
   5.000850     [D1] upsdrv_initups...
   5.003082     [D2] Checking device 1 of 6 (2357/0601)
   5.003106     [D1] Failed to open device (2357/0601), skipping: Access denied (insufficient permissions)
   5.003110     [D2] Checking device 2 of 6 (1D6B/0003)
   5.003115     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   5.003121     [D2] Checking device 3 of 6 (0403/6015)
   5.003126     [D1] Failed to open device (0403/6015), skipping: Access denied (insufficient permissions)
   5.003130     [D2] Checking device 4 of 6 (8087/0026)
   5.003135     [D1] Failed to open device (8087/0026), skipping: Access denied (insufficient permissions)
   5.003138     [D2] Checking device 5 of 6 (0001/0000)
   5.004067     [D2] - VendorID: 0001
   5.004078     [D2] - ProductID: 0000
   5.004081     [D2] - Manufacturer: MEC
   5.004083     [D2] - Product: MEC0003
   5.004086     [D2] - Serial Number: unknown
   5.004089     [D2] - Bus: 001
   5.004091     [D2] - Device: unknown
   5.004093     [D2] - Device release number: 0100
   5.004095     [D2] Trying to match device
   5.004100     [D3] match_function_regex: matching a device...
   5.004105     [D2] Device matches
   5.004108     [D2] Reading first configuration descriptor
   5.004114     [D3] libusb_kernel_driver_active() returned 0
   5.004133     [D2] Claimed interface 0 successfully
   5.004137     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   5.004144     [D5] send_to_all: SETINFO ups.vendorid "0001"
   5.004148     [D5] send_to_all: SETINFO ups.productid "0000"
   5.004151     [D2] Skipping protocol Voltronic 0.06
   5.004153     [D2] Skipping protocol Voltronic-QS 0.07
   5.004155     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   5.004158     [D2] Skipping protocol Mustek 0.07
   5.004160     [D2] Skipping protocol Megatec/old 0.07
   5.004162     [D2] Skipping protocol BestUPS 0.06
   5.004164     [D2] Skipping protocol Mecer 0.07
   5.004166     [D2] Skipping protocol Megatec 0.06
   5.004168     [D2] Skipping protocol Zinto 0.06
   5.004170     [D2] Skipping protocol Masterguard 0.02
   5.004172     [D2] Skipping protocol Hunnox 0.01
   5.004174     [D2] Skipping protocol Ablerex 0.01
   5.004178     [D3] send: Q1
   5.004180     [D4] command index: 0x03
   5.225354     [D5] read: (47 bytes) => 28 32 33 38 2e 37 20 32 33 38 2e 37 20 32 33 38 2e 37
   5.225369     [D5]  20 30 30 38 20 35 30 2e 31 20 32 36 2e 33 20 2d 2d 2e 2d 20 30 30 30 30 31
   5.225371     [D5]  30 30 31 0d
   5.225373     [D3] read: (238.7 238.7 238.7 008 50.1 26.3 --.- 00001001
   5.225382     [D5] send_to_all: SETINFO input.voltage "238.7"
   5.225385     Using protocol: Q1 0.07
   5.225390     [D5] send_to_all: SETINFO driver.version "2.8.0"
   5.225392     [D5] send_to_all: SETINFO driver.version.internal "0.32"
   5.225395     [D5] send_to_all: SETINFO driver.name "nutdrv_qx"
   5.225397     [D1] upsdrv_initinfo...
   5.225401     [D5] send_to_all: SETINFO driver.version.data "Q1 0.07"
   5.225407     [D3] send: Q1
   5.225409     [D4] command index: 0x03
   5.446946     [D5] read: (47 bytes) => 28 32 33 38 2e 37 20 32 33 38 2e 37 20 32 33 39 2e 32
   5.446968     [D5]  20 30 30 38 20 35 30 2e 31 20 32 36 2e 33 20 2d 2d 2e 2d 20 30 30 30 30 31
[... 34 lines cut ...]
   5.447136     [D2] qx_initbattery: battery runtime nominal: 178.7
   5.447139     [D2] qx_initbattery: battery runtime estimate: 178.7
   5.447141     [D2] qx_initbattery: battery charge time: 28800
   5.447144     [D2] qx_initbattery: minimum load used (idle): 0.040
   5.447148     [D5] send_to_all: SETINFO driver.parameter.pollfreq "30"
   5.447150     [D1] upsdrv_updateinfo...
   5.447153     [D1] Quick update...
   5.447155     [D3] send: Q1
   5.447158     [D4] command index: 0x03
   5.668546     [D5] read: (47 bytes) => 28 32 33 38 2e 37 20 32 33 38 2e 37 20 32 33 38 2e 37
   5.668565     [D5]  20 30 30 38 20 35 30 2e 31 20 32 36 2e 33 20 2d 2d 2e 2d 20 30 30 30 30 31
   5.668568     [D5]  30 30 31 0d
   5.668570     [D3] read: (238.7 238.7 238.7 008 50.1 26.3 --.- 00001001
   5.668580     [D5] update_status: OL
   5.668583     [D5] update_status: !LB
   5.668586     [D5] update_status: !CAL
   5.668588     [D5] update_status: !FSD
   5.668594     [D5] send_to_all: SETINFO ups.status "OL"
   5.668604     [D5] send_to_all: DATAOK
   5.668658     [D2] dstate_init: sock /run/nut/nutdrv_qx-gc2000va open on fd 9
   5.668663     [D5] send_to_all: SETINFO driver.parameter.pollinterval "2"
   5.668666     [D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   5.668668     [D5] send_to_all: SETINFO device.mfr "Green Cell"
   5.668671     [D5] send_to_all: SETINFO device.model "PowerProof 2000VA UPS05"
```

hajar97, 2024-12-20:
Here is my current output from upsc:

```
Init SSL without certificate database
battery.capacity.nominal: 18.0
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.packs: 2
battery.type: PbAcid
battery.voltage: 26.50
battery.voltage.high: 25.00
battery.voltage.low: 21.00
battery.voltage.nominal: 24
device.mfr: Green Cell
device.model: PowerProof 2000VA UPS05
device.type: ups
driver.name: nutdrv_qx
driver.parameter.chargetime: 28800
driver.parameter.idleload: 4
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.protocol: q1
driver.parameter.runtimecal: 2340,8,4740,4
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: Q1 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.frequency.nominal: 50.0
input.voltage: 238.7
input.voltage.fault: 238.7
input.voltage.nominal: 230.0
output.current.nominal: 8.0
output.frequency.nominal: 50.0
output.voltage: 238.7
output.voltage.nominal: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 8
ups.mfr: Green Cell
ups.mfr.date: 2024/12/17
ups.model: PowerProof 2000VA UPS05
ups.power.nominal: 2000
ups.product: MEC0003
ups.productid: 0000
ups.realpower.nominal: 1200
ups.status: OL
ups.type: offline / line interactive
ups.vendor: Fry's Electronics
ups.vendorid: 0001
```

hajar97, 2024-12-20:
Seems it doesn't have a serial port. This is what I get when I run dmesg | grep tty

```
[    0.077630] printk: legacy console [tty0] enabled
[    0.364178] 00:01: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    4.044402] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[    4.321387] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0
[    4.321999] dw-apb-uart.2: ttyS4 at MMIO 0x4017002000 (irq = 16, base_baud = 6250000) is a 16550A
[   16.735601] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
```

If I run screen /dev/ttyUSB0 I get: Cannot exec '/dev/ttyUSB0': No such file or directory.

So I guess no luck there either?

jimklimov, 2024-12-20:
In a remote-console use I have literally that usage, optionally followed by baud rate number. Maybe you have some other version of `screen` that trues to execute that pathname as a program?..

Does `ls /dev/ttyUSB0` show it?

Just in case you type - it is a zero in the end, not letter "Oh" ;)

jimklimov, 2024-12-20:
Not sure what happens at 16s in your dmesg - perhaps the UPS controller repurposes the connection (assuming FTDI converter is it's part)? Or a NUT driver startup attempt detaches it (try to start with NUT disabled while probing it as a serial port)?..

hajar97, 2024-12-20:
I completely removed NUT, rebooted and tried `dmesg | grep tty` again and it gave me the same result. That indicates that the only serial USB device that initially connects, but then quickly disconnects must be not UPS, but something else. 

I also tried `ls /dev/ttyUSB0` and it gives me: ls: cannot access '/dev/ttyUSB0': No such file or directory

jimklimov, 2024-12-21:
I see. Then, it indeed does not seem to create a serial port (or not for long).

With custom NUT builds, you can probably edit a subdriver source file to add tested operation calls, but I won't be able to assist with that, at least not this year.

hajar97, 2024-12-21:
I see. Well, no big deal. I am happy with the basic info I can get for now.

One question, when power is connected I get voltage of 27.40V. So I set battery.high to 27.40 and it shows battery.charge at 100%. But as soon as I remove power, that voltage immediately drops to 25V and hence my battery.charge also immediately drops from 100% to 60%. Should I set battery.high to 25V in order to avoid that sudden drop in estimated battery.charge when power goes off, or should this be handled somehow differently?

jimklimov, 2024-12-21:
Not sure, I think set it to 25V, and the driver should truncate at 100% when on wall power.

hajar97, 2024-12-21:
Got it, will do.

I think I have finalised my list of settings for my UPS, as shown below. Many thanks for your kind support. How can I record my UPS in the list of supported ones by NUT, so other users may find it when they are searching?

maxretry = 1
pollinterval = 2
battery_voltage_reports_one_pack
# debug_min = 6

```
[gc2000va]
  driver = nutdrv_qx
  protocol = q1 
  port = auto
  desc = "Green Cell PowerProof 2000VA UPS05"
  chargetime = 28800
  idleload = 4
  default.battery.type = PbAcid
  default.battery.capacity.nominal = 18.0
  default.battery.charge.low = 10
  default.battery.charge.warning = 50
  default.battery.voltage.nominal = 24
  default.battery.voltage.low = 21.00
  default.battery.voltage.high = 25.00
  default.battery.packs = 2
  runtimecal = 2340,8,4740,4
  default.input.frequency.nominal= 50.0
  default.input.voltage.nominal= 230.0
  default.output.current.nominal = 8.0
  default.output.frequency.nominal = 50.0
  default.output.voltage.nominal = 230.0
  default.ups.mfr = "Green Cell"
  default.ups.model = "PowerProof 2000VA UPS05"
  default.ups.mfr.date = "2024/12/17"
  default.ups.vendor = "Fry's Electronics"
  default.ups.product = "MEC0003"
  default.ups.power.nominal = "2000"
  default.ups.realpower.nominal = "1200"
```

jimklimov, 2024-12-22:
Thanks, a pull request to NUT DDL would be great, the syntax has a place for configuration like above, as well as "layman" comments. Please see https://github.com/networkupstools/nut/wiki/NUT-HCL-and-DDL for a starting point.

hajar97, 2024-12-27:
Hi Jim, I did submit 2 pull requests. Hopefully everything is correct there. But happy to fix if not. Thank you.

jimklimov, 2024-12-27:
Thanks! I'll take a closer look after the New Years festivities I guess, traveling mostly offline before then.