networkupstools/nut #2728: CPS HID: Some devices report `output.voltage` values too high

jimklimov, 2024-12-18:
Much higher than `input.voltage`, often similar (not always equal) to High Transfer Voltage values where available. In many examples it is stuck at `265.0`, regardless of inputs being in 120V or 220-240V ranges.

While the issue was seen on some devices with NUT v2.7.4, it is possible that USB descriptor fix-ups introduced in #1245 (released in NUT v2.8.0) and updated by #2718 (to be released in NUT v2.8.3, currently requires custom builds to test or take advantage of) fixed some issues for one set of models/firmwares, and broke the behaviors for others that were good out of the box. This aspect can now be easily verified by use of `disable_fix_report_desc` setting for `usbhid-ups` driver.

It is however possible that those devices are botched in some other way that we still mis-interpret (do not compensate for vendor errors), or that the devices just lie to us and confuse the values internally. Comparative tests with older NUT releases (2.7.4, 2.8.0, current) would be helpful.

Per https://github.com/networkupstools/nut/issues/1338#issuecomment-1085300215 :
> ````
> 0.009244     [D1] Path: UPS.Output.LowVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 0, Size: 16, Value: 207
> 0.009253     [D1] Path: UPS.Output.LowVoltageTransfer, Type: Input, ReportID: 0x10, Offset: 0, Size: 16, Value: 207
> 0.009258     [D1] Path: UPS.Output.HighVoltageTransfer, Type: Feature, ReportID: 0x10, Offset: 16, Size: 16, Value: 253
> 0.009261     [D1] Path: UPS.**Output.HighVoltageTransfer**, Type: Input, **ReportID: 0x10, Offset: 16**, Size: 16, Value: **253**
> 0.009452     [D1] Path: UPS.**Output.Voltage**, Type: Feature, **ReportID: 0x12, Offset: 0**, Size: 16, Value: **253**
> ````
> On line 4, there is ReportID 0x10 offset 16 - HighVoltageTransfer 253
> On line 5, ReportID 12, offset 0, Output Voltage 253
> Based on the debug - I'm not sure what ReportID exactly is, does NUT really read diferrent data and UPS reports the same value 253 or NUT actually reads the same data? Or ReportID 0x10 offset 16 is the same part of data as ReportID 12 offset 0 ?

Per https://github.com/networkupstools/nut/issues/982#issuecomment-787557944 this looks like something #2718 and/or #1245 already could have fixed along the way:
> NOTE  Hex 0x74 is 116 not 140
> ```
> 8.350308 Report[buf]: (3 bytes) => 12 74 00
> 8.350325 PhyMax = 0, PhyMin = 0, LogMax = 142, LogMin = 136
> 8.350342 Unit = 00f0d121, UnitExp = 7
> 8.350359 Exponent = 0
> 8.350377 Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 140
> 8.350396 send_to_all: SETINFO output.voltage "140.0"
> ```

Likewise for https://github.com/networkupstools/nut/issues/982#issuecomment-1490837532 :
> Hex 7c = 124 but the value reported is 140.
> ```
> 0.653344 Entering libusb_get_report
> 0.654006 Report[get]: (3 bytes) => 12 7c 00
> 0.654049 PhyMax = 0, PhyMin = 0, LogMax = 142, LogMin = 136
> 0.654091 Unit = 00f0d121, UnitExp = 7
> 0.654125 Exponent = 0
> 0.654295 Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 140
> ```

Or https://github.com/networkupstools/nut/issues/581#issue-343360256 which has same raw readings but may have different LogMin/LogMax (addressed by the PRs above later):
> my CP900EPFCLCD reports the output voltage as being between 260 and 270 - though it sits at 260 most of the time. The input voltage averages around 240 as would be expected. I'm not entirely sure what to make of this, though it would appear that ef (239) is somehow being interpreted as 260?
> ```
> 0.010678 Report[get]: (3 bytes) => 0f ef 00
> 0.010687 Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 239
> ...
> 0.011178 Report[get]: (3 bytes) => 12 ef 00
> 0.011188 Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 260
> ```

This issue follows up from earlier reports, some closed over time, to track the problem with this specific reading:
* #1969 PR1500ELCD
````
driver.version: 20220328-3481-gccbf3eb6c6
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
input.voltage: 236.0
input.voltage.nominal: 230
output.voltage: 265.0    #BAD: expected near 236
````
* #1191 PR1500RT2U
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 121.0
input.voltage.nominal: 120
output.voltage: 265.0    #BAD: expected near 120
````
* #520 CP1500PFCLCD
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 139
input.transfer.low: 88
input.voltage: 116.0
input.voltage.nominal: 120
output.voltage: 140.0    #BAD: expected near 120
````
* #1689 OR600ERM1U
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 231.0
input.voltage.nominal: 230
output.voltage: 263.0    #BAD: expected near 230 (note not 265.0 either)
````
* #1338 OR1500ERM1U 
````
### Version 2.7.4-5004-g1f143e5e (month before 2.8.0)
input.voltage: 232.0
input.voltage.nominal: 230
output.voltage: 253.0    #BAD: The value 253 in output.voltage is actually high transfer value. Probably easy to fix.
````
* #1338 OR1500ERM1U (same UPS, slightly newer NUT)
````
driver.version: 2.8.0-rc2-109-gbce1d5201
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 253.0
````
* #1338 CP1500PFCLCDa
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 120.0
input.voltage.nominal: 120
output.voltage: 136.0    #BAD: expected near 120
````
* #1338 OR500LCDRM1Ua
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 118.0
input.voltage.nominal: 120
output.voltage: 135.0
````
* #982 LX1500GU
````
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 116.0
input.voltage.nominal: 120
output.voltage: 140.0
````
* #581 PR1000RTXL2U 
```
input.voltage: 122.0
input.voltage.nominal: 120
output.voltage: 253.0
```
* #439 numerous ideas and readings and graphs, including some experimental heuristics that map the input voltage to error in reported output value

The dominating idea in those discussions is that for whatever reason NUT reads the output value as whatever is in that byte + 32 (may be or not be the value of some other HID report, especially one of not-yet-decoded ones), and then constrained by LogMin/LogMax values (which may be reported same as High (and maybe Low) Voltage Transfer's LogMin/LogMax values. This is directly the focus of #439 and #1245 (so really wondering if the problem is still seen on devices where the descriptor fix now does happen, especially with current master-branch NUT code base).

jimklimov, 2024-12-18:
Loosely related to #1535 which has a similar issue with another HID subdriver.

WillyTP, 2025-01-15:
CP1500EPFCLCD, with NUT running on a Synology Diskstation with DSM 7.2, shows incorrect output voltage (too high)

```
battery.charge:`3
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 77
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: CRMMxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-2-1-69057-NM-nano-4-C1-240105
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 262.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 12
ups.mfr: CPS
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CRMMY2000017
ups.status: OL CHRG LB
ups.test.result: Done and warning
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

jimklimov, 2025-01-15:
@WillyTP : Thanks, FWIW that seems to be NUT v2.7.4, 9 years old soon. You may want to poke the vendor to update their OS sometimes.

jimklimov, 2025-01-15:
@WillyTP : Are you in position to try a custom build of current NUT codebase (perhaps with some other computer/VM that this UPS can be attached to temporarily for testing)? See https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for technical details.

With #1512 solved recently, a possibly same or different issue with output voltage should no longer happen. I wonder if that solution does impact devices that expose the behavior seen here, or it is a different issue after all?..

MilhouseVH, 2025-01-22:
CP1500EPFCLCD, UK (240v/50Hz), input and output voltages look OK.

Master build (commit 6ae5fead6, 18 Dec 2024).

```
battery.charge: 100
battery.charge.low: 3
battery.charge.warning: 20
battery.mfr.date: 2023-03-28
battery.runtime: 3270
battery.runtime.low: 180
battery.status: 100%
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: CRXLX2000266
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.flag.ignorelb: enabled
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.override.battery.charge.low: 3
driver.parameter.override.battery.mfr.date: 2023-03-28
driver.parameter.override.battery.runtime.low: 180
driver.parameter.override.ups.mfr.date: 2023-03-28
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.2.1751-1751-g6ae5fead6
driver.version.data: CyberPower HID 0.82
driver.version.internal: 0.60
driver.version.usb: libusb-1.0.22 (API: 0x1000106)
input.sensitivity: normal
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 238.0
input.voltage.nominal: 230
output.voltage: 238.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 120
ups.firmware: CR01505B4
ups.load: 9
ups.mfr: CPS
ups.mfr.date: 2023-03-28
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CRXLX2000266
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

ce3a, 2025-03-16:
CyberPower (CPS) OR600ERM1U (230V,  50Hz): `output.voltage: 253.0` **not ok (too high)**.

Tested with master commit 31650d455 (16 Mar 2025).

I am not familiar with the code base, but would be happy to rebuild and retest locally. If there is anything I could try, please let me know.

[usbhid-ups_debug.txt](https://github.com/user-attachments/files/19274212/usbhid-ups_debug.txt)

```
battery.charge: 100
battery.charge.low: 30
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4275
battery.runtime.low: 180
battery.type: PbAcid
battery.voltage: 13.5
battery.voltage.nominal: 12
device.mfr: CPS
device.model: OR600ERM1U
device.serial: GA6PQ2000138
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: OR600ERM1U
driver.parameter.productid: 0601
driver.parameter.serial: GA6PQ2000138
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 227.0
input.voltage.nominal: 230
output.voltage: 253.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 20
ups.mfr: CPS
ups.model: OR600ERM1U
ups.productid: 0601
ups.realpower.nominal: 360
ups.serial: GA6PQ2000138
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

ce3a, 2025-03-16:
Yes, I was following this instruction, but I may have missed something.

Build environment is Proxmox 8.3.5 (debian bookworm, original NUT version is v2.8.0).

I have used the configure options from `/usr/share/doc/nut/packager-guide.txt.gz`.

```
./configure \
	    --enable-inplace-runtime \
            --prefix=/ \
            --sysconfdir=/etc/nut \
            --mandir=/usr/share/man \
            --libdir=/usr/lib \
            --includedir=/usr/include \
            --datadir=/usr/share/nut \
            --with-statepath=/var/run/nut \
            --with-altpidpath=/var/run/nut \
            --with-drvpath=/lib/nut \
            --with-cgipath=/usr/lib/cgi-bin/nut \
            --with-pidpath=/var/run/$name \
            --with-user=$name \
            --with-cgi \
            --without-ssl
```

Then, stopped systemd servces, intalled new nut binaries, restarted udev and systemd services.

```
upsd -V
Network UPS Tools upsd 2.8.2.2396-2396+g31650d455 (development iteration after 2.8.2)

upsc -V
Network UPS Tools upsc 2.8.2.2396-2396+g31650d455 (development iteration after 2.8.2)

upsmon -V
Network UPS Tools upsmon 2.8.2.2396-2396+g31650d455 (development iteration after 2.8.2)
```

However, **driver.version** is still **2.8.0**. Does that mean the driver wasn't updated?

```
upsc ups1
driver.version: 2.8.0
```

jimklimov, 2025-03-16:
Seems so, maybe you did not disable/enable the services to re-register new contents?

ce3a, 2025-03-17:
Thanks @jimklimov, it seems I had missed the `--with-usb` option.

CyberPower (CPS) OR600ERM1U (230V, 50Hz): output.voltage looks OK.

Tested with master commit 31650d455333b73c2a98618c4897c42363a0d4e7 (16 Mar 2025).

```
battery.charge: 97
battery.charge.low: 30
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4160
battery.runtime.low: 180
battery.status: 100%
battery.type: PbAcid
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: CPS
device.model: OR600ERM1U
device.serial: GA6PQ2000138
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.override.battery.charge.low: 30
driver.parameter.override.battery.runtime.low: 180
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: OR600ERM1U
driver.parameter.productid: 0601
driver.parameter.serial: GA6PQ2000138
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.2.2396-2396+g31650d455
driver.version.data: CyberPower HID 0.83
driver.version.internal: 0.61
driver.version.usb: libusb-1.0.26 (API: 0x01000109)
input.sensitivity: normal
input.transfer.high: 257
input.transfer.low: 195
input.voltage: 226.0
input.voltage.nominal: 230
output.voltage: 226.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 120
ups.firmware: BF01917D6
ups.load: 20
ups.mfr: CPS
ups.model: OR600ERM1U
ups.power: 127
ups.power.nominal: 600
ups.productid: 0601
ups.realpower: 68
ups.realpower.nominal: 360
ups.serial: GA6PQ2000138
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

jimklimov, 2025-03-17:
Great, thanks for reporting. So it is indeed fixed already :)

marcg1, 2025-05-05:
2.8.3 reports the correct output voltage on a Cyberpower CP1500PFCLCD, matching the value on the frontpanel display.  This is on Debian Bookworm, which was previously running the packaged version, 2.8.0.  The packaged version reported ~135V.  The correct value is ~120V.  Driver is usbhid-ups.

Thanks for the nut package and this particular fix!