networkupstools/nut #2732: apc_modbus: "upsdrvctl start" fails

moritzkraemer, 2024-12-20:
Hi,

i compiled nut with the modbus support (https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol) and copied the apc_modbus driver into "/lib/nut/" but "upsdrvctl start" now fails:

`Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
modbus_connect: unable to connect: No such device
upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)`

my ups.conf:
`
[UPS1]
        driver = apc_modbus
        port = auto
        serial = ASxxxxxxxxxx
`

jimklimov, 2024-12-20:
Can you please run with more debug, and ideally the driver program itself?

What system do you run it on?

I wonder if it's a duplicate of #2289 or a separate issue?..

moritzkraemer, 2024-12-22:
```./apc_modbus -DDDDD -s apc -x vendorid=051D -x productid=0003 -x port=auto```

results in:

```   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000124     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.000143     [D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000157     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000206     [D3] main_arg: var='vendorid' val='051D'
   0.000249     [D5] send_to_all: SETINFO driver.parameter.vendorid "051D"
   0.000268     [D3] main_arg: var='productid' val='0003'
   0.000286     [D5] send_to_all: SETINFO driver.parameter.productid "0003"
   0.000300     [D3] main_arg: var='port' val='auto'
   0.000317     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000329     [D1] Network UPS Tools version 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/ups/include/modbus --with-modbus-libs='-L/usr/local/ups/lib -lmodbus'
   0.000363     [D1] debug level is '5'
   0.000379     [D5] send_to_all: SETINFO driver.debug "5"
   0.000395     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000666     [D1] Can not become_user(nobody): not root initially, remaining UID=1000 GID=1000
   0.000703     [D1] Signalling UPS [apc]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.000730     Can't open /var/state/ups/apc_modbus-apc: No such file or directory
   0.000745     [D1] Request for other driver to exit returned code -1
   0.000760     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.000777     [D5] send_to_all: SETINFO device.type "ups"
   0.000793     [D5] send_to_all: SETINFO driver.state "init.device"
   0.009415     modbus_connect: unable to connect: No such device
   0.009445     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.009460     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
   0.009478     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true```

jimklimov, 2024-12-22:
Strange, it does not even seem to try finding the device. Also it tries to run as `nobody` (by default, as you did not `configure` any `--with-user` explicitly) but you seem to have started as someone else. Maybe the resulting account does not have permissions to the USB device FS node.

Try running via `sudo` or similar, to start as `root`, and pass `-u root` or `-x user=root` to rule put that devfs permissions are a problem.

Also, just in case - are you sure your model/firmware version should talk modbus and not USB HID or some other protocol?

moritzkraemer, 2024-12-23:
sudo and all the other options give the same result

the output of `sudo nut-scanner -U`:

```
[nutdev1]
        driver = "apc_modbus"
        port = "auto"
        vendorid = "051D"
        productid = "0003"
        product = "Smart-UPS_1500 FW:UPS 06.0 / ID=1027"
        serial = "ASxxxxxxxxxxx"
        vendor = "American Power Conversion"
        bus = "001"
        device = "002"
        busport = "004"
        ###NOTMATCHED-YET###bcdDevice = "0001"
```

usb-hid driver works but doesnt give all the values

moritzkraemer, 2025-01-02:
So i've just been able to update the firmware on the device and now the output is different:

`sudo ./apc_modbus -DDDDD -s apc -x vendorid=051D -x productid=0003 -x port=auto -x user=root`

returns:

```
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000105     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.000119     [D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000131     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000153     [D3] main_arg: var='vendorid' val='051D'
   0.000169     [D5] send_to_all: SETINFO driver.parameter.vendorid "051D"
   0.000201     [D3] main_arg: var='productid' val='0003'
   0.000215     [D5] send_to_all: SETINFO driver.parameter.productid "0003"
   0.000228     [D3] main_arg: var='port' val='auto'
   0.000242     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000270     [D3] main_arg: var='user' val='root'
   0.000283     [D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nobody'
   0.000295     [D1] Overriding previously specified user 'nobody' with 'root' specified for driver section
   0.000305     [D1] Network UPS Tools version 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/ups/include/modbus --with-modbus-libs='-L/usr/local/ups/lib -lmodbus'
   0.000332     [D1] debug level is '5'
   0.000345     [D5] send_to_all: SETINFO driver.debug "5"
   0.000358     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001215     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001247     [D1] Signalling UPS [apc]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001284     Can't open /var/state/ups/apc_modbus-apc: No such file or directory
   0.001295     [D1] Request for other driver to exit returned code -1
   0.001305     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001317     [D5] send_to_all: SETINFO device.type "ups"
   0.001328     [D5] send_to_all: SETINFO driver.state "init.device"
   0.105070     [D3] match_function_regex: matching a device...
   0.105179     [D2] match_function_regex: failed match of VendorID: 8087
   0.105203     [D1] _apc_modbus_usb_callback: Failed to match!: Success
   0.125373     [D3] match_function_regex: matching a device...
   0.125473     [D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003)
   0.125546     [D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS_1500 FW:UPS 18.0 / ID=1027 (USB VID/PID 051d:0003) report descriptors
   0.126014     [D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.126057     [D5] send_to_all: SETINFO ups.vendorid "051d"
   0.126091     [D5] send_to_all: SETINFO ups.productid "0003"
   0.141736     [D5] send_to_all: SETINFO driver.state "init.quiet"
   0.141809     [D5] send_to_all: SETINFO driver.version "2.8.2.1725-1725-gc641cff57"
   0.141843     [D5] send_to_all: SETINFO driver.version.internal "0.11"
   0.141880     [D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.141931     [D5] send_to_all: SETINFO driver.state "init.info"
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
Aborted
```

jimklimov, 2025-01-02:
Hm, so it seems the driver indeed is not compatible with all firmwares out there, to the point of not even detecting them as a device (or the firmware not representing itself as a device visible to the OS)?

From my reading, your FW version changed quite significantly - from 06.0 to 18.0, right?

> ````
> *** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
> Aborted
> ````

Now this issue is unfortunately known with some devices, although not fully understood yet, tracked in #2289 / #2609 - and after the firmware update, your case seems to have become a duplicate of those.

FWIW, there is a fair amount of log notes in those issues (especially the latter) jotted down while I was reading through code and log dumps. One question from those still stands:

> Just in case, do your devices also present as a serial port (e.g. `/dev/ttyUSB0`)? That might work differently as a serial Modbus RTU.
> An alternative experiment could be to hack around apc_modbus.c and use the second connection method from the library (also usb, but for paths).

moritzkraemer, 2025-01-02:
The FW changed fron 06.0 to 18.0, right correct.

The Device does not appear as a serial device as far as i can tell.

NUT is running on Ubuntu 24.04.1 LTS (GNU/Linux 6.8.0-51-generic x86_64).
I would be able to test some different distros any specific you would suggest to try?

jimklimov, 2025-01-03:
Well, it is good to know that the problem is not ARM-specific, at least. I don't think switching distros would help at the moment, the problem has to be between new codebases in apc_modbus driver and the fork of libmodbus, as explored in those issues...