networkupstools/nut #2735: EcoFlow River 3 Plus Support

notnarb, 2024-12-23:
EcoFlow is one of the more popular companies amongst the recent wave cheaper, consumer-facing LiFePo power station / batteries.

Their new "River 3 Plus" has a USB-B port and acts as a USB HID UPS device.

I recently purchased one and would love to see NUT support eventually.

---

The device I have it hooked up to is running an older version of NUT (2.7.x) on FreeBSD which I don't have as a compiler target but I wanted to dump what limited info I have so far:

`./nut-ddl-dump.sh` output (usbhid-ups, explore = 1, vendorId = 3746) 

<details><summary>File contents</summary>

```
EcoFlow__EF-UPS-R3P__usbhid-ups__2.7.4__01.dev generated using ups on Sun, 22 Dec 2024 19:38:06 -0800
Please revise and post as a pull request to https://github.com/networkupstools/nut-ddl/
root@twonas[~]# less EcoFlow__EF-UPS-R3P__usbhid-ups__2.7.4__01.dev 
# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Sun, 22 Dec 2024 19:38:06 -0800
# DEVICE:EOC

# :; upsc ups
device.mfr: EcoFlow
device.model: EF-UPS-R3P
device.serial: <SNIP>
device.type: ups
driver.name: usbhid-ups
driver.parameter.explore: 1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.parameter.vendorid: 3746
driver.version: 2.7.4
driver.version.data: EXPLORE HID 0.1
driver.version.internal: 0.41
ups.mfr: EcoFlow
ups.model: EF-UPS-R3P
ups.productid: ffff
ups.serial: <SNIP>
ups.status: OB
ups.vendorid: 3746

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: ups
#
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: ups
# Instant commands supported on UPS [ups]:
# DEVICE:EOC
``` 
</details> 

`/usr/local/libexec/nut/usbhid-ups -DD -a ups -x productid=ffff -x vendorid=3746 -x explore=1` output:

<details><summary>Command output:</summary>

```
   0.004043     Checking device (3746/FFFF) (/dev/usb//dev/ugen0.2)
   0.006440     - VendorID: 3746
   0.006450     - ProductID: ffff
   0.006454     - Manufacturer: EcoFlow
[... 61 lines cut ...]
   0.011309     Path: UPS.PowerSummary.PresentStatus.ShutdownImminent, Type: Input, ReportID: 0x07, Offset: 11, Size: 1, Value: 0
   0.011316     Path: UPS.PowerSummary.PresentStatus.ShutdownImminent, Type: Feature, ReportID: 0x07, Offset: 11, Size: 1, Value: 0
   0.011323     Path: UPS.PowerSummary.PresentStatus.CommunicationLost, Type: Input, ReportID: 0x07, Offset: 12, Size: 1, Value: 0
   0.011331     Path: UPS.PowerSummary.PresentStatus.CommunicationLost, Type: Feature, ReportID: 0x07, Offset: 12, Size: 1, Value: 0
   0.011338     Path: UPS.PowerSummary.PresentStatus.Overload, Type: Input, ReportID: 0x07, Offset: 13, Size: 1, Value: 0
   0.011346     Path: UPS.PowerSummary.PresentStatus.Overload, Type: Feature, ReportID: 0x07, Offset: 13, Size: 1, Value: 0
   0.011362     Report descriptor retrieved (Reportlen = 403)
   0.011366     Found HID device
   0.011370     Detected a UPS: EcoFlow/EF-UPS-R3P
   0.011376     find_nut_info: unknown info type: load.off.delay
   0.011380     find_nut_info: unknown info type: load.on.delay
   0.011383     find_nut_info: unknown info type: load.off.delay
   0.011392     upsdrv_initinfo...
   0.011400     upsdrv_updateinfo...
   0.257337     libusb_get_interrupt: Connection timed out
   0.257359     Got 0 HID objects...
   0.257365     Quick update...
   0.257469     dstate_init: sock /var/db/nut/usbhid-ups-ups open on fd 6
   0.257494     upsdrv_updateinfo...
   0.519005     libusb_get_interrupt: Connection timed out
   0.519019     Got 0 HID objects...
   0.519024     Quick update...
```

</details>

ykuksenko, 2024-12-26:
The **EcoFlow Delta 3 Plus** model seems to be almost identical. It has a USB-C port that provides UPS functions.
Product page: https://us.ecoflow.com/products/delta-3-series-portable-power-station
Using NUT 2.8.2.

```
# lsusb
Bus 001 Device 025: ID 3746:ffff EcoFlow EF-UPS-DELTA
```

<details>
<summary>`./nut-ddl-dump.sh` output</summary>

```

# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: https://us.ecoflow.com/products/delta-3-series-portable-power-station

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Wed, 25 Dec 2024 17:46:22 -0600
# DEVICE:EOC

# :; upsc ecoflow
device.mfr: EcoFlow
device.model: EF-UPS-DELTA
device.serial: <SNIP>
device.type: ups
driver.debug: 2
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.explore: 1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: auto
driver.parameter.vendorid: 3746
driver.state: updateinfo
driver.version: 2.8.2
driver.version.data: EXPLORE HID 0.2
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
ups.mfr: EcoFlow
ups.model: EF-UPS-DELTA
ups.productid: ffff
ups.serial: <SNIP>
ups.status: OB
ups.vendorid: 3746

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: ecoflow
# [driver.debug]
# Current debug verbosity level of the driver program
# Type: NUMBER
# Value: 2
#
# [driver.flag.allow_killpower]
# Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
# Type: NUMBER
# Value: 0
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: ecoflow
# Instant commands supported on UPS [ecoflow]:
#
# driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
# driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
# driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
[... 148 lines cut ...]
  1.560238     [D2] send_to_one: sending SETINFO ups.model "EF-UPS-DELTA"
  1.560260     [D2] send_to_one: sending SETINFO ups.productid "ffff"
  1.560288     [D2] send_to_one: sending SETINFO ups.serial "<SNIP>"
  1.560312     [D2] send_to_one: sending SETINFO ups.status "OB"
  1.560333     [D2] send_to_one: sending SETINFO ups.vendorid "3746"
  1.560355     [D2] send_to_one: sending ADDCMD driver.killpower
  1.560377     [D2] send_to_one: sending ADDCMD driver.reload
  1.560398     [D2] send_to_one: sending ADDCMD driver.reload-or-error
  1.560435     [D2] send_to_one: sending ADDCMD driver.reload-or-exit
  1.560458     [D2] send_to_one: sending DATAOK
  1.560479     [D2] send_to_one: sending DUMPDONE
  1.809381     [D1] upsdrv_updateinfo...
  2.559844     [D2] nut_libusb_get_interrupt: Connection timed out
  2.559913     [D1] Got 0 HID objects...
  2.559935     [D1] Quick update...
  2.809677     [D1] upsdrv_updateinfo...
  3.560098     [D2] nut_libusb_get_interrupt: Connection timed out

```

</details>

Let me know if any other output would be useful. (I can get output for River 3 Plus also)

Hope this helps. Thank you.

jimklimov, 2024-12-26:
Hello, thanks for the reports. I think an additional obe would not hurt, although I'm not at a computer to actually look at them during the hokiday season. If anyone (of you?) has time to follow through with the NUT sources' docs and script to generate the HID subdriver, you might have one sooner ;)

ykuksenko, 2024-12-27:
Here is the data for EcoFlow River 3 Plus that I have access to. I am using the same machine(pi4)/software for both the Delta and River models - just swapping USB port from one to the other.

```
# lsusb
Bus 001 Device 026: ID 3746:ffff EcoFlow EF-UPS-R3P
```

<details>
<summary>`./nut-ddl-dump.sh` output</summary>

```
cat EcoFlow__EF-UPS-R3P__usbhid-ups__2.8.2__01.dev
# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Wed, 25 Dec 2024 17:38:55 -0600
# DEVICE:EOC

# :; upsc ecoflow
device.mfr: EcoFlow
device.model: EF-UPS-R3P
device.serial: <SNIP>
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.explore: 1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.pollonly: enabled
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: auto
driver.parameter.vendorid: 3746
driver.state: quiet
driver.version: 2.8.2
driver.version.data: EXPLORE HID 0.2
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
ups.mfr: EcoFlow
ups.model: EF-UPS-R3P
ups.productid: ffff
ups.serial: <SNIP>
ups.status: OB
ups.vendorid: 3746

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: ecoflow
# [driver.debug]
# Current debug verbosity level of the driver program
# Type: NUMBER
# Value: 0
#
# [driver.flag.allow_killpower]
# Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)
# Type: NUMBER
# Value: 0
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: ecoflow
# Instant commands supported on UPS [ecoflow]:
#
# driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
# driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
# driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
# driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
[... 98 lines cut ...]
   0.046371	[D2] Found HID device
   0.046390	[D1] Detected a UPS: EcoFlow/EF-UPS-R3P
   0.046420	[D2] find_nut_info: unknown info type: load.off.delay
   0.046437	[D2] find_nut_info: unknown info type: load.on.delay
   0.046453	[D2] find_nut_info: unknown info type: load.off.delay
   0.046480	[D1] upsdrv_initinfo...
   0.046505	[D1] upsdrv_updateinfo...
   0.796924	[D2] nut_libusb_get_interrupt: Connection timed out
   0.796993	[D1] Got 0 HID objects...
   0.797015	[D1] Quick update...
   0.797392	[D2] dstate_init: sock /var/lib/nut/usbhid-ups-ecoflow open on fd 9
   0.797436	Running as foreground process, not saving a PID file
   0.797464	[D1] Driver initialization completed, beginning regular infinite loop
   0.797484	upsnotify: notify about state 2 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   0.797511	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.797528	upsnotify: logged the systemd watchdog situation once, will not spam more about it
   0.797546	[D1] upsdrv_updateinfo...
   1.547910	[D2] nut_libusb_get_interrupt: Connection timed out
   1.547981	[D1] Got 0 HID objects...
   1.548004	[D1] Quick update...
   1.797846	[D1] upsdrv_updateinfo...
   2.548295	[D2] nut_libusb_get_interrupt: Connection timed out
```

</details>

ykuksenko, 2024-12-28:
I tried to create a sub driver for the `delta 3 plus` with firmware V6.31.48.29.
- the output data is quirky.
- I was not able to control the load or beeper - not sure if it is possible or if I just do not understand how. Any suggestions welcome.
- I will try to get a PR filed when I have more time. I'll document quirks in the sub driver .c file - not sure where to do it for now.
- Also need to test with `river 3 plus` to see if quirks are the same.

<details>
<summary>here are sample output data from `upsc` with some quirks documented for the curious</summary>

```
battery.capacity: 100 //unit is percent - NUT docs say this should be Ah but percent may still be useful
battery.capacity.nominal: 100 //unit is percent
battery.charge: 31 //accurate to display, shuts of at battery.charge.low +1% (might be due to slow update rate)
battery.charge.low: 30 //matches APP's battery discharge limit setting
battery.charge.warning: 5 //fixed value AFAICT -- seems useless
battery.runtime: 32 //seems to be for total battery capacity, ignoring discharge limit setting, unit is minutes
battery.runtime.low: 120 //fixed value in minutes AFAICT -- seems useless
device.mfr: EcoFlow
device.model: EF-UPS-DELTA
device.serial: <SNIP>
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.2.1775-1775-g4e9eb9ad1
driver.version.data: EcoFlow HID 0.2
driver.version.internal: 0.60
driver.version.usb: libusb-0.1 (or compat)
ups.beeper.status: enabled
ups.mfr: EcoFlow
ups.model: EF-UPS-DELTA
ups.productid: ffff
ups.serial: <SNIP>
ups.status: OB
ups.vendorid: 3746
```

</details>

emc2cube, 2025-01-27:
A new firmware was released for the River 3+ v.1.32.76.52 that add support for power management software.

Sadly can't provide output from usbhid-ups as I'm getting a pesky "libusb1: Could not open any HID devices: insufficient permissions on everything" error, but hopefully that firmware will help for adding NUT support.

bleomycin, 2025-02-28:
> A new firmware was released for the River 3+ v.1.32.76.52 that add support for power management software.
> `/usr/lib/nut/usbhid-ups -DD -a ecoflow -x productid=ffff -x vendorid=3746 -x explore=1` output:

Sorry if I'm just being dense but does this mean you have nut successfully working with your river 3+? I'm also on the latest ecoflow firmware and running nut 2.8.2 but haven't had any success.

Do you mind sharing the relevant parts of your working configuration?

emc2cube, 2025-02-28:
No I didn't create a driver myself for my unit, I'm merely providing the debug infos in case it can help someone without a River3+ to implement it.

On a side note reading the manual for the upcoming official power manager https://manuals.ecoflow.com/eu/product/power-manager?lang=en_US&source=desktop I suspect nut may already come installed on the River3+ itself.

jimklimov, 2025-02-28:
Well, having a NUT (or compatible) data server on the UPS itself is a neat idea, just asking for itself to be implemented for ages; part of why RFC 9271 came along to codify the NUT network protocol. That way numerous computers could just be running an `upsmon`, with the UPS itself being responsible for FSD, waiting on clients to go away, and its own ultimate powerdown/powercycle.

But can you please point where in those docs you saw suggestions that NUT could be part of the UPS? I only saw mentions of adding packages/plugins into NAS appliances...

0x2501, 2025-02-28:
I don't think he's saying that NUT could be part of the UPS, but rather that NUT supports the River 3+ natively. The link he added shows how to install NUT in a Synology NAS.

emc2cube, 2025-02-28:
Both, my OP is just providing raw USB debugging infos for nut to support the ecoflow if plugged directly by USB, as previously provided by @notnarb and @ykuksenko

@jimklimov in the manual https://manuals.ecoflow.com/eu/product/power-manager?lang=en_US&source=desktop#e8dd82a5-094f-48be-a9fe-0f94739367c9 section "Connected devices" then subsection "Remote mode".
I may be misunderstanding it, and the install would activate a nutserver on the NAS plugged to the R3+ that then could be accessed by a Win computer.
Anyway judging by the install instructions looks like we may just be able to setup `upsmon` with a line such as
`MONITOR nutdev1@<ecoflow_or_NAS_IP>:3496 1 efsz <password_to_figure_out_from_GUI> <primary/secondary>` and something would act as a nut server which would be a good start.

Worst case, if the nutserver run on the NAS, it mean that they will be making a driver themselves to run on NAS, so if they don't contribute to an official nut driver maybe we can extract some infos from theirs.

greyltc, 2025-03-07:
I've just picked up a River 3 Plus. My unit is running (the latest) firmware version V1.32.76.52

At the risk of adding nothing new or useful, here's the info I've been able to gather from my unit:

<details>
  <summary>journal prints on attach:</summary>

```
Mar 06 19:38:59 box kernel: usb 3-6: new full-speed USB device number 7 using xhci_hcd
Mar 06 19:38:59 box kernel: usb 3-6: New USB device found, idVendor=3746, idProduct=ffff, bcdDevice= 1.00
Mar 06 19:38:59 box kernel: usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 06 19:38:59 box kernel: usb 3-6: Product: EF-UPS-RIVER 3 Plus
Mar 06 19:38:59 beans kernel: usb 3-6: Manufacturer: EcoFlow
Mar 06 19:38:59 box kernel: usb 3-6: SerialNumber: REDACTED
Mar 06 19:38:59 box kernel: hid-generic 0003:3746:FFFF.0003: unbalanced collection at end of report description
Mar 06 19:38:59 box kernel: hid-generic 0003:3746:FFFF.0003: probe with driver hid-generic failed with error -22
Mar 06 19:38:59 box kernel: cdc_acm 3-6:1.1: ttyACM0: USB ACM device
Mar 06 19:38:59 box mtp-probe[46033]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-6"
Mar 06 19:38:59 box mtp-probe[46033]: bus: 3, device: 7 was not an MTP device
Mar 06 19:38:59 box mtp-probe[46042]: checking bus 3, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-6"
Mar 06 19:38:59 box mtp-probe[46042]: bus: 3, device: 7 was not an MTP device
```
</details>

<details>
  <summary><code>$ sudo lsusb -vv -d 3746:</code></summary>

```
Bus 003 Device 007: ID 3746:ffff EcoFlow EF-UPS-RIVER 3 Plus
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x3746 EcoFlow
  idProduct          0xffff EF-UPS-RIVER 3 Plus
  bcdDevice            1.00
  iManufacturer           1 EcoFlow
  iProduct                2 EF-UPS-RIVER 3 Plus
  iSerial                 3 REDACTED
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
[... 211 lines cut ...]
   0.012546	[D2] find_nut_info: unknown info type: load.off.delay
   0.012549	[D2] find_nut_info: unknown info type: load.on.delay
   0.012552	[D2] find_nut_info: unknown info type: load.off.delay
   0.012563	[D1] upsdrv_initinfo...
   0.012572	[D1] upsdrv_updateinfo...
   0.762900	[D2] nut_libusb_get_interrupt: Connection timed out
   0.762918	[D1] Got 0 HID objects (1 times in a row, tolerance is -1)...
   0.762924	[D1] Quick update...
   0.763122	Listening on socket /var/lib/nut/usbhid-ups-ecoflow
   0.763127	[D2] dstate_init: sock /var/lib/nut/usbhid-ups-ecoflow open on fd 9
   0.763139	Running as foreground process, not saving a PID file
   0.763147	[D1] Driver initialization completed, beginning regular infinite loop
   0.763152	upsnotify: notify about state NOTIFY_STATE_READY_WITH_PID with libsystemd: was requested, but not running as a service unit now, will not spam more about it
   0.763157	[D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
   0.763160	upsnotify: failed to notify about state NOTIFY_STATE_READY_WITH_PID: no notification tech defined, will not spam more about it
   0.763165	upsnotify: logged the systemd watchdog situation once, will not spam more about it
   0.763168	[D1] upsdrv_updateinfo...
   1.513808	[D2] nut_libusb_get_interrupt: Connection timed out
   1.513839	[D1] Got 0 HID objects (2 times in a row, tolerance is -1)...
   1.513845	[D1] Quick update...
   2.764480	[D1] upsdrv_updateinfo...
   3.515118	[D2] nut_libusb_get_interrupt: Connection timed out
   3.515154	[D1] Got 0 HID objects (3 times in a row, tolerance is -1)...
```
</details>

greyltc, 2025-03-07:
I think I've got this working. My PR is here: https://github.com/networkupstools/nut/pull/2837

Here are some screenshots of nut-monitor for my River 3 Plus:
[image]
[image]

sondercoder, 2025-03-15:
The Windows and NAS applications have now been released, if this helps intergrate the ecoflow river.  Hopefully the Delta 3 Plus as well.  https://www.ecoflow.com/us/support/download/index

BitPopCoin, 2025-03-31:
They do not act as a USB HID UPS!

jimklimov, 2025-03-31:
@BitPopCoin : did you have a chance to check them with a recent NUT for Windows build? If the code proposed in this tread and related PRs was good for other platforms, NUT CI artifacts of `master` branch builds like https://ci.appveyor.com/project/nut-travis/nut/builds/51793727/artifacts might just work (possibly after fiddling with Zadig to attach libusb to needed VID:PID), see https://github.com/networkupstools/nut/wiki/NUT-for-Windows

peros550, 2025-04-12:
Guys, excuse my ignorance. I just received a River 3 plus, and show on their web page that it would be supposingly support connection with Synology NAS 923+.  I followed all their steps ,installed extra software but the NAS itself is not able to identify it as connected UPS device. Is the above proposed PRs going to be handy for the Synology part? Any guide on how I could install them?

jimklimov, 2025-04-12:
@peros550 : What is the "extra software" - does it seem to involve NUT? If yes - which version?

The "above proposed PRs" were merged, so are now part of NUT master branch on GitHub and will be in 2.8.3 release shortly (going through release candidate cycles now).

How to get that built into Synology specifically - sorry, got no clue on this one...

peros550, 2025-04-12:
> [@peros550](https://github.com/peros550) : What is the "extra software" - does it seem to involve NUT? If yes - which version?
>
> The "above proposed PRs" were merged, so are now part of NUT master [...]

I followed the instructions [here](https://manuals.ecoflow.com/eu/product/power-manager?lang=en_US&source=desktop). It looks, they instruct you to have some sort of NUT server running on your NAS, which runs on 3496 port. Then , their Desktop app can connect to it and pull data. 

I was succesful in connecting a Home Assistant NUT integration, after I got the port, username and password from the Desktop app. However, the Synology NAS UI couldn't connect to it, probably because it only accepts an IP address for an external NUT and it didn't accept "Syno 923 LAN IP":3496. I dont know which version runs in Ecoflow's nut server version, sorry.

greyltc, 2025-04-14:
> Hi! Any chance to get the current load?

I was seeing a live power reading appear in the `ups.power.nominal` variable while I was putting things together (see my above comment: https://github.com/networkupstools/nut/issues/2735#issuecomment-2705700983). I can't honestly remember if this number is power drawn from the wall by the UPS unit or power delivered by the UPS unit to the load.

diego351, 2025-04-14:
@greyltc It's actualyt the battery capacity in Wh:

[image]

diego351, 2025-04-14:
I actually have other problem: upsrw doesn't save value even though they though it confirmed "OK". 

[image]

Also, the `driver.flag.allow_killpower` flag value was reset after restart, is it the desired behaviour? Sorry, it's my first experience with nut.

jimklimov, 2025-04-14:
"OK" with `upsrw` and `upscmd` means only that the server connection worked and command was not instantly rejected, you may want `-w` to wait for its result.

Writes with these commands may be or not be saved by the device (depends on its firmware and whether there's R/W flash on board or only RAM), and are not saved by NUT (recommended ownership/permission on config files should forbid it too). You can explicitly configure options in ups.conf, as well as default or override values for device readings.

A nominal power/current/charge/voltage/... is just that - max safe rating burned into the device/firmware (if any).

greyltc, 2025-04-14:
> [@greyltc](https://github.com/greyltc) It's actualyt the battery capacity in Wh:

oh!

Funny coincidence that is pretty much what my typical load is.

Capacity (energy) being reported into a variable with "power" in the name is annoying nonsense...

Anyhow, I think we'll need a new firmware from ecoflow that exposes load monitoring. Is there anything newer than V1.32.76.52 yet?

jimklimov, 2025-04-14:
For named value definitions, check dosc/nut-names.adoc in sources.

"Power" is the "apparent" one, in VA. There may be "realpower" in Watts (power modulo crest factor). Capacity is in Ah, etc.

That said, there may be errors in mapping from device reports to chosen NUT-standard names, and generally speaking, there have been errors in firmwares too. Code is code.

emc2cube, 2025-04-14:
> Anyhow, I think we'll need a new firmware from ecoflow that exposes load monitoring. Is there anything newer than V1.32.76.52 yet?

I have 1.33.81.53 installed since end of March, looks pretty much the same in explore but may have other changes under the hood .

<details>
<summary>`/usr/lib/nut/usbhid-ups -u root -DD -a ecoflow -x productid=ffff -x vendorid=3746 -x explore=1` output:</summary>

```
   0.031402	[D2] Checking device 3 of 7 (3746/FFFF)
   0.048601	[D2] - VendorID: 3746
   0.048625	[D2] - ProductID: ffff
   0.048633	[D2] - Manufacturer: EcoFlow
   0.048642	[D2] - Product: EF-UPS-RIVER 3 Plus
   0.048650	[D2] - Serial Number: <snip>
   0.048658	[D2] - Bus: 003
   0.048661	[D2] - Device: unknown
   0.048664	[D2] - Device release number: 0100
   0.048666	[D2] Trying to match device
   0.048668	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.048742	[D2] Device matches
   0.048745	[D2] Reading first configuration descriptor
   0.048767	[D2] Claimed interface 0 successfully
   0.051604	[D2] Warning: two different HID descriptors retrieved (Reportlen = 403 vs. 376)
   0.051609	[D2] HID descriptor length 376
   0.059601	[D2] Report Descriptor size = 376
   0.059627	Using subdriver: EXPLORE HID 0.2
   0.059630	[D1] 51 HID objects found
   0.062606	[D1] Path: UPS.Flow.[4].ConfigActivePower, Type: Feature, ReportID: 0x01, Offset: 0, Size: 16, Value: 286
   0.065602	[D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 1
   0.068601	[D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x1f, Offset: 0, Size: 8, Value: 0
   0.071598	[D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x20, Offset: 0, Size: 8, Value: 0
   0.074602	[D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x16, Offset: 0, Size: 8, Value: 1
   0.077606	[D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x10, Offset: 0, Size: 8, Value: 1
   0.080602	[D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 1
   0.083605	[D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 100
   0.086603	[D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x17, Offset: 0, Size: 8, Value: 100
   0.089604	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x0c, Offset: 0, Size: 8, Value: 100
   0.089609	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 100
   0.092618	[D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 8, Value: 10
   0.095605	[D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x11, Offset: 0, Size: 8, Value: 0
   0.098607	[D2] refresh_report_buffer: expected 3 bytes, but got 2 instead
   0.098613	[D1] Path: UPS.PowerSummary.ManufacturerDate, Type: Feature, ReportID: 0x09, Offset: 0, Size: 16, Value: 0
   0.101604	[D1] Path: UPS.PowerSummary.AverageTimeToFull, Type: Feature, ReportID: 0x1a, Offset: 0, Size: 16, Value: 7200
   0.104604	[D1] Path: UPS.PowerSummary.AverageTimeToEmpty, Type: Input, ReportID: 0x1c, Offset: 0, Size: 16, Value: 7200
   0.104609	[D1] Path: UPS.PowerSummary.AverageTimeToEmpty, Type: Feature, ReportID: 0x1c, Offset: 0, Size: 16, Value: 7200
   0.107604	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x0d, Offset: 0, Size: 16, Value: 2229
   0.107609	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 16, Value: 2229
   0.110603	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 0, Size: 16, Value: 600
   0.110608	[D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x08, Offset: 0, Size: 16, Value: 600
   0.113599	[D1] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: -1
   0.116604	[D1] Path: UPS.PowerSummary.DelayBeforeReboot, Type: Feature, ReportID: 0x13, Offset: 0, Size: 16, Value: -1
   0.119606	[D1] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 16, Value: 13.8
   0.122608	[D1] Path: UPS.PowerSummary.Voltage, Type: Input, ReportID: 0x0b, Offset: 0, Size: 16, Value: 13
   0.122614	[D1] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 16, Value: 13
   0.125611	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Input, ReportID: 0x14, Offset: 0, Size: 8, Value: 2
   0.125616	[D1] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 2
   0.128607	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x07, Offset: 0, Size: 1, Value: 0
   0.128612	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x07, Offset: 0, Size: 1, Value: 0
   0.128616	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x07, Offset: 1, Size: 1, Value: 0
   0.128620	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x07, Offset: 1, Size: 1, Value: 0
   0.128623	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x07, Offset: 2, Size: 1, Value: 1
   0.128627	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x07, Offset: 2, Size: 1, Value: 1
[... cut ...]
   0.878886	[D1] Quick update...
   0.878937	[D2] dstate_init: sock /run/nut/usbhid-ups-ecoflow open on fd 9
   0.878970	[D1] Group and/or user account for this driver was customized ('root:nut') compared to built-in defaults. Fixing socket '/run/nut/usbhid-ups-ecoflow' ownership/access.
   0.878982	[D1] Group access for this driver successfully fixed
   0.878992	[D1] upsdrv_updateinfo...
   1.629751	[D2] nut_libusb_get_interrupt: Connection timed out
   1.629769	[D1] Got 0 HID objects...
```
</details>

greyltc, 2025-04-16:
>    0.062606	[D1] Path: UPS.Flow.[4].ConfigActivePower, Type: Feature, ReportID: 0x01, Offset: 0, Size: 16, Value: 286

God only knows why their firmware engineer(s) chose to write the design capacity into a variable called `ConfigActivePower` and then a value of `100` into the variable called `DesignCapacity.` Is there a chance there's anything wrong with the HID scanner?

> I have 1.33.81.53

After looking over that, it looks like there's not much (if anything) new and useful is exposed, unsure if I'll bother with the upgrade from V1.32.76.52...

jimklimov, 2025-04-16:
> Is there a chance there's anything wrong with the HID scanner?

Generally, AFAIK no. There are issues with certain vendors' LogMin/LogMax encodings and/or some other bogus data - mitigation of these started around #1024 and later was streamlined for `usbhid-ups` with standardized fix-up methods; see cps-hid.c and apc-hid.c (IIRC) for implemented fixups.

greyltc, 2025-04-16:
```
...
   0.048745	[D2] Reading first configuration descriptor
   0.048767	[D2] Claimed interface 0 successfully
   0.051604	[D2] Warning: two different HID descriptors retrieved (Reportlen = 403 vs. 376)
   0.051609	[D2] HID descriptor length 376
   0.059601	[D2] Report Descriptor size = 376
   0.059627	Using subdriver: EXPLORE HID 0.2
   0.059630	[D1] 51 HID objects found
...
```
Is `two different HID descriptors retrieved (Reportlen = 403 vs. 376)` of any concern?

greyltc, 2025-04-16:
Also. it might be interesting to sniff the USB traffic between [their official app](https://manuals.ecoflow.com/eu/product/power-manager?lang=en_US&source=desktop) and the device.

emc2cube, 2025-04-17:
Not the source, but synology .spk packages are .tar.gz files, with an extra 'package.tgz' inside containing the actual unix files to be installed on the Syno system.
Can peak there for files / drivers etc.

ex ups.conf from that package:
```
[nutdev1]
	driver = usbhid-ups
	pollfreq = 1
	port = auto
	usb_hid_ep_in = 1
	usb_hid_ep_out = 1
	ignorelb
	override.battery.runtime.low = -1
```

I saw the EcoFlow Vendor ID 0x3746 being used in some files but in Eaton related sections (ex udev rules, etc/usb/hotplug/libhid.usermap) and got confused fairly quickly. Maybe someone more familiar with the software will be able to figure out / decompile these.

diego351, 2025-04-17:
@emc2cube 

Huh, nice!

I'll play with it when I get back home. 

`upsmon.conf`:
```
MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h +0"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG "/etc/killpower"
OFFDURATION 30
RBWARNTIME 43200
NOCOMMWARNTIME 300
FINALDELAY 5
RUN_AS_USER root
MONITOR nutdev1@127.0.0.1:3496 1 Ecoflow Ecoflow primary
SHUTDOWNCMD "/sbin/shutdown -h +0"
```

[Entire etc dir](https://drive.google.com/file/d/1iFr3odLKUNBMsKH_wfeFWEFIzUaF5jOJ/view?usp=sharing). I removed all the comments.

greyltc, 2025-04-17:
It seems they're in violation of the licensing terms here unless they make their modifications to NUTs source code available. They can't just "sell"/distribute precompiled binaries like this.

jimklimov, 2025-04-17:
Well, IANAL but my reading is that GPL requires that customers are given sources.
* Customers are those who get/buy a product;
* To "get" they should ask for it.
* Reasonable costs" to deliver a copy may apply.

* There are pro/contra arguments about add-on recipes and methodology transfer too.

A breach of license begins when such request is ignored or denied.

Preemptive publication is a common shortcut to make life easier for all sides, but not a requirement of the license. Neither is downloadability - for all they care, CDs, punch cards or printouts also conform to the letter...

jimklimov, 2025-04-17:
Also I have no idea if they did change NUT sources, or used a 2.8.2 (per strings in binaries) tarball "as is", and integration with Synology is all scripted so not a "derived work"... I did not dig deep so just don't know.

greyltc, 2025-04-17:
> Also I have no idea if they did change NUT sources

yeah, I was just thinking they must have made modifications because the standard HID scanner tool here doesn't show any variable that reports real time power and that is something that seems to be available in their solution (by inspection of the screenshots in their documentation at https://manuals.ecoflow.com/eu/product/power-manager?lang=en_US&source=desktop although some screenshots do show `0W` while other show non zero numbers ). I don't have the supported NAS hardware to test this on my own.

earthlabinfo, 2025-04-17:
I also took a look inside the package.
Even though Synology products already have built-in UPS management software (which I believe is also based on NUT), it looks like they've gone out of their way to include a bunch of extra components.

All I really want is for the system to shut down safely when the power goes out or the battery level gets too low—I don’t need detailed information about the portable power station.

I'm not hoping for an EcoFlow-style custom implementation. I'd prefer it to just work with NUT directly.

That said, since the River 3 has Wi-Fi functionality, it would make much more sense if it could act as a NUT server and communicate over the network. Then we wouldn’t need to rely on USB at all—but that doesn't seem to be the case right now.

greyltc, 2025-04-17:
well IANAL either, but it seems to me that since they're distributing precompiled NUT binaries on their site (doesn't matter if they have modified the source or not), they must also "provide clear instructions people can follow to obtain the source"\[1] those bins were built from

\[1]: https://www.gnu.org/licenses/gpl-faq.html#SourceAndBinaryOnDifferentSites

I don't see any instructions (have I missed them? maybe the instructions are packed in with the download somehow?). So maybe just the lack of clear instructions puts them in violation already.

Not sure if this source code discussion here is off topic or not: though if they did provide the source code and they did have to modify it to work with their product, then we might be able to use that to more effectively and completely close this issue.

greyltc, 2025-04-17:
from the etc folder previously extracted and posted above in comment https://github.com/networkupstools/nut/issues/2735#issuecomment-2812602960:

```
curl -sL 'https://drive.usercontent.google.com/u/0/uc?id=1iFr3odLKUNBMsKH_wfeFWEFIzUaF5jOJ&export=download' | tar xaj --to-command 'grep -E -Hn --label="$TAR_FILENAME" Ecoflow -B12 -A4 || true'
etc/upsd.users:1:[Ecoflow]
etc/upsd.users:2:        password  = Ecoflow
etc/upsd.users-3-        actions = SET
etc/upsd.users-4-        actions = FSD
etc/upsd.users-5-        instcmds = ALL
etc/upsd.users-6-        upsmon primary
etc/upsmon.conf-1-MINSUPPLIES 1
etc/upsmon.conf-2-SHUTDOWNCMD "/sbin/shutdown -h +0"
etc/upsmon.conf-3-POLLFREQ 5
etc/upsmon.conf-4-POLLFREQALERT 5
etc/upsmon.conf-5-HOSTSYNC 15
etc/upsmon.conf-6-DEADTIME 15
etc/upsmon.conf-7-POWERDOWNFLAG "/etc/killpower"
etc/upsmon.conf-8-OFFDURATION 30
etc/upsmon.conf-9-RBWARNTIME 43200
etc/upsmon.conf-10-NOCOMMWARNTIME 300
etc/upsmon.conf-11-FINALDELAY 5
etc/upsmon.conf-12-RUN_AS_USER root
etc/upsmon.conf:13:MONITOR nutdev1@127.0.0.1:3496 1 Ecoflow Ecoflow primary
etc/upsmon.conf-14-SHUTDOWNCMD "/sbin/shutdown -h +0"
```

`0x3746` is their USB vendor ID...
```
$ curl -sL 'https://drive.usercontent.google.com/u/0/uc?id=1iFr3odLKUNBMsKH_wfeFWEFIzUaF5jOJ&export=download' | tar xaj --to-command 'grep -E -Hn --label="$TAR_FILENAME" 3746 -B7 -A3 || true'
etc/udev/rules.d/62-nut-usbups.rules-122-ATTR{idVendor}=="2e51", ATTR{idProduct}=="ffff", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-123-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0101", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-124-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0201", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-125-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0202", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-126-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0203", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-127-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0300", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-128-ATTR{idVendor}=="2e66", ATTR{idProduct}=="0302", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules:129:ATTR{idVendor}=="3746", ATTR{idProduct}=="0001", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules:130:ATTR{idVendor}=="3746", ATTR{idProduct}=="ffff", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-131-ATTR{idVendor}=="4234", ATTR{idProduct}=="0002", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-132-ATTR{idVendor}=="ffff", ATTR{idProduct}=="0000", MODE="666", GROUP="root"
etc/udev/rules.d/62-nut-usbups.rules-133-LABEL="nut-usbups_rules_end"
etc/hotplug/usb/libhid.usermap-117-libhidups      0x0003      0x2e51   0xffff    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-118-libhidups      0x0003      0x2e66   0x0101    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-119-libhidups      0x0003      0x2e66   0x0201    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-120-libhidups      0x0003      0x2e66   0x0202    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-121-libhidups      0x0003      0x2e66   0x0203    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-122-libhidups      0x0003      0x2e66   0x0300    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-123-libhidups      0x0003      0x2e66   0x0302    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap:124:libhidups      0x0003      0x3746   0x0001    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap:125:libhidups      0x0003      0x3746   0xffff    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-126-libhidups      0x0003      0x4234   0x0002    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/hotplug/usb/libhid.usermap-127-libhidups      0x0003      0xffff   0x0000    0x0000       0x0000       0x00         0x00            0x00            0x00            0x00               0x00               0x00000000
etc/devd/nut-usb.conf-1155-
etc/devd/nut-usb.conf-1156-# Eaton
etc/devd/nut-usb.conf-1157-#  various models  - usbhid-ups
etc/devd/nut-usb.conf-1158-notify 100 {
etc/devd/nut-usb.conf-1159-	match "system"		"USB";
etc/devd/nut-usb.conf-1160-	match "subsystem"	"DEVICE";
etc/devd/nut-usb.conf-1161-	match "type"		"ATTACH";
etc/devd/nut-usb.conf:1162:	match "vendor"		"0x3746";
[... cut ...]
etc/devd/nut-usb.conf-1166-#  various models  - usbhid-ups
etc/devd/nut-usb.conf-1167-notify 100 {
etc/devd/nut-usb.conf-1168-	match "system"		"USB";
etc/devd/nut-usb.conf-1169-	match "subsystem"	"DEVICE";
etc/devd/nut-usb.conf-1170-	match "type"		"ATTACH";
etc/devd/nut-usb.conf:1171:	match "vendor"		"0x3746";
etc/devd/nut-usb.conf-1172-	match "product"		"0xffff";
etc/devd/nut-usb.conf-1173-	action "chgrp root /dev/$cdev; chmod g+rw /dev/$cdev";
etc/devd/nut-usb.conf-1174-};
```
Maybe in their build, they're masquerading as an Eaton UPS?

greyltc, 2025-04-17:
One thing that's super wierd, is that in their build, they seem to have replaced the vendor ID in the manpage documentation example here
https://github.com/networkupstools/nut/blob/5d3ff46f93404c22ff1b743915e1437b0e3d9925/docs/man/usbhid-ups.txt#L285
with their own vendor ID. So maybe they've done a bunch of wild/blind find and replace on the whole codebase or something?

greyltc, 2025-04-18:
```
$ mkdir -p /tmp/econut
$ cd /tmp/econut
$ curl -s https://ecoflow-ups-us.oss-us-west-1.aliyuncs.com/PowerManager/2025/01/20/PowerManagerNUT-x86_64-1.0.1-SynologyDS923%2B.zip > econut.zip
$ binwalk -Me econut.zip

                                      /tmp/econut/extractions/econut.zip
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                ZIP archive, file count: 2, total size: 
                                                                      8014481 bytes
--------------------------------------------------------------------------------------------------------------
[+] Extraction of zip data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

          /tmp/econut/extractions/econut.zip.extracted/0/SimplePermissionManager_x64-7.0_1.0.0-9.spk
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                POSIX tar archive, file count: 22
--------------------------------------------------------------------------------------------------------------
[+] Extraction of tarball data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

        /tmp/econut/extractions/econut.zip.extracted/0/PowerManagerNUT-x86_64-1.0.1-SynologyDS923+.spk
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                POSIX tar archive, file count: 18
--------------------------------------------------------------------------------------------------------------
[+] Extraction of tarball data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

/tmp/econut/extractions/econut.zip.extracted/0/SimplePermissionManager_x64-7.0_1.0.0-9.spk.extracted/0/package.tgz
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                gzip compressed data, operating system: 
                                                                      Unix, timestamp: 1970-01-01 00:00:00, 
                                                                      total size: 1741322 bytes
--------------------------------------------------------------------------------------------------------------
[+] Extraction of gzip data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

/tmp/econut/extractions/econut.zip.extracted/0/SimplePermissionManager_x64-7.0_1.0.0-9.spk.extracted/0/package.tgz.extracted/0/decompressed.bin
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                POSIX tar archive, file count: 36
--------------------------------------------------------------------------------------------------------------
[+] Extraction of tarball data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

/tmp/econut/extractions/econut.zip.extracted/0/PowerManagerNUT-x86_64-1.0.1-SynologyDS923+.spk.extracted/0/package.tgz
--------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
--------------------------------------------------------------------------------------------------------------
0                                  0x0                                XZ compressed data, total size: 6205692 
                                                                      bytes
--------------------------------------------------------------------------------------------------------------
[+] Extraction of xz data at offset 0x0 completed successfully
--------------------------------------------------------------------------------------------------------------

[... cut ...]
battery.energysave.load
battery.energysave.delay
battery.energysave.realpower
ups.efficiency
UPS.PowerSummary.iVersion
ups.test.interval
ups.temperature
UPS.PowerSummary.Temperature
ups.power
ups.L1.power
ups.L2.power
ups.L3.power
ups.realpower
ups.L1.realpower
ups.L2.realpower
ups.L3.realpower
ups.start.battery
ups.start.reboot
ups.shutdown
UPS.PowerSummary.Time
input.L1-N.voltage
```
I guess it'll be hard to get much that's actually useful out of any of this without having access to their source code modifications.

diego351, 2025-04-18:
Not directly related but wanted to put the efficiency numbers that I measured with the River 3 Plus 230V EU version. One considering using it as an UPS might be interested.

The results for fully charged, inverter on, screen off:

**AC**:
```
0 load: 7.9W
9.8W load: 17.9W
33.4W load: 41.3W
528W load: 528W
```

**Now Inverter off, only USB PD:**
```
96,1W PD load: 121.1W
19.28 PD load: 30.6W
```

Whether your connected to wifi doesn't make any difference power efficiency wise. Also, there's no way to turn off bluetooth.

So, additional 8W on AC, not that bad, PD's terrible however. 

Comparing to APC SMT750ic:
```
Inverter on, no load: 5.6W
40.2W load: 44.2W
```

greyltc, 2025-05-17:
I've noticed a firmware update is available: `1.38.81.53` --> `1.33.91.55`. I haven't applied it yet. Changelog in the app says "Fixes some known issues."

When I have a minute, I'll do the upgrade and see if I notice anything that impacts NUT.

MeisterQ, 2025-05-18:
Is there any news here? Has anyone managed to get it running?
My Intel NUC detects the R3+ over USB, but the NUT server does not.
My River3+ is on firmware v1.33.81.53.

Yesterday support told me that you need the Power Manager to use this function. That is not satisfactory.

jimklimov, 2025-05-18:
Which NUT versions are you trying? As noted in references above, there were several PRs for River and Delta series in the past months -- they should be included in NUT v2.8.3 release (and/or later development on the master branch).

If your distro has an older version, consider https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests (you can test the driver right from the build area).

MeisterQ, 2025-05-18:
> Which NUT versions are you trying? As noted in references above, there were several PRs for River and Delta series in the past months -- they should be included in NUT v2.8.3 release (and/or later [...]

I think ive installed v2.8.0-7

do i have to enable nut somehow in the river3+ ?

jimklimov, 2025-05-18:
> I think ive installed v2.8.0-7

Then I guess you 'd have to forfeit packaged NUT until your distro delivers 2.8.3 or newer, and use a custom build - see link above (about configuring a new build to replicate the installed one as much as possible, so it would be a drop-in replacement).

> do i have to enable nut somehow in the river3+ ?

I don't think so, unless they made an UPS with a NUT server on board (there were discussions generally, but I don't think any were realized or I don't know in which devices).

Generally in smarter UPSes you may have some vendor tool or on-board web server through which to configure aspects like enabling or disabling SNMP, maybe Modbus monitoring etc. I don't think generic USB HID falls into that category that is commonly disable-able.

MeisterQ, 2025-05-18:
> Then I guess you 'd have to forfeit packaged NUT until your distro delivers 2.8.3 or newer, and use a custom build - see link above (about configuring a new build to replicate the installed one as [...]

I really have no clue how to.

i just did "apt install nut" and and "nut-scanner -U" and it wont be listed. I have no clue how to install special versions or something. Thought it would install the latest.

jimklimov, 2025-05-19:
> I really have no clue how to.

The posts just above in the discussion link to Wiki article that says how to.

> i just did "apt install nut" and ... Thought it would install the latest.

It did. Sort of. Latest as available in the distro lineage of your choice, choosing whatever they deliver by some policy (e.g. following the definition of a "sta(b)le" package repository). You get what you asked for :)

MeisterQ, 2025-05-19:
> The posts just above in the discussion link to Wiki article that says how to.
>
>
> It did. Sort of. Latest as available in the distro lineage of your choice, choosing whatever they deliver by some [...]

Thank you. I think i got it.

Ill try ist.

Marc-Eberhard, 2025-05-22:
I have just connected a new River 3 Plus to an Arch Linux install and I can see this output in dmesg:

[    1.036718] usb 1-2: New USB device found, idVendor=3746, idProduct=ffff, bcdDevice= 1.00
[    1.036750] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.036757] usb 1-2: Product: EF-UPS-RIVER 3 Plus
[    1.036763] usb 1-2: Manufacturer: EcoFlow
[    1.036767] usb 1-2: SerialNumber: <long string removed>

While this looks good, I see further down:

[    2.258067] hid-generic 0003:3746:FFFF.0001: unbalanced collection at end of report description
[    2.258092] hid-generic 0003:3746:FFFF.0001: probe with driver hid-generic failed with error -22

This doesn't sound good and I suspect this is an implementation issue in the firmware on the EcoFlow device? Is there anything I can do to debug this issue further?

Unsurprisingly, the device is not found by nut-scanner and configuring the device in /etc/nut/ups.conf as

[testups]
  driver = usbhid-ups
  port = auto

doesn't work either. Any ideas anyone?

jimklimov, 2025-05-22:
I suppose `hid-generic` is a system-provided handler and should not be attached here. Do your `udev.rules` or their equivalents say that NUT should own the device? 

UPDATE: You seem to also be using an older version of NUT than where this support was added. Try https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

Marc-Eberhard, 2025-05-22:
> [@Marc-Eberhard](https://github.com/Marc-Eberhard) : I suppose `hid-generic` is a system-provided handler and should not be attached here. Do your `udev.rules` or their equivalents say that NUT should own the device?

$ cat /etc/udev/rules.d/85-ups.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="3746", ATTR{idProduct}=="ffff", OWNER="nut", GROUP="nut"

$ ls -l /dev/bus/usb/001/007 
crw-rw---- 1 nut nut 189, 6 May 22 16:15 /dev/bus/usb/001/007

> UPDATE: You seem to also be using an older version of NUT than where this support was added. Try https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

Thanks, I'm currently using version 2.8.2 as shipped with Arch Linux. Not sure why they have not updated the package to the latest version. Will look into compiling the latest version myself and then testing again.

Marc-Eberhard, 2025-05-22:
> > UPDATE: You seem to also be using an older version of NUT than where this support was added. Try https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

@jimklimov : Thanks so much. 2.8.3 does work. I'm currently trying to get 2.8.3 into the Arch repository, so other Arch users will have this working right out of the box.

peros550, 2025-05-22:
Here is what ecoflow mentioned on a support ticket regarding the reason why Ecoflow River 3 plus is not available directly through the USB port: 

"When we developed the UPS function, we did it based on the standard NUT framework, and we had a lot of contact with Synology's official technicians during the process. According to Synology's suggestions, we developed the Power Manager host software, and plan to launch it on the Synology App Store in the future for users to use more conveniently.
During previous testing, we noticed that some unlicensed Chinese UPS devices achieved compatibility by "impersonating" the USB VID (vendor identification code) of Synology-certified UPS, essentially disguising themselves as devices of other brands. Although this method may bypass Synology's identification restrictions, there are compliance risks.
As a brand company that operates in compliance, EcoFlow will not adopt the practice of "impersonating VID". We have formally applied to the USB Association and obtained EcoFlow's USB VID for the development of UPS products that comply with the USB-HID standard. Considering the need to be compatible with multiple NAS brands including Synology, we have currently launched the Power Manager software on the EcoFlow official website first, and will promote it to the Synology App Store based on product planning and user needs.
We always adhere to compliant research and development to provide users with a stable and reliable product experience. Thank you for your understanding and support."

Maybe the above explanation matches to the observations already discussed, I am not that tech savy to judge.

rogerfar, 2025-05-26:
> However it has a different firmware. I wonder if anyone else has the RIVER 3 (UPS 10ms) and can confirm the firmware version. Mine is at 1.0.0 and there are no updates.

I have 1.33.81.53, which is the latest.

stelle007, 2025-05-26:
I don’t think the RIVER 3 (UPS 10ms) version have a USB to interface with NUT, I think it’s only the RIVER 3 Plus that have that USB
am I wrong?

BigGeorgeTx, 2025-05-31:
NUT release 2.8.3 has a driver for the EcoFlow River 3+. Although the Debian repository that my Raspberry Pi uses is still on 2.8.0-7, I was able to compile the latest on my Pi and now have NUT running and communicating with my River 3+ (and passing the information on the openHAB UPS binding).

ampersandru, 2025-06-01:
> NUT release 2.8.3 has a driver for the EcoFlow River 3+. Although the Debian repository that my Raspberry Pi uses is still on 2.8.0-7, I was able to compile the latest on my Pi and now have NUT running and communicating with my River 3+ (and passing the information on the openHAB UPS binding).

does UPS load show for you with the river 3 plus? I just connected a delta 3 plus and everything but the UPS load shows

Marc-Eberhard, 2025-06-01:
> > NUT release 2.8.3 has a driver for the EcoFlow River 3+. Although the Debian repository that my Raspberry Pi uses is still on 2.8.0-7, I was able to compile the latest on my Pi and now have NUT running and communicating with my River 3+ (and passing the information on the openHAB UPS binding).

Arch Linux has 2.8.3 in the x86 testing repo now.

> does UPS load show for you with the river 3 plus? I just connected a delta 3 plus and everything but the UPS load shows

battery.capacity: 100
battery.capacity.nominal: 100
**battery.charge: 79**
battery.charge.low: 20
battery.charge.warning: 10
**battery.runtime: 106620**
battery.runtime.low: 600
battery.type: 
battery.voltage: 13.0
battery.voltage.nominal: 13.8
device.mfr: EcoFlow
device.model: EF-UPS-RIVER 3 Plus
device.serial: <removed>
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: EF-UPS-RIVER 3 Plus
driver.parameter.productid: FFFF
driver.parameter.serial: <removed>
driver.parameter.synchronous: auto
driver.parameter.vendor: EcoFlow
driver.parameter.vendorid: 3746
driver.state: updateinfo
driver.version: 2.8.3
driver.version.data: EcoFlow HID 0.01
driver.version.internal: 0.62
driver.version.usb: libusb-1.0.28 (API: 0x0100010A)
experimental.ups.powersummary.ioeminformation: 
experimental.ups.powersummary.presentstatus.shutdownrequested: 0
experimental.ups.powersummary.presentstatus.voltagenotregulated: 0
experimental.ups.powersummary.rechargeable: 1
ups.beeper.status: enabled
ups.mfr: EcoFlow
ups.mfr.date: not set
ups.model: EF-UPS-RIVER 3 Plus
ups.power.nominal: 286
ups.productid: ffff
ups.serial: <removed>
**ups.status: OL**
ups.vendorid: 3746

Everything apart from `battery.charge`, `battery.runtime` and `ups.status` seems to be static. Especially `battery.voltage` is static at 13V all the time even during discharge. `ups.power.nominal` is the capacity in Wh according to specs and doesn't change. The `battery.runtime` is a weird one. It seems reasonable when on battery, but it jumps instantly to a much too high value when AC is connected. So, something in the calculation EcoFlow is doing seems a bit off. I guess future firmware updates might provide more information via the HID interface. At the moment it seems rather limited.

MeisterQ, 2025-06-01:
> > > NUT release 2.8.3 has a driver for the EcoFlow River 3+. Although the Debian repository that my Raspberry Pi uses is still on 2.8.0-7, I was able to compile the latest on my Pi and now have NUT running and communicating with my River 3+ (and passing the information on the openHAB UPS binding).
> 
> Arch Linux has 2.8.3 in the x86 testing repo now.
> 
> > does UPS load show for you with the river 3 plus? I just connected a delta 3 plus and everything but the UPS load shows
> 
> battery.capacity: 100 battery.capacity.nominal: 100 **battery.charge: 79** battery.charge.low: 20 battery.charge.warning: 10 **battery.runtime: 106620** battery.runtime.low: 600 battery.type: battery.voltage: 13.0 battery.voltage.nominal: 13.8 device.mfr: EcoFlow device.model: EF-UPS-RIVER 3 Plus device.serial: device.type: ups driver.debug: 0 driver.flag.allow_killpower: 0 driver.name: usbhid-ups driver.parameter.interrupt_pipe_no_events_tolerance: -1 driver.parameter.pollfreq: 30 driver.parameter.pollinterval: 2 driver.parameter.port: auto driver.parameter.product: EF-UPS-RIVER 3 Plus driver.parameter.productid: FFFF driver.parameter.serial: driver.parameter.synchronous: auto driver.parameter.vendor: EcoFlow driver.parameter.vendorid: 3746 driver.state: updateinfo driver.version: 2.8.3 driver.version.data: EcoFlow HID 0.01 driver.version.internal: 0.62 driver.version.usb: libusb-1.0.28 (API: 0x0100010A) experimental.ups.powersummary.ioeminformation: experimental.ups.powersummary.presentstatus.shutdownrequested: 0 experimental.ups.powersummary.presentstatus.voltagenotregulated: 0 experimental.ups.powersummary.rechargeable: 1 ups.beeper.status: enabled ups.mfr: EcoFlow ups.mfr.date: not set ups.model: EF-UPS-RIVER 3 Plus ups.power.nominal: 286 ups.productid: ffff ups.serial: **ups.status: OL** ups.vendorid: 3746
> 
> Everything apart from `battery.charge`, `battery.runtime` and `ups.status` seems to be static. Especially `battery.voltage` is static at 13V all the time even during discharge. `ups.power.nominal` is the capacity in Wh according to specs and doesn't change. The `battery.runtime` is a weird one. It seems reasonable when on battery, but it jumps instantly to a much too high value when AC is connected. So, something in the calculation EcoFlow is doing seems a bit off. I guess future firmware updates might provide more information via the HID interface. At the moment it seems rather limited.

This is what i have noticed too.

[21 more replies]