networkupstools/nut #2741: [HCL] APC BX950MI supported by usbhid-ups

brankko, 2024-12-30:
Product: https://www.se.com/de/de/product/BX950MI-GR/apc-backups-950-va-230-v-avr-4-schuko-ausg%C3%A4nge/

```
admin@truenas[~]$ upsc ups
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.mfr.date: 2001/01/01
battery.runtime: 1457
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.6
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: Back-UPS BX950MI 
device.serial: 9B2222A24948
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.sensitivity: medium
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 228.0
input.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: 295202G -302202G 
ups.load: 19
ups.mfr: American Power Conversion
ups.mfr.date: 2022/05/30
ups.model: Back-UPS BX950MI 
ups.productid: 0002
ups.realpower.nominal: 520
ups.serial: 9B2222A24948
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

```
admin@truenas[~]$ upsrw ups
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.mfr.date]
Battery manufacturing date
Type: STRING
Maximum length: 10
Value: 2001/01/01

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 120

[input.sensitivity]
Input power sensitivity
Type: STRING
Maximum length: 10
Value: medium

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 295

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 145

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20
```

```
admin@truenas[~]$ upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
shutdown.reboot - Shut down the load briefly while rebooting the UPS
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

So far everything works without any tweaks.

jimklimov, 2024-12-30:
Thanks for the report. Just wanted to clarify: does NUT regularly report this device going on battery and back, or something similarly funky, that would match the #2347 / #2565 use-case?

I see your device (or firmware) seems to be from 2022 and the problem addressed by that PR (and some related to it) was associated with "2023 and newer".

brankko, 2024-12-30:
@jimklimov I am not sure, as I just installed it, but I'll check and report back. And I'll also check the firmware updates as it only worked on Windows, so I had to pass-trough the USB to virtual machine so I can use APC's software to disable beeping sounds.

btw this is on `TrueNAS Scale ElectricEel-24.10.1`. So far I set it as a master and I have two Unraid machines running as slave clients and they both get the data and report normally so far.

```
admin@truenas[~]$ uname -a
Linux truenas 6.6.44-production+truenas #1 SMP PREEMPT_DYNAMIC Mon Dec 16 20:59:32 UTC 2024 x86_64 GNU/Linux
```

Phils80, 2025-01-03:
Hi ! I'm new in NUT, but I have just installed it on Ubuntu 24.04 Server with BX950MI. I gave up after 1 week fight with apcupsd, where I was struggling battery swiching on and back and flooding logs, additionally warnings about battery replacemnent necessity and randomly low battery warnings regardless 100% charge status reported. On NUT is better (no battery switching), but time to time I'm recieving warnings about battery exchange and low battery status on 100% charged. I'm using deb packege in v2.8.1. I haven't tried already sollutions from #2565. No unexpected shutdowns observed so far... If I can support somehow let me know, but I'm rather user than developer :).

jimklimov, 2025-01-03:
Well, essentially that PR is the solution... until/unless proven otherwise and fixed further by a later PR :)

So the "on-site" fix would be to use the newer version of NUT, with `lbrb_log_delay_sec` and `lbrb_log_delay_without_calibrating` settings to keep it quiet in this situation. Since that code is not part of a release yet (would be in 2.8.3 when it happens), currently the remedy is to custom-build NUT as documented on Wiki and elsewhere, or to tolerate the noise (hopefully reduced by those other workarounds) until new NUT release version is issued *and* distros pick it up (that may lag another year or two behind and is out of our influence).

Phils80, 2025-01-03:
Thanks @jimklimov I suppose I' forced to tolerate and wait... I tried to install it from scratch using GitHub master branch, and configure options:
`./configure --with-user=nut --with-group=nut --with-usb`
but regardless succesful installation I was unable to launch services probably I should define more options about paths or something. Anyway I'm not skilled enough to go through it without more simple instructions how to do it in Ubuntu 24.04 to get similar result to deb package installation. If somebody can guide me a bit I will be thankful :)

brankko, 2025-01-03:
I can confirm that with older firmware (original one that came with UPS `ups.firmware: 295202G -302202G`) there is no log flooding. I'll check if there is an option to downgrade the firmware and if that's possible, I'll try to update it to the latest to confirm if it works as expected. If not, I'll rollback because the current state works for me.

jimklimov, 2025-01-03:
@Phils80 : did you go over https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt ? It is linked from the in-place build instructions ;) Generally the in-place scripting should take care of discovering the significant paths used by the existing (e.g. packaged) installation, such as configuration location, and user/group accounts, so the two builds can manage same devices (primarily useful to test the built drivers from the build workspace, without installing them over OS delivered files until you like them better).

@brankko : thanks :)

jimklimov, 2025-01-03:
Can't think of much at the moment...

Perhaps, if you can get new (current) NUT running, check that it (still) supports the device well also with the old firmware, e.g. no regressions in NUT about that (notably with USB code changes and LogMin/LogMax hot-fixes in particular)?

And that the UPS does begin to misbehave per #2565 with the new firmware (confirming whether this is more of a firmware than contemporary hardware bug), and that `lbrb_log_delay_*` options alleviate that.

Phils80, 2025-01-03:
> @Phils80 : did you go over https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt ? It is linked from the in-place build instructions ;) Generally the in-place scripting should [...]

Thanks @jimklimov finally succesful after going through it and installation with 
` ./configure --enable-inplace-runtime --with-doc=no`
so now starting testing :)