networkupstools/nut #2742: nut on Raspberry PI keeps giving stale data after USB disconnect (usbhid-ups driver)

slavatarsyurimi, 2024-12-30:
Greetings. I've got a single Smart-UPS X 1500 connected via USB to a Raspberry PI with nut installed with fairly basic configuration. Aside from locally getting data via the **upsc** command I also have [nut-http](https://github.com/dietah/nut-http/) set up to provide the same data via a HTTP API (end usecase is making a bitfocus companion module that is going to query that data). Nut is talking to the UPS via the **usbhid-ups** driver.

If I disconnect the usb cable - nut keeps reporting last known data, seemly unaware of the disconnect (both when asked via upsc or the http api). Expected behaviour after a usb disconnect would be seeing a "Driver not connected" error. I've tinkered with a bunch of settings, googled issues and chatgpt suggestions for a few hours now, but whatever I do it doesn't change this behavior. Here's some stuff I tried:
 - `pollonly = "enabled"` in the `ups.conf` UPS setting
 - playing around with `pollinterval` and `pollfreq` in the same config
 - playing around with `MAXAGE` in `upsd.conf`
 - manual UPS port selection in `ups.conf`
 
The curious thing is that it keeps returning stale data even after restarting the nut server (`sudo systemctl resart nut-server`). The only 'soft' way I've found for it to figure out it's gone was either making an adjustment in `ups.conf` or doing a hard usb driver restart for this UPS (`/bin/systemctl restart nut-driver@[upsname].service`).
To poke it around some more I installed nut on my macbook, and plugged the UPS right in. It looks like it doesn't automatically keep the usb driver running after configuring the ups on mac like it does on linux. When running the usb driver manually and disconnecting the usb cable I see instantly see a `nut_libusb_get_report: No such device (it may have been disconnected)` error, and the driver crashing shortly after - which is nice.

To explore some more I discovered you can tail the usb driver logs like so `sudo journalctl -u nut-driver@[upsname].service -f`.
What I observe there is that the driver is infact well aware of the fact that a disconnect took place. Only in this case it keeps repeating the following:
```
  96.184957     [D1] Can't retrieve Report 0f: No such device
  96.184996     nut_libusb_get_report: No such device (it may have been disconnected)
```
Meanwhile the driver is still running, and both **upsc** and **nut-http** keep dumping old data. I haven't let it take quite a lot of time to see if it realizes eventually, but it definitely lasts for 5+ minutes which is unaccetable for my use case. Also, logically the `MAXAGE` setting should tell nut to declare data as stale as quickly as after 15 seconds.
Anyone any clue what's going on here and what else I can try? Many thanks!

jimklimov, 2024-12-30:
Thanks for the report, indeed seems like a serious issue. Can you please clarify, which version of NUT you encounter it with (on Pi and on Mac) - or if you've tried a build of current master to see if it behaves better? 

IIRC some time during NUT v2.8.x timeframe (maybe leading up to 2.8.0) there were changes about driver reconnection after communication loss, so maybe this was either fixed by those changes, or a regression brought in by them (e.g. not marking data as stale while we're trying to reconnect - which the earlier crashing approach sort of avoided?)

Also to clarify, your use-case does not insist on specifically the driver exiting due to problems - just on having the NUT stack report that up-to-date information is not available, right?

slavatarsyurimi, 2024-12-30:
@jimklimov Thanks for the prompt response! I'm not around the PI area that the moment, so I'll be able to report the PI version, and attempt a build from master either this Wednesday or Thursday. In any case, it was installed with `apt` on the PI and with brew on the mac. OSX version: `==> nut: stable 2.8.2 (bottled), HEAD`

> IIRC some time during NUT v2.8.x timeframe (maybe leading up to 2.8.0) there were changes about driver reconnection after communication loss, so maybe this was either fixed by those changes, or a [...]

Good to know, thanks! I'll try tinkering around with different version when I have the chance.

> Also to clarify, your use-case does not insist on specifically the driver exiting due to problems - just on having the NUT stack report that up-to-date information is not available, right?

That is pretty correct, as long as there's a reliable way to know if the reported info is 'fresh' or not via the http API - all good.

jimklimov, 2025-01-02:
Looking at https://github.com/networkupstools/nut/blame/0f543e745e7830dac714e28592972319fae04888/drivers/usbhid-ups.c#L1180 and history before it, the main loop in `upsdrv_updateinfo()` with detection of (various names of) `LIBUSB_ERROR_NO_DEVICE` should have been in place for years. Likewise, in 
https://github.com/networkupstools/nut/blame/0f543e745e7830dac714e28592972319fae04888/drivers/usbhid-ups.c#L1140-L1144 there is logic, for the past 18 years, to mark it stale: `if (!reconnect_ups()) { ... dstate_datastale(); }`

Since #1771 (released as part of NUT v2.8.1) code should also report the situation via `dstate_setinfo("driver.state", "reconnect.trying");` - do you see that in e.g. `upsc` query results? 

You can also bump debug logging verbosity to check if messages from https://github.com/networkupstools/nut/blame/0f543e745e7830dac714e28592972319fae04888/drivers/usbhid-ups.c#L2095 method are seen?

Also, USB-capable drivers since NUT v2.8.0 should report which libusb version they were built against (there's different code for libusb-0.1 and libusb-1.0 API, and differently capable libraries on the other side). The "crashing" you see on MacOS is probably one of `fatalx()` exits from `reconnect_ups()` calling an `open` method like https://github.com/networkupstools/nut/blob/bdc139d541ef37cfcf41672f1188c22e7b879caf/drivers/libusb1.c#L177 or https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L213 - probably higher debug would confirm that, and at which point it decides to exit the driver.

In fact, *maybe* it makes sense to extend these methods with optional ability to not `exit()` when reconnecting vs. initial attempt (or even initially), since not all platforms run frameworks like systemd or SMF that would start another copy of the driver and provide data when the broken connection is reinstated.

jimklimov, 2025-01-02:
In short, ideally you should check if the *current* NUT codebase behaves well on your Raspberry (so there's nothing to fix for your original report, in fact -- the idea in my post above is a separate thing), see e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about building your own.

slavatarsyurimi, 2025-01-02:
Hi @jimklimov looks like your hunch is currect. The "problematic" nut version is 2.8.0. I've just spent a while manually building 2.8.2 - after unplugging the usb cable I get the `Error: Data stale` with the upsc command after 5 seconds.
Haven't tried it with a http API yet, but I'm confident that it would give an error as well.

My understanding of C is pretty non-existent, so I'm going to have to take your word for most of the technical stuff in your last message. Is there anything in particular you'd like me to try out? I've got the PI instance with 2.8.0 on the separate sd card.

> Since #1771 (released as part of NUT v2.8.1) code should also report the situation via `dstate_setinfo("driver.state", "reconnect.trying");` - do you see that in e.g. `upsc` query results?

Now I can't get that to happen as of yet. The normale state I get when the ups is plugged in is `driver.state: quiet`, and it stays that way even during the 5 seconds after the cable is unplugged before the data is declared stale.

For now it seems like I can move forward with my module development. Hand building the current codebase is fine for our use-case.

jimklimov, 2025-01-03:
The "during the 5 seconds" part may be due to the frequency of the big loop in `drivers/main.c`, dictated by `pollinterval` setting.

So the driver does some work in `upsdrv_updateinfo()`, effectively sleeps for the remainder of `timeout` in `dstate_poll_fds()` (if no FD like those for a serial port gets bytes incoming), and only then gets to run `upsdrv_updateinfo()` again and perhaps notice the absence of the device as reported by libusb.

manuel-arguelles, 2025-05-21:
Hi, I'm experiencing the same issue, nut 2.8.1, Ubuntu 24.04, usbhid-ups doesn't seem to reconnect if the usb cable is unplugged.

If the usb cable is disconnected and connected, the logs just keep showing this endlessly:

```
usbhid-ups[389913]: D1] upsdrv_updateinfo...
usbhid-ups[389913]: [D1] Not using interrupt pipe...
usbhid-ups[389913]: [D1] Quick update...
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 12: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 13: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
```

Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46

If the service is started with the usb cable disconnected and then connected after a while I works as expected, it seems that only the re-connection is missing.

manuel-arguelles, 2025-05-21:
> Since [#1771](https://github.com/networkupstools/nut/pull/1771) (released as part of NUT v2.8.1) code should also report the situation via `dstate_setinfo("driver.state", "reconnect.trying");` - do you see that in e.g. `upsc` query results?
> 

It seems that `dstate_setinfo("driver.state", "reconnect.trying");` is only set for `use_interrupt_pipe`, driver.state goes to quiet and updateinfo, `reconnect_ups` doesn't gets called.

More logs with 3 iterations of upsdrv_updateinfo, first one connected, the other 2 disconnected: 

```
usbhid-ups[389913]: [D1] upsdrv_updateinfo...
usbhid-ups[389913]: [D1] Not using interrupt pipe...
usbhid-ups[389913]: [D1] Quick update...
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: [D2] refresh_report_buffer: expected 5 bytes, but got 3 instead
usbhid-ups[389913]: [D3] Report[err]: (3 bytes) => 12 ff ff
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 32767, LogMin = -1
usbhid-ups[389913]: [D5] Unit = 00001001, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x12, Offset: 0, Size: 32, Value: -1
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: [D3] Report[get]: (5 bytes) => 13 ff ff ff ff
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 0, LogMin = -1
usbhid-ups[389913]: [D5] Unit = 00001001, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.DelayBeforeReboot, Type: Feature, ReportID: 0x13, Offset: 0, Size: 32, Value: -1
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: [D3] Report[get]: (3 bytes) => 07 0d 00
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
usbhid-ups[389913]: [D5] Unit = 00000000, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x07, Offset: 2, Size: 1, Value: 1
usbhid-ups[389913]: [D5] hu_find_infoval: found online (value: 1)
usbhid-ups[389913]: [D5] process_boolean_info: online
usbhid-ups[389913]: [D3] Report[buf]: (3 bytes) => 07 0d 00
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
usbhid-ups[389913]: [D5] Unit = 00000000, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x07, Offset: 1, Size: 1, Value: 0
usbhid-ups[389913]: [D5] hu_find_infoval: found !dischrg (value: 0)
usbhid-ups[389913]: [D5] process_boolean_info: !dischrg
usbhid-ups[389913]: [D3] Report[buf]: (3 bytes) => 07 0d 00
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
usbhid-ups[389913]: [D5] Unit = 00000000, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x07, Offset: 0, Size: 1, Value: 1
usbhid-ups[389913]: [D5] hu_find_infoval: found chrg (value: 1)
usbhid-ups[389913]: [D5] process_boolean_info: chrg
usbhid-ups[389913]: [D3] Report[buf]: (3 bytes) => 07 0d 00
usbhid-ups[389913]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
usbhid-ups[389913]: [D5] Unit = 00000000, UnitExp = 0
usbhid-ups[389913]: [D5] Exponent = 0
usbhid-ups[389913]: [D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 4, Size: 1, Value: 0
usbhid-ups[389913]: [D5] hu_find_infoval: found !lowbatt (value: 0)
usbhid-ups[389913]: [D5] process_boolean_info: !lowbatt
usbhid-ups[389913]: [D5] send_to_all: SETINFO driver.state "quiet"
usbhid-ups[389913]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
usbhid-ups[389913]: [D5] send_to_all: SETINFO driver.state "updateinfo"
usbhid-ups[389913]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"

usbhid-ups[389913]: [D1] upsdrv_updateinfo...
usbhid-ups[389913]: [D1] Not using interrupt pipe...
usbhid-ups[389913]: [D1] Quick update...
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 12: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 13: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
[... 29 lines cut ...]
usbhid-ups[389913]: [D1] Can't retrieve Report 13: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 07: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
usbhid-ups[389913]: nut_libusb_get_report: No such device (it may have been disconnected)
usbhid-ups[389913]: [D1] Can't retrieve Report 07: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D4] Entering libusb_get_report
[... cut ...]
usbhid-ups[389913]: [D1] Can't retrieve Report 07: No such device
usbhid-ups[389913]: [D1] HIDGetDataValue unknown retcode '-19'
usbhid-ups[389913]: [D5] send_to_all: SETINFO driver.state "quiet"
usbhid-ups[389913]: [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
usbhid-ups[389913]: [D5] send_to_all: SETINFO driver.state "updateinfo"
usbhid-ups[389913]: [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"

usbhid-ups[389913]: [D1] upsdrv_updateinfo...
```

jimklimov, 2025-05-22:
@manuel-arguelles : the mention of polling vs. interrupts does ring a bell; there was a long-queued fix proposed for that, merged into main code base just recently (after NUT v2.8.3 release) as #1630

Are you in position to try a custom newer build of NUT per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests please, to check in practice that the proposed solution does work for use-cases like yours too?

manuel-arguelles, 2025-05-26:
@jimklimov sorry for the delay, I can try, would let you know how soon, thanks.

Koli0842, 2026-09-06:
I have a similar issue and it's been pretty frustrating lately. I live in an area with unstable power, so my UPS often switches to battery for a brief moment. It was fine in the first year or so, but lately I get an USB disconnect whenever it switches to battery. There's a single line of log `usbhid-ups[2015]: nut_libusb_get_report: No such device (it may have been disconnected)` in my nut-driver and `upsc` keeps reporting stale data indefinitely. My `dmesg` shows the usb disconnect, and 0.2s later the reconnect. My config only really shows pollonly=true

jimklimov, 2026-09-06:
@Koli0842 : sorry to hear that. Just in case, which NUT version are you using? Distros tend to ship some 3 to 10 year old code, so certain problems may have been fixed since. Also, can you see in `dmesg` whether the re-discovered device has the same or different identification (device number, bus number, etc.) -- at least some versions of NUT would reconnect to the same OS-provided numbers, so if the kernel re-enumerated the "newly plugged" device differently, it would be never seen until you restart the driver. See internet lore for `udev` driven tricks about ensuring that with the older NUT releases.

> It was fine in the first year or so, but lately I get an USB disconnect whenever it switches to battery

This is rather a problem with the UPS comms controller, essentially it reboots during the inverter switcheroo. Most infamous on CPS, but I recently had that with an Eaton whose battery went old. So maybe you need to replace it, especially if it gets tear and wear in your conditions. Normally for PbAc the working lifetime is 2-3 years (some special long-life technologies claim 5 or 10 years).

Koli0842, 2026-09-08:
Hey @jimklimov, thanks for the reply. I'm on arch, package 2.8.5-2, so I think it should be fairly recent, and from memory I'd say it's the same device id from dmesg. Restarting nut-driver actually restored my connection, if that tells you anything. The battery theory might stand true though, I'm running Eaton 5E-s that I bought together, and I had to replace the battery recently for my main PC, the server room one still seems to hold charge well, but maybe it's also on the way out.