networkupstools/nut #2754: How do I effectively build nut without any support or links to systemd?

PatVax, 2025-01-05:
I am on artix linux(systemd free distro). I am trying to make nut build on that system without any connection to systemd(since there is nothing from systemd in this distro).

And yet `sudo upsdrvctl -F start` prints this error `/usr/lib/nut/usbhid-ups: error while loading shared libraries: libsystemd.so.0: cannot open shared object file: No such file or directory`.

I am currently trying to run configure with the following arguments:
```
  ./configure \
    --prefix=/usr \
    --datadir=/usr/share/nut \
    --libexecdir=/usr/lib/nut \
    --sbindir=/usr/bin \
    --sysconfdir=/etc/nut \
    --disable-static \
    --with-user=nut \
    --with-group=nut \
    --with-altpidpath=/run/nut \
    --with-cgipath=/usr/share/nut/cgi \
    --with-drvpath=/usr/lib/nut \
    --with-htmlpath=/usr/share/nut/html \
    --with-pidpath=/run/nut \
    --with-statepath=/var/lib/nut \
    --with-systemdsystemunitdir=no \
    --with-libsystemd=no \
    --with-udev-dir=/usr/lib/udev \
    --with-cgi \
    --with-dev \
    --with-doc=man \
    --with-libltdl \
    --with-neon \
    --with-openssl \
    --with-serial \
    --with-snmp \
    --with-usb \
    --without-avahi \
    --without-ipmi \
    --without-freeipmi \
    --without-powerman \
    --without-wrap
```

[This is my github repo were I am trying to make nut work without systemd](https://github.com/PatVax/artix-pkgbuilds/tree/nut)

So is there some argument I can pass to configure so that existence of systemd on the build system isn't even checked?

jimklimov, 2025-01-05:
Can you please post the summary NUT `configure` script shows after build? Are you certain the driver program you called does come from your build and not some other source (trying to wedge another distro's packages/files into this system etc.)?

The reference to `libsystemd.so.0` in `usbhid-ups` or other binaries would only come via linker, and the exact file name from the build environment. If this platform lacks systemd, this binary was not built here. I'm say 99% sure :)

PatVax, 2025-01-06:
@jimklimov Here is the configure summary. It is before make is run:
```
NUT Configuration summary:
==========================

* configured version:	2.8.2 (v2.8.2)
* build serial drivers:	yes 
* build USB drivers:	yes (libusb-1.0)
* build neon based XML driver:	yes 
* enable Avahi support:	no 
* build Powerman PDU client driver:	no 
* build Modbus driver:	no 
* build IPMI driver:	no 
* build GPIO driver:	no 
* build Mac OS X meta-driver:	no 
* build i2c based drivers:	no 
* enable SSL support:	yes (OpenSSL)
* enable libwrap (tcp-wrappers) support:	no 
* enable libltdl (Libtool dlopen abstraction) support:	yes 
* build nut-scanner:	yes 
* build CGI programs:	yes 
* install NUT-Monitor desktop application:	no 
* install PyNUT binding module:	no 
* build and install documentation:	man 
* build specific documentation format(s):	yes  man
* build and install the development files:	yes 
* consider basic SMF support:	no
* consider basic systemd support:	no 
* build with tighter systemd support:	no 
* build C++11 codebase (client library, etc.):	yes 
* build C++ tests with CPPUNIT:	no 
* build and install the nutconf tool (experimental, may lack support for recent NUT options):	yes 
* build SNMP drivers:	yes 
* build SNMP drivers with statically linked lib(net)snmp:	no 
* User to run as:	nut
* Group of user to run as:	nut
```

However, I know have verified that systemd is detected when the build is run in github workflow. When I run everything the same but on a native artix linux machine systemd remains undetected(how it should be). I am running the workflow on the standard ubuntu-latest github runner but inside an artix linux docker container that is started on the github runner. My guess is that for whatever reason something about systemd leaks over to the docker container which causes systemd to be detected.

Honestly a configure option to disable systemd detection whatsoever would be probably the easiest solution.

PatVax, 2025-01-06:
I compared the summary on github workflow with the one on my native machine and they are exactly the same.

jimklimov, 2025-01-06:
So as far as NUT build goes,

````
checking whether to install systemd unit files... no 

checking whether to consider basic systemd support... no 

checking whether to install systemd shutdown files... no 

checking whether to install systemd tmpfiles files... no 

checking for systemd-tmpfiles... /usr/bin/systemd-tmpfiles 

checking for libsystemd version via pkg-config... none found 

checking for libsystemd cflags... 

checking for libsystemd ldflags... -lsystemd 

checking for systemd/sd-daemon.h... no 

checking for sd_notify... no 

checking whether requested and can build binaries with tighter systemd integration support... no 

checking for systemd-analyze... /usr/bin/systemd-analyze 

checking whether to build with tighter systemd support... no 
````

I do not disagree that a single option to perhaps skip others could be useful though. FWIW, most of them are about generation and delivery of resource files like unit definitions; only the `libsystemd` (`sd_notify` et al) part impacts the binaries.

Note that at least systemd tools were found.

jimklimov, 2025-01-06:
I took your package from GH at https://github.com/PatVax/artix-pkgbuilds/releases/download/nut-2.8.2-2/nut-2.8.2-2-x86_64.pkg.tar.zst and `ldd` does not find references to libsystemd:

```
~/artix $ ldd usr/lib/nut/usbhid-ups                                                     libusb-1.0.so.0 => not found                                                     libm.so.6 => not found                                                           libc.so.6 => not found
```

Never mind `not found`'s, it is another OS I have under hand, but `ldd` seems to work as such; reproducing the test on your side is welcome :)

PatVax, 2025-01-07:
This is how it looks at my side. `artix-pkgbuilds/nut-2.8.2-2-x86_64.pkg.tar.zst` is my locally compiled package. What OS are you using that accepts pacman packages? Normally it should make no difference with the same package but maybe it does idk.

```bash
$ sudo pacman -U https://github.com/PatVax/artix-pkgbuilds/releases/download/nut-2.8.2-2/nut-2.8.2-2-x86_64.pkg.tar.zst
loading packages...
warning: nut-2.8.2-2 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) nut-2.8.2-2

Total Installed Size:  8.26 MiB
Net Upgrade Size:      0.38 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                                                                                                                           [########################################################################################################] 100%
(1/1) checking package integrity                                                                                                                                         [########################################################################################################] 100%
(1/1) loading package files                                                                                                                                              [########################################################################################################] 100%
(1/1) checking for file conflicts                                                                                                                                        [########################################################################################################] 100%
(1/1) checking available disk space                                                                                                                                      [########################################################################################################] 100%
:: Processing package changes...
(1/1) reinstalling nut                                                                                                                                                   [########################################################################################################] 100%
:: Running post-transaction hooks...
(1/3) Creating system user accounts...
/usr/lib/sysusers.d/util-linux.conf:1: Unknown modifier 'u!'.
(2/3) Creating temporary files...
/usr/lib/tmpfiles.d/nut.conf:1: Duplicate line for path "/run/nut", ignoring.
/usr/lib/tmpfiles.d/nut.conf:2: Duplicate line for path "/var/lib/nut", ignoring.
(3/3) Reloading device manager configuration...
$ ldd /usr/lib/nut/usbhid-ups
        linux-vdso.so.1 (0x000078608478d000)
        libsystemd.so.0 => not found
        libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x0000786084719000)
        libm.so.6 => /usr/lib/libm.so.6 (0x0000786084629000)
        libc.so.6 => /usr/lib/libc.so.6 (0x000078608443b000)
        libudev.so.1 => /usr/lib/libudev.so.1 (0x00007860843f4000)
        /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x000078608478f000)
        libcap.so.2 => /usr/lib/libcap.so.2 (0x00007860843e6000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007860843b9000)
$ sudo pacman -U artix-pkgbuilds/nut-2.8.2-2-x86_64.pkg.tar.zst
loading packages...
warning: nut-2.8.2-2 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) nut-2.8.2-2

Total Installed Size:   7.88 MiB
Net Upgrade Size:      -0.38 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                                                                                                                           [########################################################################################################] 100%
(1/1) checking package integrity                                                                                                                                         [########################################################################################################] 100%
(1/1) loading package files                                                                                                                                              [########################################################################################################] 100%
(1/1) checking for file conflicts                                                                                                                                        [########################################################################################################] 100%
(1/1) checking available disk space                                                                                                                                      [########################################################################################################] 100%
:: Processing package changes...
(1/1) reinstalling nut                                                                                                                                                   [########################################################################################################] 100%
:: Running post-transaction hooks...
(1/3) Creating system user accounts...
[... cut ...]
        linux-vdso.so.1 (0x00007b5742353000)
        libusb-1.0.so.0 => /usr/lib/libusb-1.0.so.0 (0x00007b57422e1000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007b57421f1000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007b5742003000)
        libudev.so.1 => /usr/lib/libudev.so.1 (0x00007b5741fbc000)
        /lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007b5742355000)
        libcap.so.2 => /usr/lib/libcap.so.2 (0x00007b5741fb0000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007b5741f83000)
```

jimklimov, 2025-01-07:
I un-tarred the package and looked into the binaries it brings. Just like you did in the last part of the post above.

jimklimov, 2025-01-07:
Off-topic: are you sure those patch-files in your recipe are still needed? Some are against code over a decade old, I'm surprised they even still apply.

PatVax, 2025-01-07:
The patch files come from the original arch package I copied. All I changed are references to systemd in the configure call. I didn't really looked into the patch files.

jimklimov, 2025-01-08:
Just thinking: did you always reinstall it over some build/package you had in the system? Maybe it refused to replace some files so they remain as they were (and possibly as built for another system)?

What does `ls -la` or `cksum` say about the binaries in the system (after pacman) and in the just-unpacked tarball, are they identical or not?

Maybe un-install the package, make sure NUT files are gone, and install it again?

PatVax, 2025-01-08:
I tried removing the package and installing the other then to no effect.
Checksums differ between the installed one vs the extracted one.
There must be something in the pacman pipeline that adds the link to systemd. But again this doesn't happen at all with locally built package. Obviously I can just build the package locally and use that so this isn't much of a problem. At this point I am more curious why things happen like they do than anything else. I would need to somehow reproduce the github workflows environment to test it. Debugging via commits and logs is a hassle.

jimklimov, 2025-01-08:
Yeah, I feel for you.

I *think* you can somehow feed the downloaded file to `pacman` to see if it also gets mangled this way, and/or if pacman finds the package name from CLI args and installs one from remote repo and not the URL(/file?) you passed it.

Also try to investigate what the different pre/post-processing scripts do; I would be surprised if they re-link any installed programs, but who knows...

PatVax, 2025-01-08:
Only thing I found was a call to ldconfig but I have no idea if it can do anything to a binary directly. However before yesterday I wouldn't have thought that a binary might get modified during installation of a prebuild package so obviously I am not that experienced what exactly might modify the binary.

jimklimov, 2025-01-08:
So, looking at https://wiki.archlinux.org/title/Pacman#Additional_commands you can pass the filename to `pacman -U FILENAME` same as you would pass an URL.

So I propose a further experiment - having removed older traces of NUT from that system, perhaps then clearing local package cache and disconnecting/blocking the internet to avoid interference from package repos, and finally installing this way from the downloaded archive in which we know that the binary programs do not refer to `libsystemd`, would you end up with it linked there anyway?

PatVax, 2025-01-08:
That experiment did it. The issue was completely on my machine. When I first tried nut I installed the arch package to test if it works. Since it didn't because of systemd obviously I made my own version for artix and wanted to share it to the world so I made the github repo.
Because I already installed nut from the arch repository it was in my cache. When pacman installs the package from a local file it doesn't check and doesn't write to the cache. Which is why my locally built package worked. I guess I never tested installing the package pre-downloaded from github. When pacman gets the package remotely(e.g. via https) it does check the cache. So the whole time I installed via https I actually installed the arch package.
The build system of nut therefor is correct. I thank you for your help and sorry for taking your time for my local issue. Presumably turning of systemd detection during the configure step could still be interesting but at least in my case it wasn't what caused the problem.

closed 2025-01-08 by PatVax

jimklimov, 2025-01-08:
Thanks for the summary, and happy I could help!