networkupstools/nut #2755: Tripp Lite SMART1500RM2UL Support

scuppasteve, 2025-01-05:
​
I am getting the error below when i try to startup nut.

TrippLite device (09ae:3018) is not (or perhaps not yet) supported

and it just spams the log with, any idea what i am doing wrong? Is it just about the device being different and not supported? Surprised its that different from all the other TRIPP LITE SMART1500's.

```
Jan  4 22:13:34 MovieServer kernel: hid-generic 0003:09AE:3018.0C45: hiddev97,hidraw1: USB HID v1.10 Device [Tripp Lite  TRIPP LITE SMART1500RM2UL ] on usb-0000:00:14.0-11.2/input0
Jan  4 22:13:49 MovieServer kernel: usb 1-11.2: USB disconnect, device number 95
Jan  4 22:13:49 MovieServer kernel: usb 1-11.2: new low-speed USB device number 96 using xhci_hcd
Jan  4 22:13:50 MovieServer kernel: hid-generic 0003:09AE:3018.0C46: hiddev97,hidraw1: USB HID v1.10 Device [Tripp Lite  TRIPP LITE SMART1500RM2UL ] on usb-0000:00:14.0-11.2/input0
Jan  4 22:14:04 MovieServer kernel: usb 1-11.2: USB disconnect, device number 96
Jan  4 22:14:04 MovieServer kernel: usb 1-11.2: new low-speed USB device number 97 using xhci_hcd
Jan  4 22:14:04 MovieServer kernel: hid-generic 0003:09AE:3018.0C47: hiddev97,hidraw1: USB HID v1.10 Device [Tripp Lite  TRIPP LITE SMART1500RM2UL ] on usb-0000:00:14.0-11.2/input0
Jan  4 22:14:19 MovieServer kernel: usb 1-11.2: USB disconnect, device number 97
```

​

clepple, 2025-01-07:
Not terribly surprising; Tripp Lite has already used the SMART1500 name for two completely different protocol families ([USB PID 0001](https://github.com/networkupstools/nut/blob/09ad17e5b802eacbc654d7fbd86aaf3cbf47dbb9/data/driver.list.in#L1415) and more-or-less standard [USB HID PDC](https://github.com/networkupstools/nut/blob/09ad17e5b802eacbc654d7fbd86aaf3cbf47dbb9/data/driver.list.in#L1419).

You should be able to get further by specifying `productid = 3018` in `ups.conf` - the driver normally doesn't need the PID for known models, but this is a sign that nobody has tested for any quirks in the protocol.

Also, do those disconnection times correspond to when you unplugged the USB cable? If not, you may be in for some amount of annoyance. The 3016 protocol UPSes were similar - they did not work well with modern USB controllers (even though they only use USB 1.1 low-speed) and would fall off of the bus when least convenient. I offloaded my 3016 UPS to someone who just needed battery backup, and didn't need monitoring support.

ciuchs, 2025-01-15:
looks like the SMART2200RM2UL uses the same 

ID 09ae:3018 Tripp Lite TRIPP LITE SMART2200RM2UL

jimklimov, 2026-02-18:
@RDshift : which NUT version? Current should know about the ID I think. Also note that `bus` (also `busport` etc.) is link-dependent and may vary with re-enumerations (something un/re/plugged sonewhere, bus resets, etc.

scuppasteve, 2026-02-18:
> [@RDshift](https://github.com/RDshift) : which NUT version? Current should know about the ID I think. Also note that `bus` (also `busport` etc.) is link-dependent and may vary with re-enumerations (something un/re/plugged sonewhere, bus resets, etc.

I am making no claims to having solved anything special, it also worked for @RDshift. This is what worked for me for version: 
nut-2.8.4-x86_64-1stable.ssl31 built into Unraid.  The busport is link dependent, but for USB it largely stays consistent once you identify it.