networkupstools/nut #2757: modbus_connect: unable to connect: No such device when trying to connect apc_modbus via tcp

moritzkraemer, 2025-01-07:
I am coming from this issue #2732 

I am now trying to connect to the ups via tcp modbus. If i start the driver manually with:
`sudo ./apc_modbus -DDDDD -s apc -x vendorid=051D -x productid=0003 -x porttype=tcp -x port=192.168.11.131 -x user=root`
the driver starts reading out all the information correctly.

But `sudo upsdrvctl start` fails with 'modbus_connect: unable to connect: No such device'.

my ups.conf:

```
[USV2]
    driver = apc_modbus
    porttype = tcp
    port = 192.168.11.131
    vendorid = 051D
    productid = 0003
    user = root
```

when i start the driver manually with `sudo ./apc_modbus -a USV2` also the `modbus_connect: unable to connect: No such device` error occurs.

moritzkraemer, 2025-01-07:
When i run `sudo ./apc_modbus -DDDDD -a USV2`

i get the following output:

```
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) - NUT APC Modbus driver with USB support 0.11
   0.000116     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.000128     [D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.000137     [D1] This build of the driver is USB-capable; also Serial and TCP Modbus RTU are supported
   0.000190     [D5] do_upsconf_args: confupsname=(null), var=pollinterval, val=1
   0.000202     [D5] do_upsconf_args: call do_global_args()
   0.000217     [D3] do_global_args: var='pollinterval' val='1'
   0.000229     [D1] testval_reloadable: setting 'pollinterval' exists and differs: new value '1' vs. '2'
   0.000242     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000257     [D5] do_upsconf_args: call do_global_args()
   0.000269     [D3] do_global_args: var='maxretry' val='3'
   0.000285     [D5] do_upsconf_args: confupsname=USV2, var=driver, val=apc_modbus
   0.000297     [D5] do_upsconf_args: call main_arg()
   0.000309     [D3] main_arg: var='driver' val='apc_modbus'
   0.000319     [D5] do_upsconf_args: not a main_arg()
   0.000331     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000343     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000356     [D5] do_upsconf_args: confupsname=USV2, var=port, val=auto
   0.000367     [D5] do_upsconf_args: call main_arg()
   0.000379     [D3] main_arg: var='port' val='auto'
   0.000391     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000405     [D5] do_upsconf_args: confupsname=USV2, var=vendorid, val=051D
   0.000416     [D5] do_upsconf_args: call main_arg()
   0.000427     [D3] main_arg: var='vendorid' val='051D'
   0.000437     [D5] do_upsconf_args: not a main_arg()
   0.000448     [D5] do_upsconf_args: process as value
   0.000463     [D5] send_to_all: SETINFO driver.parameter.vendorid "051D"
   0.000477     [D5] do_upsconf_args: confupsname=USV2, var=productid, val=0003
   0.000489     [D5] do_upsconf_args: call main_arg()
   0.000500     [D3] main_arg: var='productid' val='0003'
   0.000511     [D5] do_upsconf_args: not a main_arg()
   0.000523     [D5] do_upsconf_args: process as value
   0.000535     [D5] send_to_all: SETINFO driver.parameter.productid "0003"
   0.000549     [D5] do_upsconf_args: confupsname=USV2, var=serial, val=AS23xxxxx
   0.000581     [D5] do_upsconf_args: call main_arg()
   0.000592     [D3] main_arg: var='serial' val='AS2313351576'
   0.000603     [D5] do_upsconf_args: not a main_arg()
   0.000615     [D5] do_upsconf_args: process as value
   0.000629     [D5] send_to_all: SETINFO driver.parameter.serial "AS23xxxxx"
   0.000644     [D5] do_upsconf_args: confupsname=USV2, var=vendor, val=American Power Conversion
   0.000656     [D5] do_upsconf_args: call main_arg()
   0.000669     [D3] main_arg: var='vendor' val='American Power Conversion'
   0.000682     [D5] do_upsconf_args: not a main_arg()
   0.000691     [D5] do_upsconf_args: process as value
   0.000710     [D5] send_to_all: SETINFO driver.parameter.vendor "American Power Conversion"
   0.000724     [D5] do_upsconf_args: confupsname=USV2, var=bus, val=001
   0.000736     [D5] do_upsconf_args: call main_arg()
   0.000748     [D3] main_arg: var='bus' val='001'
   0.000761     [D5] do_upsconf_args: not a main_arg()
   0.000776     [D5] do_upsconf_args: process as value
   0.000798     [D5] send_to_all: SETINFO driver.parameter.bus "001"
   0.000814     [D5] do_upsconf_args: confupsname=USV2, var=device, val=002
   0.000826     [D5] do_upsconf_args: call main_arg()
   0.000838     [D3] main_arg: var='device' val='002'
   0.000850     [D5] do_upsconf_args: not a main_arg()
   0.000861     [D5] do_upsconf_args: process as value
   0.000875     [D5] send_to_all: SETINFO driver.parameter.device "002"
   0.000893     [D1] Network UPS Tools version 2.8.2.1725-1725-gc641cff57 (development iteration after 2.8.2) built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/usr/local/ups/include/modbus --with-modbus-libs='-L/usr/local/ups/lib -lmodbus'
   0.000921     [D1] debug level is '5'
   0.000936     [D5] send_to_all: SETINFO driver.debug "5"
   0.000949     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001814     [D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.001853     [D1] Signalling UPS [USV2]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001876     Can't open /var/state/ups/apc_modbus-USV2: No such file or directory
   0.001888     [D1] Request for other driver to exit returned code -1
   0.001902     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001925     [D5] send_to_all: SETINFO device.type "ups"
   0.001939     [D5] send_to_all: SETINFO driver.state "init.device"
   0.009289     modbus_connect: unable to connect: No such device
   0.009308     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.009321     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
   0.009344     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
```

so it looks like it is reading some information from a different config file?

jimklimov, 2025-01-07:
Seems so. Which `ups.conf` did you edit? By default it would be under `${prefix}/etc` e.g. `/usr/local/ups/etc` as you don't seem to use a custom `--with-sysconfdir=/etc/nut` here?

moritzkraemer, 2025-01-07:
Oh yes because of the custom build driver i edited the wrong ups.conf

closed 2025-01-07 by moritzkraemer