networkupstools/nut #2758: Is NUT 2.8.3 coming?

teoberi, 2025-01-08:
I'm asking this because you're at 87% and I have to install a new server?

jimklimov, 2025-01-08:
Well, just today I was cleaning up the queue pushing some issues to 2.8.4, so percentage grew :)

As for a new server, note that distributions (especially stable ones and older release baselines) can take years to pick up a new release. You may be better off with a custom build if you want one quickly, in any case - see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details.

teoberi, 2025-01-08:
I've been building NUT from source since I can remember so I only need the sources for a new release (maybe a git clone).
I wanted to know if I should build from github or wait for a release in the near future.

jimklimov, 2025-01-08:
Well... I keep promising myself it's high time to release, just gotta squash that recently reported nasty bug, so the release is nice and clean... rinse and repeat as new ones get reported and seem serious.

I can recommend a git clone nowadays. With NUT CI churning the code, master should be pretty solid and portable.

dvdesolve, 2025-01-09:
Also I'm working on Q2/Q6 protocols these days - perhaps I will be able to provide them soon