networkupstools/nut #2760: No Support for Minutman UPS?

almulder, 2025-01-09:
Unable to get any connection working to my Minutman UPS with this plugin. If I stop this one and use the Unraid one, that can connect and report status.

Would like to use this plugin instead of unraids. Are we able to get support for the Minutman UPS

almulder, 2025-01-09:
I have tried all three users listed. and all give connection refused, but the built-in unraid one works so really confused.

[image]

jimklimov, 2025-01-09:
Whoa-whoa, what "this plugin", for which appliance/ecosystem? We here are about the upstream NUT project...

Generally I'd ask about logs from the NUT driver, started with debug verbosity settings, as well as NUT version involved and if you have a chance to custom-build current code to test with it. Can still come to that, if we'd need to dig deeper later.

At the moment, since your UPS has logins with passwords and not plain old communities, you definitely should try the settings for SNMPv3 - those sort of credentials are its concepts.

almulder, 2025-01-10:
Sorry, Its the Unraid plugin I instleed form the app store and for support it linked me kere. If i use the unraid built in one as seen in the secon set of screenshots, it pulls the info without user name and passwords.

In the 3rd set of screenshots is this plugin and even with user name and password entered it will not pull the battery levels and runtime.

Not sure what you need or if this is the right place.

jimklimov, 2025-01-10:
Then I'm also CC'ing @desertwitch to chime in here :)

Alas, GitHub won't open me the images in a new tab as such, so I have to make do with the tiny ones in the ticket. Anyhow...

The first pair of screenshots (with driver 3.14.14 from 2016) apparently refers to `apcupsd` software, that's their last release.

The second pair exposes a recent NUT release 2.8.2, so it should be capable of SNMPv3 and various types of crypto that come with it. Check the Examples section in https://networkupstools.org/historic/v2.8.2/docs/man/snmp-ups.html
* Upon second reading, given that the driver does discover some info, maybe SNMPv3 auth is not the culprit, and SNMPv1/2 communities also work for you. Not really sure what the "users" screenshot is about now, is it from the UPS itself or from the appliance's setup? Are these accounts for Web(/SSH/...) access or also/specifically for SNMP (then v3) to the UPS?

Devices are expected to implement the IETF MIB as the lowest common denominator, this is the last lock pick a driver tries in automatic mode. More/different information may be available from vendor extension branches of SNMP OID tree, and NUT subdrivers are coded to know about some such branches and the data points they deliver (and this mapping may be getting out of date as new vendor MIB versions are released and supported by their firmwares).

In your case the "delta_ups" MIB seems to have been served and thus used, see https://github.com/networkupstools/nut/blob/v2.8.2/drivers/delta_ups-mib.c#L94 for the maximum of information that it can deliver today, which is frankly not much - most of the data points (discovered in a data walk back knows when) are tucked away as "unmapped". Maybe more can be discovered from the newer device following https://github.com/networkupstools/nut/blob/master/docs/snmp-subdrivers.txt instructions. Ultimately, this needs someone to take a look at contents of these data points, give a proper name (and maybe measurement scaling) according to https://github.com/networkupstools/nut/blob/v2.8.2/docs/nut-names.txt, ideally test it with a local build of NUT - not necessarily on the same appliance, SNMP can run from anywhere, and submit a pull request. That's how open-source works, especially as nobody has all the devices under hand.

FWIW, the Minuteman (note the name is with "e") is not explicitly mentioned in SNMP drivers, but is in USB, as "Delta/Minuteman" in Tripplite HID subdriver, so the discovered MIB for SNMP looks right.

desertwitch, 2025-01-10:
I think this is mostly an issue of the "wrong" (or not ideal) MIB being chosen by the automatic mechanisms, but this can easily be switched in the Unraid plugin's GUI for (hopefully) better results. I don't think the Unraid OS "UPS Settings" (`apcupsd`) have any specific MIB for your UPS either, so I'm thinking you'll see best results with the IETF standard MIB as Jim already pointed out.

**Please "Reset Config" first (!) and then set up your NUT Settings as follows:**
[image]
If `ietf` doesn't provide more values, you can test other MIBs selectable under `SNMP Definition MIB`.

**Additionally I'd suggest inserting your nominal values so missing values can be calculated from them:**
[image]
I assumed these for you, so you'll have to put in the actual nominal values of your UPS here.

jimklimov, 2025-01-10:
But still, if you are in position to roll up the sleeves to make Delta MIB more useful - that would be great and welcome too :)

almulder, 2025-01-11:
Let me know how I can help, more than willing to (just need to guide me)

jimklimov, 2025-01-11:
Primarily check https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about build prerequisites and general procedure to check out and build NUT (for your case, plain `./ci_build.sh` should suffice to wrap the common case), and  https://github.com/networkupstools/nut/blob/master/docs/snmp-subdrivers.txt about iterating with the new/updated SNMP MIB against a device.

You can use any interactive VM or container for this, not necessarily a production machine, as long as you can get by network from it to the UPS.