networkupstools/nut #2763: SMX1500RM2U & AP9620

FugginOld, 2025-01-12:
Hello...I am trying to figure out which cable and driver will provide the most feedback and control over the UPS operation within the NUT plugin. The documentation, to me, is not clear. Can someone please advise?

jimklimov, 2025-01-13:
Hello, plugin for what? This here is the upstream NUT project, packaged and embedded into many operating systems and appliances...

Generally speaking, do you know which NUT version you are running there? Is building a custom one an option?

What sort of cables are offered? Which protocols/drivers seem to work at all (since you ask "which provides the most control")?

Looking at https://www.apc.com/us/en/product/AP9620/protocol-translation-to-upslink-management-options-legacy-communications-smartslot-card-ups-support-for-upslink-protocol-serial-communications/ more closely, it seems to only offer Serial and USB ports. 

I could not quickly find the offered protocols, and APC has quite a few to choose from. By URL and docs on that page, the card and/or its latest firmware from 2017 seems to "add support for UPS-Link" protocol, which is not mentioned in current NUT sources :\

Generally, talking over USB you might have luck with `usbhid-ups` driver, but since 2010 or so, APC were winding down support for the standard protocol in favor of Modbus. Current NUT does deliver an `apc_modbus` driver, and over the Serial port it may be usable "as is", but for USB support you would likely need a custom build with a patched libmodbus - see https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol

Some other options may include `apcsmart` or `apcsmart-old`; for some companies that APC bought up and digested into its common product lines, `microsol-apc` or `solis` might be the answer.

Also note that with Ethernet port/cable (if you have some other AP cards to try), you might be able to use `snmp-ups` (with the SNMP v1/v2c community as set up on the UPS side, by default `public` for R/O or `private` for R/W with commands and settings, or SNMP v3 user/pass credentials). Note that some APC UPSes have a network port usable only for the UPS to "dial home" into their cloud, and not for site-local monitoring/management.

FugginOld, 2025-01-13:
Hello...thank you for getting back to me. I am using a NUT plugin in a file server application called Unraid. There is a NUT plugin that will scan the USB ports (or you can specify the IP if using an ethernet module).
Based on the model of the UPS it detects, it will pick the driver or you can select manually based on your configuration. 
But my question is which driver/cable configuration yields the most information transfer to a plugin?

jimklimov, 2025-01-14:
Just in case, CCing @rysz about the plugin (and current APC expertise).

For "which one yields most info" - it depends on many factors...

One is, which protocol(s) the device talks - so back to my follow-up question, which drivers worked for you and what info they gave (again, assuming from how you worded the OP that at least one did)? The drivers do potentially offer different maximum sets of data points for a given build (may be extended via pull requests for future NUT releases, as more mappings or protocol queries are discovered), some of which may remain unpopulated and hidden based on what the device tells them. 

So, say, an APC USB HID subdriver has 48 unique mapped names (with commands, settings and all), as of current code. And some other driver has maybe 10 data points. But say APC wind-down of USB HID support reached your device's firmware version, so it only exposes some 5 most-critical data points over USB HID... If both drivers' protocols are what your device does talk, suddenly the one with 10 data points in code is "better" than the one with 48, for your device. Something not easy to say hands-off without experimenting with your device (and your version of NUT).