networkupstools/nut #2766: usbhid-ups: missing battery.voltage on Powercom BNT-800AP

Vladdrako, 2025-01-16:
Battery voltage reports by UPSMON PRO, but not NUT.
ups.cof
```
pollinterval = 1

[powercom]
desc = Powercom BNT-800AP Schuko
driver = usbhid-ups
ignorelb
interruptsize = 8
offdelay = 60
ondelay = 70
pollfreq = 5
port = auto
productid = 0004
vendorid = 0d9f
```
upsd.conf
```
LISTEN 0.0.0.0 
MAXAGE 25
STATEPATH /var/run/nut
```
upsd.users
```
[root]
  password = ********
  actions = SET
  actions = FSD
  instcmds = ALL
  upsmon master
```
upsc powercom
```
battery.charge: 100
battery.charge.warning: 30
battery.runtime: 800
battery.temperature: 25.0
battery.type: PbAc
device.type: ups
driver.debug: 1
driver.flag.allow_killpower: 0
driver.flag.ignorelb: enabled
driver.name: usbhid-ups
driver.parameter.interruptsize: 8
driver.parameter.offdelay: 60
driver.parameter.ondelay: 70
driver.parameter.pollfreq: 5
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.powercom_sdcmd_byte_order_fallback: 1
driver.parameter.productid: 0004
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0d9f
driver.state: updateinfo
driver.version: 2.8.1
driver.version.data: PowerCOM HID 0.71
driver.version.internal: 0.55
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 50.0
input.voltage: 226.0
input.voltage.nominal: 220
output.frequency: 50.0
output.voltage: 226.0
output.voltage.nominal: 220
ups.beeper.status: disabled
ups.date: 2014/01/15
ups.delay.shutdown: 60
ups.delay.start: 70
ups.load: 15
ups.productid: 0004
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: 0
ups.timer.start: 740400
ups.vendorid: 0d9f
```

jimklimov, 2025-01-16:
Cheers, and thanks for the report.

I see some `battery.voltage*` mappings in https://github.com/networkupstools/nut/blob/master/drivers/powercom-hid.c but maybe your device/firmware reports them at a different USB HID branch.

A driver start with high verbosity can help track them down, or generally https://github.com/networkupstools/nut/blob/master/docs/hid-subdrivers.txt procedure.

Vladdrako, 2025-01-16:
@jimklimov Here debug output
```
root@MR7350:[~]: # /lib/nut/usbhid-ups -DDDD -a powercom
Network UPS Tools - Generic HID driver 0.55 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D3] do_global_args: var='pollinterval' val='1'
   0.000043     [D1] testval_reloadable: setting 'pollinterval' exists and differs: new value '1' vs. '2'
   0.000072     [D3] main_arg: var='desc' val='Powercom'
   0.000091     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000107     [D3] main_arg: var='ignorelb' val='<null>'
   0.000129     [D3] main_arg: var='interruptsize' val='8'
   0.000157     [D3] main_arg: var='offdelay' val='60'
   0.000180     [D3] main_arg: var='ondelay' val='70'
   0.000201     [D3] main_arg: var='pollfreq' val='5'
   0.000225     [D3] main_arg: var='port' val='auto'
   0.000246     [D3] main_arg: var='productid' val='0004'
   0.000270     [D3] main_arg: var='vendorid' val='0d9f'
   0.000299     [D3] main_arg: var='powercom_sdcmd_byte_order_fallback' val='1'
   0.000368     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r30324-c4c1e9f469) 13.3.0 and configured with flags: --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --sysconfdir=/etc/nut --datadir=/usr/share/nut --with-dev --with-usb --without-avahi --without-snmp --without-serial --without-doc --with-neon --without-powerman --without-wrap --with-hotplug-dir=/etc/hotplug --without-cgi --without-ipmi --without-freeipmi --without-linux-i2c --with-ssl --with-openssl --without-libltdl --without-macosx_ups --with-statepath=/var/run/nut --with-pidpath=/var/run --with-drvpath=/lib/nut --with-user=root --with-group=root
   0.000449     [D1] debug level is '4'
   0.000999     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001108     [D1] upsdrv_initups (non-SHUT)...
   0.001150     [D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.003706     [D2] Checking device 1 of 9 (05E3/0626)
   0.242582     [D2] - VendorID: 05e3
   0.242685     [D2] - ProductID: 0626
   0.242725     [D2] - Manufacturer: GenesysLogic
   0.242763     [D2] - Product: USB3.1 Hub
   0.242799     [D2] - Serial Number: unknown
   0.242836     [D2] - Bus: 002
   0.242876     [D2] - Bus Port: 004
   0.242915     [D2] - Device: 013
   0.242952     [D2] - Device release number: 0656
   0.242989     [D2] Trying to match device
   0.243031     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.243105     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.243147     [D2] Device does not match - skipping
   0.350709     [D2] Checking device 2 of 9 (152D/578E)
   0.351665     [D2] - VendorID: 152d
   0.351742     [D2] - ProductID: 578e
   0.351781     [D2] - Manufacturer: TOSHIBA
   0.351818     [D2] - Product: DT01ACA300
   0.351857     [D2] - Serial Number: 00027D0SNUAS
   0.351895     [D2] - Bus: 002
   0.351935     [D2] - Bus Port: 001
   0.351970     [D2] - Device: 011
   0.352007     [D2] - Device release number: 0515
   0.352043     [D2] Trying to match device
   0.352086     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.352141     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.352190     [D2] Device does not match - skipping
   0.352253     [D2] Checking device 3 of 9 (0D9F/0004)
   0.358368     [D2] - VendorID: 0d9f
   0.358421     [D2] - ProductID: 0004
   0.358459     [D2] - Manufacturer: POWERCOM Co.,LTD
   0.358497     [D2] - Product: HID UPS Battery
   0.358537     [D2] - Serial Number: 004-0D9F-000
   0.358574     [D2] - Bus: 001
   0.358610     [D2] - Bus Port: 004
   0.358648     [D2] - Device: 056
   0.358684     [D2] - Device release number: 0002
   0.358723     [D2] Trying to match device
   0.358765     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.358820     [D3] match_function_regex: matching a device...
   0.358882     [D2] Device matches
   0.358922     [D2] Reading first configuration descriptor
   0.358977     [D3] libusb_kernel_driver_active() returned 0: Success
   0.359060     [D2] Claimed interface 0 successfully
   0.359102     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.359829     [D2] Retrieved HID descriptor (expected 9, got 9)
[... 1040 lines cut ...]
  20.010470     [D3] Report[buf]: (3 bytes) => 0a 0c 00
  20.010500     [D2] Path: UPS.PowerSummary.PresentStatus.VoltageNotRegulated, Type: Input, ReportID: 0x0a, Offset: 7, Size: 1, Value: 0
  20.010524     [D3] NUT doesn't use this HID object
  20.010542     [D3] Report[buf]: (3 bytes) => 0a 0c 00
  20.010566     [D2] Path: UPS.PowerSummary.PresentStatus.ShutdownRequested, Type: Input, ReportID: 0x0a, Offset: 8, Size: 1, Value: 0
  20.010645     [D3] NUT doesn't use this HID object
  20.010667     [D3] Report[buf]: (3 bytes) => 0a 0c 00
[... cut ...]
  20.011005     [D4] Entering libusb_get_report
  20.011575     [D2] refresh_report_buffer: expected 3 bytes, but got 2 instead
  20.011604     [D3] Report[err]: (2 bytes) => 0f 00
  20.011628     [D2] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 0
  20.011652     [D3] powercom_shutdown_fun: value = 0, buf = 0
^C  20.365177   [D1] set_exit_flag: raising exit flag due to signal 2
  20.365229     Signal 2: exiting
  20.365257     [D1] upsdrv_cleanup...
```

Vladdrako, 2025-01-16:
Also
```
root@MR7350:[~]: # /lib/nut/usbhid-ups -DD -u root -x explore -x vendorid=0d9f -x port=auto -s ups
Network UPS Tools - Generic HID driver 0.55 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000001     [D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
   0.000092     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r30324-c4c1e9f469) 13.3.0 and configured with flags: --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --sysconfdir=/etc/nut --datadir=/usr/share/nut --with-dev --with-usb --without-avahi --without-snmp --without-serial --without-doc --with-neon --without-powerman --without-wrap --with-hotplug-dir=/etc/hotplug --without-cgi --without-ipmi --without-freeipmi --without-linux-i2c --with-ssl --with-openssl --without-libltdl --without-macosx_ups --with-statepath=/var/run/nut --with-pidpath=/var/run --with-drvpath=/lib/nut --with-user=root --with-group=root
   0.000161     [D1] debug level is '2'
   0.000511     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000594     [D1] upsdrv_initups (non-SHUT)...
   0.000640     [D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.002890     [D2] Checking device 1 of 9 (05E3/0626)
   0.238988     [D2] - VendorID: 05e3
   0.239101     [D2] - ProductID: 0626
   0.239155     [D2] - Manufacturer: GenesysLogic
   0.239193     [D2] - Product: USB3.1 Hub
   0.239229     [D2] - Serial Number: unknown
   0.239269     [D2] - Bus: 002
   0.239305     [D2] - Bus Port: 004
   0.239342     [D2] - Device: 013
   0.239378     [D2] - Device release number: 0656
   0.239414     [D2] Trying to match device
   0.239456     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.239535     [D2] match_function_regex: failed match of VendorID:  5e3
   0.239572     [D2] Device does not match - skipping
   0.347534     [D2] Checking device 2 of 9 (152D/578E)
   0.348508     [D2] - VendorID: 152d
   0.348586     [D2] - ProductID: 578e
   0.348625     [D2] - Manufacturer: TOSHIBA
   0.348666     [D2] - Product: DT01ACA300
   0.348701     [D2] - Serial Number: 00027D0SNUAS
   0.348740     [D2] - Bus: 002
   0.348777     [D2] - Bus Port: 001
   0.348813     [D2] - Device: 011
   0.348851     [D2] - Device release number: 0515
   0.348886     [D2] Trying to match device
   0.348929     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.348985     [D2] match_function_regex: failed match of VendorID: 152d
   0.349024     [D2] Device does not match - skipping
   0.349105     [D2] Checking device 3 of 9 (0D9F/0004)
   0.355217     [D2] - VendorID: 0d9f
   0.355268     [D2] - ProductID: 0004
   0.355305     [D2] - Manufacturer: POWERCOM Co.,LTD
   0.355345     [D2] - Product: HID UPS Battery
   0.355383     [D2] - Serial Number: 004-0D9F-000
   0.355420     [D2] - Bus: 001
   0.355458     [D2] - Bus Port: 004
   0.355498     [D2] - Device: 056
   0.355537     [D2] - Device release number: 0002
   0.355575     [D2] Trying to match device
   0.355620     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.355680     [D2] Device matches
   0.355718     [D2] Reading first configuration descriptor
   0.355813     [D2] Claimed interface 0 successfully
   0.356556     [D2] Retrieved HID descriptor (expected 9, got 9)
   0.356603     [D2] HID descriptor length 996
   0.388280     [D2] Report Descriptor size = 996
   0.388569     Using subdriver: EXPLORE HID 0.2
   0.388620     [D1] 102 HID objects found
   0.389269     [D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.389918     [D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.390528     [D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 4
   0.391223     [D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 1
   0.391916     [D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 2
   0.392471     [D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x09, Offset: 0, Size: 8, Value: 100
   0.392529     [D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x09, Offset: 8, Size: 8, Value: 100
   0.393223     [D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x0e, Offset: 0, Size: 8, Value: 100
   0.393282     [D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 100
   0.393916     [D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 8, Value: 30
   0.394471     [D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 10
[... 260 lines cut ...]
  50.631931     [D1] upsdrv_updateinfo...
  51.382175     [D2] nut_libusb_get_interrupt: Connection timed out
[... cut ...]
  53.012966     [D2] Path: UPS.PowerSummary.PresentStatus.CommunicationLost, Type: Input, ReportID: 0x0a, Offset: 10, Size: 1, Value: 0
  53.012997     [D2] Path: UPS.PowerSummary.PresentStatus.Overload, Type: Input, ReportID: 0x0a, Offset: 11, Size: 1, Value: 0
  53.013027     [D2] Path: UPS.PowerSummary.PresentStatus.ff860080, Type: Input, ReportID: 0x0a, Offset: 15, Size: 1, Value: 0
  53.013053     [D1] Quick update...
^C  54.086955   [D1] set_exit_flag: raising exit flag due to signal 2
  54.087009     Signal 2: exiting
  54.087036     [D1] upsdrv_cleanup...
```

jimklimov, 2025-01-16:
So it does have those readings, although probably not scaled well:
````
   0.408663     [D4] Entering libusb_get_report
   0.409173     [D3] Report[get]: (2 bytes) => 18 64
   0.409199     [D1] Path: UPS.Battery.RemainingCapacity, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 100
   0.409220     [D4] Entering libusb_get_report
   0.409749     [D3] Report[get]: (3 bytes) => 19 dc 00
   0.409774     [D1] Path: UPS.Battery.ConfigVoltage, Type: Feature, ReportID: 0x19, Offset: 0, Size: 16, Value: 3072
   0.409797     [D4] Entering libusb_get_report
   0.410354     [D3] Report[get]: (3 bytes) => 1a 0d 07
   0.410379     [D1] Path: UPS.Battery.Voltage, Type: Feature, ReportID: 0x1a, Offset: 0, Size: 16, Value: 2304
````

I think a higher debug verbosity, 5 or 6, should report also how it parsed these reports (base values and exponents involved). 

My current guess is that the results were so wildly out of range that they were ignored. It may be that the vendor/firmware got the USB HID encoding wrong and so the values got shifted, in decimal or binary...

Vladdrako, 2025-01-16:
>I think a higher debug verbosity, 5 or 6, should report also how it parsed these reports (base values and exponents involved).

```
root@MR7350:[~]: # /lib/nut/usbhid-ups -DDDDDD -u root -x explore -x vendorid=0d9f -x port=auto -s ups
Network UPS Tools - Generic HID driver 0.55 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
   0.000100     [D3] main_arg: var='explore' val='<null>'
   0.000157     [D5] send_to_all: SETINFO driver.flag.explore "enabled"
   0.000196     [D3] main_arg: var='vendorid' val='0d9f'
   0.000247     [D5] send_to_all: SETINFO driver.parameter.vendorid "0d9f"
   0.000288     [D3] main_arg: var='port' val='auto'
   0.000329     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000373     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000414     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000439     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r30324-c4c1e9f469) 13.3.0 and configured with flags: --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --sysconfdir=/etc/nut --datadir=/usr/share/nut --with-dev --with-usb --without-avahi --without-snmp --without-serial --without-doc --with-neon --without-powerman --without-wrap --with-hotplug-dir=/etc/hotplug --without-cgi --without-ipmi --without-freeipmi --without-linux-i2c --with-ssl --with-openssl --without-libltdl --without-macosx_ups --with-statepath=/var/run/nut --with-pidpath=/var/run --with-drvpath=/lib/nut --with-user=root --with-group=root
   0.000512     [D1] debug level is '6'
   0.000555     [D5] send_to_all: SETINFO driver.debug "6"
   0.000598     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000995     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001083     [D5] send_to_all: SETINFO device.type "ups"
   0.001132     [D5] send_to_all: SETINFO driver.state "init.device"
   0.001192     [D1] upsdrv_initups (non-SHUT)...
   0.001235     [D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.003509     [D2] Checking device 1 of 9 (05E3/0626)
   0.245172     [D2] - VendorID: 05e3
   0.245271     [D2] - ProductID: 0626
   0.245311     [D2] - Manufacturer: GenesysLogic
   0.245348     [D2] - Product: USB3.1 Hub
   0.245385     [D2] - Serial Number: unknown
   0.245423     [D2] - Bus: 002
   0.245458     [D2] - Bus Port: 004
   0.245494     [D2] - Device: 013
   0.245530     [D2] - Device release number: 0656
   0.245566     [D2] Trying to match device
   0.245607     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.245669     [D3] match_function_regex: matching a device...
   0.245724     [D2] match_function_regex: failed match of VendorID:  5e3
   0.245762     [D2] Device does not match - skipping
   0.353728     [D2] Checking device 2 of 9 (152D/578E)
   0.354678     [D2] - VendorID: 152d
   0.354756     [D2] - ProductID: 578e
   0.354795     [D2] - Manufacturer: TOSHIBA
   0.354835     [D2] - Product: DT01ACA300
   0.354871     [D2] - Serial Number: 00027D0SNUAS
   0.354906     [D2] - Bus: 002
   0.354946     [D2] - Bus Port: 001
   0.354981     [D2] - Device: 011
   0.355018     [D2] - Device release number: 0515
   0.355054     [D2] Trying to match device
   0.355098     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.355140     [D3] match_function_regex: matching a device...
   0.355194     [D2] match_function_regex: failed match of VendorID: 152d
   0.355233     [D2] Device does not match - skipping
   0.355294     [D2] Checking device 3 of 9 (0D9F/0004)
   0.361330     [D2] - VendorID: 0d9f
   0.361383     [D2] - ProductID: 0004
   0.361421     [D2] - Manufacturer: POWERCOM Co.,LTD
   0.361464     [D2] - Product: HID UPS Battery
   0.361502     [D2] - Serial Number: 004-0D9F-000
   0.361538     [D2] - Bus: 001
   0.361577     [D2] - Bus Port: 004
   0.361613     [D2] - Device: 056
   0.361650     [D2] - Device release number: 0002
   0.361687     [D2] Trying to match device
   0.361731     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.361775     [D3] match_function_regex: matching a device...
   0.361829     [D2] Device matches
   0.361867     [D2] Reading first configuration descriptor
   0.361921     [D3] libusb_kernel_driver_active() returned 0: Success
[... 1522 lines cut ...]
  24.807208     [D3]  00 00 00 00 d0 9b 50 08 00 00 00 00 00 70 21 00 00 00 00 00 00 20 21 00 00
  24.807247     [D3]  00 00 00 10 56 30 e5 7f 00 00 00 b8 8e 76 ac 7f 00 00 00 20 56 30 e5 7f 00
  24.807286     [D3]  00 00 b8 8e 76 ac 7f 00 00 00 0e 00 00 00 00 00 00 00 01 00 00 00 00 00 00
  24.807327     [D3]  00 30 9b 50 08 00 00 00 00 00 90 7f ac 7f 00 00 00 01 00 00 00 00 00 00 00
  24.807381     [D3]  00 b0 7f ac 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
  24.807644     [D5] hid_lookup_path: 00850066 -> RemainingCapacity
  24.807664     [D2] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x0e, Offset: 0, Size: 8, Value: 100
  24.807689     [D3] NUT doesn't use this HID object
  24.807707     [D1] Quick update...
  24.807735     [D5] send_to_all: SETINFO driver.state "quiet"
  26.633645     Signal 2: exiting
  26.633719     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  26.633736     [D1] upsdrv_cleanup...
  26.634143     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

jimklimov, 2025-01-16:
According to https://eleccelerator.com/usbdescreqparser/ these bytes meant:
````
...
0x85, 0x18,        //     Report ID (24)
0x09, 0x66,        //     Usage (0x66)
0x75, 0x08,        //     Report Size (8)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x64,        //     Logical Maximum (100)
0xB1, 0xA2,        //     Feature (Data,Var,Abs,No Wrap,Linear,No Preferred State,No Null Position,Volatile)
0x05, 0x84,        //     Usage Page (Power Pages)
0x85, 0x19,        //     Report ID (25)
0x09, 0x40,        //     Usage (0x40)
0x75, 0x10,        //     Report Size (16)
0x16, 0x00, 0x0C,  //     Logical Minimum (3072)
0x26, 0x00, 0x30,  //     Logical Maximum (12288)
0xB1, 0x22,        //     Feature (Data,Var,Abs,No Wrap,Linear,No Preferred State,No Null Position,Non-volatile)
0x85, 0x1A,        //     Report ID (26)
0x09, 0x30,        //     Usage (0x30)
0x75, 0x10,        //     Report Size (16)
0x16, 0x00, 0x09,  //     Logical Minimum (2304)
0x26, 0x00, 0x24,  //     Logical Maximum (9216)
0xB1, 0xA2,        //     Feature (Data,Var,Abs,No Wrap,Linear,No Preferred State,No Null Position,Volatile)
...
````

So the values reported are the LogMin - which are probably passed wrong by the device, or scaled like 100x?.. Not sure how 12V (nominal) and 11V-14V (range) fits into those numbers with any decimal factors.

Vladdrako, 2025-01-16:
>So the values reported are the LogMin - which are probably passed wrong by the device, or scaled like 100x?.. Not sure how 12V (nominal) and 11V-14V (range) fits into those numbers with any decimal factors.

To note, UPSMON PRO reports 13.7-13.8V.
[image]

jimklimov, 2025-01-16:
Thanks, so in your driver's report it's effectively the same log min/max...
```
   0.417542     [D4] Entering libusb_get_report
   0.418321     [D3] Report[get]: (2 bytes) => 18 64
   0.418346     [D5] PhyMax = 0, PhyMin = 0, LogMax = 100, LogMin = 0
   0.418362     [D5] Unit = 00000000, UnitExp = 0
   0.418376     [D5] Exponent = 0
   0.418391     [D5] hid_lookup_path: 00840004 -> UPS
   0.418407     [D5] hid_lookup_path: 00840012 -> Battery
   0.418423     [D5] hid_lookup_path: 00850066 -> RemainingCapacity
   0.418443     [D1] Path: UPS.Battery.RemainingCapacity, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 100
   0.418463     [D4] Entering libusb_get_report
   0.419042     [D3] Report[get]: (3 bytes) => 19 dc 00
   0.419062     [D5] PhyMax = 0, PhyMin = 0, LogMax = 12288, LogMin = 3072
   0.419080     [D5] Unit = 00000000, UnitExp = 0
   0.419093     [D5] Exponent = 0
   0.419108     [D5] hid_lookup_path: 00840004 -> UPS
   0.419122     [D5] hid_lookup_path: 00840012 -> Battery
   0.419138     [D5] hid_lookup_path: 00840040 -> ConfigVoltage
   0.419156     [D1] Path: UPS.Battery.ConfigVoltage, Type: Feature, ReportID: 0x19, Offset: 0, Size: 16, Value: 3072
   0.419176     [D4] Entering libusb_get_report
   0.419749     [D3] Report[get]: (3 bytes) => 1a 0d 07
   0.419767     [D5] PhyMax = 0, PhyMin = 0, LogMax = 9216, LogMin = 2304
   0.419783     [D5] Unit = 00000000, UnitExp = 0
   0.419796     [D5] Exponent = 0
   0.419810     [D5] hid_lookup_path: 00840004 -> UPS
   0.419825     [D5] hid_lookup_path: 00840012 -> Battery
   0.419841     [D5] hid_lookup_path: 00840030 -> Voltage
   0.419859     [D1] Path: UPS.Battery.Voltage, Type: Feature, ReportID: 0x1a, Offset: 0, Size: 16, Value: 2304
```

I don't see any reports where they would encode a meaningful `Unit = ...` either.

For comparison, from another device and ticket (#1208 randomly found in the tracker), reasonable reports for these data points look like this:
````
   0.186788     [D4] Entering libusb_get_report
   0.192544     [D3] Report[get]: (3 bytes) => 25 60 09
   0.192618     [D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.192638     [D5] Unit = 00f0d121, UnitExp = 5
   0.192656     [D5] Exponent = -2
   0.192677     [D5] hid_lookup_path: 00840004 -> UPS
   0.192698     [D5] hid_lookup_path: 00840012 -> Battery
   0.192719     [D5] hid_lookup_path: 00840040 -> ConfigVoltage
   0.192761     [D1] Path: UPS.Battery.ConfigVoltage, Type: Feature, ReportID: 0x25, Offset: 0, Size: 16, Value: 24

   0.192783     [D4] Entering libusb_get_report
   0.198525     [D3] Report[get]: (3 bytes) => 26 aa 0a
   0.198597     [D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.198617     [D5] Unit = 00f0d121, UnitExp = 5
   0.198636     [D5] Exponent = -2
   0.198660     [D5] hid_lookup_path: 00840004 -> UPS
   0.198680     [D5] hid_lookup_path: 00840012 -> Battery
   0.198701     [D5] hid_lookup_path: 00840030 -> Voltage
   0.198779     [D1] Path: UPS.Battery.Voltage, Type: Feature, ReportID: 0x26, Offset: 0, Size: 16, Value: 27.3
````

jimklimov, 2025-01-16:
In NUT drivers, we do have a way to fix-up broken report descriptors, but need a pattern to identify the breakage and confirm a correct value by.

Alternately, if there were more readings and some other reported the correct value, we could add that into the source mapping (so whichever works - wins).

But by quick search through this page's text so far, I did not find any "137", "138", "13.7" or "13.8" values in the debug reports...

Vladdrako, 2026-01-14:
With this hack, it works
```
--- a/drivers/powercom-hid.c
+++ b/drivers/powercom-hid.c
@@ -25,6 +25,113 @@
 #include "usbhid-ups.h"
 #include "powercom-hid.h"
 #include "usb-common.h"
+#include <ctype.h>
+#include <libusb-1.0/libusb.h>
+
+/* FAIL-SAFE HACK: Use global udev handle from usbhid-ups.h */
+/* Custom conversion function that reads Report 0xa4 directly via libusb */
+static const char *powercom_hack_voltage(double value)
+{
+	static char buf[32];
+	static char last_valid[32] = "0.0"; /* Cache last valid reading */
+	int ret;
+	unsigned char data[8];
+
+	/* Global 'udev' is defined in usbhid-ups.h and initialized by main driver */
+	if (!udev)
+		return last_valid;
+
+	/* Direct read of Report 0xa4 (Feature) */
+	/* 0xA1: Dir=IN, Type=Class, Recp=Interface */
+	ret = libusb_control_transfer(udev,
+		0xA1, 
+		0x01, /* GET_REPORT */
+		(0x03 << 8) | 0xa4, /* Feature + 0xa4 */
+		0, 
+		data, sizeof(data), 1000);
+
+	if (ret > 0) {
+		char msg[32];
+		int len = (ret > 8) ? 8 : ret;
+		int j=0;
+		/* Skip byte 0 (Report ID) */
+		for(int i=1; i<len; i++) msg[j++] = data[i];
+		msg[j] = '\0';
+
+		/* Parse " 13.7 2" -> "13.7" */
+		char *start = msg;
+		while (*start && !isdigit(*start) && *start != '.') start++;
+		char *end = start;
+		while (*end && (isdigit(*end) || *end == '.')) end++;
+		*end = '\0';
+
+		/* VALIDATION: Check for decimal point to avoid "0100" garbage */
+		if (strlen(start) > 0 && strchr(start, '.') != NULL) {
+			snprintf(buf, sizeof(buf), "%s", start);
+			/* Update cache */
+			snprintf(last_valid, sizeof(last_valid), "%s", start);
+			return buf;
+		}
+	}
+
+	/* Return last valid value if read failed or data was invalid (to avoid flickering 0.0) */
+	return last_valid;
+}
+
+static info_lkp_t powercom_hack_voltage_lkp[] = {
+	{ 0, NULL, powercom_hack_voltage, NULL }
+};
+
+/* FAIL-SAFE HACK: Use global udev handle from usbhid-ups.h */
+/* Custom conversion function that reads Report 0xa4 directly via libusb */
+static const char *powercom_hack_voltage(double value)
+{
+	static char buf[32];
+	int ret;
+	unsigned char data[8];
+
+	/* Global 'udev' is defined in usbhid-ups.h and initialized by main driver */
+	if (!udev)
+		return "0.0";
+
+	/* Direct read of Report 0xa4 (Feature) */
+	/* 0xA1: Dir=IN, Type=Class, Recp=Interface */
+	ret = libusb_control_transfer(udev,
+		0xA1, 
+		0x01, /* GET_REPORT */
+		(0x03 << 8) | 0xa4, /* Feature + 0xa4 */
+		0, 
+		data, sizeof(data), 1000);
+
+	if (ret > 0) {
+		char msg[32];
+		int len = (ret > 8) ? 8 : ret;
+		int j=0;
+		/* Skip byte 0 (Report ID) */
+		for(int i=1; i<len; i++) msg[j++] = data[i];
+		msg[j] = '\0';
+
+		/* Parse " 13.7 2" -> "13.7" */
+		char *start = msg;
+		while (*start && !isdigit(*start) && *start != '.') start++;
+		char *end = start;
+		while (*end && (isdigit(*end) || *end == '.')) end++;
+		*end = '\0';
+
+		/* VALIDATION: Check for decimal point to avoid "0100" garbage */
+		if (strlen(start) > 0 && strchr(start, '.') != NULL) {
+			snprintf(buf, sizeof(buf), "%s", start);
+			return buf;
+		}
+	}
+
+	/* Return 0.0 if read failed or invalid data */
+	return "0.0";
+}
+
+static info_lkp_t powercom_hack_voltage_lkp[] = {
+	{ 0, NULL, powercom_hack_voltage, NULL }
+};
 
 #define POWERCOM_HID_VERSION	"PowerCOM HID 0.72"
 /* FIXME: experimental flag to be put in upsdrv_info */
@@ -457,14 +509,11 @@
 /*	{ "BOOL", 0, 0, "UPS.PresentStatus.Tested", NULL, "%.0f", 0, NULL }, */
 /*	{ "BOOL", 0, 0, "UPS.PresentStatus.VoltageNotRegulated", NULL, "%.0f", 0, NULL }, */
 
-/*
- * According to the HID PDC specifications, the below values should report battery.voltage(.nominal)
- * PowerCOM duplicates the output.voltage(.nominal) here, so we ignore them
- *	{ "battery.voltage", 0, 0, "UPS.PowerSummary.Voltage", NULL, "%.2f", 0, NULL },
- *	{ "battery.voltage", 0, 0, "UPS.Battery.Voltage", NULL, "%.2f", 0, NULL },
- *	{ "battery.voltage.nominal", 0, 0, "UPS.PowerSummary.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, NULL },
- *	{ "battery.voltage.nominal", 0, 0, "UPS.Battery.ConfigVoltage", NULL, "%.0f", HU_FLAG_STATIC, NULL },
- */
+	/* HACK: Map battery.voltage to *ACPresent* (Report 0x0a) which is always present.
+	   But use our HACK conversion function to ignore the value and read Report 0xa4 instead! 
+	   Value 0 in lookup table matches any value.
+	*/
+	{ "battery.voltage", 0, 0, "UPS.PowerSummary.PresentStatus.ACPresent", NULL, "%s", 0, powercom_hack_voltage_lkp },
 	{ "battery.charge", 0, 0, "UPS.PowerSummary.RemainingCapacity", NULL, "%.0f", 0, NULL },
 	{ "battery.charge", 0, 0, "UPS.Battery.RemainingCapacity", NULL, "%.0f", 0, NULL },
 	{ "battery.charge.low", 0, 0, "UPS.PowerSummary.RemainingCapacityLimit", NULL, "%.0f", 0, NULL },
```
@jimklimov Maybe this patch will help you

closed 2026-01-18 by jimklimov