networkupstools/nut #2769: PowerMaster / PowerGuide (legacy?) serial Protocol support

3v1n0, 2025-01-16:
Hi, I've got an [Atlantis Land OnePower A03-S1200](https://www.atlantis-land.com/download/SCHEDATECNICA/A03-S1200.pdf) UPS and it doesn't seem to be recognized by the classic serial protocols so far (nor I see anything similar in the source code of the drivers I've quickly looked at).

The device shows up as a serial converter only on the USB side `Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics CH340 serial converter` so I don't think there's any HID support (although I can't fully confirm it since in the NAS I'm running it, I've a limited number of kenel modules and I'm currently remote, so I can't plug it elsewhere :)).

However, thanks to [this read](https://infeeeee.github.io/sysadmin/njoy-ups.html) I ended up to try the [powerguide-1.0.0](https://www.powermonitor.software/#PowerGuideSoftware) software (easier to strace and debug compared to the PowerMaster java blobs) which gave me a some simple protocol information...

```
WRITE: "\r"
READ: "#-1\r"
WRITE: "K19CuRVvhy5ZyMgqZgH\r"
READ: "#0\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "X19\r"
READ: "#C930D6DF88DB1F30\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I237.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "X41\r"
READ: "#1200,PE02022.002,000000000000,000000000000000000\r"
WRITE: "X72\r"
READ: "#1200,0720,230,40,70,05.20\r"
WRITE: "X27\r"
READ: "#230,290,162,10,300\r"
WRITE: "X15\r"
READ: "#10\r"
WRITE: "X87\r"
READ: "#\307\227\360\360\r"
WRITE: "P19\r"
READ: "#-1\r"
WRITE: "K601\r"
READ: "#0\r"
WRITE: "K411\r"
READ: "#-1\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I237.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I237.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
WRITE: "\r"
READ: "#-1\r"
WRITE: "B\r"
READ: "#I235.0O000.0L000B100V27.0F50.2H00.0R060S\200\204\320\200\300\200\r"
```

That leads to this status info:
```
        Current UPS status:
                State........................ Normal
                Power Supply by.............. Utility Power
                Input Voltage................ 235 V
                Output Voltage............... 0 V
                Utility Frequency............ 50.2 Hz
                Output Frequency............. 0.0 Hz
                Battery Capacity............. 100 %
                Load......................... 0 %
                Line Interaction............. None
                Test Result.................. Unknown
                Last Power Event............. None
```

Now... The analysis of it seems quite easy and fun, but... Before I start looking on implementing this `nutdrv_hashx` (just or any better name, as every reply is `#`-prefixed), is this something that can be added to other driver, or that some other driver already implements?

jimklimov, 2025-01-16:
Does not quickly ring a bell, check if https://networkupstools.org/ups-protocols.html shows anything similar?

3v1n0, 2025-01-23:
So... Looking at the docs I didn't find anything relevant, so I'll start hacking something... Any driver I should base on that is preferred?

jimklimov, 2025-01-23:
Not really... probably blazer or nutdrv_qx as best-evolved serial-capable ones. However it may be best to read through a few sources to see common or different ways of approaching some similar problems.

And the new-drivers doc for general structure (what methods are for what purpose, etc.), and nut-names.txt for standard data point and command names.