networkupstools/nut #2772: CDP R-Smart1510 Driver failed to start (exit status=1).

binarydepth, 2025-01-19:
When I run` .\upsdrvctl.exe start` I get `Driver failed to start (exit status=1)`. But I mainly don't know if the program is working on Standalone mode

My configuration that works on Linux

nut.conf
MODE=standalone

ups.conf

[rsmart-1510]
driver = "blazer_usb"
port = auto
langid_fix="0x409"
desc="R-Smart-1510"
vendorid = "0001"
productid = "0000"

upsmon.conf

default sample file

upssched.conf

default conf file

binarydepth, 2025-01-19:
.\nut-scanner.exe
Cannot load USB library (libusb) : The specified module could not be found.. USB search disabled.
Cannot load SNMP library (libnetsnmp) : The specified module could not be found.. SNMP search disabled.
Cannot load XML library (libneon) : The specified module could not be found.. XML search disabled.
Cannot load NUT library (libupsclient) : The specified module could not be found.. NUT search disabled.

jimklimov, 2025-01-19:
Just in case: which NUT for Windows build is this, a recent regular from Appveyor, or the 15-year-old MSI based on NUT 2.6.5?

binarydepth, 2025-01-19:
> Just in case: which NUT for Windows build is this, a recent regular from Appveyor CI, or the 15-year-old MSI based on NUT 2.6.5?

Latest download binary, Nut 2.6.5-6

jimklimov, 2025-01-20:
Please try a recent successful CI build, see https://github.com/networkupstools/nut/wiki/NUT-for-Windows

Still on the lookout how to create MSI...

binarydepth, 2025-01-22:
> Please try a recent successful CI build, see https://github.com/networkupstools/nut/wiki/NUT-for-Windows
> 
> Still on the lookout how to create MSI...

Here? https://ci.appveyor.com/project/nut-travis/nut/build/artifacts

jimklimov, 2025-01-23:
Yes, but note there are all sorts of builds mixed there - both "master" branch ones and pull requests alike :(

It may be easier to go from master branch's commit history on NUT github, find one with a green/successful status, and click that icon for links to CI jobs involved - and from there get to the  "correct" recent AppVeyor build and then to its artifacts.

binarydepth, 2025-01-27:
> Yes, but note there are all sorts of builds mixed there - both "master" branch ones and pull requests alike :(
>
> It may be easier to go from master branch's commit history on NUT github, find one [...]

I installed the old version to get the Windows service and copied the build in the NUT folder, overwriting files. I think that is the procedure, correct me if wrong. Then tried:

```
.\upsdrvctl.exe start
Network UPS Tools - UPS driver controller Windows-v2.6.5-5-7-g72f380c
Network UPS Tools - Megatec/Q1 protocol USB driver 0.10 (Windows-v2.6.5-5-7-g72f380c)
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer).```

.\nut-scanner.exe
Cannot load USB library (libusb) : The specified module could not be found.. USB search disabled.
Cannot load SNMP library (libnetsnmp) : The specified module could not be found.. SNMP search disabled.
Cannot load XML library (libneon) : The specified module could not be found.. XML search disabled.
Cannot load NUT library (libupsclient) : The specified module could not be found.. NUT search disabled.

```[rsmart-1510]
driver = "blazer_usb"
port = auto
langid_fix="0x409"
desc="R-Smart-1510"
vendorid = "0001"
productid = "0000"```

binarydepth, 2025-02-04:
I switched to Linux, all working excellent.

closed 2025-02-04 by binarydepth