networkupstools/nut #2783: RIELLO UPS IN MODE SMART = RED

wisepds, 2025-01-25:
Good morning, everyone. 
I have several RIELLO SENTINELL PRO 3000 VA UPS and I have them configured in “SMART” mode which according to their manual analyzes the quality of the network and if it is good is automatically configured in BYPASS and if there are fluctuations activates the double conversion ONLINE mode.

Unfortunately NUT does not recognize this state in their drivers and their CGI statistics resulting in BYPASS being equivalent to ONBATTERY.

How can we solve this?
Thank you.

[image]

[image]

jimklimov, 2025-01-25:
Generally speaking, you solve this by getting a new/updated NUT driver; in this case it should be additionally coded to recognize the situation - hopefully based on direct statement from the device "I am in SMART/ECO/..." mode rather than heuristics or config toggle like "ONLINE+BYPASS on this device means SMART/ECO/..." which has a chance to misfire when it is really BYPASS for UPS maintenance, with no intention of returning to battery in case of problems.

You might want to contact riello-ups.com (general support, maybe the addresses in driver sources/commits - but they were written a decade ago...) to get hold of the updated protocol for your device and ask if it exposes the state information as such. Or sniff with USB version of wireshark etc. to see if something pops up. Maybe sniff what the official software does (asks, sees) on the wire.

To that extent, running the NUT driver with higher debug verbosity may have it dump the USB packet contents too, not sure if that was done only in USB HID or all USB drivers - check https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity in any case.

Finally, check the recent issues and PRs tagged with the "ECO/..." label linked just above this post, as this was addressed in approx autumn 2024 for Eaton devices with a similar feature. Note there is further work to be done about this area soon on the NUT project side, specifically to re-shuffle the naming and location of this status to be more vendor/marketing-agnostic before releasing NUT v2.8.3, but the subject matter of device driver changes will remain similar.

jimklimov, 2025-01-28:
Thanks for this bit of information too :)

As a quick fix, probably the configuration toggle(s?) similar to `onlinedischarge*` bunch in `usbhid-ups` would help, generally at a risk of misfire noted above (might not be a practical problem if e.g. this UPS can not be put into real bypass for maintenance like battery replacement). I'll keep it in mind when I get to finalizing the "ECO mode" situation before NUT v.2.8.3 release.

Having information or even sendable commands to switch the state would be preferable, but can take some time to get the knowledge from vendor or elsewhere.

wisepds, 2025-01-29:
Great! If you need anything, just ask.
I'm sorry I can't be of much help because I don't know about programming, compilation, etc...
Riello uses a driver called “Riello-usb”, not the standard usb-hid.
If  you want i can upload a lusb result or any test do you want!