networkupstools/nut #2784: CyberPower SL900UC supported by usbhid-ups

neilwaybright, 2025-01-25:
CyberPower SL900UC

upsc cyberpower900
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 14100
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 14.1
battery.voltage.nominal: 12
device.mfr: CPS
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 003
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 140
input.transfer.low: 96
input.voltage: 116.0
input.voltage.nominal: 0
output.voltage: 116.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.productid: 0501
ups.realpower.nominal: 480
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

I have been unable to conduct shutdown testing yet, but all of the output has been promising and when the USB cable became disconnected I did get the communication failure notifications.

Manufacturer's Website (manuals on downloads tab): https://www.cyberpowersystems.com/product/ups/standby/sl900uc/

neilwaybright, 2025-01-25:
I am enough of a noob that I was unable to figure out adding the label "HCL" to the issue.  If someone more clever than I could, I would appreciate it.

jimklimov, 2025-01-25:
Thanks, it may be less of a noob and more of permissions situation :)