networkupstools/nut #2786: CyberPower CST135UC2 supported by usbhid-ups

neilwaybright, 2025-01-25:
CyberPower CST135UC2

upsc cyberpower1350
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4875
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 26.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CST135UC2
device.serial: CDZNW7021381
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 003
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 117.0
input.voltage.nominal: 120
output.voltage: 117.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 10
ups.mfr: CPS
ups.model: CST135UC2
ups.productid: 0601
ups.realpower.nominal: 810
ups.serial: CDZNW7021381
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

I have been unable to remove power (plug is in a super-inconvenient place for me), but pulling the USB cable does result in a Communications failure/Communications Restored.

Manufacturer's Website (manuals are on downloads tab): https://www.cyberpowersystems.com/product/ups/battery-backup/cst135uc2/

j3ich0, 2025-03-22:
is this why I'm getting
`[D2] match_function_regex: failed match of Product: CST135UC2`
when trying to connect my Pi to the UPS?

sudo nut-scanner -U gave me hope 
Scanning USB bus.
```
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0601"
        product = "CST135UC2"
        serial = "CDZPY7002744"
        vendor = "CPS"
        bus = "001"
```

lsusb said
```
Bus 001 Device 002: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

but it kept erroring out
```
upsc garage-ont
Init SSL without certificate database
Error: Driver not connected
```

```
sudo /lib/nut/usbhid-ups -a garage-ont -u root -DDDD
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '4'
   0.001058     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.001227     [D1] upsdrv_initups (non-SHUT)...
   0.009868     [D2] Checking device 1 of 2 (0764/0601)
   0.014907     [D2] - VendorID: 0764
   0.015067     [D2] - ProductID: 0601
   0.015190     [D2] - Manufacturer: CPS
   0.015304     [D2] - Product: CST135UC2
   0.015425     [D2] - Serial Number: CDZPY7002744
   0.015539     [D2] - Bus: 001
   0.015653     [D2] - Device: unknown
   0.015771     [D2] - Device release number: 0200
   0.015910     [D2] Trying to match device
   **0.016025     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.016150     [D3] match_function_regex: matching a device...
   0.016567     [D2] match_function_regex: failed match of Product: CST135UC2**
   0.016717     [D2] Device does not match - skipping
   0.016900     [D2] Checking device 2 of 2 (1D6B/0002)
   0.017764     [D2] - VendorID: 1d6b
   0.017897     [D2] - ProductID: 0002
   0.018010     [D2] - Manufacturer: Linux 6.6.74+rpt-rpi-v8 dwc_otg_hcd
   0.018132     [D2] - Product: DWC OTG Controller
   0.018250     [D2] - Serial Number: 3f980000.usb
   0.018362     [D2] - Bus: 001
   0.018475     [D2] - Device: unknown
   0.018590     [D2] - Device release number: 0606
   0.018702     [D2] Trying to match device
   0.018823     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.018962     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.019076     [D2] Device does not match - skipping
   0.019228     [D2] libusb1: No appropriate HID device found
   0.019345     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.019464     No matching HID UPS found
```

jimklimov, 2025-03-22:
Interesting, maybe something is going on wrong about passing the matching string there, or some memory corruption hit along the way? Repetitively?

The matcher function tries to compare as exact string, and as a regular expression (so being a substring it would also match). For kicks, can also try `".*CST135UC2.*"` to make sure it is a substring possibly surrounded by zero or more characters (at least, Java matchers tend to require that - but this should not be relevant for NUT as a C program... although it does rely on whatever the system library implementation does).

The easier workaround however could be to comment away that line in `ups.conf`, at least as long as you are not using two devices of same model, a minimum unique set of criteria should suffice (often just `vendorid` and `productid` numbers). Likewise, `bus` (also `port` and `device`) is a value that depends on external run-time circumstances and is best left commented away; in newer NUT releases, `nut-scanner` does not offer it by default.

j3ich0, 2025-03-22:
@jimklimov thanks for looking at this. 
in ups.conf, i commented out product and bus
```
[garage-ont]
    driver = usbhid-ups
#   product = UPS
    desc = "UPS in garage for ONT"
    port = auto
    vendorid = 0764
    productid = 0601
#    bus = 001
```
and now it works!! 
here is running upsc garage-ont
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 25600
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.3
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CST135UC2
device.serial: CDZPY7002744
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 117.0
input.voltage.nominal: 120
output.voltage: 117.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: CST135UC2
ups.productid: 0601
ups.realpower.nominal: 810
ups.serial: CDZPY7002744
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
Thanks @jimklimov su