networkupstools/nut #2788: PyNUTClient package is available for Arch Linux

dvdesolve, 2025-01-26:
Just a quick note: I've created PKGBUILD for PyNUTClient python module. It's available [here](https://aur.archlinux.org/packages/python-pynutclient)

jimklimov, 2025-01-26:
Thanks! Note: the license there should be "GPLv3+" not "BSD3".

jimklimov, 2025-01-26:
Also note #2774 and some other recent PRs about possible changes expected for NUT v2.8.3+ packaging.

dvdesolve, 2025-01-26:
> Thanks! Note: the license there should be "GPLv3+" not "BSD3".

Oh, my fault! Forgot to update `.SRCINFO`. Fixed

dvdesolve, 2025-01-26:
> Also note [#2774](https://github.com/networkupstools/nut/pull/2774) and some other recent PRs about possible changes expected for NUT v2.8.3+ packaging.

I'll change URL as soon as new version will be pushed. Thanks!

closed 2025-01-26 by dvdesolve