networkupstools/nut #2791: PR1500LCDRT2U Cyber Power UPS - serial number not reported so cannot run more than 1 UPS per server - tried device/port

fmagliaro, 2025-01-29:
Hello,
Have 2 PR1500LCDRT2U Cyber Power UPS hooked up via usb on the same server (proxmox).
Installed nuts via apt-get.
nut scanner and/or lsusb do not report serial number for these particular Cyber Power UPSes to differentiate the 2.
Tried using device: 001 port 002 / device 001 port 003 in ups.conf and get libusb1 permission error.

Tried doing rules via udev
/etc/udev/rules.d/99-nut-ups.rules

===========BEGIN=Contents of /etc/udev/rules.d/99-nut-ups.rules
SUBSYSTEM!="usb", GOTO="nut-usbups_rules_end"
# TrippLite
#  e.g. TrippLite SMART1500LCD - usbhid-ups
ACTION=="add|change", SUBSYSTEM=="usb|usb_device", SUBSYSTEMS=="usb|usb_device", ATTR{idVendor}=="09ae", ATTR{idProduct}=="2012", MODE="0664", GROUP="nut", RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"
ACTION=="add|change", SUBSYSTEM=="usb|usb_device", SUBSYSTEMS=="usb|usb_device", ATTR{idVendor}=="0764", ATTR{idProduct}=="0501", MODE="0664", GROUP="nut", RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"
ACTION=="add|change", SUBSYSTEM=="usb|usb_device", SUBSYSTEMS=="usb|usb_device", ATTR{idVendor}=="051d", ATTR{idProduct}=="0002", MODE="0664", GROUP="nut", RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"
===========END=Contents of /etc/udev/rules.d/99-nut-ups.rules
 
then ran command service udev restart

===========OUTPUT OF upsdevctl start
root@pve11:/etc/nut# upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Duplicate driver instance detected (PID file /run/nut/usbhid-ups-apc01.pid exists)! Terminating other driver!
Using subdriver: APC HID 0.98
root@pve11:/etc/nut# nano /etc/nut/ups.conf

===========BEGIN=ContentsOf=/etc/nut/ups.conf
pollinterval = 1
mxretry = 3
allow_duplicates

[cyberpower01]
  driver = usbhid-ups
  port = auto
  desc = "Cyber Power PR1500LCDRT2U UPS"
  vendor = "CPS"
  vendorid = 0764
  productid = 0601
  product = "OR2200LCDRT2U"
  bus = "001"  
  device = "003"

[cyberpower02]
  driver = usbhid-ups
  port = auto
  desc = "Cyber Power PR1500LCDRT2U UPS"
  vendor = "CPS"  
  vendorid = 0764
  productid = 0601
  product = "OR2200LCDRT2U"
  bus = "001"
  device = "002"

[apc01]
  driver = usbhid-ups
  port = auto
  desc = "APC UPS 1"
  vendorid = 051d
  productid = 0002
  bus = 001
#  device = 005
===========END=ContentsOf=/etc/nut/ups.conf

Output of nut-scannerScanning USB bus.
===========BEGIN=========
No start IP, skipping SNMP
Scanning XML/HTTP bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
Scanning IPMI bus.
Failed to create client: Daemon not running
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0601"
        product = "OR2200LCDRT2U"
        vendor = "CPS"
        bus = "001"
[nutdev2]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        product = "Back-UPS RS 1500G FW:865.L7 .D USB FW:L7"
        serial = "4B1813P07085"
        vendor = "American Power Conversion"
        bus = "001"
[nutdev3]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0601"
        product = "OR2200LCDRT2U"
        bus = "001"
===========END=========

Any ideas?

As soon as I try to put bus/device I get permission errors.  Tried changing port from auto to /dev/...usb.. and get error that I should be using auto.

Any help is appreciated.

Thank you,
Franco

jimklimov, 2025-01-29:
A few random ideas first:
* For systems with NUT v2.8.x and systemd, `upsdrvctl` is effectively obsolete when NDE manages systemd units wrapping each driver: the manually executed drivers would conflict with those under the service. For more details see https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE)
* /etc/udev/rules.d/99-nut-ups.rules which goes `/sbin/upsdrvctl stop; /sbin/upsdrvctl start` for all IDs would restart all NUT drivers on re-plug or other rediscovery of any matching device
* can you run the driver programs themselves with elevated debug verbosity, e.g. `/lib/nut/usbhid-ups -a cyberpower01 -DDDDDD -d1` to see what the logs would bring up - how they fail to access the devices?
* when you run just one of those drivers, does it start? (permissions errors might be or not be due to them trying to access the same devfs node simultaneously)
* newer NUT versions (2.8.1+) have an `allow_duplicates` option just for that, consider installing that or building from git per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

fmagliaro, 2025-01-29:
If I disable the entry for the 2nd Cyberpower UPS then it works.

I downloaded and ran the [tools/nut-ddl-dump.sh](https://raw.githubusercontent.com/networkupstools/nut/master/tools/nut-ddl-dump.sh)

========BEGIN=CPS__OR2200LCDRT2U__usbhid-ups__2.8.0__01.dev 
# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Wed, 29 Jan 2025 13:54:07 -0500
# DEVICE:EOC

# :; upsc cyberpower01@127.0.0.1
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1962
battery.runtime.low: 300
battery.type:
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: OR2200LCDRT2U
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.product: OR2200LCDRT2U
driver.parameter.productid: 0601
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 140
input.transfer.low: 90
input.voltage: 119.0
input.voltage.nominal: 120
output.voltage: 119.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 48
ups.mfr: CPS
ups.model: OR2200LCDRT2U
ups.productid: 0601
ups.realpower.nominal: 296
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: cyberpower01@127.0.0.1
# [battery.charge.low]
# Remaining battery level when UPS switches to LB (percent)
# Type: STRING
# Maximum length: 10
# Value: 10
#
# [battery.runtime.low]
# Remaining battery runtime when UPS switches to LB (seconds)
# Type: STRING
# Maximum length: 10
# Value: 300
#
# [input.transfer.high]
# High voltage transfer point (V)
# Type: STRING
# Maximum length: 10
# Value: 140
#
# [input.transfer.low]
# Low voltage transfer point (V)
# Type: STRING
# Maximum length: 10
# Value: 90
#
# [ups.delay.shutdown]
# Interval to wait after shutdown with delay command (seconds)
# Type: STRING
# Maximum length: 10
# Value: 20
#
# [ups.delay.start]
# Interval to wait before (re)starting the load (seconds)
# Type: STRING
# Maximum length: 10
# Value: 30
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: cyberpower01@127.0.0.1
# Instant commands supported on UPS [cyberpower01]:
#
# beeper.disable - Disable the UPS beeper
# beeper.enable - Enable the UPS beeper
# beeper.mute - Temporarily mute the UPS beeper
# beeper.off - Obsolete (use beeper.disable or beeper.mute)
# beeper.on - Obsolete (use beeper.enable)
# load.off - Turn off the load immediately
# load.off.delay - Turn off the load with a delay (seconds)
# load.on - Turn on the load immediately
# load.on.delay - Turn on the load with a delay (seconds)
# shutdown.return - Turn off the load and return when power is back
# shutdown.stayoff - Turn off the load and remain off
# shutdown.stop - Stop a shutdown in progress
# test.battery.start.deep - Start a deep battery test
# test.battery.start.quick - Start a quick battery test
# test.battery.stop - Stop the battery test
# DEVICE:EOC

=========END=CPS__OR2200LCDRT2U__usbhid-ups__2.8.0__01.dev

fmagliaro, 2025-01-29:
I ran the driver command with just one cyber power ups and one APC
root@pve11:/tmp# /lib/nut/usbhid-ups -a cyberpower01 -DDDDDD -d1
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000567     [D5] send_to_all: SETINFO device.type "ups"
   0.000574     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000578     [D1] upsdrv_initups (non-SHUT)...
   0.003948     [D2] Checking device 1 of 11 (05E3/0612)
   0.003963     [D1] Failed to open device (05E3/0612), skipping: Access denied (insufficient permissions)
   0.003965     [D2] Checking device 2 of 11 (1D6B/0003)
   0.003969     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.003971     [D2] Checking device 3 of 11 (0764/0601)
   0.005005     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005329     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005688     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005698     [D2] - VendorID: 0764
   0.005706     [D2] - ProductID: 0601
   0.005715     [D2] - Manufacturer: CPS
   0.005724     [D2] - Product: OR2200LCDRT2U
   0.005732     [D2] - Serial Number: unknown
   0.005742     [D2] - Bus: 001
   0.005750     [D2] - Device: unknown
   0.005759     [D2] - Device release number: 0200
   0.005768     [D2] Trying to match device
   0.005776     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005803     [D3] match_function_regex: matching a device...
   0.005854     [D2] Device matches
   0.005857     [D2] Reading first configuration descriptor
   0.005863     [D3] libusb_kernel_driver_active() returned 0
   0.005868     [D2] failed to claim USB device: Resource busy
   0.005872     [D2] Kernel driver already detached
   0.005875     [D2] failed to claim USB device: Resource busy
   0.005879     [D2] Kernel driver already detached
   0.005882     [D2] failed to claim USB device: Resource busy
   0.005885     [D2] Kernel driver already detached
   0.005888     [D2] failed to claim USB device: Resource busy
   0.005892     [D2] Kernel driver already detached
   0.005895     Can't claim USB device [0764:0601]@0/0: Entity not found

fmagliaro, 2025-01-29:
PS Sorry for the text being large, bold, etc.  Its all lower case and when I hit the comment button it posts all crazy.

jimklimov, 2025-01-29:
> large text

Block quotes can be surrounded by lines with 3 or 4 backticks, I edited the earlier posts

> Resource busy (still)

I wonder if the systemd-wrapped drivers still run. Try to first `systemctl stop nut-driver@cyberpower01 nut-driver@cyberpower02` and then re-run the test with `usbhid-ups` and debugging verbosity.

fmagliaro, 2025-01-29:
ran 
systemctl stop nut-driver@cyberpower01 nut-driver@cyberpower02

then ran (had to use |& tee outfile to catpure stderr and stdout)

/lib/nut/usbhid-ups -a cyberpower01 -DDDDDD -d1 |& tee outfile

   0.000000	[D1] debug level is '6'
   0.000597	[D5] send_to_all: SETINFO device.type "ups"
   0.000603	[D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000605	[D1] upsdrv_initups (non-SHUT)...
   0.004006	[D2] Checking device 1 of 11 (05E3/0612)
   0.004020	[D1] Failed to open device (05E3/0612), skipping: Access denied (insufficient permissions)
   0.004023	[D2] Checking device 2 of 11 (1D6B/0003)
   0.004028	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.004029	[D2] Checking device 3 of 11 (0764/0601)
   0.005078	[D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005448	[D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005805	[D1] nut_libusb_open get iSerialNumber failed, retrying...
   0.005807	[D2] - VendorID: 0764
   0.005808	[D2] - ProductID: 0601
   0.005810	[D2] - Manufacturer: CPS
   0.005811	[D2] - Product: OR2200LCDRT2U
   0.005812	[D2] - Serial Number: unknown
   0.005813	[D2] - Bus: 001
   0.005814	[D2] - Device: unknown
   0.005815	[D2] - Device release number: 0200
   0.005816	[D2] Trying to match device
   0.005818	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.005821	[D3] match_function_regex: matching a device...
   0.005870	[D2] Device matches
   0.005872	[D2] Reading first configuration descriptor
   0.005876	[D3] libusb_kernel_driver_active() returned 0
   0.005894	[D2] Claimed interface 0 successfully
   0.005896	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.006034	[D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 90 02
   0.006036	[D3] HID descriptor length (method 1) 656
   0.006038	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.006044	[D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 90 02
   0.006045	[D3] HID descriptor length (method 2) 656
   0.006046	[D2] HID descriptor length 656
   0.007505	[D2] Report Descriptor size = 656
   0.007510	[D3] Report Descriptor: (656 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.007516	[D3]  75 08 95 01 15 00 26 ff 00 b1 23 85 02 09 ff b1 23 85 1b 06 01 ff 95 02 09
   0.007521	[D3]  d0 09 d2 b1 22 95 01 85 03 05 85 09 89 b1 23 85 04 09 8f b1 23 85 05 09 8b
   0.007526	[D3]  b1 23 85 06 09 2c b1 23 85 07 75 08 95 06 25 64 09 83 09 8d 09 8e 09 8c 09
   0.007531	[D3]  29 09 67 b1 23 85 08 75 08 95 01 65 00 09 66 81 23 09 66 b1 a3 09 68 75 10
   0.007536	[D3]  27 ff ff 00 00 66 01 10 81 23 09 68 b1 a3 09 2a 26 58 02 81 23 09 2a b1 a2
   0.007541	[D3]  85 09 75 08 26 ff 00 05 84 09 40 67 21 d1 f0 00 55 06 b1 23 85 0a 09 30 b1
   0.007546	[D3]  a3 09 02 a1 02 65 00 55 00 85 0b 75 01 95 06 25 01 05 85 09 d0 09 44 09 45
   0.007551	[D3]  09 42 09 46 09 43 81 23 09 d0 09 44 09 45 09 42 09 46 09 43 b1 a3 75 02 95
   0.007555	[D3]  01 81 01 b1 01 c0 85 0c 05 84 09 5a 75 08 15 01 25 03 b1 a2 09 5a 81 22 85
   0.007558	[D3]  0d 09 fd 15 00 26 ff 00 b1 23 c0 05 84 09 1a a1 00 85 0e 05 84 09 40 75 08
   0.007561	[D3]  67 21 d1 f0 00 55 07 b1 23 85 0f 09 30 b1 a3 85 10 75 10 09 53 15 58 25 5c
   0.007563	[D3]  b1 a2 09 53 81 23 09 54 16 8a 00 26 8e 00 b1 a2 09 54 81 23 c0 09 1c a1 00
   0.007566	[D3]  85 12 75 08 09 30 15 00 26 ff 00 b1 a3 65 00 55 00 85 13 09 35 b1 a3 85 14
   0.007569	[D3]  09 58 25 06 b1 a2 09 58 81 22 85 15 09 57 75 10 15 ff 26 ff 7f 35 c4 47 c4
   0.007572	[D3]  ff 1d 00 66 01 10 b1 a2 85 16 09 56 b1 a2 85 17 09 6e 75 01 15 00 25 01 35
   0.007575	[D3]  00 45 00 65 00 b1 a3 09 65 b1 a3 75 06 b1 01 85 18 75 10 09 44 26 84 03 66
   0.007577	[D3]  21 d1 55 07 b1 a3 85 1a 06 01 ff 65 00 55 00 75 08 15 00 25 02 09 43 b1 a2
   0.007580	[D3]  09 43 81 23 c0 06 01 ff 09 1d a1 00 09 19 a1 02 85 25 09 20 65 00 55 00 15
   0.007583	[D3]  00 25 01 95 01 75 01 b1 82 75 07 b1 03 85 2c 09 21 75 01 b1 83 09 21 81 03
   0.007586	[D3]  75 07 b1 83 81 03 c0 09 1a a1 02 85 26 09 01 75 01 b1 82 75 07 b1 83 85 27
   0.007589	[D3]  09 02 25 03 75 08 b1 03 c0 09 1b a1 02 85 28 09 16 65 00 55 00 15 00 25 3e
   0.007592	[D3]  75 08 95 01 b1 83 09 16 81 03 09 18 26 ff 00 95 3e b2 83 01 09 18 82 03 01
   0.007595	[D3]  85 29 09 15 15 01 25 3e 95 01 b1 02 09 15 91 02 09 17 15 00 26 ff 00 95 3e
   0.007597	[D3]  b2 02 01 09 17 92 02 01 85 2d 75 01 95 03 25 01 09 10 09 1e 09 1f b1 83 09
   0.007600	[D3]  10 09 1e 09 1f 81 03 75 05 95 01 b1 83 81 03 85 2a 09 13 75 01 b1 82 75 07
   0.007602	[D3]  b1 83 85 2b 09 14 75 01 b1 82 75 07 b1 83 c0 c0 c0
   0.007653	Using subdriver: CyberPower HID 0.6
[... 1399 lines cut ...]
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 140
input.transfer.low: 90
input.voltage: 119.0
input.voltage.nominal: 120
output.voltage: 119.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 46
ups.mfr: CPS
ups.model: OR2200LCDRT2U
ups.productid: 0601
ups.realpower.nominal: 296
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

jimklimov, 2025-01-29:
Great, so that part did work and the device is supported.

It may be that they mis-encoded locale for `iSerial` property, there was a fix late last year in master branch for some broken strings like that. See #2604 or #2613 for more context on that bit.

I remembered now there was also a bug in 2.8.0 (maybe 2.8.1 too?) that the `device` was not properly detected by drivers and so could not be matched. Generally note that it is risky to rely on bus/device numbers as they are subject to re-enumeration by system.

So to support both CPS UPSes at once - you'd have to follow that wiki link and custom-build a replacement version of NUT.

fmagliaro, 2025-01-29:
I agree - don't want to use device/bus.  Reading the cutom-build replacement of NUT...
Thank you.

closed 2025-01-29 by fmagliaro