networkupstools/nut #2794: Nut shutdown assuming dead ups

electrofloat, 2025-02-01:
Hi!

I don't know if this is a bug or a misconfiguration but here is what happened.
I have 2 machines, one is `netserver` other is `netclient`.
The `netserver` is connected through usb to a `device.model: Back-UPS RS 550G` with the following config
```
[ups]
  driver = usbhid-ups
  port = auto
  onlinedischarge_calibration = 1
```

Both of the machines running `Ubuntu 24.04.1 LTS` with nut version `2.8.1-3.1ubuntu2`.

The ups seems to calibrate itself every two weeks and the last one was on:
```
2025-01-29T14:39:04.808340+01:00 server nut-monitor[2772730]: UPS ups@localhost: administratively OFF or asleep
2025-01-29T14:39:04.816059+01:00 server nut-monitor[1551623]: Network UPS Tools upsmon 2.8.1
2025-01-29T14:39:14.810120+01:00 server nut-monitor[2772730]: UPS ups@localhost: no longer administratively OFF or asleep
2025-01-29T14:39:14.818648+01:00 server nut-monitor[1551648]: Network UPS Tools upsmon 2.8.1
```

Ubuntu has this function that when a lib used by a process is upgraded it restarts the corresponding systemd unit so it uses the updated lib.
Recently a bunch of libs were updated and when nut restarted on the `netserver` the `netclient` machine shut itself down.
At first I did not understand what happened, so after manually starting `netclient` and checked the logs I could see this:
```
2025-01-30T19:57:28.515107+01:00 client nut-monitor[3178]: Poll UPS [ups@10.90.1.1] failed - Write error: Broken pipe
2025-01-30T19:57:28.515556+01:00 client nut-monitor[3178]: Communications with UPS ups@10.90.1.1 lost
2025-01-30T19:57:28.515684+01:00 client nut-monitor[3178]: UPS [ups@10.90.1.1] was last known to be calibrating and currently is not communicating, assuming dead
2025-01-30T19:57:28.515770+01:00 client nut-monitor[3178]: Executing automatic power-fail shutdown
2025-01-30T19:57:28.520256+01:00 client nut-monitor[2709826]: Network UPS Tools upsmon 2.8.1
2025-01-30T19:57:28.520422+01:00 client nut-monitor[3178]: Auto logout and shutdown proceeding
2025-01-30T19:57:28.524341+01:00 client nut-monitor[2709831]: Network UPS Tools upsmon 2.8.1
2025-01-30T19:57:33.521944+01:00 client nut-monitor[3178]: Network UPS Tools upsmon 2.8.1
2025-01-30T19:57:33.540056+01:00 client shutdown[2709872]: Shutdown scheduled for Thu 2025-01-30 19:57:33 CET, use 'shutdown -c' to cancel.
2025-01-30T19:57:33.541551+01:00 client nut-monitor[3147]: Network UPS Tools upsmon 2.8.1
2025-01-30T19:57:33.543370+01:00 client systemd[1]: nut-monitor.service: Deactivated successfully.
2025-01-30T19:57:33.543530+01:00 client systemd[1]: nut-monitor.service: Consumed 3min 45.465s CPU time, 4.5M memory peak, 556.0K memory swap peak.
```

So it seems that the `netclinet` was thinking that the ups is still in calibrating mode when the `netserver` restarted it's nut systemd service, and it was assuming the ups is dead and it immediately shutdown.

Of course I did not check what nut thinks about the state of the ups _before_ the libs upgraded, but after that I could see that both `netclient` and `netserver` reports: `ups.status: OL` correctly.

Also yesterday and today again a bunch of libs were upgraded and nut was restarted on `netserver` and this time it did not trigger the shutdown (and I've checked this time and the status was still OL)

Now.. is this a bug that somehow the `netclient` did not get the memo that the ups is not in calibrating state, or something is misconfigured?

jimklimov, 2025-02-02:
Interesting situation. My guess would be that it works as designed, at least. 

There are things known to you and unknown to the servers, such as that the service restart was "intentional". I have little idea how to even propagate the concept (is it known to services they are being restarted and will return in a second or minute?)

Perhaps upsmon could wait a cycle or two on connection loss (maybe tied to knowledge that the driver or its data server program is gracefully going down and it was not power yanked from their machine or network gear - something that could be developed e.g. around `driver.state` already reported for the curious), but in power-event case, the safe default assumption is the pessimistic one - that we have seconds to live. At least it keeps the data safe and filesystems consistent, as much as we can ensure.

Note that knowledge of being calibrated does not necessarily allow us to discount the on-battery situation as safe. I did see UPSes losing power for the load because its earlier guess was too optimistic and calibration thought the battery was still 20-30% full. Regular calibration with same load probably reduces this possibility, but still...

electrofloat, 2025-02-02:
I think the main issue here is that the netclient thought that the ups is still in calibrating state as stated by this log line:
`2025-01-30T19:57:28.515684+01:00 client nut-monitor[3178]: UPS [ups@10.90.1.1] was last known to be calibrating and currently is not communicating, assuming dead`

But it was not. The last calibration was on 2025.01.29 and the service restart happened on 2025.01.30. So 1 day later.
Also the calibration takes only 10 seconds.

So the UPS was in a simple OL state (I know it was physically seeing it, but as I said I have not checked the nutserver to be sure it also thinks it is in OL state) when the service restart happened on the server, but since the client thought that the ups was in ST_CAL it shut down the machine.

So either the server thought that the ups is still in st_cal and that state propagated to the netclient, or the server correctly thought it is on OL state but that state change somehow did not get to the client for more than a full day somehow.

jimklimov, 2025-02-04:
Well... either new calibration kicked in right during those seconds, or... From messages above, I assume you are running a NUT v2.8.1 release build? looking at https://github.com/networkupstools/nut/blob/v2.8.1/clients/upsmon.c#L2111-L2170 it might be that `ST_CAL` flag was only raised and never un-set (no `clearflag` mentions it). This was fixed in v2.8.2 by introduction of `ups_is_notcal()` and similar methods for other states: https://github.com/networkupstools/nut/blob/v2.8.2/clients/upsmon.c#L1234-L1242

Can you try a newer package or ideally a custom build per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to check that the current codebase actually does not have this practical buggy use-case, please?

electrofloat, 2025-02-04:
It was not calibrating at that moment that is for sure.
And yes I am using version `2.8.1-3.1ubuntu2` which is the official version in `Ubuntu Noble (24.04.1)`. (so `2.8.1` with modifications from debian/ubuntu).

I'll wait for the next calibration first and check the statuses on both server and client to see what they are reporting.

Then I'll maybe recompile ubuntu's version with the added https://github.com/networkupstools/nut/blob/v2.8.2/clients/upsmon.c#L1234-L1242 function and https://github.com/networkupstools/nut/blob/v2.8.2/clients/upsmon.c#L2181-L2182 call and see if that helps.

jimklimov, 2025-02-04:
Well, IMHO monkey-patching sources like that may be a bit risky - just too easy to miss something. If you do go that route, use git blame (or github UI) to track down the commits and whole PRs that delivered the change, to reduce that particular risk. It may be that the changes relied on some other work, possibly in other source files, that would not be in your patched history and codebase though.

It may be more fruitful to use a source tarball from 2.8.2 (or generate one from current master branch with `make dist`) and adjust the packaging recipes to use it whole instead of 2.8.1 tarball.

The "in-place" builds as detailed on Wiki should overlay much of the packaged installation, especially of a recent one that reports its `configure` flags, but the main focus is on functional replacement (same config files and runtime user/group names) so some "unreferenced" files might remain if e.g. drivers are newly installed into standard dirs by default, and older packaged `/lib/nut/` ones remain too but nobody would call them anymore.

jimklimov, 2025-04-22:
@electrofloat : Did you have a chance to check if a newer NUT release resolves the issue? 2.8.2 was expected to have it already fixed, and now v2.8.3 is out, finally :)

electrofloat, 2025-04-22:
Sorry, I'm a bit busy these days, I only removed the `onlinedischarge_calibration = 1` config from ups.conf hoping that this bypasses the issue since it would no longer think that being in onlinedischarge mode means it is calibrating (even though it is really calibrating).

I'll set this config back to 1 once I have the chance to try out the new version.

sic79, 2026-08-21:
I can reproduce the same shutdown path on NUT 2.8.5 (Unraid) with an APC Back-UPS BGM2200B-GR connected locally via USB.

The UPS has a firmware/HID quirk where it reports `OL DISCHRG` while mains power is present and battery charge is 100%. I had enabled:

```ini
onlinedischarge_calibration
```

This mapped the erroneous `OL DISCHRG` state to `CAL DISCHRG`.

During the incident, the exact sequence was:

```text
usbhid-ups: nut_libusb_get_report: Input/Output Error
upsd: Data for UPS [UPS] is stale - check driver
upsmon: Poll UPS [UPS@127.0.0.1] failed - Data stale
upsmon: Communications with UPS UPS@127.0.0.1 lost
upsmon: UPS [UPS@127.0.0.1] was last known to be calibrating and currently is not communicating, assuming dead
upsmon: Too few UPS(es) are healthy (0<1), initiating forced shutdown
upsd: Client monuser@127.0.0.1 set FSD on UPS [UPS]
upsmon: Executing automatic power-fail shutdown
shutdown: shutting down for system halt
```

There was no `ONBATT` or `LOWBATT` event. This was not a low-battery shutdown path. The fatal combination was:

`OL DISCHRG` reported while online
→ `onlinedischarge_calibration`
→ last known state `CAL`
→ USB/HID I/O failure and stale data
→ UPS assumed dead
→ immediate FSD / automatic shutdown.

I removed `onlinedischarge_calibration`. The raw current state is now again `OL DISCHRG` with battery charge 100% and normal input power, rather than `CAL DISCHRG`.

This appears related to #2652 (BGM1500B / Back-UPS Pro Gaming family), but demonstrates that the shutdown behavior in this issue is still reproducible on NUT 2.8.5.

Hannsre, 2026-09-11:
I noticed the same on Proxmox. 

Did you end up backporting the fix to Proxmox or did you fix it at all? Apparently it still not made it into the packaged version of trixie.

jimklimov, 2026-09-11:
https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

biergaizi, 2026-09-11:
My proposal: NUT's CI/CD infrastructure should generate some build artifacts with options that are compatible with Debian (not necessarily packages, raw binary is better than none). This will allow users to apply hotfixes and will greatly accelerate end-user testing of PRs.

jimklimov, 2026-09-12:
That's more or less what OBS builds can do and are for (other than trying more CI environments). Recipes (spec/dsc) may need some revision, though - those in tree are based on 10+ year old snapshots.

bigon, 2026-09-14:
> Upstream Debian Bug active since 2023: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1056190
> 
> But it's never backported!

Yes sorry, I'm ENOTIME.

I'll try to backport that fix later this week

GoDigitalizeMe, 2026-09-17:
I can confirm this on completely different hardware/setup than the original report, which suggests the root cause is generic rather than tied to the Ubuntu library-update restart scenario:

- **Setup**: two independent Proxmox VE hosts (Debian 13 "trixie", `nut-client` 2.8.1-5), each in `MODE=netclient`, monitoring a networked Ubiquiti UPS (via its UniFi-Network NUT bridge) as `secondary`.
- **Trigger**: a brief (a few seconds to a few minutes) network interruption to the UPS's NUT server while `ups.status` last reported `CAL` — in our case caused by real, if transient, network flakiness on the UPS's own uplink (confirmed via independent monitoring: the UPS's network-presence tracking shows repeated ~15s connect/disconnect flapping for several minutes right before the event), not a service restart.
- **Result**: `upsmon` immediately logs `"was last known to be calibrating and currently is not communicating, assuming dead"` and executes `SHUTDOWNCMD`, powering off the host (and, since it's a hypervisor, all its guest VMs) — even though `DEADTIME` was set to 60s and mains power was never actually lost.

I traced this to `clients/upsmon.c`: the `ST_CAL` check (`if (flag_isset(ups->status, ST_CAL)) { ... return 1; }`) is unconditional — unlike the sibling checks for `ST_OFF` (gated by `OFFDURATION`), `ST_OVER` (gated by `OVERDURATION`), and `ST_ALARM` (gated by `ALARMCRITICAL`), which all give the admin a way to require sustained loss-of-contact before declaring the UPS dead. `ST_CAL` has no equivalent — any comms loss while the last-known state included `CAL`, however brief, is treated as fatal, and `DEADTIME` is never consulted.

Given calibration/self-test is a normal, often-scheduled UPS operation (weekly in our case), and any network hiccup during that window — regardless of cause — now triggers an unconditional shutdown, this seems like a real gap rather than intended behavior. A `CALDURATION`-style option (analogous to `OFFDURATION`/`OVERDURATION`) that requires the outage to persist for a configurable duration before assuming the UPS is dead would close it while preserving the existing safety-first behavior for anyone who wants immediate shutdown.

Happy to test a patch against our setup if one gets proposed.

jimklimov, 2026-09-17:
A `CALDURATION` sounds like a reasonable idea and name. Gotta double-check if changes in the *current* code base with various throttle options added in the recent years do not implement that, at least in spirit.

* Just in case, if the `master` branch works and the old package does not, the bug is about running (delivering) obsolete code when fixes are available :)

bigon, 2026-09-17:
OK I pushed yesterday the fix (just the clean of the CAL flag) to debian stable/trixie, but that still needs to be [accepted](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1148073) by the release team

GoDigitalizeMe, 2026-10-09:
Follow-up from our earlier comment: after a 4th recurrence of this exact failure mode (5+ weeks since the Debian backport was promised, still not shipped as of `nut-client 2.8.1-5`), we went ahead and patched `clients/upsmon.c` locally rather than waiting further. Posting the diff here in case it's useful as a starting point for an upstream fix.

Confirmed root cause (matches what we described before, now pinned down exactly): in `is_ups_critical()`, the `commstate == 0` block's `ST_CAL` branch returns dead unconditionally, with no duration gate at all. `DEADTIME` is never consulted anywhere in this function — grepping the whole file shows `deadtime` is only read in `recalc()`, gating the unrelated ONBATT→LOWBATT promotion. So even with `DEADTIME` raised well above its default, a CAL-state comms blip of any length still triggers an instant forced shutdown.

The patch adds a `CALDURATION` config directive (mirroring `DEADTIME`'s own declaration/parsing pattern) and gates the `ST_CAL` branch with it, using the already-existing `ups->lastpoll` timestamp (zero-initialized at struct creation, updated on every successful poll — safe to use here with no extra state tracking needed):

```c
/* declaration, alongside deadtime */
static int caldurationtime = 60;

/* config parsing, alongside DEADTIME */
if (!strcmp(arg[0], "CALDURATION")) {
    caldurationtime = atoi(arg[1]);
    return 1;
}

/* is_ups_critical(), gating the ST_CAL branch */
if (flag_isset(ups->status, ST_CAL)) {
    if (caldurationtime < 0 || (now - ups->lastpoll) > caldurationtime) {
        upslogx(LOG_WARNING,
            "UPS [%s] was last known to be calibrating "
            "and currently is not communicating, assuming dead",
            ups->sys);
        return 1;
    }
    upsdebugx(1,
        "UPS [%s] lost comms while calibrating, within "
        "CALDURATION (%ds) grace period (%d s since last poll) "
        "- not yet declaring dead",
        ups->sys, caldurationtime, (int)(now - ups->lastpoll));
}
```

`-1` disables the grace period (restores the current instant-declare behavior) for anyone who wants that. Default of 60s chosen to match our own `DEADTIME` value, tolerating the short (~10s) comms blips we've observed without masking a genuinely dead UPS for long.

Built and running on both our PVE hosts since today against `nut 2.8.1-5` (Debian trixie) with zero false shutdowns so far. Happy to open a proper PR against this repo if that's useful instead of just posting the diff here — let us know.

jimklimov, 2026-10-09:
@GoDigitalizeMe : looks promosing, thanks! A PR would be welcome, impacting a lot of places as these settings do: docs, `conf/*` samples, nutconf (C++) and augeas parsers, maybe I missed naming something. Better double-check with earlier PRs that modified something of this sort.

I think the chosen name could be confusing however, I had initially read it as "allowed(?) duration for calibration itself"; maybe `CALDEADTIME` would be more to the point.

bigon, 2026-11-05:
The package has finally been accepted to debian stable, please test

https://tracker.debian.org/news/1819532/accepted-nut-281-5deb13u1-source-into-proposed-updates/