networkupstools/nut #2795: Deconstruct Unknown UPS megatec sort of

aplayerv1, 2025-02-02:
Ok so I have this annoying UPS that constantly gives me Zeros on nut and it was pissing me off that I ignored it for a long while.

I captured data I reversed the bullshit from UPSmart. I am sure some of you know what I am talking about.

Now I am no expert and nut kept giving me 0 "ZEROS"

        000.0 000.0 000.0 000 00.0 0.00 00.0 00000000

on what ever driver I tried and protocol.

I used python this code 

          import usb.core
          import usb.util
          import time
          
          VENDOR_ID = 0x0001    # Vendor ID of the device
          PRODUCT_ID = 0x0000   # Product ID of the device
          TIMEOUT = 60000       # Timeout in milliseconds
          RETRY_LIMIT = 5
          RETRY_DELAY = 2       # seconds
          
          def log_debug(message):
              """Helper function to log debug messages with timestamps."""
              print(f"{time.strftime('%Y-%m-%d %H:%M:%S')} - {message}")
          
          def initialize_ups(dev):
              log_debug("Initializing MEC0003 UPS...")
              setup_commands = [
                  (0x80, 0x06, 0x0300, 0x0000, 255),  # Get String Descriptor
                  (0x80, 0x06, 0x0303, 0x0409, 255)   # Get Specific String Descriptor
              ]
              for request in setup_commands:
                  try:
                      log_debug(f"Sending setup command: {request}")
                      response = dev.ctrl_transfer(*request, TIMEOUT)
                      log_debug(f"Response: {list(response)}")
                  except usb.core.USBError as e:
                      log_debug(f"[ERROR] Failed to initialize UPS with command {request}: {e}")
          
          def find_ups():
              log_debug("Searching for MEC0003 UPS device...")
              dev = usb.core.find(idVendor=VENDOR_ID, idProduct=PRODUCT_ID)
              if dev is None:
                  log_debug("[ERROR] MEC0003 UPS device not found.")
                  return None
          
              log_debug("MEC0003 UPS device found! Resetting and configuring...")
              dev.reset()
              
              # Detach kernel driver if necessary
              if dev.is_kernel_driver_active(0):
                  log_debug("Kernel driver is active. Detaching...")
                  dev.detach_kernel_driver(0)
          
              dev.set_configuration()
              cfg = dev.get_active_configuration()
              intf = cfg[(0, 0)]
              
              # Claim the interface for our use
              usb.util.claim_interface(dev, 0)
              initialize_ups(dev)
          
              # Find the interrupt endpoints (for completeness)
              ep_in = usb.util.find_descriptor(
                  intf,
                  custom_match=lambda e: usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_IN
              )
              ep_out = usb.util.find_descriptor(
                  intf,
[... 284 lines cut ...]
Receive | (226.0 000.0 226.0 029 50.1 27.1 29.0 00001000
  |  
Send | 80 06 0c 03 09 04 00 66
Receive | #
  |  
Send | 80 06 03 03 09 04 00 66
Receive | (223.0 000.0 224.0 029 50.0 27.1 29.0 00001000
  |  
Send | 80 06 0d 03 09 04 00 66
Receive | #000.0 0.0 00.00 00.0
  |  
Send | 80 06 03 03 09 04 00 66
Receive | 
Send | 80 06 03 03 09 04 00 66
Receive | (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  |  
Receive | #
  |  
Send | 80 06 03 03 09 04 00 66

</body>

</html>

I hope that some people with better knowledge pertaining to nut can implement something. I tried 2.7 and compiled from git and nothing seemed to work.

jimklimov, 2025-02-04:
Thanks a lot for the data and script! Do I understand the logs above correctly, that same queries sometimes return good or bogus data or nothing at all, depending on... phase of the moon? mood? got any pattern there? :)

Is the cabling reliable (e.g. just wires, or a twisted cable wrapped into foil/net grounded by connectors to shield against EMI)?

aplayerv1, 2025-02-05:
> Thanks a lot for the data and script! Do I understand the logs above correctly, that same queries sometimes return good or bogus data or nothing at all, depending on... phase of the moon? mood? got [...]

I switched the cable and tried multiple others, including a very short one. I even disassembled the UPS to inspect the internals for any failed components. I also captured USB data, and while I'm just guessing, it seems like the UPS is too slow to send the data, causing the computer to read zeros instead.

The UPS needs to be initialized; otherwise, it will continuously send zeros—literally nothing but zeros.

jimklimov, 2025-02-06:
I think it would not hurt. Not much of a low-level developer myself, but if someone gets around to coding this - any inputs can help.

aplayerv1, 2025-02-06:
[https://other.mvia.ca/uspusbcapture.pcapng](url) can get the pcapng for wireshark here