networkupstools/nut #2804: upsdrvctl - error: Driver not connected

matthys70, 2025-02-16:
I got Ubuntu server with Ubuntu 24.04.2 LTS and today I changed my UPS with Cyber Power System model VP700ELCD and is no longer working.

First I notice that nut-scanner was no longer working and I had to symbolic link the library:
/usr/lib/libusb-1.0.so -> /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.4.0

Now it's working are report the correct information:
# nut-scanner -U
Cannot load SNMP library (libnetsnmp.so) : file not found. SNMP search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load AVAHI library (libavahi-client.so) : file not found. AVAHI search disabled.
Cannot load IPMI library (libfreeipmi.so) : file not found. IPMI search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0501"
        product = "VP700ELCD"
        serial = ""
        vendor = "CPS"
        bus = "001"
        device = "002"
        busport = "002"
        ###NOTMATCHED-YET###bcdDevice = "0002"

# WARNING: <empty> "serial" reported in some devices: nutdev1

However if I try to start the driver I get this (my ups is called ups and not nutdev1 in the config):
# upsdrvctl -DD -d start ups
Network UPS Tools - UPS driver controller 2.8.1
   0.000003     [D1] upsdrvctl commanding one driver (ups): start
   0.000103     [D1] Starting UPS: ups
   0.000136     [D1] WARNING: Requested a debugging level but not explicitly a backgrounding mode - driver may never try to fork away; however the upsdrvctl tool will fork it and not wait.
   0.000311     [D2] 3 remaining attempts
   0.000379     [D2] exec:  /lib/nut/usbhid-ups -DD -a ups
   0.000902     [D2] Starting driver with debug but without explicit backgrounding: will not wait for it to fork and detach, continuing...
   0.001068     [D1] upsdrvctl: successfully finished
   0.001097     [D1] Completed the job of upsdrvctl tool, cleaning up and exiting now
   0.001124     [D1] Completed the job of upsdrvctl tool, clean-up finished, exiting now
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
root@vienna:/#    0.000000      [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with gcc (Ubuntu 13.2.0-23ubuntu2) 13.2.0 and configured with flags: --build=x86_64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/x86_64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix= --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=${prefix}/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-doc=man,html-single,html-chunked,pdf
   0.000090     [D1] debug level is '2'
   0.004068     [D1] Succeeded to become_user(nut): now UID=121 GID=134
   0.004130     [D1] upsdrv_initups (non-SHUT)...
   0.004142     [D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.010770     [D2] Checking device 1 of 7 (1D6B/0002)
   0.010812     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.010830     [D2] Checking device 2 of 7 (1D6B/0001)
   0.010843     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.010857     [D2] Checking device 3 of 7 (1D6B/0001)
   0.010869     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.010876     [D2] Checking device 4 of 7 (1D6B/0001)
   0.010888     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.010931     [D2] Checking device 5 of 7 (1D6B/0001)
   0.010953     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.010982     [D2] Checking device 6 of 7 (0764/0501)
   0.024271     [D2] - VendorID: 0764
   0.024286     [D2] - ProductID: 0501
   0.024295     [D2] - Manufacturer: CPS
   0.024308     [D2] - Product: VP700ELCD
   0.024323     [D2] - Serial Number:
   0.024332     [D2] - Bus: 001
   0.024341     [D2] - Bus Port: 002
   0.024352     [D2] - Device: 002
   0.024367     [D2] - Device release number: 0002
   0.024383     [D2] Trying to match device
   0.024395     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.024499     [D2] match_function_regex: failed match of Device: 002
   0.024517     [D2] Device does not match - skipping
   0.024561     [D2] Checking device 7 of 7 (1D6B/0001)
   0.024582     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.024674     [D2] libusb1: No appropriate HID device found
   0.024691     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.024700     No matching HID UPS found
   0.024719     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it

Why is it checking device 1 to 5 and 7? And only device 6 is the configured one at 0764/0501, which is also in the udev rules:
# cat /etc/udev/rules.d/62-nut-usbups.rules | grep 0764
ATTR{idVendor}=="0764", ATTR{idProduct}=="0005", MODE="664", GROUP="nut"
ATTR{idVendor}=="0764", ATTR{idProduct}=="0501", MODE="664", GROUP="nut"
ATTR{idVendor}=="0764", ATTR{idProduct}=="0601", MODE="664", GROUP="nut"

So it does find the device but (not match?) fails to start, maybe because of the other devices or HID?
Or is this already been fixed but not with Ubuntu itself?

jimklimov, 2025-02-17:
Yes, probably that release still had a bug about `device` matching. Also newer versions do not by default suggest enumeration-dependent fields anymore (they can change at run-time due to OS/HW reasons).

Note that frequency also seems wrong (some devices report in Hz; others in 0.1Hz, but fail to report the scaling factor too); newer NUT builds should deal with that better.

Probably "CPS" as battery date is mis-reported in USB HID field for the date, but I've not noticed *such* mistakes bbefore.

matthys70, 2025-02-18:
Seems the battery.mfr.date is a common problem for Cyber Power System, as it also mention in #2786 and #2784 

Looking forward to the new improved version.

domasist, 2025-05-29:
On Ubuntu 24.04.2 LTS my ups eaton was not even listed in usb devices (lsusb). In Ubuntu 22.04.5 LTS the same UPS is working just fine. 

What could be the problem?

jimklimov, 2025-05-29:
Same computer? Some kernel driver problem?..

Different computers? Hardware issues?

domasist, 2025-05-30:
same computer, same hardware. 

I use the same combination pc hw and ups hw:
- on Ubuntu Desktop 22.04.5 LTS working well
- on Ubuntu Server 24.04.2 not working, multiple pcs with the same result