networkupstools/nut #2805: NUT is not picking up accurate and timely data from APC UPS

ausfas, 2025-02-16:
I'm using NUT add-on and integration in Home assistant. My UPS is APC  BX1200.
The integration data is delayed and doesn't update after a certain point. In this situation, the charge level is stuck at 99% while I unplugged the USB and attached the PC with Power Chute personal edition and got it as 100%. Then reconnecting is back to RPi it shows 100%.Turn off the input to UPS so it dropped to 95-96% and after a while it again stuck at 99% showing charging as not 100% despite it being 100% (confirmed from Power Chute) but the integration doesn't pick it up. Also any changes are displayed with a delay of 1-2 mins.

Not sure why it reports up to 99% fine and not after that. Yesterday it got stuck at 95% then did the same process of connecting to Power Chute and back.

The Add-on always have log entries of can't write to PID fopen issues so not sure if that is the root cause or not as others have the same PID fopen error but have not reported this data stale issue.
I raised this issue with HA https://github.com/home-assistant/core/issues/125723
and after investigation advised the issue is NUT package and should raise it in here.
```
Logs from the UPS NUT Add-on

Using subdriver: APC HID 0.98
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[17:05:25] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000023	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000239	[D1] debug level is '1'
   0.000321	listening on 0.0.0.0 port 3493
   0.000490	Connected to UPS [APC]: usbhid-ups-APC
   0.001020	Running as foreground process, not saving a PID file
[17:05:26] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000023	Could not find PID file to see if previous upsmon instance is already running!

   0.000253	Using power down flag file /etc/killpower
   0.000387	UPS: APC@localhost (primary) (power value 1)
   0.000401	[D1] debug level is '1'
   0.000427	[D1] debug level is '1'
   0.000432	Warning: running as one big root process by request (upsmon -p)
   0.000489	[D1] Saving PID 451 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.002375	[D1] Trying to connect to UPS [APC@localhost]
   0.509793	User upsmonmaster@127.0.0.1 logged into UPS [APC]
   0.003079	[D1] Logged into UPS APC@localhost
```
more
```
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[01:16:21] INFO: �[32mStarting the UPS information server...�[0m
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000033	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000275	[D1] debug level is '1'
   0.000385	listening on 0.0.0.0 port 3493
   0.000577	Connected to UPS [APC]: usbhid-ups-APC
   0.001142	Running as foreground process, not saving a PID file
[01:16:21] INFO: �[32mStarting the UPS monitor and shutdown controller...�[0m
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000019	Could not find PID file to see if previous upsmon instance is already running!

   0.000161	Using power down flag file /etc/killpower
   0.000262	UPS: APC@localhost (primary) (power value 1)
   0.000274	[D1] debug level is '1'
   0.000295	[D1] debug level is '1'
   0.000299	Warning: running as one big root process by request (upsmon -p)
   0.000346	[D1] Saving PID 459 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.001996	[D1] Trying to connect to UPS [APC@localhost]
   0.009844	User upsmonmaster@127.0.0.1 logged into UPS [APC]
   0.002661	[D1] Logged into UPS APC@localhost
[... 189 lines cut ...]
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.sensitivity: high
input.transfer.high: 290
input.transfer.low: 150
input.voltage: 232.0
input.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.load: 16
ups.mfr: American Power Conversion
ups.mfr.date: 2024/01/29
ups.model: Back-UPS BX1200MI
ups.productid: 0002
ups.realpower.nominal: 650
ups.serial: xx
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

jimklimov, 2025-02-22:
I can only guess the driver reports what the UPS controller firmware tells it, and maybe that controller gets stuck. Probably the act of USB connection re-plug (when you switched to PC with powerchute and back, or in the later test) resets the state of that controller or even reboots it.

If that is the case, you might have some luck with software-driven reset of the USB connection without physical re-plugging, see e.g. https://github.com/networkupstools/nut/tree/master/scripts/usb_resetter

Note that this may need extra hardware (an honestly-made USB hub that honours the individual port reset commands) if your built-in ports do not implement them as required by spec, or act as whole-hub reset and mess up your other devices on other ports. Too many vendors cut corners or otherwise can not read the spec, alas...

giacman, 2025-04-08:
After many attempts to fix this, I'm still experiencing the intermittent "Data stale" problem described here. My specific model identifies as USB ID 051d:0002 (American Power Conversion), which seems affected similarly to the BX1600MI-GR mentioned here.

I see v2.8.3-rc2 is out with usbhid-ups fixes. Do you know if this release candidate is expected to address the problem discussed in this issue?

ausfas, 2025-05-08:
Its unfortunate, the there is no HACS or add-on to reset USB port. I end up buying a USB dongle on/off with automation that has resolved the issue for me ( but I look forward to some one creating a add-on for that).

giacman, 2025-05-25:
Hello,

I'm still experiencing the issues described in this thread with my APC UPS (USB ID 051d:0002) on Ubuntu 24.04 (Noble). This includes:

- Intermittent "Data stale" messages.
- Occasional `libusb1: Could not open any HID devices: insufficient permissions on everything` errors logged by `usbhid-ups`.
- Communication outages that can last several minutes.

I'm keen to upgrade to the newly released **NUT v2.8.3** to see if it resolves these problems, as Ubuntu's Noble repositories currently provide v2.8.1. I noted another user's comment about using a hardware USB power cycler as a workaround, which underscores the persistence of these types of issues.

My main question is about the recommended method to install/upgrade to NUT v2.8.3 on Ubuntu 24.04 (Noble):

1. Are there any community-endorsed or official PPAs that provide v2.8.3 for Noble? (I found ppa:mamarley/updates but am unsure of its status for NUT).

1. If not, is compiling from source the best approach? If so, any specific ./configure flags recommended for a standard Ubuntu setup would be helpful.

Thanks for any advice you can offer!

jimklimov, 2025-05-26:
As for the outages themselves, please check in your `dmesg` output. If that is the physical communications getting lost, there's not much NUT can do about flaky hardware. Try a different cable (ideally shielded), perhaps a different USB port of motherboard chipset hub; check if there are EMI sources near the cabling, like luminescent lighting or motors.

ausfas, 2025-05-26:
I have installed a ZigBee dongle to turn on/off the connection when no update below 100% for over 2 hours.

Still wondering for such a simple thing why no one has developed an add-on or HACS to reset USB port in homeassistant.