networkupstools/nut #2806: fopen /run/nut/upsmon.pid: No such file or directory & nut-monitor.service: Supervising process 861 which is not our child. We'll most likely not notice when it exits.

Diktiorwalle, 2025-02-17:
Dear Support,

Running: Linux Ubuntu 22.04 
nut-client: 
Package: nut-client
Version: 2.7.4-14ubuntu2

Running systemclt status nut-client show the following:

● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2025-02-17 14:51:36 CET; 50s ago
    Process: 816 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 861 (upsmon)
      Tasks: 2 (limit: 9086)
     Memory: 4.1M
        CPU: 16ms
     CGroup: /system.slice/nut-monitor.service
             ├─858 /lib/nut/upsmon
             └─861 /lib/nut/upsmon

Feb 17 14:51:35 <hostname> systemd[1]: Starting Network UPS Tools - power device monitor and shutdown controller...
Feb 17 14:51:36 <hostname> upsmon[816]: **fopen /run/nut/upsmon.pid: No such file or directory**
Feb 17 14:51:36 <hostname> upsmon[816]: UPS: <name ups>@<IP address> (slave) (power value 1)
Feb 17 14:51:36 <hostname> upsmon[816]: Using power down flag file /etc/killpower
Feb 17 14:51:36 <hostname> upsmon[858]: Startup successful
Feb 17 14:51:36 <hostname> systemd[1]: **nut-monitor.service: Can't open PID file /run/nut/upsmon.pid (yet?) after start: Operation not permitted**
Feb 17 14:51:36 <hostname> systemd[1]: **nut-monitor.service: Supervising process 861 which is not our child. We'll most likely not notice when it exits.**
Feb 17 14:51:36 <hostname> systemd[1]: Started Network UPS Tools - power device monitor and shutdown controller.

Are those error messages  or just information and the nut-client is working normally?: 
**fopen /run/nut/upsmon.pid: No such file or directory**
nut-monitor.service: Can't open PID file /run/nut/upsmon.pid (yet?) after start: Operation not permitted
**nut-monitor.service: Supervising process 861 which is not our child. We'll most likely not notice when it exits.**

Kind Regards,

Rob

jimklimov, 2025-02-17:
For one, this should be mostly harmless. Several NUT daemons leave PID files so that later invocations for signal-sending (reload, etc.) know where to send them. With systemd, its reloader/restarter can be used instead.

Daemons also drop privileges (in case of `upsmon`, it also tries to save the new PID file after that); apparently `/run/nut` ownership/permission setting in that package (maybe via systemd-tmpfiles) is wrong - so the unprivileged program fails to write there (or maybe to replace the file it made earlier as `root`). No great loss, but any fixes should be asked via distro packaging.

Finally note that NUT v2.7.4 is about 9 years old. It *was* the newest release for too long, until 2022 (probably that's why it is still shipped in Ubuntu 22.04), but we've had several releases since then.

Diktiorwalle, 2025-02-18:
Thanks!

Yesterday I have updated the server to  Ubuntu 24.04:
nut-client: 2.8.1-3.1ubuntu2 

Running systemclt status nut-client show the following:

nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/usr/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-02-18 10:30:10 CET; 3min 9s ago
    **Process: 21868 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=1/FAILURE)**
   Main PID: 21869 (upsmon)
      Tasks: 2 (limit: 9130)
     Memory: 840.0K (peak: 1.1M)
        CPU: 31ms
     CGroup: /system.slice/nut-monitor.service
             ├─21869 /lib/nut/upsmon -F
             └─21871 /lib/nut/upsmon -F

Feb 18 10:30:10 <hostname> systemd[1]: Starting nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller...
**Feb 18 10:30:10 <hostname> systemd-tmpfiles[21868]: Failed to open '/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf': No such file or directory**
Feb 18 10:30:10 <hostname> systemd[1]: Started nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller.
Feb 18 10:30:10 <hostname> nut-monitor[21869]: fopen /run/nut/upsmon.pid: No such file or directory
Feb 18 10:30:10<hostname> nut-monitor[21869]: Could not find PID file to see if previous upsmon instance is already running!
Feb 18 10:30:10 <hostname> nut-monitor[21869]: UPS: <ups name>@<IP address> (secondary) (power value 1)
Feb 18 10:30:10 <hostname> nut-monitor[21869]: Using power down flag file /etc/killpower
Feb 18 10:30:10 <hostname> nut-monitor[21871]: Init SSL without certificate database
Feb 18 10:30:10 <hostname> nut-monitor[21871]: upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it

Following messages are shown: how can I solve those or are they are just informational? 

**Process: 21868 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf 
(code=exited, status=1/FAILURE)**
**Feb 18 10:30:10 <hostname> systemd-tmpfiles[21868]: Failed to open '/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf': No such file or directory**
Could not find PID file to see if previous upsmon instance is already running!

Failed to open '/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf: this file doesn't exists in this location. 

running sudo upsc <name ups>@<IP address>:

Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: not set
battery.mfr.date: 2016/10/14
battery.runtime: 1702
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.6
battery.voltage.nominal: 12.0
device.mfr: APC
device.model: Back-UPS ES 700G
device.serial: 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-2-1-69057-NM-nano-4-C1-240105
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 266
input.transfer.low: 180
input.voltage: 238.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 871.O4 .I
ups.firmware.aux: O4 
ups.load: 17
ups.mfr: APC
ups.mfr.date: 2016/10/14
ups.model: Back-UPS ES 700G
ups.productid: 0002
ups.serial: 
ups.status: OL
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d

Kind Regards,

Rob

jimklimov, 2025-02-18:
> Failed to open '/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf: this file doesn't exists in this location.

This seems significant, although here does not block the client from starting (may keep it unable to save the PID file). Probably the config file is placed into some other package by the distro?

jimklimov, 2025-02-22:
Also, the driver version identifies as `driver.version: DSM7-2-1-69057-NM-nano-4-C1-240105` so I suppose your server is not plain Ubuntu but a Synology DiskStation Manager? They and some other vendors tend to inject "firmware" version over the original software versions, making identification and troubleshooting a bit harder.

jimklimov, 2025-02-24:
Thanks. So we're back to Ubuntu/Debian packaging of `upsmon` client lacking the `/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf` systemd tmpfiles config for the locations it should write PID files etc. into when just `nut-client` package is installed, right?

According to https://serverfault.com/questions/30737/how-do-i-find-the-package-that-contains-a-given-program-on-ubuntu you can use `apt-file` (may need to install it) to search for which package delivers which file name.

Alternately, you can check their package sources for the string: 
````
:; git clone https://salsa.debian.org/debian/nut nut-debian-salsa
:; cd nut-debian-salsa

:; git grep /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf
debian/nut-server.install:debian/tmp/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf
````

So they've messed up a bit, this should have been a file shared by several packages (typically in some `nut-common` package depended on by several others).

CC @bigon @zeha

Diktiorwalle, 2025-02-25:
Thanks. So we're back to Ubuntu/Debian packaging of upsmon client lacking the /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf systemd tmpfiles config for the locations it should write PID files etc. into when just nut-client package is installed, right?

https://packages.ubuntu.com/noble/amd64/nut-client/filelist:

/bin/upsc
/bin/upscmd
/bin/upslog
/bin/upsrw
/bin/upssched-cmd
/etc/init.d/nut-client
/etc/init.d/ups-monitor
/etc/nut/nut.conf
/etc/nut/upsmon.conf
/etc/nut/upssched.conf
/lib/nut/upsmon
/lib/systemd/system/nut-client.service
/lib/systemd/system/nut-monitor.service
/lib/systemd/system/nut.target
/lib/systemd/system/ups-monitor.service
/sbin/upsmon
/sbin/upssched
/usr/lib/tmpfiles.d/nut-client.conf
/usr/share/apport/package-hooks/source_nut-client.py
/usr/share/augeas/lenses/nuthostsconf.aug
/usr/share/augeas/lenses/nutnutconf.aug
/usr/share/augeas/lenses/nutupsconf.aug
/usr/share/augeas/lenses/nutupsdconf.aug
/usr/share/augeas/lenses/nutupsdusers.aug
/usr/share/augeas/lenses/nutupsmonconf.aug
/usr/share/augeas/lenses/nutupsschedconf.aug
/usr/share/augeas/lenses/nutupssetconf.aug
/usr/share/augeas/lenses/tests/test_nut.aug
/usr/share/bash-completion/completions/nut
/usr/share/doc/nut-client/AUTHORS.gz
/usr/share/doc/nut-client/NEWS.adoc.gz
/usr/share/doc/nut-client/README.adoc.gz
/usr/share/doc/nut-client/TODO.adoc.gz
/usr/share/doc/nut-client/changelog.Debian.gz
/usr/share/doc/nut-client/copyright
/usr/share/lintian/overrides/nut-client
/usr/share/man/man5/nut.conf.5.gz
/usr/share/man/man5/upsmon.conf.5.gz
/usr/share/man/man5/upssched.conf.5.gz
/usr/share/man/man8/upsc.8.gz
/usr/share/man/man8/upscmd.8.gz
/usr/share/man/man8/upslog.8.gz
/usr/share/man/man8/upsmon.8.gz
/usr/share/man/man8/upsrw.8.gz
/usr/share/man/man8/upssched.8.gz

Process: 21868 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=1/FAILURE)
 systemd-tmpfiles[21868]: Failed to open '/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf'

Unable to create the file so the next failed to open seems logic. The reason why the file cannot be created is unknown.

Thanks

jimklimov, 2025-02-25:
Hm... The http://ftp.de.debian.org/debian/pool/main/n/nut/nut-client_2.8.1-4_amd64.deb package seems to deliver a `/usr/lib/tmpfiles.d/nut-client.conf` (as your lookup above confirms) with these contents:
````
d /run/nut 0770 root nut - -
````

Compared to the common file, it lacks a subdir needed (by default) for `upssched` that may be called from `upsmon` so specifically a client concern; you can use custom configuration to get around that lack of directory though:
````
# Default PIPEFN and LOCKFN locations per upssched.conf:
d /run/nut/upssched 0770 nut nut - -
````

Also lacks the protection against auto-removal (IIRC after 10 days of age by default):
````
X /run/nut
````

But at least for initial start-up (shortly after boot) this should work. Restart of the service after some weeks of uptime may be trickier.

In their `/lib/systemd/system/nut-monitor.service` I see:
```
...
SyslogIdentifier=%N
ExecStartPre=-/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf
ExecStart=/usr/sbin/upsmon -F
ExecReload=/usr/sbin/upsmon -c reload
...
```

So the `nut-client.conf` is never re-applied with the service restarts (is only handled as part of `systemd-tmpfiles` service or a manual `systemd-tmpfiles --create` run to process all known configuration files.

The lack of `nut-common-tmpfiles.conf` per se should not be fatal to this service however, because `ExecStartPre=-...` with the minus meaning to ignore the result, IIRC.

Overall, I think the messages can be treated as "just informative" at least in short term (until you want to restart something later on, or use `upssched`, and dirs are not there).

> Running `systemctl status nut-client` shows the following:
>
> ````
> nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
> Loaded: loaded (/usr/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
> Active: active (running) since Tue 2025-02-18 10:30:10 CET; 3min 9s ago
> ...
> Main PID: 21869 (upsmon)
> ...
> CGroup: /system.slice/nut-monitor.service
> ├─21869 /lib/nut/upsmon -F
> └─21871 /lib/nut/upsmon -F
> ````

The excerpt above says the `upsmon` daemon did start and fork (to drop privileges for the main part of code loop) and is running for 3 minutes - so did not crash there and then on startup.

Diktiorwalle, 2025-02-27:
Thanks again:

apt-cache show nut-client
Package: nut-client
Architecture: amd64
Version: 2.8.1-3.1ubuntu2
Priority: optional
Section: admin
Source: nut
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Laurent Bigonville <bigon@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 438
Provides: ups-monitor
Pre-Depends: init-system-helpers (>= 1.54~)
Depends: adduser, libc6 (>= 2.38), libupsclient6t64 (>= 2.8.0)
Recommends: bash-completion
Suggests: nut-monitor
Conflicts: ups-monitor
Replaces: ups-monitor
Filename: pool/main/n/nut/nut-client_2.8.1-3.1ubuntu2_amd64.deb
Size: 109158
MD5sum: 8d141ec7a5962229beab6c9bfe4352d8
SHA1: 00f6fc643f0ac415338dbc90e681ed7c795e736e
SHA256: 8f6d168ff84393e38ed9db9e17d606240ce323ac89b903024052146fe393b190
SHA512: 76409b342cc82cfa022b01b822fad2bc4e81e905552c06865d4bf934dd69f68eae512a1fdcab4ede58d7ac39cac954d9eb746a4270b6214d1e578a99d7a52a46
Homepage: https://networkupstools.org/
Description-en: network UPS tools - clients
 Network UPS Tools (NUT) is a client/server monitoring system that
 allows computers to share uninterruptible power supply (UPS) and
 power distribution unit (PDU) hardware. Clients access the hardware
 through the server, and are notified whenever the power status
 changes.
 .
 This package provides NUT's clients that allows the user to control
 and monitor the UPS.
Description-md5: 30f1061db3258a7a5c46e18466afa6c0

To whom should this be reported? 
Bugs: https://bugs.launchpad.net/ubuntu/+filebug ?

jimklimov, 2025-02-27:
I guess so, hopefully then it would bubble through the community to find someone to deliver a fix.

I *think* much of the time Ubuntu packages are a downstream of Debian ones, so preparing and posting a pull request to that (or both) can help you see it fixed in their repos more quickly.

The fix IMHO should focus on:
1) Delivering relevant content (effectively same as `nut-common-tmpfiles.conf` in their case) in `nut-client.conf` which something they invented (but we here might as well consider picking up for future releases, the `nut-common-tmpfiles.conf` file is generated in `configure.ac` based on a number of checks whether different directory variables have same values or not - so in the general case, a `nut-client.conf` may differ)
2) Avoiding `nut-client.conf` in the first place and making sure that `nut-common-tmpfiles.conf` is delivered with all NUT packages that deliver daemons, perhaps by having (introducing if not yet?) a `nut-common` package that everything else requires, to deliver stuff like this configuration file, maybe basic docs like the FAQ, or user/group account definitions.

Diktiorwalle, 2025-02-28:
thanks 
I have reported the issue. 

Kind Regards,

jimklimov, 2025-03-12:
@Diktiorwalle : can you please post your distro issue report for reference?

Debian and Ubuntu trackers seem to have quite a few (~30 together) issues reported for NUT, although some are 15 years old and not marked solved there:
* https://bugs.debian.org/cgi-bin/pkgreport.cgi?pkg=nut
* https://launchpad.net/ubuntu/+source/nut/+bugs

Some more cross-links that seemed relevant:
* #2210
* https://bugs.launchpad.net/ubuntu/+source/nut/+bug/2081755

Diktiorwalle, 2025-03-12:
Hi,

I am not sure what you request me to do? 

I have contacted the following but didn't get any response or reply:

bigon@debian.org
ubuntu-devel-discuss@lists.ubuntu.com

https://help.ubuntu.com/community/ReportingBugs

Kind Regards

closed 2025-03-12 by Diktiorwalle

jimklimov, 2025-03-12:
Yeah, thanks. The bug in a tracker is a bit less prone to getting lost, than an e-mail...

jimklimov, 2025-07-02:
Is the client always installed? Is there some package like `nut-common` (or can it appear) which others depend on, to cover the few shared items like this file, user/group accounts, etc.?