networkupstools/nut #2809: [HCL] EAST / Energine EG-UPSO-6000 / USB-SERIAL CH340

pdtabor, 2025-02-20:
Hello,
We have a UPS in our company: Energine EG-UPSO-6000
I have a problem with it because it is only monitored through a program on Windows - upsSmartView.
It is connected via USB - Serial CH340
Serial port - COM
Baud rate: 9600
Parity: n
Data bit: 8
Stop bit: 1
Timeout: 9999

In the program, the device shows as:
Somm, type: Serial
Dev protocol: EA900_g4_6-20k
Addr:1

I tried different drivers in NUT but failed to connect it. 

Translated with DeepL.com (free version)

https://energenie.com/item.aspx?id=12484
https://en.eastups.com/OnlineTransformerlessUPS/6756.jhtml
https://energenie.com/Repository/CommonFiles/cf_driver---844304e8-5115-4811-815a-0e651367f942.zip

pdtabor, 2025-02-21:
There's no rush. it's been running for so long (barely but it works on that damn windows) , I'll wait and keep checking and testing. I have tried probably all the drivers that are in the list of compatible hardware as well as from the Various tab. I searched the forum for UPS from East and Energine jenak I found information only for simple under-desk devices. And what I have theoretically is a higher server series. 

Translated with DeepL.com (free version)

jimklimov, 2025-02-22:
The name rings a bell, but I've now searched the sources of NUT code and NUT DDL collection, and neither "Energenie" nor "East" pop up there - only a few issues discussed here in GitHub.

In #1836 the blazer or nutdrv_qx drivers worked for the device that their accompanying "UPSmart" identified as "mega-usb" (so likely a variant of Megatec Qx protocol).

Which brings up again the question of how fresh or old your NUT programs are, and also if a custom build per e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests instructions are an option? A build environment like that would also be useful if you end up walking down the road of listening to the serial-port communications of their software to decipher the protocol, and writing a new driver...