networkupstools/nut #2814: [HCL] Tripp Lite ECO850LCD supported by usbhid-ups

dormitionskete, 2025-02-24:
You have listed on the NUT HCL that the Tripp Lite ECO850LCD is supported by usbhid-ups.  

This used to be accurate.

However, a couple of years ago, Tripp Lite redesigned their ECO850LCD UPS's, and the new ones do not appear to be compatible with this driver.  In fact, I tried to connect to one of the new version's using the following drivers, without success:

[usbhid-ups](https://networkupstools.org/docs/man/usbhid-ups.html)
[tripplite](https://networkupstools.org/docs/man/tripplite.html)
[tripplite_usb](https://networkupstools.org/docs/man/tripplite_usb.html)
[tripplitesu](https://networkupstools.org/docs/man/tripplitesu.html)
[nutdrv_qx](https://networkupstools.org/docs/man/nutdrv_qx.html)
[blazer_usb](https://networkupstools.org/docs/man/blazer_usb.html)
[bcmxcp_usb](https://networkupstools.org/docs/man/bcmxcp_usb.html)
[riello_usb](https://networkupstools.org/docs/man/riello_usb.html)
[nutdrv_atcl_usb](https://networkupstools.org/docs/man/nutdrv_atcl_usb.html)
[megatec_usb](no web page given)

I tried it with pretty much every driver on the NUT site that I thought had any chance of working.

All of the drivers above were tested with NUT 2.7.4 on Ubuntu 22.04 LTS.
The top four drivers above were also tested with NUT 2.6.5 on a very old Scientific Linux 6.10.

Please also note that the new version of this UPS also does not appear to have any way to mute it from its hardware menu like the old version does. 

Attached are photos to show the difference between the old and new versions of this UPS. They do not look even remotely alike, even though they have the same model number.

[image]

[image]

jimklimov, 2025-02-24:
Can you try modern NUT, possibly a custom build of master branch (see wiki on in-place builds), to see if various changes in it impacted the visibility of these devices? 2.7.4 is 9 years old...

Otherwise, what do detailed debug logs at driver startup show (also see wiki on debugging)?

I saw some newer Tripplites claim new model and/or protocol IDs (a 4-digit number) that is just not known to, and so rejected by, the NUT driver code. And the solution was to add the number to the list.

dormitionskete, 2025-02-25:
Jim -- I don't expect you to remember me, but this is Father Peter from the OpenIndiana list many, many years ago.  It's nice to see you are still around, and helping idiots like me with our computer problems.  Many thanks again!

It took some doing, but I have been able to confirm that both the old and new versions of the Tripp-Lite ECO850LCD work with NUT 2.8.1 as shipped with Ubuntu 24.04 LTS.

Who would have ever imagined that Ubuntu would ship such an old version of NUT and not update it for so long?  Amazing!!!

closed 2025-02-25 by dormitionskete