networkupstools/nut #2815: Debian Package to >=2.8.2 ?

spacelama, 2025-02-24:
In the spirit of #1461 and #1433 , no one's been able to get feedback from debian bugs to get 2.8.2 into unstable before the freeze in a month (eg https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1092106 ) and clearly the debian maintainer could do with some help.

Anyone able to offer the maintainer some patches to get the current version into debian before the freeze?

I tried backporting it but couldn't even convince it to compile.

jimklimov, 2025-02-24:
I suppose your best bet is contacting people recently active in https://tracker.debian.org/pkg/nut - @bigon, @zeha; there were also commits to https://salsa.debian.org/debian/nut in the past year from @bdrung and @mbiebl (if I am not mistaken about github handle)

mbiebl, 2025-02-24:
Those autoremovals work on a per source package basis, i.e. all binary packages are affected.

jimklimov, 2025-02-24:
One alternative to keep in mind is the long-rolling plan to include more "reference" packaging recipes for different ecosystems in NUT sources (probably under `scripts/` or another dedicated directory so as to not conflict with production recipes of distros), and absorb as many hotfixes and patches applied by numerous packages, so that distributions would have less custom work to do... presumably.

Ideally the distro recipes would then devolve to copying the reference recipe and perhaps providing `configure` script tuning to enable/disable "absorbed" fixes. But I know the real world is not that simple and we won't be able to provide all the nuance fixes as quickly as distros change their policies (e.g. the `time64_t` related package variants and names), so the real recipes would not be too trivial in fact. But at least it would allow us to introduce new sub-package names consistently across the board as new connection media drivers are introduced (modbus, gpio, sysfs-polling, etc. families recently) instead of dozens of occasional maintainers noticing that a new NUT release brought new gifts like that.

spacelama, 2025-02-25:
Given #2792 is merged presumably allowing that RC bug to be closed in debian, and perhaps if the debian maintainer's has a desire to go with a release rather than git master (dunno, seen many packages just take a git master version when it's known that the master branch is a clean tested tree), is a 2.8.3 tag due anytime soon?

Mind you, the existence of a tag isn't sufficient to trigger a maintainer to realise that a build for debian unstable is warranted, given 2.8.2 has been out for a year.

jimklimov, 2025-02-25:
Well, one big non-functional blocker before a 2.8.3 release is to address issue #2708 so as to not publish a long-term snapshot with standard names for `ups.status` values and/or certain NUT variables we decided we do not want in the way they are in code now. So they should be changed to something more sustainable before people who base their work off of official releases would not use something that appears now and disappears in the next release.

But work on that one uncovered some entanglements here and there, so I bite at that problem in small chunks for a few weeks now to get it done with.

bigon, 2025-06-30:
2.8.3 is waiting in the NEW queue and will enter experimental as soon as it has been cleared

jimklimov, 2025-06-30:
Great to hear from you! FYI: Does the package include fixes to published errata - see https://github.com/networkupstools/nut/releases/tag/v2.8.3 ?

cypherstream, 2025-07-03:
Following, because I'd love to see the latest nut as a debian package, including apc-modbus with libusb support built in.