networkupstools/nut #2819: Delta Amplon supported by usbhid-ups

fernandoacf, 2025-02-25:
Hello there,

I'm facing some issues trying to make it work with Delta UPS [1]

```

[myups]
  driver = usbhid-ups
  product = UPS
  port = auto
  bus = 001
  vendorid = 1a86
  productid = 5523 

```

`Bus 001 Device 021: ID 1a86:5523 QinHeng Electronics CH341 in serial mode, usb to serial port converter`

```
[root@localhost]# upsdrvctl -DDDDDD start
Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.2190-2190-g53a539bab (development iteration after 2.8.2)
   0.000000     [D2] 
If you're not a NUT core developer, chances are that you're told to enable debugging
to see why a driver isn't working for you. We're sorry for the confusion, but this is
the 'upsdrvctl' wrapper, not the driver you're interested in.

Below you'll find one or more lines starting with 'exec:' followed by an absolute
path to the driver binary and some command line option. This is what the driver
starts and you need to copy and paste that line and append the debug flags to that
line (less the 'exec:' prefix).

Alternately, provide an additional '-d' (lower-case) parameter to 'upsdrvctl' to
pass its current debug level to the launched driver, and '-B' keeps it backgrounded.

   0.000255     [D1] upsdrvctl commanding all drivers (1 found): start
   0.000261     [D1] Starting UPS: myups
   0.000276     [D2] 3 remaining attempts
   0.000282     [D2] exec:  /usr/lib/nut/usbhid-ups -a myups
Network UPS Tools 2.8.2.2190-2190-g53a539bab (development iteration after 2.8.2) - Generic HID driver 0.60
USB communication driver (libusb 1.0) 0.50
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
upsnotify: notify about state NOTIFY_STATE_STOPPING with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
   0.010552     Driver failed to start (exit status=1)
```

I also tried with different drivers (genericups, belkin, blazer_usb). No luck

Am I missing something? Is there any other information that I can provide?

[1]https://www.deltapowersolutions.com/en/mcis/1kva-3kva-single-phase-ups-rt-gen3-series.php

jimklimov, 2025-02-25:
Are these tests on different systems?

In first I see:
```
   0.002000     [D1] Succeeded to become_user(nut): now UID=950 GID=950
```

But in second the UID/GID differ:
```
   0.000860     [D1] Succeeded to become_user(nut): now UID=961 GID=961
```

Either way, it seems that the `nut` user/group lacks permission to devfs nodes for `1a86:5523` and `04e2:1410` respectively. Usually this is set up by udev rules (in Linux, or equivalent on other platforms), so you may need to `sudo systemctl restart udev` or `sudo udevadm control --reload-rules ; sudo udevadm trigger` if you installed NUT from scratch, and the `udev` rules files were generated and delivered by NUT `make install` but not made known to `udev`.

A quick-test alternative is to add `user=root` to the driver settings, or run as `/usr/lib/nut/usbhid-ups -a myups -u root -DDDDDD` - either way to avoid dropping privileges to `nut` and so being able to access any devices to see if they are supported on protocol level. Or to generally wreak havoc on the system in case of bugs, so running as `root` is not recommended for production.

For more details see around https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-systemd-deployment

These two USB IDs are also implicated in `huawei-ups2000` driver (one was also seen for "Vinsench" devices and `nutdrv_qx` though). See its man page about gory details (for the Exar chip you may be *required* to use Linux 5.12 or newer with `xr_serial` kernel module attached). CH341 should be easier to use, long on the market.

fernandoacf, 2025-02-25:
hey @jimklimov, thank you so much for your input. I recompiled to add --with-modbus and triggered udev to reload. Worked liked a charm with nutdrv_qx on both systems:

xr_serial:
```
# upsc delta-network
battery.charge: 100
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 2340
battery.voltage: 026.9
device.model: WPLVR1K0
device.type: ups
driver.debug: 2
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2.2211-2211-g325e0a1cd
driver.version.data: Voltronic 0.09
driver.version.internal: 0.40
driver.version.usb: libusb-1.0.27 (API: 0x0100010A)
input.frequency: 60.0
input.frequency.high: 63.0
input.frequency.low: 57.0
input.phases: 1
input.transfer.high: 133
input.transfer.high.max: 139
input.transfer.high.min: 130
input.transfer.low: 121
input.transfer.low.max: 124
input.transfer.low.min: 115
input.voltage: 133.0
input.voltage.nominal: 127.0
output.current: 1.8
output.frequency: 60.0
output.phases: 1
output.power.maximum.percent: 22
output.power.minimum.percent: 15
output.powerfactor: 0.9
output.voltage: 126.1
output.voltage.nominal: 127.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 01991.03
ups.firmware.aux: P01
ups.load: 22
ups.power.nominal: 1000
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL
ups.temperature: 35.3
ups.type: online
```

ch341:
```
# upsc delta-office
battery.charge: 100
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 1980
battery.voltage: 027.1
device.model: WPLVR1K0
device.type: ups
driver.debug: 2
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2.2211-2211-g325e0a1cd
driver.version.data: Voltronic 0.09
driver.version.internal: 0.40
driver.version.usb: libusb-1.0.27 (API: 0x0100010A)
input.frequency: 60.0
input.frequency.high: 63.0
input.frequency.low: 57.0
input.phases: 1
input.transfer.high: 133
input.transfer.high.max: 139
input.transfer.high.min: 130
input.transfer.low: 115
input.transfer.low.max: 124
input.transfer.low.min: 115
input.voltage: 130.2
input.voltage.nominal: 127.0
output.current: 2.2
output.frequency: 60.0
output.phases: 1
output.power.maximum.percent: 28
output.power.minimum.percent: 17
output.powerfactor: 0.9
output.voltage: 126.5
output.voltage.nominal: 127.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 01991.02
ups.firmware.aux: P01
ups.load: 27
ups.power.nominal: 1000
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL
ups.temperature: 31.1
ups.type: online
```

closed 2025-02-25 by fernandoacf

fernandoacf, 2025-02-27:
Hey @jimklimov, I'm reaching out again just to ask if you know why I can't use `port = auto` with this device? Even with udev rules in place and correct vendorid/productid they never match on the driver. I also tried setting subdriver but no subdriver worked, the one that appears to work but without results is snr. I'm asking because I hit this bug https://github.com/torvalds/linux/commit/40827729cff16269b5e99fc8e12bf53d848407de where the ttyUSB0 disconnects and connect again but this time as ttyUSB1. That breaks the nut-server and nut-monitor

jimklimov, 2025-02-27:
For USB-capable devices, `port=auto` means detection via libusb (using vendorid and half a dozen other possible match criteria), and in case of Megatec dialects, talking their shim of serial protocol over USB packets. The `port=/dev/ttyUSB...` is about UPSes that actually have a USB-to-serial converter chip in the controller, so they essentially have a serial port that connects with USB cable. On the wire this is a different protocol for the driver to talk.

As for renumbering of the resulting serial port, I think this can be handled by udev rules that would assign a name for the device based on its ID, but this is out of scope for NUT.