networkupstools/nut #2820: Extend NUT server-client communications to optionally use a Unix socket (or WIN32 named pipe)

jimklimov, 2025-02-25:
An idea I did not find in the issue tracker but I think saw floating around some documentation during recent reformatting effort: for security and certain simplification, it may make sense for `upsd` data server and clients (`upsmon`, `upsc` etc.) both running on a local system to not require a functional IP stack and support communications with a Unix socket file as well.

This could be useful for containers without networking (or networked into different LAN segments), embedded systems (especially if they only need the IP stack for NUT programs to chatter), and any number of other conjured-up reasons.

There is quite a bit of industry-wide precedent, e.g. with popular databases. After all, decades of sockets over FS/IP/... abstractions are there for a reason.

Initial-access security (similar to IP firewall) would be managed by POSIX permission (or ACL) for the file system based sockets, or eventually SELinux or equivalent constraints on newer OSes. Not sure about Windows named pipes.

Syntax-wise, we could support it as a re-use of the host specification starting with a slash or backslash depending on platform, and probably ignoring any `:port` part:
```
# POSIX
:; upsc myups@/run/nut/upsd.sock

# WIN32
:; upsc -l \\.\pipe\upsd.sock
```

Implementation seems like a low-hanging fruit, building upon all we've done before...

CC @arnaudquette-eaton @clepple : WDYT ?

capr, 2025-09-10:
so currently, I can't use an UPS and have jailed apps at the same on the machine time that can't connect to it and mess with it ??

capr, 2025-09-10:
AFAIU nut server ignores user/password on localhost, and it doesn't work with unix sockets either, so any non-root app can connect to it. I have jailed app deployments on the machine that I don't want to mess with my ups but I also have a root app that monitors the ups. Currently I need to add a firewall rule to block connections from non-root processes.