networkupstools/nut #2826: UPSILON2000 nutdrv/blazer wont work

DiSHTiX, 2025-02-27:
Hi guys
I got this new UPS and thought it would work since i saw Upsilon2k configuration; I have tried a ton of possible cfg combos but whatever i chuck at it... it wont respond. 
Perhaps someone with more knowledge can help out please.
Thank you

- Brand: PDSD http://www.pudsd.com
- Upsilon config uses Megatec driver: http://www.pudsd.com/uploadfiles/files/20231205150512_0064.jpg)
- Windows driver: http://www.pudsd.com/uploadfiles/files/20231206111612_5860.rar)

blazer and lsusb:
```
./blazer_usb -s ups2 -x port="auto" -x bus="007" -x vendorid="075d" -x productid="0300" -x serial="HHtFi.*" -x langid_fix="0x409" -DDDDD -u root

   0.026809     [D3] match_function_regex: matching a device...
   0.026815     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.026820     [D2] Device does not match - skipping
   0.026908     [D2] Checking device 2 of 12 (075D/0300)
   0.044186     [D2] - VendorID: 075d
   0.044203     [D2] - ProductID: 0300
   0.044206     [D2] - Manufacturer: -UPS-
   0.044209     [D2] - Product: Smart-UPS
   0.044212     [D2] - Serial Number: HHtFiJmIpMtLtAj4041:40
   0.044215     [D2] - Bus: 007
   0.044218     [D2] - Device: unknown
   0.044220     [D2] - Device release number: 0001
   0.044223     [D2] Trying to match device
...

Bus 007 Device 003: ID 075d:0300 -UPS- Smart-UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x075d
  idProduct          0x0300
  bcdDevice            0.01
  iManufacturer           1 -UPS-
  iProduct                2 Smart-UPS
  iSerial                 3 HHtFiJmIpMtLtAj4041:40
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               44mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     250
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              33
Device Status:     0x0901
  Self Powered
```

jimklimov, 2025-02-27:
Which NUT version are you running? With `Device: unknown` it seems that this is a slightly older NUT v2.8.0 or 2.8.1?

Can you try a newer one?

Also, in any case try `nutdrv_qx` driver as well, it should support more Megatec dialects. Several were added on master branch recently, so if you can build your own per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests - it can help better than older packaged code.

Your excerpt from the log ended at the interesting part - where it matched the USB metadata and hopefully began probing the device (and maybe decided it can't talk to it). I can't guess what it said there, and that might be important for troubleshooting.

Just in case, with NUT v2.8.x your driver might have got automatically wrapped into a systemd unit which can conflict with the driver you run manually - see https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) for more details.

DiSHTiX, 2025-02-27:
Jim, thanks for replying so quick. 

Proxmox uses nut/stable 2.8.0-7, and its working with the APC ups thats also attached.

Yes I did also try nutdrv_qx, as i saw the references when browsing the source code here but it didnt do anything different.  Also tried subvendors, novendor, noscanlangid etc but nada.

There's no further excerpt, it moved on to the next usb port.. oh i see I cp'd wrong including data of the previous port. My bad..
The command I posted shows blazer_usb however i think the logoutput came from nutdrv_qx because its searching to match "VendorID: 1d6b" instead of the specified "075d" ; not sure why it did that.

TBH i didnt expect a quick response, i moved the ups to my desktop. 
I'll try build it on my laptop with arch tomorrow

ups.conf currently only has the apc, the below looks ok i think
```shell
ls -ld `find /etc/systemd | grep -w nut | sort`
lrwxrwxrwx 1 root root   30 Oct 13 22:03 /etc/systemd/system/multi-user.target.wants/nut.target -> /lib/systemd/system/nut.target
drwxr-xr-x 2 root root 4096 Feb 27 14:48 /etc/systemd/system/nut-driver@apc.service.d
-rw-r--r-- 1 root root   74 Feb 27 14:48 /etc/systemd/system/nut-driver@apc.service.d/nut-driver-enumerator-generated-checksum.conf
-rw-r--r-- 1 root root  361 Feb 27 14:48 /etc/systemd/system/nut-driver@apc.service.d/nut-driver-enumerator-generated.conf
drwxr-xr-x 2 root root 4096 Oct 13 22:03 /etc/systemd/system/nut-driver@.service.d
-rw-r--r-- 1 root root   81 Jan 15 04:29 /etc/systemd/system/nut-driver@.service.d/nut-driver-enumerator-generated-checksum.conf
drwxr-xr-x 2 root root 4096 Feb 27 16:15 /etc/systemd/system/nut-driver.target.wants
lrwxrwxrwx 1 root root   39 Feb 27 14:48 /etc/systemd/system/nut-driver.target.wants/nut-driver@apc.service -> /lib/systemd/system/nut-driver@.service
drwxr-xr-x 2 root root 4096 Oct 13 22:03 /etc/systemd/system/nut.target.wants
lrwxrwxrwx 1 root root   46 Oct 13 22:03 /etc/systemd/system/nut.target.wants/nut-driver-enumerator.path -> /lib/systemd/system/nut-driver-enumerator.path
lrwxrwxrwx 1 root root   49 Oct 13 22:03 /etc/systemd/system/nut.target.wants/nut-driver-enumerator.service -> /lib/systemd/system/nut-driver-enumerator.service
lrwxrwxrwx 1 root root   37 Oct 13 22:03 /etc/systemd/system/nut.target.wants/nut-driver.target -> /lib/systemd/system/nut-driver.target
lrwxrwxrwx 1 root root   39 Oct 13 22:03 /etc/systemd/system/nut.target.wants/nut-monitor.service -> /lib/systemd/system/nut-monitor.service
lrwxrwxrwx 1 root root   38 Oct 13 22:03 /etc/systemd/system/nut.target.wants/nut-server.service -> /lib/systemd/system/nut-server.service

```

jimklimov, 2025-02-27:
> "VendorID: 1d6b" instead of the specified "075d"

It walks all devices found by libusb, in your case there are apparently 12 to try. The first one had wrong IDs so did not match and was quickly skipped.

jimklimov, 2025-02-27:
FWIW, Vendor ID `075d` is only seen in `drivers/idowell-hid.c` currently in NUT sources, and the device you have has a small but non-trivial report HID descriptor of 250 bytes, so it may be worth a shot to try `usbhid-ups` also.

DiSHTiX, 2025-02-27:
> > "VendorID: 1d6b" instead of the specified "075d"
> It walks all devices found by libusb, in your case there are apparently 12 to try. The first one had wrong IDs so did not match and was quickly skipped.
yes, there are quite a few ports on this motherboard.

> FWIW, Vendor ID `075d` is only seen in `drivers/idowell-hid.c` currently in NUT sources, and the device you have has a small but non-trivial report HID descriptor of 250 bytes, so it may be worth a shot to try `usbhid-ups` also.
hah that would be a fun joke; yes i didnt even try the default usbhid driver.

thank you

jimklimov, 2025-02-28:
Well, not much (but that HID descriptor was small after all) - however it seems to report the relevant UPS status and battery charge, so should suffice for shutdowns...

Strange about the master-branch build. On the one hand, the error looks similar what I saw in a brewing PR with some recipes I am optimizing now; on another - I don't think any of that was committed into the master branch yet. Or I was now finding repercussions of some older mistakes?..

Namely, not sure where the `build` part of `build/nut/docs/man` comes from. Is that from a simple `make`, or from a `make distcheck` (that one is not really needed when building for consumption, more of a developer thing)?

DiSHTiX, 2025-03-01:
> Well, not much (but that HID descriptor was small after all) - however it seems to report the relevant UPS status and battery charge, so should suffice for shutdowns...

Yes thats all i'm after, thanks again

> Strange about the master-branch build. On the one hand, the error looks similar what I saw in a brewing PR with some recipes I am optimizing now; on another - I don't think any of that was committed [...]

I was gonna post what i did then realized it was somewhat user error.
Your guide under Pre-requisites links - https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt
As i was checking that i automatically ended up following the build there which uses the following and fails on **make all**

```
:; mkdir -p nut && cd nut && \
    git clone https://github.com/networkupstools/nut/ -b master .
:; ./autogen.sh && \
    ./configure --with-doc=all --with-all --with-cgi && \
    make all && make check && make spellcheck
```

https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#testing-with-ci-helper
I tested master again, and if i use **./ci_build.sh inplace** on the master branch it works 

Sorry for the worries

DiSHTiX, 2025-03-02:
thanks jim

closed 2025-03-02 by DiSHTiX

jimklimov, 2025-03-02:
Good thoughts come late, can you please also check what `nut-scanner -U` would say about this device? Does it know the ID in your NUT version and would it offer `usbhid-ups` right away?