networkupstools/nut #2827: Trying to configure Nut with native macOS Sequoia UPS integration running

EagleDTW, 2025-02-28:
Hi,

Trying to configure Nut with native macOS Sequoia UPS integration running, when I run upsc it comes back with connection refused.

TIA - Daniel

EagleDTW, 2025-02-28:
Yes I have the macosx-ups driver running for the main ups integration into macOS I believe. I installed using homebrew and edited the corresponding files and started nut, which looked to successfully start. I'll post my files a bit later after work but followed a pretty straightforward install guide which seemed very comprehensive.

EagleDTW, 2025-03-01:
Ahhh...I am not using the macosx-ups driver in nut, I'm using the USBHID driver default for an APC UPS, is it as simple as changing the config for NLUT to use this driver in place of the APC one?

EagleDTW, 2025-03-01:
following this guide, loosely: https://www.youtube.com/watch?v=vyBP7wpN72c

The ups.conf file contains now with your suggestion:
pollinterval = 1
maxretry = 3

[apc-network]
    driver = **macosx-ups**
    port = auto
    desc = "APC Back-UPS XS 1500M"
    vendorid = 051d
    productid = 0002
    serial = redacted

I get the error code in the terminal of % upsc apc-network@localhost      
Error: Connection failure: Connection refused

jimklimov, 2025-03-01:
@EagleDTW : just to recap, I think Tim's guide also mentions that: NUT has a 3-layer architecture with NUT driver program(s) talking to devices with a vendor's protocol over some media on one side and to a NUT data server (`upsd`) on the same machine, and that data server talks common NUT protocol to clients like `upsc`, `upsmon`, various monitoring integrations, etc. over TCP/IP (on localhost or remote).

You can check NUT-related running processes with something like:
````
:; ps -ef | grep -v grep | grep -E 'nut|ups'
````

Note that the MacOS `ps` may be not showing symbolic user names (just their ID numbers in first column), so if your NUT daemons run as they should, with a dedicated unprivileged account like `nut`, you may have to add it into the `grep` above, e.g.
````
:; id nut
uid=501(nut) gid=20(staff) groups=20(staff),501(nut),...

:; ps -ef | grep -v grep | grep -E 'nut|ups|501'
````

The `upsc` "Connection failure: Connection refused" means it does not see an `upsd` to talk to, regardless of whether a driver runs or not. This may be because of its service not starting, some mis-configuration, firewalls, etc. So the first thing to check here is if the `ps` output above shows `upsd` as running. If it is not, investigate why (is there a launchd service for it? is it enabled? are there any logs indicating if it tried to start or why it failed?)

As for the NUT driver itself, you can start it in command line, with raised debug verbosity, to check how it sees the device:
````
:; sudo usbhid-ups -s test -x port-auto -u root -x vendorid=051d -x productid=0002 -DDDDDD -d 1
````

NOTES:
* Make sure no other drivers are actually running for the device, to avoid conflicts between them (and the service management framework like launchd trying to revive them and breaking your experiments).
* In this example, `sudo` and `-u root` parts should let the driver start as `root` and not drop privileges, so you don't have to worry about hardware access permissions. Ideally the permissions should be set up for the `nut` run-time user and running as `root` is not required, but it is easier to quickly test for basic existence of support like this.
* `-d 1`  tells it to collect one loop of data updates, print them, and exit the driver.
* `-DDDDDD` is debug verbosity (6 here)
* driver program location depends on how/where you got NUT installed

NOTE: With your configuration as posted above, I think the service would also fail to start because `vendorid` etc. parameters are not known to `macosx-ups` program. This driver literally watches (read-only) what the OS reports about power devices that *its* built-in UPS/Battery support says about the device, primarily relevant for USB HID devices.

clepple, 2025-03-01:
Yes, as Jim mentioned, there are slightly different options for different drivers, and https://networkupstools.org/docs/man/macosx-ups.html shows that you can only specify `model` for matching devices with that driver (macOS doesn't list provide the vendor or any of the numeric IDs in the API that `macosx-ups` uses). You should be able to just use the following configuration in `ups.conf`:

```
[apc-network]
    driver = macosx-ups
    port = auto
    desc = "APC Back-UPS XS 1500M"
```

(You're using them correctly, but the reference for `port` and `desc` is https://networkupstools.org/docs/man/ups.conf.html )

Due to difficulties in talking directly to USB devices, I can't personally recommend any of the USB drivers on macOS.

clepple, 2025-03-01:
Also, full disclosure, UPS hardware usually fails before the power fails here, so I have not done much testing with `macosx-ups` recently. "Should work" but at any rate, starting that driver manually with the `-D` debug flags (no `-u root` needed) should indicate whether something broke since the last time I tested:

```
/opt/homebrew/Cellar/nut/*/bin/macosx-ups -a apc-network -d 1 -D
```

Like USB drivers on macOS, you might need at least one `-D` flag (or `-F`) to prevent the driver from crashing. (This prevents the driver from forking, and many APIs on macOS are broken with respect to forking.) I have not looked to see how Homebrew suggests you start the drivers - I used to use launchd plists when I was developing the drivers in Fink on Macs. (The launchd plists work better if the driver stays in the foreground, too.) Here's some more discussion on that: https://github.com/networkupstools/nut/issues/2642#issuecomment-2380721920

EagleDTW, 2025-03-03:
Grep returned this:
    0   388     1   0 23Feb25 ??         3:12.65 /usr/libexec/ioupsd
    0 64173     1   0  4:46PM ??         0:00.08 /usr/sbin/cupsd -l
  502 65904 65247   0  4:58PM ttys000    0:00.01 grep ups

So looks like there is a cuspd isntance and the built in ups integration working

Did your command above and got this output:
Network UPS Tools - Mac OS X UPS meta-driver 1.41 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000001	[D1] testval_reloadable: setting 'pollinterval' exists and differs: new value '1' vs. '2'
   0.000045	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 15.0.0 (clang-1500.3.9.4); Target: arm64-apple-darwin23.4.0; Thread model: posix and configured with flags: --disable-dependency-tracking --prefix=/opt/homebrew/Cellar/nut/2.8.2 --localstatedir=/opt/homebrew/var --sysconfdir=/opt/homebrew/etc/nut --with-statepath=/opt/homebrew/var/state/ups --with-pidpath=/opt/homebrew/var/run --with-systemdtmpfilesdir=/opt/homebrew/Cellar/nut/2.8.2/share/nut --with-openssl --with-serial --with-usb --without-avahi --without-cgi --without-dev --without-doc --without-ipmi --without-libltdl --without-neon --without-nss --without-nut_monitor --without-powerman --without-pynut --without-snmp --without-wrap --with-macosx_ups
   0.000053	[D1] debug level is '1'
   0.000437	[D1] Can not become_user(nobody): not root initially, remaining UID=502 GID=20
   0.001043	[D1] Number of power supplies found: 1
   0.001062	[D1]         model name: Back-UPS XS 1500M FW:947.d7 .D USB FW:d7     
   0.001081	[D1] upsdrv_initinfo()
   0.001332	[D1] upsdrv_updateinfo()
   0.001533	Running as foreground process, not saving a PID file
   0.001546	[D1] Driver initialization completed, beginning data dump (1 loops)
   0.001549	[D1] upsdrv_updateinfo()
   0.001736	[D1] upsdrv_updateinfo()
battery.charge: 100
battery.runtime: 3660
battery.voltage: 120.000
device.mfr: (unknown)
device.model: Back-UPS XS 1500M FW:947.d7 .D USB FW:d7     
device.type: ups
driver.debug: 1
driver.flag.allow_killpower: 0
driver.name: macosx-ups
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: dumping
driver.version: 2.8.2
driver.version.internal: 1.41
ups.mfr: (unknown)
ups.model: Back-UPS XS 1500M FW:947.d7 .D USB FW:d7     
ups.status: OL CHRG
   0.001928	[D1] Cleanup: release references

EagleDTW, 2025-03-03:
Used the straight command on the old driver and got this long printout:
:; sudo usbhid-ups -s test -x port=3493 -u root -x vendorid=051d -x productid=0002 -DDDDDD -d 1
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
   0.000786	[D1] upsdrv_makevartable...
   0.001095	[D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x100010a)"
   0.001102	[D1] Using USB implementation: libusb-1.0.27 (API: 0x100010a)
   0.001136	[D3] main_arg: var='port' val='3493'
   0.001139	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=3493, reloadable=0, reload_flag=0
   0.001141	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.001143	[D5] send_to_all: SETINFO driver.parameter.port "3493"
   0.001144	[D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.001146	[D3] main_arg: var='vendorid' val='051d'
   0.001148	[D5] send_to_all: SETINFO driver.parameter.vendorid "051d"
   0.001150	[D3] main_arg: var='productid' val='0002'
   0.001153	[D5] send_to_all: SETINFO driver.parameter.productid "0002"
   0.001154	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 15.0.0 (clang-1500.3.9.4); Target: arm64-apple-darwin23.4.0; Thread model: posix and configured with flags: --disable-dependency-tracking --prefix=/opt/homebrew/Cellar/nut/2.8.2 --localstatedir=/opt/homebrew/var --sysconfdir=/opt/homebrew/etc/nut --with-statepath=/opt/homebrew/var/state/ups --with-pidpath=/opt/homebrew/var/run --with-systemdtmpfilesdir=/opt/homebrew/Cellar/nut/2.8.2/share/nut --with-openssl --with-serial --with-usb --without-avahi --without-cgi --without-dev --without-doc --without-ipmi --without-libltdl --without-neon --without-nss --without-nut_monitor --without-powerman --without-pynut --without-snmp --without-wrap --with-macosx_ups
   0.001157	[D1] debug level is '6'
   0.001158	[D5] send_to_all: SETINFO driver.debug "6"
   0.001160	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.002062	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002066	[D5] send_to_all: SETINFO device.type "ups"
   0.002067	[D5] send_to_all: SETINFO driver.state "init.device"
   0.002069	[D1] upsdrv_initups (non-SHUT)...
   0.002070	[D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.47')
   0.002072	WARNING: warn_if_bad_usb_port_filename(): port argument specified to
  the driver is "3493" but USB drivers do not use it and rely on
  libusb walking all devices and matching their identification metadata.
  NUT documentation recommends port="auto" for USB devices to avoid confusion.
   0.011154	[D2] Checking device 1 of 20 (0BDA/8153)
   0.011570	[D2] - VendorID: 0bda
   0.011573	[D2] - ProductID: 8153
   0.011574	[D2] - Manufacturer: Realtek
   0.011575	[D2] - Product: USB 10/100/1000 LAN
   0.011577	[D2] - Serial Number: 000001
   0.011578	[D2] - Bus: 002
   0.011579	[D2] - Bus Port: 003
   0.011580	[D2] - Device: 013
   0.011581	[D2] - Device release number: 3000
   0.011582	[D2] Trying to match device
   0.011584	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.011717	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.011719	[D2] Device does not match - skipping
   0.011733	[D2] Checking device 2 of 20 (051D/0002)
   0.017014	[D2] - VendorID: 051d
   0.017017	[D2] - ProductID: 0002
   0.017019	[D2] - Manufacturer: American Power Conversion
   0.017020	[D2] - Product: Back-UPS XS 1500M FW:947.d7 .D USB FW:d7     
   0.017021	[D2] - Serial Number: 3B1849X23910  
   0.017022	[D2] - Bus: 002
   0.017023	[D2] - Bus Port: 003
   0.017024	[D2] - Device: 012
   0.017026	[D2] - Device release number: 0090
   0.017027	[D2] Trying to match device
   0.017028	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.017032	[D3] match_function_regex: matching a device...
   0.017037	[D2] Device matches
   0.017039	[D2] Reading configuration descriptor 1 of 1
   0.017121	[D3] libusb_kernel_driver_active() returned 1 (driver active)
   0.017124	[D2] successfully set kernel driver auto-detach flag
   0.022281	[D2] Claimed interface 0 successfully
   0.022299	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.022985	[D2] Retrieved HID descriptor (expected 9, got 9)
   0.022992	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 21 01 22 19 04
   0.022994	[D3] HID descriptor length (method 1) 1049
   0.022995	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.022998	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 21 01 22 19 04
   0.022999	[D3] HID descriptor length (method 2) 1049
   0.023000	[D2] HID descriptor length 1049
[... 2270 lines cut ...]
input.sensitivity: medium
input.transfer.high: 139
input.transfer.low: 88
input.transfer.reason: input voltage out of range
input.voltage: 120.0
input.voltage.nominal: 120
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: 947.d7 .D
ups.firmware.aux: d7     
ups.load: 13
ups.mfr: American Power Conversion
ups.mfr.date: 2018/12/06
ups.model: Back-UPS XS 1500M
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: 3B1849X23910  
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
   0.796884	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   0.796886	[D1] upsdrv_cleanup...
   0.927667	[D5] send_to_all: SETINFO driver.state "cleanup.exit"

EagleDTW, 2025-03-04:
That makes sense...I'm using my normal user for my account with auto login methods - I wonder how it's pulling that user = nobody...that's probably at the heart of the issue (as I have seen files/programs for upsd and upsdrvctl) I will try setting the User set to root in the config file, can I also just use this setting to set the user as my login user as well to gain permissions to the running process?

So like user = daniel... in the ups.conf file will work potentially, right then permissions may be set correctly?

jimklimov, 2025-03-04:
The `user=nobody` is built-in when relevant `configure` options do not specify another (same for group). Usually packages deliver an unique service account like `nut` and run service programs as it, assign FS permissions for it, etc.

Maybe consistently running as a `nobody` is also an option, though.

Otherwise, NUT daemons have ways of overriding the built-in default user/group via CLI or config files.

EagleDTW, 2025-03-04:
Could this be a permissions error like this thread? Also what would be my strategy to enable the default nut account to run this:
https://www.truenas.com/community/threads/cannot-monitor-ups-locally-or-remotely-upsc-gives-error-connection-failure-connection-refused.100999/

EagleDTW, 2025-03-04:
Trying to check if the service is running and getting that the nut service is turned off, which now makes sense that the connection is refused

EagleDTW, 2025-03-10:
Hi @jimklimov and @clepple, What would be the best strategy to configure the nut service to run and confirm that the service is indeed running before trying to upsc to it...I have my configs in the upper threads, would you mind perusing them and seeing if anything needs to be changed to get the service up and running - I would like to try for the usbhid-ups driver as I do imagine controlling the UPS features downstream (rather than the macosX driver that just monitors). If I'm not even getting the service to launch and show an up status, what are my most likely troubleshooting steps to resolve the service issue - is it driver config or multiple instances of the drivers using the same hardware? TIA!

jimklimov, 2025-03-11:
Sorry, can't help much beyond those notes and referring to google. My experience with the platform is limited by a few hours of making a build agent from a VM I was given access to. :\

EagleDTW, 2025-03-11:
> Sorry, can't help much beyond those notes and referring to google. My experience with the platform is limited by a few hours of making a build agent from a VM I was given access to. :\

Thanks Jim! much appreciated for the help...do you know if I can say runas: then my username instead of root or does that put it into a weird state where permissions will cause it to fail instantly (kind of like what I'm currently seeing)?

EagleDTW, 2025-03-13:
Running "brew services info nut" after configuring and restarting the service displays this...how can I fully check what is going on for why it is not running?:

[image]

gkd720, 2025-03-29:
Yeah, me too. I was getting the "Can't claim USB device [051d:0002]@0/0/0: Access denied (insufficient permissions)", and "upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it" from a "upsdrvctl start". Also got both errors from a "usbhid-ups -a myups". And a "ps -ef|grep ups" shows the Mac's driver: "/usr/libexec/ioupsd". You had mentioned that it quickly restarts when killed, so any other ideas on what to do? Thanks.

gonzonia, 2025-07-02:
I'm going through all this now. If I start with "brew services start nut" I immediately get 

Communications with UPS rackups@localhost lost

 I can verify that it's running using 

:; ps -ef | grep -v grep | grep -E 'nut|ups'

This returns: 
    0 55231     1   0  1:17PM ??         0:00.13 /usr/libexec/ioupsd
  501 60356     1   0  1:31PM ??         0:00.01 /usr/local/opt/nut/sbin/upsmon -D
  501 60359 60356   0  1:31PM ??         0:00.01 /usr/local/opt/nut/sbin/upsmon -D

I can also use "brew services info nut"

nut (homebrew.mxcl.nut)
Running: ✔
Loaded: ✔
Schedulable: ✘
User: me
PID: 60356

Witht the service stopped I can run the driver directly with 
"/usr/local/Cellar/nut/2.8.2/bin/macosx-ups -a rackups -d 1 -D" 

Network UPS Tools - Mac OS X UPS meta-driver 1.41 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 15.0.0 (clang-1500.3.9.4); Target: x86_64-apple-darwin23.3.0; Thread model: posix and configured with flags: --disable-dependency-tracking --prefix=/usr/local/Cellar/nut/2.8.2 --localstatedir=/usr/local/var --sysconfdir=/usr/local/etc/nut --with-statepath=/usr/local/var/state/ups --with-pidpath=/usr/local/var/run --with-systemdtmpfilesdir=/usr/local/Cellar/nut/2.8.2/share/nut --with-openssl --with-serial --with-usb --without-avahi --without-cgi --without-dev --without-doc --without-ipmi --without-libltdl --without-neon --without-nss --without-nut_monitor --without-powerman --without-pynut --without-snmp --without-wrap --with-macosx_ups
   0.000010	[D1] debug level is '1'
   0.000157	[D1] Can not become_user(nobody): not root initially, remaining UID=501 GID=20
   0.000454	[D1] Number of power supplies found: 1
   0.000462	[D1]         model name: OL1500RTXL2UN
   0.000470	[D1] upsdrv_initinfo()
   0.000616	[D1] upsdrv_updateinfo()
   0.000766	Running as foreground process, not saving a PID file
   0.000775	[D1] Driver initialization completed, beginning data dump (1 loops)
   0.000778	[D1] upsdrv_updateinfo()
   0.000885	[D1] upsdrv_updateinfo()
battery.charge: 100
battery.runtime: 1080
battery.voltage: 118.000
device.mfr: (unknown)
device.model: OL1500RTXL2UN
device.type: ups
driver.debug: 1
driver.flag.allow_killpower: 0
driver.name: macosx-ups
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: dumping
driver.version: 2.8.2
driver.version.internal: 1.41
ups.mfr: (unknown)
ups.model: OL1500RTXL2UN
ups.status: OL CHRG
   0.001017	[D1] Cleanup: release references

So I know the driver works. 

ups.conf:

maxretry = 3

[rackups]
driver = macosx-ups
port = auto
desc = "Rack UPS - OL1500RTXL2UN"

nut.conf:

MODE=netserver

upsmon.conf:

RUN_AS_USER nutmon

MONITOR rackups@localhost 1 monuser pass primary

MINSUPPLIES 1

POLLFREQ 5

HOSTSYNC 15

DEADTIME 15

POWERDOWNFLAG "/etc/killpower"

OFFDURATION 30

RBWARNTIME 43200

NOCOMMWARNTIME 300

FINALDELAY 5

upsd.users: 

   [monuser]
       password  = pass
       upsmon primary

upsd.conf: 

LISTEN * 3493

gonzonia, 2025-07-02:
Yes. I started with  `usbhid-ups` but was getting the same thing so switched to `macosx-ups` hoping that would fix it. 

Running the  `usbhid-ups` driver directly: 

```
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
   0.000000	[D1] upsdrv_makevartable...
   0.000026	[D1] Using USB implementation: libusb-1.0.29 (API: 0x100010a)
   0.000089	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with Apple clang version 15.0.0 (clang-1500.3.9.4); Target: x86_64-apple-darwin23.3.0; Thread model: posix and configured with flags: --disable-dependency-tracking --prefix=/usr/local/Cellar/nut/2.8.2 --localstatedir=/usr/local/var --sysconfdir=/usr/local/etc/nut --with-statepath=/usr/local/var/state/ups --with-pidpath=/usr/local/var/run --with-systemdtmpfilesdir=/usr/local/Cellar/nut/2.8.2/share/nut --with-openssl --with-serial --with-usb --without-avahi --without-cgi --without-dev --without-doc --without-ipmi --without-libltdl --without-neon --without-nss --without-nut_monitor --without-powerman --without-pynut --without-snmp --without-wrap --with-macosx_ups
   0.000094	[D1] debug level is '1'
   0.000239	[D1] Can not become_user(nobody): not root initially, remaining UID=501 GID=20
   0.000252	[D1] upsdrv_initups (non-SHUT)...
   0.019089	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.019501	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.019897	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.020309	[D1] failed to detach kernel driver from USB device: Access denied (insufficient permissions)
   0.020316	Can't claim USB device [0764:0601]@0/0/0: Access denied (insufficient permissions)
```

jimklimov, 2025-07-03:
FWIW, `Can not become_user(nobody): not root initially, remaining UID=501 GID=20` indicates that the package builder did not customize `--with-user` and perhaps `--with-group` (and probably did not deliver a dedicated set of those to the OS, so NUT would run in its own bubble).

Either way, in that direct run above, the process was likely owned by your user account and had no permissions to access USB hardware - something that is solved by running as either `root` (with options like `-x user=root` to prevent dropping privileges), or a dedicated user account like `nut` which owns the device filesystem nodes (using some OS subsystem which manages that - like devd, hotplug, udev, upower...)

CC @daeho-ro

sojc701, 2025-08-09:
After finishing to setup the configurations, I got the same error, `Connection failure: Connection refused` when executing `upsc <ups name>@localhost` or `/opt/homebrew/opt/nut/sbin/upsmon -D`.
```
%  /opt/homebrew/opt/nut/sbin/upsmon -D
Network UPS Tools upsmon 2.8.2
kill: No such process
   0.000000	[D1] Just failed to send signal, no daemon was running
   0.000758	UPS: cp1500@localhost (primary) (power value 1)
   0.000977	Using power down flag file /etc/killpower
   0.001557	[D1] debug level is '1'
   0.004485	[D1] Saving PID 19716 into /opt/homebrew/var/run/upsmon.pid
   0.004597	[D1] Can not become_user(nobody): not root initially, remaining UID=501 GID=20
   0.008607	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.008614	[D1] Trying to connect to UPS [cp1500@localhost]
   0.010329	UPS [cp1500@localhost]: connect failed: Connection failure: Connection refused
   0.010337	Communications with UPS cp1500@localhost lost

% upsc cp1500@localhost
Error: Connection failure: Connection refused
```

I found the way to resolve the issue. I got the help from [NUT on Mac OS X](https://github.com/networkupstools/nut/wiki/NUT-on-Mac-OS-X) wiki introduced on [Network UPS Tools](https://networkupstools.org/documentation.html).
According to the wiki, `upsmon` is not the only one to be executed to start the server but also `upsdrvctl` and `upsd` need to be executed prior to it.

Here are the commands I execute to start the NUT server and then `upsc cp1500@localhost` succeeds.

```
% /opt/homebrew/opt/nut/sbin/upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Mac OS X UPS meta-driver 1.41 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

% /opt/homebrew/opt/nut/sbin/upsd
Network UPS Tools upsd 2.8.2
fopen /opt/homebrew/var/state/ups/upsd.pid: No such file or directory
Could not find PID file '/opt/homebrew/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
/opt/homebrew/etc/nut/upsd.conf is world readable
listening on ::0 port 3493
listening on 0.0.0.0 port 3493
/opt/homebrew/var/state/ups is world readable
Connected to UPS [cp1500]: macosx-ups-cp1500
Found 1 UPS defined in ups.conf
/opt/homebrew/etc/nut/upsd.users is world readable

% /opt/homebrew/opt/nut/sbin/upsmon
Network UPS Tools upsmon 2.8.2
kill: No such process
UPS: cp1500@localhost (primary) (power value 1)
Using power down flag file /etc/killpower

% upsc cp1500@localhost 
battery.charge: 100
battery.runtime: 4740
battery.voltage: 121.000
device.mfr: (unknown)
device.model: CP1500PFCRM2U
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: macosx-ups
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.internal: 1.41
ups.mfr: (unknown)
ups.model: CP1500PFCRM2U
ups.status: OL
```

I used Automator app to create an Application app running a script executing 3 commands above in order to start the NUT server automatically on boot.

[image]

For the reference, here are the file settings.

- nut.conf
`MODE=netserver`

- ups.conf
```
[<UPS name>]
  driver = macosx-ups
  port = auto
  desc = "<UPS model name>"
```

- upsd.conf
`LISTEN * 3493`

- upsd.users
```
[<user name>]
  password = <password>
  upsmon master
```

- upsmon.conf
`MONITOR <UPS name>@localhost 1 <user name> <passord> master`

EagleDTW, 2025-08-09:
> After finishing to setup the configurations, I got the same error, `Connection failure: Connection refused` when executing `upsc <ups name>@localhost` or `/opt/homebrew/opt/nut/sbin/upsmon -D`.
> 
> ```
> %  /opt/homebrew/opt/nut/sbin/upsmon -D
> Network UPS Tools upsmon 2.8.2
> kill: No such process
>    0.000000	[D1] Just failed to send signal, no daemon was running
>    0.000758	UPS: cp1500@localhost (primary) (power value 1)
>    0.000977	Using power down flag file /etc/killpower
>    0.001557	[D1] debug level is '1'
>    0.004485	[D1] Saving PID 19716 into /opt/homebrew/var/run/upsmon.pid
>    0.004597	[D1] Can not become_user(nobody): not root initially, remaining UID=501 GID=20
>    0.008607	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
>    0.008614	[D1] Trying to connect to UPS [cp1500@localhost]
>    0.010329	UPS [cp1500@localhost]: connect failed: Connection failure: Connection refused
>    0.010337	Communications with UPS cp1500@localhost lost
> 
> % upsc cp1500@localhost
> Error: Connection failure: Connection refused
> ```
> 
> I found the way to resolve the issue. I got the help from [NUT on Mac OS X](https://github.com/networkupstools/nut/wiki/NUT-on-Mac-OS-X) wiki introduced on [Network UPS Tools](https://networkupstools.org/documentation.html). According to the wiki, `upsmon` is not the only one to be executed to start the server but also `upsdrvctl` and `upsd` need to be executed prior to it.
> 
> Here are the commands I execute to start the NUT server and then `upsc <ups name>@localhost` succeeds.
> 
> ```
> % /opt/homebrew/opt/nut/sbin/upsdrvctl start
> Network UPS Tools - UPS driver controller 2.8.2
> Network UPS Tools - Mac OS X UPS meta-driver 1.41 (2.8.2)
> Warning: This is an experimental driver.
> Some features may not function correctly.
> 
> % /opt/homebrew/opt/nut/sbin/upsd
> Network UPS Tools upsd 2.8.2
> fopen /opt/homebrew/var/state/ups/upsd.pid: No such file or directory
> Could not find PID file '/opt/homebrew/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
> /opt/homebrew/etc/nut/upsd.conf is world readable
> listening on ::0 port 3493
> listening on 0.0.0.0 port 3493
> /opt/homebrew/var/state/ups is world readable
> Connected to UPS [cp1500]: macosx-ups-cp1500
> Found 1 UPS defined in ups.conf
> /opt/homebrew/etc/nut/upsd.users is world readable
> 
> % /opt/homebrew/opt/nut/sbin/upsmon
> Network UPS Tools upsmon 2.8.2
> kill: No such process
> UPS: cp1500@localhost (primary) (power value 1)
> Using power down flag file /etc/killpower
> 
> % upsc cp1500@localhost 
> battery.charge: 100
> battery.runtime: 4740
> battery.voltage: 121.000
> device.mfr: (unknown)
> device.model: CP1500PFCRM2U
> device.type: ups
> driver.debug: 0
> driver.flag.allow_killpower: 0
> driver.name: macosx-ups
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.synchronous: auto
> driver.state: quiet
> driver.version: 2.8.2
> driver.version.internal: 1.41
> ups.mfr: (unknown)
> ups.model: CP1500PFCRM2U
> ups.status: OL
> ```
> 
> I used Automator app to create an Application app running a script executing 3 commands above in order to start the NUT server automatically on boot.
> <img alt="Image" width="920" height="529" src="https://private-user-images.githubusercontent.com/187454/476248904-64e599ca-00da-4d04-84fb-38d30ed023d6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ3MTczNTQsIm5iZiI6MTc1NDcxNzA1NCwicGF0aCI6Ii8xODc0NTQvNDc2MjQ4OTA0LTY0ZTU5OWNhLTAwZGEtNGQwNC04NGZiLTM4ZDMwZWQwMjNkNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwOVQwNTI0MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZWQwOWJmMDBhNjNlOWJjMjJhNWVmZTE3MjExZDg3ODFlMjg2YTczMjExNzMyNWRiY2UwZDJkYTM2NjJkZDYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zaaKd70_lO_S1IbE_jTpkGswjMOLjnLRTMYFuioiBIM">
> 
> For the reference, here are the file settings.
> 
>     * nut.conf
>       `MODE=netserver`
> 
>     * ups.conf
> 
> 
> ```
> [<UPS name>]
>   driver = macosx-ups
>   port = auto
>   desc = "<UPS model name>"
> ```
> 
>     * upsd.conf
>       `LISTEN * 3493`
> 
>     * upsd.users
> 
> 
> ```
> [<user name>]
>   password = <password>
>   upsmon master
> ```
> 
>     * upsmon.conf
>       `MONITOR <UPS name>@localhost 1 <user name> <passord> master`

I'll be trying this tomorrow!!! Many thanks!