networkupstools/nut #2833: Update generic_gpio_libgpiod driver to latest libgpiod versions.

tomeko12, 2025-03-04:
Hi.
Please update generic_gpio_libgpiod driver to latest libgpiod versions.
Currently, this driver require libgpiod version 1.x.x (latest is 1.6.5), but newest distribution don't include this version,  2.x.x only (latest is 2.2.1), and driver can't be build.

jimklimov, 2025-03-05:
Thanks for the report. Apparently none of the CI farm systems, including those in GHA and similar providers who define the OS we get, exposed this to be a problem. Maybe they don't have the library at all and build `--with-all=auto`, or have older distro versions and older library :-\

@modrisb : would you have a chance to look at this anytime soon? (I hope to finish one other PR and cut a release to slip into distros' release window, so it is a matter of hours/days to fix before that too...)

@tomeko12 : can you suggest which distro/release this problem can be triggered on?

jimklimov, 2025-03-05:
I took a quick look at it, and it seems the v2.x revamp of https://web.git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/ / https://github.com/brgl/libgpiod was even bigger than libusb0 going to libusb1 (where most of the concepts were equivalent but with different names and data types).

Here it is (seems) to be like a separate project dabbling in the same area as libgpiod v1.x

jimklimov, 2025-03-05:
FWIW, some good history (and pointers about future) here:
* https://forums.raspberrypi.com/viewtopic.php?t=377766

And some precedent update to new API in another randomly googled project:
* https://github.com/LinuxCNC/linuxcnc/issues/2842

jimklimov, 2025-03-05:
Laid some groundwork to detect the libgpiod variant and provide the fixes, but not sure I'd be able to follow up on those quickly (I don't have time now to understand the API changes responsibly, nor hardware to test against).

A commit would be very welcome to complete the PR, feel free to use https://github.com/jimklimov/nut/tree/issue-2833 as branch as the starting point.

Currently the configuration passes, `#define WITH_LIBGPIO_VERSION 0x00020000` appears in `config.h`, and the actual build reported problems only with this driver and none others (nor in gpio related tests):

````
Making all in drivers
  CC       generic_gpio_libgpiod.o
In file included from generic_gpio_libgpiod.c:27:
generic_gpio_libgpiod.h:30:33: error: field ‘gpioLines’ has incomplete type
   30 |         struct gpiod_line_bulk  gpioLines;      /* libgpiod lines to monitor */
      |                                 ^~~~~~~~~
generic_gpio_libgpiod.h:31:33: error: field ‘gpioEventLines’ has incomplete type
   31 |         struct gpiod_line_bulk  gpioEventLines; /* libgpiod lines for event monitoring */
      |                                 ^~~~~~~~~~~~~~
generic_gpio_libgpiod.c: In function ‘reserve_lines_libgpiod’:
generic_gpio_libgpiod.c:80:50: error: storage size of ‘config’ isn’t known
   80 |                 struct gpiod_line_request_config config;
      |                                                  ^~~~~~
generic_gpio_libgpiod.c:84:47: error: ‘GPIOD_LINE_REQUEST_EVENT_BOTH_EDGES’ undeclared (first use in this function)
   84 |                         config.request_type = GPIOD_LINE_REQUEST_EVENT_BOTH_EDGES;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
generic_gpio_libgpiod.c:84:47: note: each undeclared identifier is reported only once for each function it appears in
generic_gpio_libgpiod.c:87:47: error: ‘GPIOD_LINE_REQUEST_DIRECTION_INPUT’ undeclared (first use in this function); did you mean ‘GPIOD_LINE_DIRECTION_INPUT’?
   87 |                         config.request_type = GPIOD_LINE_REQUEST_DIRECTION_INPUT;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               GPIOD_LINE_DIRECTION_INPUT
generic_gpio_libgpiod.c:91:26: warning: implicit declaration of function ‘gpiod_line_request_bulk’; did you mean ‘gpiod_line_request_get_fd’? [-Wimplicit-function-declaration]
   91 |                 gpioRc = gpiod_line_request_bulk(&libgpiod_data->gpioLines, &config, NULL);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~
      |                          gpiod_line_request_get_fd
generic_gpio_libgpiod.c:80:50: warning: unused variable ‘config’ [-Wunused-variable]
   80 |                 struct gpiod_line_request_config config;
      |                                                  ^~~~~~
generic_gpio_libgpiod.c: In function ‘gpio_open’:
generic_gpio_libgpiod.c:113:41: warning: implicit declaration of function ‘gpiod_chip_open_by_name’; did you mean ‘gpiod_chip_info_get_name’? [-Wimplicit-function-declaration]
  113 |         libgpiod_data->gpioChipHandle = gpiod_chip_open_by_name(gpioupsfdlocal->chipName);
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~
      |                                         gpiod_chip_info_get_name
generic_gpio_libgpiod.c:113:39: warning: assignment to ‘struct gpiod_chip *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  113 |         libgpiod_data->gpioChipHandle = gpiod_chip_open_by_name(gpioupsfdlocal->chipName);
      |                                       ^
generic_gpio_libgpiod.c:123:50: warning: implicit declaration of function ‘gpiod_chip_num_lines’; did you mean ‘gpiod_chip_request_lines’? [-Wimplicit-function-declaration]
  123 |                 gpioupsfdlocal->chipLinesCount = gpiod_chip_num_lines(libgpiod_data->gpioChipHandle);
      |                                                  ^~~~~~~~~~~~~~~~~~~~
      |                                                  gpiod_chip_request_lines
generic_gpio_libgpiod.c:134:17: warning: implicit declaration of function ‘gpiod_line_bulk_init’ [-Wimplicit-function-declaration]
  134 |                 gpiod_line_bulk_init(&libgpiod_data->gpioLines);
      |                 ^~~~~~~~~~~~~~~~~~~~
generic_gpio_libgpiod.c:136:26: warning: implicit declaration of function ‘gpiod_chip_get_lines’; did you mean ‘gpiod_chip_get_info’? [-Wimplicit-function-declaration]
  136 |                 gpioRc = gpiod_chip_get_lines(
      |                          ^~~~~~~~~~~~~~~~~~~~
      |                          gpiod_chip_get_info
generic_gpio_libgpiod.c: In function ‘gpio_get_lines_states’:
generic_gpio_libgpiod.c:180:41: error: storage size of ‘event’ isn’t known
  180 |                 struct gpiod_line_event event;
      |                                         ^~~~~
generic_gpio_libgpiod.c:198:24: warning: implicit declaration of function ‘gpiod_line_event_wait_bulk’ [-Wimplicit-function-declaration]
  198 |                 monRes=gpiod_line_event_wait_bulk(
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
generic_gpio_libgpiod.c:209:52: warning: implicit declaration of function ‘gpiod_line_bulk_num_lines’ [-Wimplicit-function-declaration]
  209 |                         int num_lines_local = (int)gpiod_line_bulk_num_lines(&libgpiod_data->gpioEventLines);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
generic_gpio_libgpiod.c:231:16: warning: implicit declaration of function ‘gpiod_line_get_value_bulk’ [-Wimplicit-function-declaration]
  231 |         gpioRc=gpiod_line_get_value_bulk(
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
generic_gpio_libgpiod.c:253:17: warning: implicit declaration of function ‘gpiod_line_release_bulk’ [-Wimplicit-function-declaration]
  253 |                 gpiod_line_release_bulk(&libgpiod_data->gpioLines);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:2188: generic_gpio_libgpiod.o] Error 1
make[1]: Target 'all' not remade because of errors.
````

modrisb, 2025-03-05:
@jimklimov not clear yet what would be the best solution: add ifdef's to generic_gpio-gpiolib.c to build for both library versions or create generic_gpio-gpiolib2.c to be used with v2 library. Probably 1st is possible, but may require significant rework and testing.

jimklimov, 2025-03-05:
Well, it depends on how large the API difference is. I hoped for a dozen tactical ifdef's to solve the issue, if most of the other logic remains. Easier to keep the two variant builds in sync this way, going forward (new NUT on older distros).

modrisb, 2025-03-07:
@jimklimov  I have some progress with changes to support v2, but need more time for testing, hope will have by Monday. Test tools might be under questions as needs new  mock for library. Let me know if see reason to publish recent code to git.

jimklimov, 2025-03-07:
Sounds encouraging, thanks! If it is buildable in at least one case, makes sense to push and check by CI (although that would mostly reveal that ifdef's did not break syntax for existing platforms). Maybe we can add a scenario using (packaged or locally built) libgpiod2 for more relevant checks too :) but I'll be also away for the weekend or much of it.

CC @mbiebl @bgermann : sorry for poking you guys - when is the Debian/Ubuntu packaging freeze for April releases, and how strict is it? Any chance of NUT v2.8.3 making it there if released e.g. next week, or is it too late already?..

bgermann, 2025-03-07:
I do not use nut myself, just fixed a trivial bug that would have kept the existing package version from being released with trixie.
This was a good enough reason to upload not being the package maintainer.

For Ubuntu, I do not know the date, for Debian it is [April 15th](https://release.debian.org/testing/freeze_policy.html), so the new package version should enter on the 5th at the latest. If a SONAME bump (and with it, a package rename) is involved, you are probably out of luck.

The right place to discuss this is the [Debian bug](https://bugs.debian.org/1092106).

modrisb, 2025-03-08:
I opened pull request for solution.

modrisb, 2025-03-09:
How did you checked presence of define for GPIOD_API? V2 git installation adds gpiod.h to /usr/local/include and not to /usr/include, may that's a trick? I lost at least hour on that. I added to code printout that depends on this define and got proper printouts.

jimklimov, 2025-03-09:
I `grep`'ed the two sets of headers :)

Actually seen in git too:
```
:; cd libgpiod-git

:; git grep GPIOD_API v2.2.1 | grep define
v2.2.1:lib/internal.h:#define GPIOD_API __attribute__((visibility("default")))

:; git grep GPIOD_API v1.6.5 | grep define
v1.6.5:include/gpiod.h:#define GPIOD_API                __attribute__((visibility("default")))
```

jimklimov, 2025-03-09:
I'm preparing the branch with your and my changes together, checking locally that the two builds pass and will post a PR to upstream NUT - would be grateful if you can test that against HW too :)

jimklimov, 2025-03-09:
PR #2841 posted, thanks to @modrisb for the actual changes. I just added the fancy bells and whistles :)

modrisb, 2025-03-09:
@jimklimov what would be right way to get code for testing?

jimklimov, 2025-03-09:
`git clone https://github.com/jimklimov/nut -b issue-2833+modrisb nut-issue-2833`

modrisb, 2025-03-09:
@jimklimov v2 works fine, except test tool prints lib version in decimal as 131072 (0x00020000); v1 does not compile, I see lib version 0x00020000 - compiling v2 code against v1 library. tested with bookworm on rpi0 2 w, gpiod 1.6.3 + libgpiod-dev 1.6.3

jimklimov, 2025-03-10:
Aha, missed that hex :\ Thanks for checking!

Your initial post mentioned issues with v1 build - was that an operator error? :)

modrisb, 2025-03-10:
I did not properly cleared build before v1 - my fault.

PR #2841 (2025-03-11): Fix `generic_gpio_libgpiod` driver builds with libgpio 2.x API changes
Closes: #2833

API of the library has massively changed between 1.6.x available in older distros, and 2.x in newer ones (and matching Linux kernel functionality changed a lot as well). The driver was not buildable in newer/upcoming distribution releases because of that.

Great thanks to @modrisb for initial driver development, and for this subsequent fix (on short notice at that). And to @tomeko12 for raising the alarm :)

For my part, it should now be easy to identify which API (1 or 2) the program got built against:
```
# [... cut]