networkupstools/nut #2838: riello_usb NPW1000 with serial to USB converter

d1nuc0m, 2025-03-07:
Hi, this is a parallel to #2291 (haven't been able to replace the USB port yet...)

I'm using a Riello NPW1000 with a bad USB port, so I tried to connect it to the PC using a RS-232 to USB converter.

The adapter is reported by lsusb as `ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC` and accessible as  `/dev/ttyUSB0`

I tried to configure NUT to use the serial with this configuration:

```
[riellonpw1000_serial]
  driver = riello_ser
  port   = /dev/ttyUSB0
  user   = root
```

But the UPS is not detected:

```
$ sudo riello_ser -a riellonpw1000_serial -DDDDDD -d1
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Riello serial driver 0.10 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000124     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000129     [D5] do_upsconf_args: call do_global_args()
   0.000133     [D3] do_global_args: var='maxretry' val='3'
   0.000165     [D5] do_upsconf_args: confupsname=riellonpw1000, var=driver, val=riello_usb
   0.000168     [D5] do_upsconf_args: confupsname=riellonpw1000, var=port, val=/dev/ttyUSB0
   0.000173     [D5] do_upsconf_args: confupsname=riellonpw1000, var=user, val=root
   0.000176     [D5] do_upsconf_args: confupsname=riellonpw1000_serial, var=driver, val=riello_ser
   0.000179     [D5] do_upsconf_args: call main_arg()
   0.000182     [D3] main_arg: var='driver' val='riello_ser'
   0.000186     [D5] do_upsconf_args: not a main_arg()
   0.000189     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000193     [D6] testval_reloadable: var=driver, oldval=riello_ser, newval=riello_ser, reloadable=0, reload_flag=0
   0.000197     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000200     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000203     [D5] do_upsconf_args: confupsname=riellonpw1000_serial, var=port, val=/dev/ttyUSB0
   0.000206     [D5] do_upsconf_args: call main_arg()
   0.000209     [D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000211     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyUSB0, reloadable=0, reload_flag=0
   0.000214     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000218     [D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.000221     [D5] do_upsconf_args: confupsname=riellonpw1000_serial, var=user, val=root
   0.000223     [D5] do_upsconf_args: call main_arg()
   0.000226     [D3] main_arg: var='user' val='root'
   0.000228     [D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000232     [D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000234     [D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000237     [D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000242     [D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3) and configured with flags: --build=x86_64-redhat-linux-gnu --host=x86_64-redhat-linux-gnu --program-prefix= --disable-dependency-tracking --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib64 --libexecdir=/usr/libexec --localstatedir=/var --sharedstatedir=/var/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-all --without-powerman --with-libltdl --without-wrap --with-modbus --without-linux-i2c --with-cgi --with-python=/usr/bin/python3 --with-python3=/usr/bin/python3 --without-python2 --datadir=/usr/share/nut --with-user=nut --with-group=dialout --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --sysconfdir=/etc/ups --with-cgipath=/var/www/nut-cgi-bin --with-drvpath=/usr/sbin --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-pkgconfig-dir=/usr/lib64/pkgconfig --disable-static --with-udev-dir=/usr/lib/udev --libdir=/usr/lib64
   0.000249     [D1] debug level is '6'
   0.000252     [D5] send_to_all: SETINFO driver.debug "6"
   0.000255     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001586     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001593     [D5] send_to_all: SETINFO device.type "ups"
   0.001597     [D5] send_to_all: SETINFO driver.state "init.device"
   0.001600     [D2] entering upsdrv_initups()
   0.007892     [D2] set baudrate 9600
   0.008884     [D2] try to detect SENTR
   5.013472     [D2] try to detect GPSER
  10.018493     [D2] set baudrate 1200
  10.019021     [D2] try to detect SENTR
  15.024215     [D2] try to detect GPSER
  20.029480     Can't connect to the UPS on port /dev/ttyUSB0!

  20.029499     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

**System**
Almalinux 9.5, nut-client-2.8.2-2, nut-2.8.2-2, Riello USB driver 0.12 (2.8.2), Riello serial driver 0.10 (2.8.2)

jimklimov, 2025-03-07:
Just in case, some sanity-checks:
* do you have any other programs (`screen`, `getty` etc.) trying to use the `/dev/ttyUSB0` as well?
  * Potentially, the `riello_usb` if it is also running? In fact, the `nut-driver-enumerator` might auto-create you instances of the service units to wrap the drivers, and those could also conflict with each other and command-line runs, see https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE)
* are you sure there are no other ttyUSB's there? :)
* is the USB link to the UPS detached? (Some vendors/devices allow only one of USB/Serial links, if both are plugged)
  * ...and the port broken on the UPS side might impact this, e.g. think it is always connected?..

AminAliz, 2025-06-01:
You have to check your list. 
$cd /
$cd /dev
$ls
By connecting and disconnecting your USB, you could find the right configuration