networkupstools/nut #2840: NUT displays wrong battery voltage

Aaron451, 2025-03-08:
Just installed client on Win11 PC. Everything seems to work but display shows 8 volts for battery. When I look at UPS variables it shows 26.1 volts. Is this a bug? Something I'm doing wrong?

[image]

[image]

jimklimov, 2025-03-08:
Could be a bug, but that program is a third-party project - please bring this up in their tracker so it can be actually investigated and fixed.

Aaron451, 2025-03-08:
not sure how to do that @jimklimov

jimklimov, 2025-03-09:
Determine which one you have, likely https://github.com/gawindx/WinNUT-Client or https://github.com/nutdotnet/WinNUT-Client but others by the name also exist, and ask there :)

gbakeman, 2025-03-31:
Hello,

I'm currently maintaining the WinNUT Client, you can see the project [here](https://github.com/nutdotnet/WinNUT-Client). More specific to your issue, I've been slowly working on addressing problems with the gauges with so I hope to have this fixed at some point. You can find more info our [issues](https://github.com/nutdotnet/WinNUT-Client/issues?q=is%3Aissue%20state%3Aopen%20gauge).